zigbeefordomoticz/Domoticz-Zigbee #1201: beta6 does not start with non-standard API port

mitkodotcom, 2022-05-29:
If domoticz is running on non-standard port (not 8080), the current beta6 branch is unable to star. The plugin is trying to open the hardware database via API call but the API call fails with an exception here:

Domoticz-Zigate/Classes/DomoticzDB.py:        url = "http://127.0.0.1:%s/%s" %(self.pluginconf.pluginConf["port"], DOMOTICZ_HARDWARE_API)

As a workaround, start domoticz on port 8080 first, then enter the plugin setup, change the api port and restart domoticz on the new port. May be a better exception handling is required.

pipiche38, 2022-05-29:
Good catch @mitkodotcom , and I'm afraid that we are in the chicken and egg issue.

I think that best workaround would be to : 

1/ start the plugin which will fail
2/ edit the file Conf/PluginConf-xx.json

add a line

`"port": "the not by default domoticz port",`

Otherwise , This would have to be an additional parameter on the Hardware menu when starting the plugin

mitkodotcom, 2022-05-29:
Moving this parameter from plugin settings to Hardware settings seems the right solution. You can't change it from the plugin's web gui anyway as the web gui won't start.

mitkodotcom, 2022-05-29:
Besides, you will also need two more parameters for API user and password unless 127.0.0.1 is added to Local Networks which is not always the case.

pipiche38, 2022-05-29:
Unfortunately there is not enough Parameters

mitkodotcom, 2022-05-29:
Zigbee for domoticz has 8 parameters, but I have plugins with more than that (for example Modbus has 11 parameters). You can use just one parameter for API base url: http[s]://user:pass@ip:port/

pipiche38, 2022-05-29:
good point API base is definitively the right way to do

pipiche38, 2022-05-30:
Could you please check against f6464046 (on branch https://github.com/zigbeefordomoticz/Domoticz-Zigbee/tree/pipiche38/urlbaseapi )

pipiche38, 2022-06-01:
Did you have a chance to test that one

pipiche38, 2022-06-02:
@mitkodotcom did you have a chance to test that one ?

mitkodotcom, 2022-06-05:
It looks that urlencoding works now. Both serial port with a plus sign and the API base url were OK after restarting the plugin. I haven't tested the API url with a password, though.

closed 2022-06-06 by pipiche38