zigbeefordomoticz/Domoticz-Zigbee #1203: plus sign in device name gets translated to space

mitkodotcom, 2022-05-29:
Device name:

lrwxrwxrwx 1 root root 13 May 27 12:43 usb-FTDI_ZiGate_ZIGATE+-if00-port0 -> ../../ttyUSB0

Error message:

"Cannot open Zigate port /dev/serial/by-id/usb-FTDI_ZiGate_ZIGATE -if00-port0 error: [Errno 2] could not open port /dev/serial/by-id/usb-FTDI_ZiGate_ZIGATE -if00-port0: [Errno 2] No such file or directory: '/dev/serial/by-id/usb-FTDI_ZiGate_ZIGATE -if00-port0'"

may be the port name should be urlencoded

pipiche38, 2022-05-29:
feel free to submit a PR against beta6 branch

mitkodotcom, 2022-05-30:
I don't know if it deserves a pull request -  it's just one line in Modules/restartPlugin.py

    url += "&port=%s" % self.pluginParameters["Port"]
    
    to
    
    url += "&port=%s" % urllib.urlencode(self.pluginParameters["Port"])
    
In fact, all parameters, that may contain special characters should be urlencoded - for example the new API base url.

mitkodotcom, 2022-05-30:
The whole parameter list in Modules/restartPlugin.py has to be urlencoded, especially the new parameter for API base url. 
A quick patch for those of us with ZiGate+ is to replace the line

    url += "&serialport=%s" % self.pluginParameters["SerialPort"]

to

    url += "&serialport=%s" % self.pluginParameters["SerialPort"].replace("+", "%2B") 
    
This quick patch does not deserve pull request IMHO. A more complete solution will require rewriting this function and using array of parameters and urllib.parse.urlencode instead of just creating the url as string.

pipiche38, 2022-06-01:
can you try branch pipiche38/urlbaseapi ( git checkout pipiche38/urlbaseapi ) and let me know how this work ?

pipiche38, 2022-06-01:
Duplicate #1201