zigbeefordomoticz/Domoticz-Zigbee #1205: Unable to remove broken devices

tomm1ed, 2022-06-03:
I had an Aqara temperature sensor die on me and I seem unable to remove it from the Zigate. Deleting the device from Domoticz left it in provREQ Status (unclear what that means exactly)
[https://tweakers.net/fotoalbum/image/lwwByUnMh2Sds6gewTA9Sev2.png](url)
Can anyone tell me how to fix this?

tomm1ed, 2022-06-03:
I also tried removing three Aqara Motion Sensors (which I am no longer using) and even though I receive this in the log:
2022-06-03 10:40:00,155 INFO    : [       MainThread] onDeviceRemoved - removing End Device
2022-06-03 10:40:00,170 INFO    : [       MainThread] onDeviceRemoved - removing End Device
2022-06-03 10:40:00,177 INFO    : [       MainThread] onDeviceRemoved - removing End Device
The devices themselves do not get removed from the Zigate

Is there any way to remove these manually @pipiche38 ? I have a ZiGate Classic with WiFi

pipiche38, 2022-06-03:
The best is now, to restart the plugin , all entries will be removed, and then while making sure the Zigate is not in pairing mode, you simply reset the Aqara temperature device. In that case it will send a leave message and get removed from the Zigate.

In all cases, you cannot do more as , only the device itself can me removed from the coordinator

pipiche38, 2022-06-03:
https://github.com/zigbeefordomoticz/wiki/blob/master/en-eng/HowTo_Remove-device.md