zigbeefordomoticz/Domoticz-Zigbee #1209: [zigpy-integration] - error at plungin start

pipiche38, 2022-06-03:
```
2022-06-03 17:09:15,941 INFO    :ZNP is already configured, not forming a new network
2022-06-03 17:09:20,024 ERROR   :Task exception was never retrieved
future: <Task finished name='Task-6' coro=<ControllerApplication.on_af_message() done, defined at /var/lib/domoticz/plugins/Domoticz-Zigate/zigpy_znp/zigbee/application.py:763> exception=KeyError(00:12:4b:00:25:6c:12:07)>
Traceback (most recent call last):
  File "/usr/lib/python3.9/asyncio/tasks.py", line 256, in __step
    result = coro.send(None)
  File "/var/lib/domoticz/plugins/Domoticz-Zigate/zigpy_znp/zigbee/application.py", line 779, in on_af_message
    if msg.DstEndpoint in self.zigpy_device.endpoints:
  File "/var/lib/domoticz/plugins/Domoticz-Zigate/zigpy_znp/zigbee/application.py", line 816, in zigpy_device
    return self.devices[self.ieee]
KeyError: 00:12:4b:00:25:6c:12:07
2022-06-03 17:09:20,069 ERROR   :Task exception was never retrieved
future: <Task finished name='Task-8' coro=<ControllerApplication.on_af_message() done, defined at /var/lib/domoticz/plugins/Domoticz-Zigate/zigpy_znp/zigbee/application.py:763> exception=KeyError(00:12:4b:00:25:6c:12:07)>
Traceback (most recent call last):
  File "/usr/lib/python3.9/asyncio/tasks.py", line 256, in __step
    result = coro.send(None)
  File "/var/lib/domoticz/plugins/Domoticz-Zigate/zigpy_znp/zigbee/application.py", line 779, in on_af_message
    if msg.DstEndpoint in self.zigpy_device.endpoints:
  File "/var/lib/domoticz/plugins/Domoticz-Zigate/zigpy_znp/zigbee/application.py", line 816, in zigpy_device
    return self.devices[self.ieee]
KeyError: 00:12:4b:00:25:6c:12:07
2022-06-03 17:09:20,101 ERROR   :Task exception was never retrieved
future: <Task finished name='Task-10' coro=<ControllerApplication.on_af_message() done, defined at /var/lib/domoticz/plugins/Domoticz-Zigate/zigpy_znp/zigbee/application.py:763> exception=KeyError(00:12:4b:00:25:6c:12:07)>
Traceback (most recent call last):
  File "/usr/lib/python3.9/asyncio/tasks.py", line 256, in __step
    result = coro.send(None)
  File "/var/lib/domoticz/plugins/Domoticz-Zigate/zigpy_znp/zigbee/application.py", line 779, in on_af_message
    if msg.DstEndpoint in self.zigpy_device.endpoints:
  File "/var/lib/domoticz/plugins/Domoticz-Zigate/zigpy_znp/zigbee/application.py", line 816, in zigpy_device
    return self.devices[self.ieee]
KeyError: 00:12:4b:00:25:6c:12:07
2022-06-03 17:09:20,133 ERROR   :Task exception was never retrieved
future: <Task finished name='Task-12' coro=<ControllerApplication.on_af_message() done, defined at /var/lib/domoticz/plugins/Domoticz-Zigate/zigpy_znp/zigbee/application.py:763> exception=KeyError(00:12:4b:00:25:6c:12:07)>
Traceback (most recent call last):
  File "/usr/lib/python3.9/asyncio/tasks.py", line 256, in __step
    result = coro.send(None)
  File "/var/lib/domoticz/plugins/Domoticz-Zigate/zigpy_znp/zigbee/application.py", line 779, in on_af_message
    if msg.DstEndpoint in self.zigpy_device.endpoints:
  File "/var/lib/domoticz/plugins/Domoticz-Zigate/zigpy_znp/zigbee/application.py", line 816, in zigpy_device
    return self.devices[self.ieee]
KeyError: 00:12:4b:00:25:6c:12:07
2022-06-03 17:09:20,180 ERROR   :Task exception was never retrieved
future: <Task finished name='Task-14' coro=<ControllerApplication.on_af_message() done, defined at /var/lib/domoticz/plugins/Domoticz-Zigate/zigpy_znp/zigbee/application.py:763> exception=KeyError(00:12:4b:00:25:6c:12:07)>
Traceback (most recent call last):
  File "/usr/lib/python3.9/asyncio/tasks.py", line 256, in __step
    result = coro.send(None)
  File "/var/lib/domoticz/plugins/Domoticz-Zigate/zigpy_znp/zigbee/application.py", line 779, in on_af_message
    if msg.DstEndpoint in self.zigpy_device.endpoints:
  File "/var/lib/domoticz/plugins/Domoticz-Zigate/zigpy_znp/zigbee/application.py", line 816, in zigpy_device
    return self.devices[self.ieee]
KeyError: 00:12:4b:00:25:6c:12:07
```

pipiche38, 2022-06-03:
@puddly we have quiet often some errors when the plugin is starting. Unfortunately this happening most likely on prod, where debug are not enabled. 

From those logs is there any indication, what could be the issue ? I suspecting that at plugin start we receive already reports and other messages from Devices while the zigpy libs is not yet ready to handle them. Is that something possible .

The main issue, it looks like it is outside of the plugin in task spawn by the lib itself and I cannot catch the Exception

pipiche38, 2022-06-24:
@puddly more information captured, hope it helps

```
2022-06-24 11:10:30,736 DEBUG   :Received command: ZDO.MsgCallbackRegister.Rsp(Status=<Status.SUCCESS: 0>)
2022-06-24 11:10:30,739 DEBUG   :Sending request: ZDO.MsgCallbackRegister.Req(ClusterId=32785)
2022-06-24 11:10:30,752 DEBUG   :Received command: ZDO.MsgCallbackRegister.Rsp(Status=<Status.SUCCESS: 0>)
2022-06-24 11:10:30,755 DEBUG   :Sending request: ZDO.MsgCallbackRegister.Req(ClusterId=32786)
2022-06-24 11:10:30,768 DEBUG   :Received command: ZDO.MsgCallbackRegister.Rsp(Status=<Status.SUCCESS: 0>)
2022-06-24 11:10:30,771 DEBUG   :Sending request: ZDO.MsgCallbackRegister.Req(ClusterId=32788)
2022-06-24 11:10:30,784 DEBUG   :Received command: ZDO.MsgCallbackRegister.Rsp(Status=<Status.SUCCESS: 0>)
2022-06-24 11:10:30,787 DEBUG   :Sending request: ZDO.MsgCallbackRegister.Req(ClusterId=32789)
2022-06-24 11:10:30,800 DEBUG   :Received command: ZDO.MsgCallbackRegister.Rsp(Status=<Status.SUCCESS: 0>)
2022-06-24 11:10:30,804 DEBUG   :Sending request: ZDO.MsgCallbackRegister.Req(ClusterId=32790)
2022-06-24 11:10:30,816 DEBUG   :Received command: ZDO.MsgCallbackRegister.Rsp(Status=<Status.SUCCESS: 0>)
2022-06-24 11:10:30,819 DEBUG   :Sending request: ZDO.MsgCallbackRegister.Req(ClusterId=32791)
2022-06-24 11:10:30,832 DEBUG   :Received command: ZDO.MsgCallbackRegister.Rsp(Status=<Status.SUCCESS: 0>)
2022-06-24 11:10:30,836 DEBUG   :Sending request: ZDO.MsgCallbackRegister.Req(ClusterId=32792)
2022-06-24 11:10:30,848 DEBUG   :Received command: ZDO.MsgCallbackRegister.Rsp(Status=<Status.SUCCESS: 0>)
2022-06-24 11:10:30,851 DEBUG   :Sending request: ZDO.MsgCallbackRegister.Req(ClusterId=32793)
2022-06-24 11:10:30,864 DEBUG   :Received command: ZDO.MsgCallbackRegister.Rsp(Status=<Status.SUCCESS: 0>)
2022-06-24 11:10:30,868 DEBUG   :Sending request: ZDO.MsgCallbackRegister.Req(ClusterId=32794)
2022-06-24 11:10:31,040 DEBUG   :Received command: AF.IncomingMsg.Callback(GroupId=0x0000, ClusterId=1026, SrcAddr=0x3234, SrcEndpoint=1, DstEndpoint=1, WasBroadcast=<Bool.false: 0>, LQI=109, SecurityUse=<Bool.false: 0>, TimeStamp=434821, TSN=0, Data=b'\x18\x2D\x0A\x00\x00\x29\x20\x0A', MacSrcAddr=0x3234, MsgResultRadius=29)
2022-06-24 11:10:31,044 DEBUG   :Received command: AF.IncomingMsg.Callback(GroupId=0x0000, ClusterId=1029, SrcAddr=0x3234, SrcEndpoint=1, DstEndpoint=1, WasBroadcast=<Bool.false: 0>, LQI=109, SecurityUse=<Bool.false: 0>, TimeStamp=435067, TSN=0, Data=b'\x18\x2E\x0A\x00\x00\x21\x17\x17', MacSrcAddr=0x3234, MsgResultRadius=29)
2022-06-24 11:10:31,049 DEBUG   :Received command: AF.IncomingMsg.Callback(GroupId=0x0000, ClusterId=1027, SrcAddr=0x3234, SrcEndpoint=1, DstEndpoint=1, WasBroadcast=<Bool.false: 0>, LQI=109, SecurityUse=<Bool.false: 0>, TimeStamp=435274, TSN=0, Data=b'\x18\x2F\x0A\x00\x00\x29\xD7\x03\x14\x00\x28\xFF\x10\x00\x29\x6B\x26', MacSrcAddr=0x3234, MsgResultRadius=29)
2022-06-24 11:10:31,052 DEBUG   :Received command: ZDO.MsgCallbackRegister.Rsp(Status=<Status.SUCCESS: 0>)
2022-06-24 11:10:31,056 DEBUG   :Sending request: ZDO.MsgCallbackRegister.Req(ClusterId=32795)
2022-06-24 11:10:31,064 ERROR   :Task exception was never retrieved
future: <Task finished name='Task-4' coro=<ControllerApplication.on_af_message() done, defined at /var/lib/domoticz/plugins/Domoticz-Zigate/zigpy_znp/zigbee/application.py:763> exception=KeyError(00:12:4b:00:25:6c:12:07)>
Traceback (most recent call last):
  File "/usr/lib/python3.9/asyncio/tasks.py", line 256, in __step
    result = coro.send(None)
  File "/var/lib/domoticz/plugins/Domoticz-Zigate/zigpy_znp/zigbee/application.py", line 779, in on_af_message
    if msg.DstEndpoint in self.zigpy_device.endpoints:
  File "/var/lib/domoticz/plugins/Domoticz-Zigate/zigpy_znp/zigbee/application.py", line 816, in zigpy_device
    return self.devices[self.ieee]
KeyError: 00:12:4b:00:25:6c:12:07
2022-06-24 11:10:31,073 ERROR   :Task exception was never retrieved
future: <Task finished name='Task-5' coro=<ControllerApplication.on_af_message() done, defined at /var/lib/domoticz/plugins/Domoticz-Zigate/zigpy_znp/zigbee/application.py:763> exception=KeyError(00:12:4b:00:25:6c:12:07)>
Traceback (most recent call last):
  File "/usr/lib/python3.9/asyncio/tasks.py", line 256, in __step
    result = coro.send(None)
  File "/var/lib/domoticz/plugins/Domoticz-Zigate/zigpy_znp/zigbee/application.py", line 779, in on_af_message
    if msg.DstEndpoint in self.zigpy_device.endpoints:
  File "/var/lib/domoticz/plugins/Domoticz-Zigate/zigpy_znp/zigbee/application.py", line 816, in zigpy_device
    return self.devices[self.ieee]
KeyError: 00:12:4b:00:25:6c:12:07
2022-06-24 11:10:31,076 ERROR   :Task exception was never retrieved
future: <Task finished name='Task-6' coro=<ControllerApplication.on_af_message() done, defined at /var/lib/domoticz/plugins/Domoticz-Zigate/zigpy_znp/zigbee/application.py:763> exception=KeyError(00:12:4b:00:25:6c:12:07)>
Traceback (most recent call last):
  File "/usr/lib/python3.9/asyncio/tasks.py", line 256, in __step
    result = coro.send(None)
  File "/var/lib/domoticz/plugins/Domoticz-Zigate/zigpy_znp/zigbee/application.py", line 779, in on_af_message
    if msg.DstEndpoint in self.zigpy_device.endpoints:
  File "/var/lib/domoticz/plugins/Domoticz-Zigate/zigpy_znp/zigbee/application.py", line 816, in zigpy_device
    return self.devices[self.ieee]
KeyError: 00:12:4b:00:25:6c:12:07
```

puddly, 2022-06-24:
It just means that a few packets were received before the radio was fully initialized. I can try to address this in a future release.

pipiche38, 2022-06-25:
Hi,

If you can handle it in future that would be good as this is a frequent issue when restarting the plugin on a large network with especially metering devices

Thanks in advance

closed 2022-07-13 by pipiche38