zigbeefordomoticz/Domoticz-Zigbee #1226: [zigate] - un expected behaviour of Zigate firmware on read Configuration Report

pipiche38, 2022-07-08:
When doing a read configuration report with 2 unsupported attributes, 

Zigate -firmware send some unexpected data after the first attribute !!! This is not expected, either the firmware send the next attribute or do not send such extra data ( 4 bytes )

Paylaod received by plugin
Read Configure Reporting response - de1aec0107028600000400000000
Sqn: de
Nwkid: 1aec
Ep 0&
Cluster 0702
status 86
direction 00
attribute 0004
remaining bytes : 00 00 00 00

pipiche38, 2022-07-08:
@fairecasoimeme FYI

fairecasoimeme, 2022-07-08:
If the read configure Reporting response is command : 0x8122 the structure is  : 

Sqn: de
Nwkid: 1aec
Ep 0&
Cluster 0702
status 86 --> E_ZCL_CMDS_UNSUPPORTED_ATTRIBUTE
**datatype** 00
attribute 0004
**max interval** 0000 
**min interval** 0000

Logically if the second attribute is unsupported too, you may be received an another packet.

pipiche38, 2022-07-09:
Shame that it doesn't match to the Zigbee specification

> If the status field is not set to SUCCESS, all fields except the direction and attribute identifier fields SHALL
> be omitted.

fairecasoimeme, 2022-07-09:
I'll verify the source code to match to the specification

closed 2022-07-11 by pipiche38