zigbeefordomoticz/Domoticz-Zigbee #1227: [zigate] do not handle correctly data type 0x25 on read configuratiuon attributes

pipiche38, 2022-07-08:
Command sent
`0120 021aec010107020000000002002500000005012C0FFF000000000000000a002a04000005012C0FFF0000000a`

Command seen on the sniffer
[image]

pipiche38, 2022-07-08:
@fairecasoimeme FYI

fairecasoimeme, 2022-07-08:
For specific data type like 24bits, 40bits or 48bits, default encoding from source is cast to:

24bits --> 32bits
40/48bits --> 64bits

From source code : 
```
 if( eAttributeDataType >= E_ZCL_UINT8 &&
        eAttributeDataType <= E_ZCL_INT64 )
    {
        switch ( eAttributeDataType )
        {
        case E_ZCL_UINT8:
        case E_ZCL_INT8:
            puAttributeReportable->zuint8ReportableChange = pu8Buffer[(*pu8Offset)++];
            break;
        case E_ZCL_UINT16:
        case E_ZCL_INT16:
            puAttributeReportable->zuint16ReportableChange =  ZNC_RTN_U16_OFFSET ( pu8Buffer, *pu8Offset,  *pu8Offset);
            break;
        case E_ZCL_UINT24:
        case E_ZCL_INT24:
        case E_ZCL_UINT32:
        case E_ZCL_INT32:
            puAttributeReportable->zuint32ReportableChange =  ZNC_RTN_U32_OFFSET ( pu8Buffer, *pu8Offset,  *pu8Offset);
            break;
        case E_ZCL_UINT40:
        case E_ZCL_INT40:
        case E_ZCL_UINT48:
        case E_ZCL_INT48:
        case E_ZCL_UINT56:
        case E_ZCL_INT56:
        case E_ZCL_UINT64:
        case E_ZCL_INT64:
            puAttributeReportable->zuint64ReportableChange =  ZNC_RTN_U64_OFFSET ( pu8Buffer, *pu8Offset,  *pu8Offset);
            break;
        default:
            break;
        }
    }
```

0120 021aec010107020000000002002500000005012C0FFF **000000000000000a** 002a04000005012C0FFF **0000000a**
**000000000000000a** ->  64bits  for datatype 0x25 (48 bits)                                                                 
**0000000a** -> 32bits for datatype 0x2a (24 bits)

pipiche38, 2022-07-08:
Ok. Will not support this feature on Zigate.

closed 2022-07-08 by pipiche38