zigbeefordomoticz/Domoticz-Zigbee #1235: Tracking Zigpy issues

pipiche38, 2022-07-19:
zigpy

- [ ] https://github.com/zigpy/zigpy/discussions/1011
- [ ] https://github.com/zigpy/zigpy/discussions/999
- [ ] https://github.com/zigpy/zigpy/discussions/985
- [ ] https://github.com/zigpy/zigpy/discussions/949
- [ ] https://github.com/zigpy/zigpy/discussions/970

zigpy/bellows

- [ ] https://github.com/zigpy/bellows/issues/458
- [ ] https://github.com/zigpy/bellows/issues/465
- [ ] https://github.com/zigpy/bellows/issues/468

zigpy/znp
- [ ] https://github.com/zigpy/zigpy-znp/issues/155
- [ ] https://github.com/zigpy/zigpy-znp/issues/154
- [ ] https://github.com/zigpy/zigpy-znp/issues/156

zigpy/deconz

- [ ] https://github.com/zigpy/zigpy-deconz/issues/187
- [ ] https://github.com/zigbeefordomoticz/Domoticz-Zigbee/issues/1145
- [ ] https://github.com/zigpy/zigpy-deconz/issues/192

pipiche38, 2022-12-30:
up

Hedda, 2023-12-06:
> We are tracking here all of the open issues against zigpy and zigpy related radio modules, which are impacting the plugin
> 
> zigpy
> 
> * [ ]  [aligning ZNP and Bellows configuration zigpy/zigpy#1011](https://github.com/zigpy/zigpy/discussions/1011)
> * [ ]  [What handling recommandation when receiving  ControllerException zigpy/zigpy#1000](https://github.com/zigpy/zigpy/discussions/1000)

If you want these to be followed as issues by the upstream zigpy project then suggest (re)submitting them as new issues instead:

https://github.com/zigpy/zigpy/issues

Issues tend to be more formal tasks, feature requests and/or bug reports which can more easily be tracked with a status.

https://docs.github.com/en/issues/tracking-your-work-with-issues/about-issues

Discussions on GitHub are just an informal communication channel which are not tracked or followed up as issues are.

https://docs.github.com/en/discussions/guides/best-practices-for-community-conversations-on-github

https://docs.github.com/en/discussions/collaborating-with-your-community-using-discussions/about-discussions

Discussions can still be cross-referenced in issues and vice versa, and people with triage permission to a repository can even create an issue from a discussion, (but then you have to get a project admin to do it so probably easier to just create new issues).

https://docs.github.com/en/issues/tracking-your-work-with-issues/creating-an-issue#creating-an-issue-from-discussion

pipiche38, 2024-04-04:
new issue reported with zigpy 0.63.4 

https://github.com/zigpy/zigpy/discussions/1375
https://github.com/crate-py/rpds/issues/63