zigbeefordomoticz/Domoticz-Zigbee #1243: Error during firmware and plugin check

strethill, 2022-07-26:
I have a Zigate-USB v1 plugged in to an RPi 3b running Domoticz.
I have just updated the plugin to version 4.11.017 and the firmware to 3.21(ZiGate_v3.21-legacy.bin) and now get the error:

> _Error: Zigate: DNS error while checking Plugin and Firmware version: None of DNS query names exist: zigate.pipiche.net._

Should I set the firmware to a previous version? If so which version?

closed 2022-07-27 by pipiche38

strethill, 2022-07-27:
> You are running an unsupported version of the plugin. You need to upgrade to 6 . Here is the wiki to move from 5 to 6 https://zigbeefordomoticz.github.io/wiki/en-eng/Plugin_Version-6.html
> As moving from 4 to 6 this might work.
>
> Better to do backup of DomoticZ and plugin folder before starting

Many thanks for your help – this has fixed the issue.

 

I knew that I had got behind with my plugin and firmware updates but not quite as far two major versions.

 

Thanks again.

pipiche38, 2022-07-27:
Great news.
You'll see quiet a number of enhancements.