zigbeefordomoticz/Domoticz-Zigbee #1293: Improve OTA mecanism

pipiche38, 2022-09-28:
When a device is doing an OTA query we have information on Image Type, Manufacturer code, Image Version
it would be nice if we could check against the online  https://github.com/Koenkk/zigbee-OTA/blob/master/index.json

and in case we find a newest firmware , we could notify the user for him to download it and put it in the right place.

I do consider that a manual step (downloading) is required so it is a human action allowing the firmware update over the air with the corresponding and accepted risks

@koenkk I do hope you do not mind that we use your file online ?

pipiche38, 2022-09-28:
[image]

pipiche38, 2022-09-28:
- Loading the zigbee OTA repository once at plugin startup. Then every time there is an OTA query next image and we don't find an existing OTA file , we check against the loaded repository.

Hedda, 2022-09-29:
@pipiche38 Please also see this related zigpy discussion about OTA provider in zigpy for @Koenkk's repo -> https://github.com/zigpy/zigpy/issues/535

That is a discussion/suggestion/request for adding a GitHub OTA Firmware Provider to zigpy so that it can download files from GitHub repository if there is a formatted index such as Koenkk 's "index.json" to -> https://github.com/zigpy/zigpy/blob/dev/zigpy/ota/provider.py

There is as well an indirectly related zigpy discussion about ideas for GUI/UI experience for OTA FW in ZHA -> https://github.com/zigpy/zigpy/discussions/951

@Koenkk FYI, mentioned there that the lead developer of OpenHAB ZigBee Binding is also looking at getting Zigbee OTA firmware images from GitHub repository/repositories and specifically parsing https://github.com/Koenkk/zigbee-OTA/blob/master/index.json and grabbing OTA files from https://github.com/Koenkk/zigbee-OTA/tree/master/images

First, OpenHAB community they discussed that here. https://community.openhab.org/t/ikea-tradfri-firmware-updates/132042/63 

And now the same developer of [OpenHAB's ZigBee Binding](https://www.openhab.org/addons/bindings/zigbee/) (who is also the developer of [com.zsmartsystems.zigbee](https://github.com/zsmartsystems/com.zsmartsystems.zigbee)  https://github.com/zsmartsystems/com.zsmartsystems.zigbee/ Zigbee Cluster Library Java framework/libraries it depends on) is apparently working on a GitHub Zigbee Firmware Provider backed by GitHub repository/repositories, using a GitHub Library Reader to parse "index.json" and will by default point to Koenkk's zigbee-OTA repo.

See -> https://github.com/openhab/org.openhab.binding.zigbee/pull/767

> This PR implements a `FirmwareProvider` for Zigbee devices. It gets the firmware files from the web and caches them locally for use based on requests from the device. The provider gets the firmware IDs from the `ZigBeeThingHandler` based on these periodic requests from the device, and provides them as a list so the user can allow the upgrade via the standard OH firmware management system.
> 
> Requires an updated ZSS `1.4.7` library to provide callback when the client sends a request. This is a little cleaner than the other approach of adding a `ZclCommandListener` as it better integrates in with the server. Also requires further testing

PS: OpenHAB's Zigbee developers and users are in an OpenHAB forum thread also discuss GUI/UI management features for such Zigbee OTA firmware updates in the existing unified firmware management system in OpenHAB 3.x / OH3 which apparently already existed in OpenHAB 2.x / OH2) versions but that not yet been migrated to the web-GUI of OpenHAB 3.x as mentioned here, see:

https://community.openhab.org/t/ikea-tradfri-firmware-updates/132042/67

and

https://github.com/openhab/openhab-webui/issues/1427

> Firmware update management is not implemented in OH3:  The firmware API, which reads the list of available firmware for a thing, and allows the user to select a firmware, and then provides the status information when firmware is loading is not implemented in OH3. This is a core feature of OH so probably should have support included in the UI as it was in OH2.

closed 2022-09-30 by pipiche38