zigbeefordomoticz/Domoticz-Zigbee #1295: Tuya TS0601 smoke detector support

Achterhoeker, 2022-09-30:
I've tried to pair a tuya smoke detector. It seems not supported correctly, it is recognized as:

* zigbee - TS0601_Temp-a4c138740d8646ce-01
* zigbee - TS0601_ThermoSetpoint-a4c138740d8646ce-01
* zigbee - TS0601_ThermoMode_2-a4c138740d8646ce-01 

I would expect a smoke detector.

Log if i try to re-pair it again: 
2022-09-30 11:39:21.532  Status: zigbee: [None] NEW OBJECT: fc33 None
2022-09-30 11:39:24.168  Status: zigbee: [-] NEW OBJECT: fc33 Active Endpoint Response Ep: 01 LQI: 167
2022-09-30 11:39:24.172  Status: zigbee: [-] NEW OBJECT: fc33 Request Simple Descriptor for Ep: 01
2022-09-30 11:39:25.611  Status: zigbee: [-] NEW OBJECT: fc33 Simple Descriptor Response EP: 0x01 LQI: 163
2022-09-30 11:39:25.612  Status: zigbee: [-]    NEW OBJECT: fc33 ProfileID 0104
2022-09-30 11:39:25.613  Status: zigbee: [-]    NEW OBJECT: fc33 ZDeviceID 0051
2022-09-30 11:39:25.613  Status: zigbee: [-]    NEW OBJECT: fc33 Application Version 0001
2022-09-30 11:39:25.614  Status: zigbee: [-]    NEW OBJECT: fc33 Cluster IN Count: 04
2022-09-30 11:39:25.615  Status: zigbee: [-]       NEW OBJECT: fc33 Cluster In 1: 0004 (Groups)
2022-09-30 11:39:25.616  Status: zigbee: [-]       NEW OBJECT: fc33 Cluster In 2: 0005 (Scenes)
2022-09-30 11:39:25.617  Status: zigbee: [-]       NEW OBJECT: fc33 Cluster In 3: ef00
2022-09-30 11:39:25.617  Status: zigbee: [-]       NEW OBJECT: fc33 Cluster In 4: 0000 (Basic)
2022-09-30 11:39:25.618  Status: zigbee: [-]    NEW OBJECT: fc33 Cluster OUT Count: 02
2022-09-30 11:39:25.619  Status: zigbee: [-]       NEW OBJECT: fc33 Cluster Out 1: 0019 (Over-the-Air Upgrade)
2022-09-30 11:39:25.620  Status: zigbee: [-]       NEW OBJECT: fc33 Cluster Out 2: 000a (Time)
2022-09-30 11:39:27.118  Status: zigbee: [1] NEW OBJECT: fc33 Model Name: TS0601
2022-09-30 11:39:27.125  Status: zigbee: [1] NEW OBJECT: fc33 Manufacturer Name: _TZE200_ntcy3xu1
2022-09-30 11:39:27.126  Status: zigbee: [None] NEW OBJECT: fc33 Request Node Descriptor

Last device in raw device list:
[export.json.log](https://github.com/zigbeefordomoticz/Domoticz-Zigbee/files/9683122/export.json.log)

https://nl.aliexpress.com/item/1005004533181972.html

Thanks for your help,
Ronald.

pipiche38, 2022-09-30:
could you kindly:

1- remove the object from Domoticz and plugin Db by removing all Domoticz widget associated to this Smoke sensor
2- switch to beta6 branch of the plugin. In case you are on Stable6 version, please follow this https://github.com/zigbeefordomoticz/Domoticz-Zigbee/blob/beta6/VERSION-6.3.md#pre-requisities as there is a configuration update that you have to do in Domoticz for the plugin
3- restart Domoticz 
4- redo the pairing

And let me know. In all cases I would be interested to have the export.json after pairing and after few tests.

pipiche38, 2022-10-03:
@Achterhoeker have you been able to test ?

Achterhoeker, 2022-10-03:
@pipiche38  Sorry, I was not able yet to test the beta plugin. I hope that I have some time this evening.

Achterhoeker, 2022-10-03:
@pipiche38 Great work, i think i have a working smoke detector now :-)
I believe that there was still some information from the old pairing device, but i see now a smoke detector in domoticz that is correctly updated with an alarm status.

The log from the pairing:

> 2022-10-03 20:45:32.913  Status: zigbee: Plugin with Zigpy, Coordinator CC1352/CC2652, Z-Stack 3.30+ (znp) firmware Znp - (....0219) correctly initialized
> 2022-10-03 20:47:02.344  Status: zigbee: [None] NEW OBJECT: 1c38 None
> 2022-10-03 20:47:05.043  Status: zigbee: [-] NEW OBJECT: 1c38 Active Endpoint Response Ep: 01 LQI: 182
> 2022-10-03 20:47:05.047  Status: zigbee: [-] NEW OBJECT: 1c38 Request Simple Descriptor for Ep: 01
> 2022-10-03 20:47:06.390  Status: zigbee: [-] NEW OBJECT: 1c38 Simple Descriptor Response EP: 0x01 LQI: 182
> 2022-10-03 20:47:06.391  Status: zigbee: [-]    NEW OBJECT: 1c38 ProfileID 0104
> 2022-10-03 20:47:06.392  Status: zigbee: [-]    NEW OBJECT: 1c38 ZDeviceID 0051
> 2022-10-03 20:47:06.393  Status: zigbee: [-]    NEW OBJECT: 1c38 Application Version 0001
> 2022-10-03 20:47:06.400  Status: zigbee: [-]    NEW OBJECT: 1c38 Cluster IN Count: 04
> 2022-10-03 20:47:06.403  Status: zigbee: [-]       NEW OBJECT: 1c38 Cluster In 1: 0004 (Groups)
> 2022-10-03 20:47:06.404  Status: zigbee: [-]       NEW OBJECT: 1c38 Cluster In 2: 0005 (Scenes)
> 2022-10-03 20:47:06.405  Status: zigbee: [-]       NEW OBJECT: 1c38 Cluster In 3: ef00
> 2022-10-03 20:47:06.407  Status: zigbee: [-]       NEW OBJECT: 1c38 Cluster In 4: 0000 (Basic)
> 2022-10-03 20:47:06.410  Status: zigbee: [-]    NEW OBJECT: 1c38 Cluster OUT Count: 02
> 2022-10-03 20:47:06.411  Status: zigbee: [-]       NEW OBJECT: 1c38 Cluster Out 1: 0019 (Over-the-Air Upgrade)
> 2022-10-03 20:47:06.412  Status: zigbee: [-]       NEW OBJECT: 1c38 Cluster Out 2: 000a (Time)
> 2022-10-03 20:47:07.110  Status: zigbee: [0] NEW OBJECT: 1c38 Model Name: TS0601-smoke
> 2022-10-03 20:47:07.116  Status: zigbee: [0] NEW OBJECT: 1c38 Manufacturer Name: _TZE200_ntcy3xu1
> 2022-10-03 20:47:07.123  Status: zigbee: [None] NEW OBJECT: 1c38 Request Node Descriptor
> 2022-10-03 20:47:39.388  Status: zigbee:  (1c38/a4c138740d8646ce) send a Leave indication and will be outside of the network. LQI: 0
> 2022-10-03 20:47:57.882  zigbee: Enable Assisted pairing
> 2022-10-03 20:47:57.888  Status: zigbee: Request Accepting new Hardware for 240 seconds
> 2022-10-03 20:47:57.937  zigbee: Assisted Pairing: Polling: []
> 2022-10-03 20:47:57.937  zigbee: --> Empty queue
> 2022-10-03 20:47:57.999  Status: zigbee: Accepting new Hardware: Enable (On)
> 2022-10-03 20:48:03.667  Status: zigbee: reconnectNWkDevice - Update Status from inDB to 'inDB' for NetworkID : 8945
> 2022-10-03 20:48:03.961  Status: zigbee: NetworkID: 8945 is replacing 1c38 for object: a4c138740d8646ce
> 2022-10-03 20:48:05.457  zigbee: UpdateDevice - (zigbee - TS0601-smoke_Smoke-a4c138740d8646ce-01) 1:On
> 2022-10-03 20:48:05.472  zigbee: UpdateDevice - (zigbee - TS0601-smoke_Tamper-a4c138740d8646ce-01) 1:Tamper

Raw devices .json list:
[export_.zip](https://github.com/zigbeefordomoticz/Domoticz-Zigbee/files/9700264/export_.zip)

pipiche38, 2022-10-04:
Happy man.

Could you kindly let me know which one is it ?
https://zigbee.blakadder.com/Tuya_SA12IZL.html
https://zigbee.blakadder.com/Tuya_YG400A.html
https://zigbee.blakadder.com/Tuya_Qaynn.html
https://zigbee.blakadder.com/Woox_R7049.html
https://zigbee.blakadder.com/Smart9_S9ZB-SA02.html

So I can add the plugin to that list .

Thanks in advance

Achterhoeker, 2022-10-04:
@pipiche38 It is this one: https://zigbee.blakadder.com/Tuya_YG400A.html

pipiche38, 2022-10-04:
Which version of Domoticz are you running ? Are you on 2022.1 or above , or on 2021 ?
In any case, update to the beta6 of today, I just push fix which should make the battery working for all Domoticz version, but you'll need to wait for an update from the device (like a status telling no smoke detected )

Achterhoeker, 2022-10-04:
I've updated to the latest version, i see now correctly a battery status 90% 

Thanks for all you work.

pipiche38, 2022-10-05:
I guess we can close the Issue then.
Thanks for your time and sharing the data

closed 2022-10-05 by pipiche38