zigbeefordomoticz/Domoticz-Zigbee #1300: Error on startup and plugin stops

mcmikev, 2022-10-09:
Error on startup as with updated beta6 plugin on october 9 2022
Version :

```
 2022-10-09 10:35:08.552 Error: Zigbee: Error at startup Expecting value: line 1 column 1 (char 0)
2022-10-09 10:35:08.779 Error: Zigbee: [print_stack | /home/pi/domoticz/plugins/Domoticz-Zigbee/Modules/tools.py:1553
2022-10-09 10:35:08.784 Error: Zigbee: [radio_start | /home/pi/domoticz/plugins/Domoticz-Zigbee/Classes/ZigpyTransport/zigpyThread.py:245
2022-10-09 10:35:08.786 Error: Zigbee: [_run | /usr/lib/python3.7/asyncio/events.py:88
2022-10-09 10:35:08.789 Error: Zigbee: [_run_once | /usr/lib/python3.7/asyncio/base_events.py:1775
2022-10-09 10:35:08.791 Error: Zigbee: [run_forever | /usr/lib/python3.7/asyncio/base_events.py:539
2022-10-09 10:35:08.793 Error: Zigbee: [run_until_complete | /usr/lib/python3.7/asyncio/base_events.py:571
2022-10-09 10:35:08.796 Error: Zigbee: [zigpy_thread | /home/pi/domoticz/plugins/Domoticz-Zigbee/Classes/ZigpyTransport/zigpyThread.py:85
2022-10-09 10:35:08.799 Error: Zigbee: [run | /usr/lib/python3.7/threading.py:865
2022-10-09 10:35:08.801 Error: Zigbee: [_bootstrap_inner | /usr/lib/python3.7/threading.py:917
2022-10-09 10:35:08.804 Error: Zigbee: [_bootstrap | /usr/lib/python3.7/threading.py:885
2022-10-09 10:36:55.096 Error: Zigbee: [120] I have hard time to get Coordinator Version. Mostlikly there is a communication issue
2022-10-09 10:37:05.111 Error: Zigbee: [130] I have hard time to get Coordinator Version. Mostlikly there is a communication issue
2022-10-09 10:37:15.075 Error: Zigbee: [140] I have hard time to get Coordinator Version. Mostlikly there is a communication issue
2022-10-09 10:37:25.091 Error: Zigbee: [150] I have hard time to get Coordinator Version. Mostlikly there is a communication issue
2022-10-09 10:37:35.075 Error: Zigbee: [160] I have hard time to get Coordinator Version. Mostlikly there is a communication issue
2022-10-09 10:37:36.113 Error: Zigbee: [ ] Stopping the plugin and lease do check the Coordinator connectivity. 
```

mcmikev, 2022-10-09:
Update: After the plugin stopped itself , I left domoticz running for a few minutes to get stable in running.
The hardware of the plugin was Off in domoticz, I enabled it again and pressed update and now the plugin is running again.
Weird. Almost like the plugin was doing something in the background before it was allowed to start

pipiche38, 2022-12-22:
suggest to close. please re-open in case it comes again

closed 2022-12-22 by pipiche38