zigbeefordomoticz/Domoticz-Zigbee #1348: [Model Certification]

deennoo, 2022-11-18:
1. Please do provide device information such as Brand and Model name.
WWWC LED STRIP DRIVER 

2. Provide the URL where you have purchase the device
https://fr.aliexpress.com/item/1005003938851288.html?spm=a2g0o.order_list.0.0.58545e5b1lcKu2&gatewayAdapt=glo2fra

3. Finaly copy-paste all information provided in the log

{"Version": "3", "ZDeviceName": "Sous Evier", "Status": "inDB", "SQN": "7e", "Ep": {"01": {"0000": {"0004": "eWeLight", "0005": "ZB-CT01", "0000": "3", "0001": "3", "0002": "2", "0003": "0", "0006": "20211229", "0007": "1", "4000": "2.0.1"}, "0003": {}, "0004": {}, "0005": {}, "0006": {"0000": "00", "4001": "0", "4002": "0"}, "0008": {"0000": "7c"}, "0300": {"0000": "0", "0001": "0", "0003": "24939", "0004": "24701", "0007": "454", "0008": "2", "400a": "16"}, "1000": {}, "ClusterType": {"26422": "ColorControlWW"}}}, "Heartbeat": "240", "RIA": "4", "LQI": 105, "Battery": {}, "Model": "ZB-CT01", "ForceAckCommands": [], "MacCapa": "8e", "IEEE": "a4c1389a81dbff45", "Type": "Switch/LvlControl/ColorControl", "ProfileID": "0104", "ZDeviceID": "010c", "App Version": "3", "Attributes List": {"Request": {"01": {"0000": 1668642177.1142936, "0003": 1668642177.1143613, "0004": 1668642177.1144233, "0005": 1668642177.1144845, "0006": 1668642177.1145456, "0008": 1668642177.1146162, "0300": 1668642177.1146774, "1000": 1668642177.1148932}}}, "DeviceType": "FFD", "HW Version": "0", "Last Cmds": [], "LogicalType": "Router", "Manufacturer": "100b", "Manufacturer Name": "eWeLight", "NbEp": "1", "PowerSource": "Main", "ReadAttributes": {"Ep": {"01": {"0000": {"TimeStamp": 1668601231, "iSQN": {"0004": 1224, "0005": 1224, "0000": 1224, "0001": 1224, "0002": 1224, "0003": 1225, "0006": 1225, "0007": 1225, "000a": 1225, "000f": 1225, "0010": 1226, "0015": 1226, "4000": 1226, "f000": 1226}, "Attributes": {"0004": "00", "0005": "00", "0000": "00", "0001": "00", "0002": "00", "0003": "00", "0006": "00", "0007": "00", "000a": "86", "000f": "86", "0010": "86", "0015": "86", "4000": "00", "f000": "86"}, "ZigateRequest": {}}, "0006": {"TimeStamp": 1668776622, "iSQN": {"4001": 1220, "4002": 1220, "0000": 16298}, "Attributes": {"4001": "00", "4002": "00", "0000": "00"}, "ZigateRequest": {}}, "0008": {"TimeStamp": 1668776406, "iSQN": {"0000": 16299}, "Attributes": {"0000": "00"}, "ZigateRequest": {}}, "0300": {"TimeStamp": 1668601230, "iSQN": {"0000": 1222, "0001": 1222, "0003": 1222, "0004": 1222, "0007": 1222, "0008": 1223, "400a": 1223}, "Attributes": {"0000": "00", "0001": "00", "0003": "00", "0004": "00", "0007": "00", "0008": "00", "400a": "00"}, "ZigateRequest": {}}}}}, "ReceiveOnIdle": "On", "Stack Version": "2", "Stamp": {"time": 1668776622.594148, "Time": "2022-11-18 14:03:42", "MsgType": "8102", "LastSeen": 1668776622}, "ZCL Version": "3", "Health": "Live", "Epv2": {"01": {"ClusterIn": {"0000": {}, "0003": {}, "0004": {}, "0005": {}, "0006": {}, "0008": {}, "0300": {}, "1000": {}}, "ClusterOut": {}, "ProfileID": "0104", "ZDeviceID": "010c"}}, "ConfigSource": "8043", "Capability": ["Full-Function Device", "Main Powered", "Receiver during Idle", "Standard security", "NwkAddr should be allocated"], "ConfigureReporting": {"Ep": {"01": {"0006": {"TimeStamp": 1668601228.092499, "iSQN": {"0000": 1217}, "Attributes": {"0000": "00"}, "ZigateRequest": {}}, "0008": {"TimeStamp": 1668601228.093898, "iSQN": {"0000": 1218}, "Attributes": {"0000": "00"}, "ZigateRequest": {}}, "0300": {"TimeStamp": 1668601228.0959618, "iSQN": {"0008": 1219}, "Attributes": {"0008": "86"}, "ZigateRequest": {}}}}}, "Param": {}, "GroupMemberShip": {"01": {"0002": {"Status": "OK"}}}, "SWBUILD_3": "2.0.1", "Bind": {"01": {"0000": {"Target": "0000", "Stamp": 1668601228, "Phase": "binded", "Status": "00", "i_sqn": 1208}, "0003": {"Target": "0000", "Stamp": 1668601228, "Phase": "binded", "Status": "00", "i_sqn": 1209}, "0004": {"Target": "0000", "Stamp": 1668601228, "Phase": "binded", "Status": "00", "i_sqn": 1210}, "0005": {"Target": "0000", "Stamp": 1668601228, "Phase": "binded", "Status": "00", "i_sqn": 1211}, "0006": {"Target": "0000", "Stamp": 1668601228, "Phase": "binded", "Status": "00", "i_sqn": 1212}, "0008": {"Target": "0000", "Stamp": 1668601229, "Phase": "binded", "Status": "00", "i_sqn": 1213}, "0300": {"Target": "0000", "Stamp": 1668601229, "Phase": "binded", "Status": "00", "i_sqn": 1214}, "1000": {"Target": "0000", "Stamp": 1668601229, "Phase": "binded", "Status": "00", "i_sqn": 1215}}}, "ReadConfigureReporting": {"Ep": {"01": {"0006": {"0000": {"TimeStamp": 1668601231.5245566, "Status": "00", "DataType": "10", "MinInterval": "0001", "MaxInterval": "012c"}}, "0008": {"0000": {"TimeStamp": 1668601231.6772392, "Status": "00", "DataType": "20", "MinInterval": "0005", "MaxInterval": "012c"}}}}, "TimeStamp": 1668601231.6772318}, "WriteAttributes": {"Ep": {"01": {"0000": {"TimeStamp": 0, "iSQN": {}, "Attributes": {}, "ZigateRequest": {}}, "0003": {"TimeStamp": 0, "iSQN": {}, "Attributes": {}, "ZigateRequest": {}}, "0004": {"TimeStamp": 0, "iSQN": {}, "Attributes": {}, "ZigateRequest": {}}, "0005": {"TimeStamp": 0, "iSQN": {}, "Attributes": {}, "ZigateRequest": {}}, "0006": {"TimeStamp": 0, "iSQN": {}, "Attributes": {}, "ZigateRequest": {}}, "0008": {"TimeStamp": 0, "iSQN": {}, "Attributes": {}, "ZigateRequest": {}}, "0300": {"TimeStamp": 0, "iSQN": {}, "Attributes": {}, "ZigateRequest": {}}, "1000": {"TimeStamp": 0, "iSQN": {}, "Attributes": {}, "ZigateRequest": {}}}}}, "BindingTable": {"SQN": 0, "Status": "Success", "TimeStamp": 1668601232.1907868, "BindingTableSize": 8, "Devices": [{"a4c1389a81dbff45": {"sourceEp": "01", "Cluster": "0006", "targetIEEE": "00158d0002bcfce3", "targetNickName": "Zigate", "targetEp": "01"}}, {"a4c1389a81dbff45": {"sourceEp": "01", "Cluster": "0000", "targetIEEE": "00158d0002bcfce3", "targetNickName": "Zigate", "targetEp": "01"}}, {"a4c1389a81dbff45": {"sourceEp": "01", "Cluster": "0003", "targetIEEE": "00158d0002bcfce3", "targetNickName": "Zigate", "targetEp": "01"}}, {"a4c1389a81dbff45": {"sourceEp": "01", "Cluster": "0004", "targetIEEE": "00158d0002bcfce3", "targetNickName": "Zigate", "targetEp": "01"}}, {"a4c1389a81dbff45": {"sourceEp": "01", "Cluster": "0005", "targetIEEE": "00158d0002bcfce3", "targetNickName": "Zigate", "targetEp": "01"}}, {"a4c1389a81dbff45": {"sourceEp": "01", "Cluster": "0008", "targetIEEE": "00158d0002bcfce3", "targetNickName": "Zigate", "targetEp": "01"}}, {"a4c1389a81dbff45": {"sourceEp": "01", "Cluster": "0300", "

pipiche38, 2022-11-18:
invalid json, can you kindly drop the export.json file

pipiche38, 2022-11-18:
to be tested

closed 2022-11-18 by pipiche38