zigbeefordomoticz/Domoticz-Zigbee #1349: [Model Certification]

deennoo, 2022-11-18:
1. Please do provide device information such as Brand and Model name.
Gateway to drive 2.4ghz light from L&S for Emotion Franchfurt Light Series with zigbee coordinator

L&S Emotion Home Base 

2. Provide the URL where you have purchase the device

https://www.opo.ch/fr/home-base-ls-emotion-gateway-12-24-v/p/bp5059230.html

3. Finaly copy-paste all information provided in the log

{"Version": "3", "ZDeviceName": "Sur Evier", "Status": "inDB", "SQN": "0d", "Ep": {"01": {"0000": {"0004": "LS Deutschland GmbH", "0005": "Emotion", "0000": "3", "0001": "0", "0002": "0", "0003": "1", "0006": "", "0007": "4", "000a": "", "4000": "1.0.0_r00000"}, "0003": {}, "0004": {}, "0005": {}, "0006": {"0000": "00", "4001": "0", "4002": "0"}, "0008": {"0000": "79"}, "0300": {"0007": "241", "0008": "2", "400a": "16"}, "ClusterType": {"26428": "ColorControlWW"}}, "f2": {"0021": {}}}, "Heartbeat": "240", "RIA": "4", "LQI": 120, "Battery": {}, "Model": "Emotion", "ForceAckCommands": [], "MacCapa": "8e", "IEEE": "b43a31fffe778d20", "Type": "Switch/LvlControl/ColorControl", "ProfileID": "a1e0", "ZDeviceID": "0061", "App Version": "0", "Attributes List": {"Request": {"01": {"0000": 1668642177.1161082, "0003": 1668642177.1161726, "0004": 1668642177.1162357, "0005": 1668642177.1163075, "0006": 1668642177.1163702, "0008": 1668642177.1164322, "0300": 1668642177.116493}, "f2": {"0021": 1668642177.1165679}}}, "DeviceType": "FFD", "HW Version": "1", "Last Cmds": [], "LogicalType": "Router", "Manufacturer": "124e", "Manufacturer Name": "LS Deutschland GmbH", "NbEp": "2", "PowerSource": "Main", "ReadAttributes": {"Ep": {"01": {"0000": {"TimeStamp": 1668614210, "iSQN": {"0004": 1291, "0005": 1291, "0000": 1291, "0001": 1292, "0002": 1292, "0003": 1292, "0006": 1293, "0007": 1293, "000a": 1293, "000f": 1294, "0010": 1294, "0015": 1294, "4000": 1295, "f000": 1295}, "Attributes": {"0004": "00", "0005": "00", "0000": "00", "0001": "00", "0002": "00", "0003": "00", "0006": "00", "0007": "00", "000a": "00", "000f": "86", "0010": "86", "0015": "86", "4000": "00", "f000": "86"}, "ZigateRequest": {}}, "0006": {"TimeStamp": 1668776629, "iSQN": {"4001": 1286, "4002": 1286, "0000": 16302}, "Attributes": {"4001": "00", "4002": "00", "0000": "00"}, "ZigateRequest": {}}, "0008": {"TimeStamp": 1668776630, "iSQN": {"0000": 16303}, "Attributes": {"0000": "00"}, "ZigateRequest": {}}, "0300": {"TimeStamp": 1668747460, "iSQN": {"0000": 1288, "0001": 1288, "0003": 1288, "0004": 1289, "0007": 1289, "0008": 1289, "400a": 1290}, "Attributes": {"0000": "86", "0001": "86", "0003": "86", "0004": "86", "0007": "00", "0008": "00", "400a": "00"}, "ZigateRequest": {}}}, "f2": {"0000": {"TimeStamp": 0, "iSQN": {}, "Attributes": {}, "ZigateRequest": {}}}}}, "ReceiveOnIdle": "On", "Stack Version": "0", "Stamp": {"time": 1668776630.4394205, "Time": "2022-11-18 14:03:50", "MsgType": "8011", "LastSeen": 1668776630}, "ZCL Version": "3", "Health": "Live", "Epv2": {"01": {"ClusterIn": {"0000": {}, "0003": {}, "0004": {}, "0005": {}, "0006": {}, "0008": {}, "0300": {}}, "ClusterOut": {}, "ProfileID": "0104", "ZDeviceID": "010c"}, "f2": {"ClusterIn": {}, "ClusterOut": {"0021": {}}, "ProfileID": "a1e0", "ZDeviceID": "0061"}}, "ConfigSource": "8043", "Capability": ["Full-Function Device", "Main Powered", "Receiver during Idle", "Standard security", "NwkAddr should be allocated"], "ConfigureReporting": {"Ep": {"01": {"0006": {"TimeStamp": 1668614207.122511, "iSQN": {"0000": 1283}, "Attributes": {"0000": "00"}, "ZigateRequest": {}}, "0008": {"TimeStamp": 1668614207.122795, "iSQN": {"0000": 1284}, "Attributes": {"0000": "00"}, "ZigateRequest": {}}, "0300": {"TimeStamp": 1668614207.123027, "iSQN": {"0008": 1285}, "Attributes": {"0008": "00"}, "ZigateRequest": {}}}}}, "Param": {}, "GroupMemberShip": {"01": {"0002": {"Status": "OK"}}}, "SWBUILD_3": "1.0.0_r00000", "Bind": {"01": {"0000": {"Target": "0000", "Stamp": 1668614207, "Phase": "binded", "Status": "00", "i_sqn": 1275}, "0003": {"Target": "0000", "Stamp": 1668614207, "Phase": "binded", "Status": "00", "i_sqn": 1276}, "0004": {"Target": "0000", "Stamp": 1668614207, "Phase": "binded", "Status": "00", "i_sqn": 1277}, "0005": {"Target": "0000", "Stamp": 1668614207, "Phase": "binded", "Status": "00", "i_sqn": 1278}, "0006": {"Target": "0000", "Stamp": 1668614207, "Phase": "binded", "Status": "00", "i_sqn": 1279}, "0008": {"Target": "0000", "Stamp": 1668614208, "Phase": "binded", "Status": "00", "i_sqn": 1280}, "0300": {"Target": "0000", "Stamp": 1668614208, "Phase": "binded", "Status": "00", "i_sqn": 1281}}}, "ReadConfigureReporting": {"Ep": {"01": {"0006": {"0000": {"TimeStamp": 1668614210.8800304, "Status": "00", "DataType": "10", "MinInterval": "0001", "MaxInterval": "012c"}}, "0008": {"0000": {"TimeStamp": 1668614211.0331244, "Status": "00", "DataType": "20", "MinInterval": "0005", "MaxInterval": "012c"}}, "0300": {"0008": {"TimeStamp": 1668614211.1860402, "Status": "00", "DataType": "30", "MinInterval": "0001", "MaxInterval": "012c"}}}}, "TimeStamp": 1668614211.186033}, "WriteAttributes": {"Ep": {"01": {"0000": {"TimeStamp": 0, "iSQN": {}, "Attributes": {}, "ZigateRequest": {}}, "0003": {"TimeStamp": 0, "iSQN": {}, "Attributes": {}, "ZigateRequest": {}}, "0004": {"TimeStamp": 0, "iSQN": {}, "Attributes": {}, "ZigateRequest": {}}, "0005": {"TimeStamp": 0, "iSQN": {}, "Attributes": {}, "ZigateRequest": {}}, "0006": {"TimeStamp": 0, "iSQN": {}, "Attributes": {}, "ZigateRequest": {}}, "0008": {"TimeStamp": 0, "iSQN": {}, "Attributes": {}, "ZigateRequest": {}}, "0300": {"TimeStamp": 0, "iSQN": {}, "Attributes": {}, "ZigateRequest": {}}}, "f2": {"0021": {"TimeStamp": 0, "iSQN": {}, "Attributes": {}, "ZigateRequest": {}}}}}, "BindingTable": {"SQN": 0, "Status": "Success", "TimeStamp": 1668614211.3412542, "BindingTableSize": 7, "Devices": [{"b43a31fffe778d20": {"sourceEp": "01", "Cluster": "0000", "targetIEEE": "00158d0002bcfce3", "targetNickName": "Zigate", "targetEp": "01"}}, {"b43a31fffe778d20": {"sourceEp": "01", "Cluster": "0003", "targetIEEE": "00158d0002bcfce3", "targetNickName": "Zigate", "targetEp": "01"}}, {"b43a31fffe778d20": {"sourceEp": "01", "Cluster": "0004", "targetIEEE": "00158d0002bcfce3", "targetNickName": "Zigate", "targetEp": "01"}}, {"b43a31fffe778d20": {"sourceEp": "01", "Cluster": "0005", "targetIEEE": "00158d0002bcfce3", "targetNickName": "Zigate", "targetEp": "01"}}, {"b43a31fffe778d20": {"sourceEp": "01", "Cluster": "0006", "targetIEEE": "00158d0002bcfce3", "targetNickName": "Zigate", "targetEp": "01"}}

pipiche38, 2022-11-18:
invalid Json. Can you kindly upload the export.json ?

pipiche38, 2022-11-18:
Would be interested to redo the pairing with the config file now. And test the impact of the parameter PowerOnAfterOffOn
255 -> nack to previous state after electrical Off On
0 -> turn to Off
1 -> turn to On