zigbeefordomoticz/Domoticz-Zigbee #1350: [Model Certification] -

deennoo, 2022-11-18:
1. Please do provide device information such as Brand and Model name.

Muller Licht Tint Extended Color from ALDI Smart Home Solution

2. Provide the URL where you have purchase the device

https://www.mueller-licht.de/tint-smartes-licht/produktvielfalt/

3. Finaly copy-paste all information provided in the log

{"Version": "3", "ZDeviceName": "Barre Bureau", "Status": "inDB", "SQN": "4c", "Ep": {"01": {"0000": {"0004": "MLI", "0005": "tint-ExtendedColor"}, "0003": {}, "0004": {}, "0005": {}, "0006": {"0000": "01"}, "0008": {"0000": "fe"}, "0300": {}, "1000": {}, "100f": {}, "0019": {}, "ClusterType": {"25961": "ColorControlRGBWW"}}, "f2": {}}, "Heartbeat": "115", "RIA": "4", "LQI": 0, "Battery": {}, "Model": "tint-ExtendedColor", "ForceAckCommands": [], "MacCapa": "8e", "IEEE": "a4c13800aa301221", "Type": "Switch/LvlControl/ColorControl", "ProfileID": "0104", "ZDeviceID": "010d", "App Version": "", "Attributes List": {"Request": {"01": {"0000": 1667513991.0875313, "0003": 1667513991.0883458, "0004": 1667513991.0890052, "0005": 1667513991.0896063, "0006": 1667513991.0902236, "0008": 1667513991.0908165, "0300": 1667513991.0914054, "1000": 1667513991.0920258, "100f": 1667513991.0926375, "0019": 1667513991.0932379}, "f2": {}}}, "DeviceType": "FFD", "HW Version": "", "Last Cmds": [], "LogicalType": "Router", "Manufacturer": "121b", "Manufacturer Name": "MLI", "NbEp": "2", "PowerSource": "Main", "ReadAttributes": {"Ep": {"01": {"0000": {"TimeStamp": 1667489884, "iSQN": {"0004": "0a", "0005": "0a", "0000": "0a", "0001": "0a", "0002": "0a", "0003": "0b", "0006": "0b", "0007": "0b", "000a": "0b", "000f": "0b", "0010": "0c", "0015": "0c", "4000": "0c", "f000": "0c"}, "Attributes": {}, "ZigateRequest": {}}, "0019": {"TimeStamp": 1667489884, "iSQN": {"0001": "04", "0002": "04", "0003": "04", "0004": "04", "0005": "04", "0006": "05", "0007": "05", "0008": "05", "0009": "05", "000a": "05"}, "Attributes": {}, "ZigateRequest": {}}, "0006": {"TimeStamp": 1668776026, "iSQN": {"4001": "06", "4002": "06", "0000": "7e"}, "Attributes": {"0000": "00"}, "ZigateRequest": {}}, "0008": {"TimeStamp": 1668776026, "iSQN": {"0000": "7f"}, "Attributes": {"0000": "00"}, "ZigateRequest": {}}, "0300": {"TimeStamp": 1667489884, "iSQN": {"0000": "08", "0001": "08", "0003": "08", "0004": "08", "0007": "08", "0008": "09", "400a": "09"}, "Attributes": {}, "ZigateRequest": {}}}, "f2": {"0000": {"TimeStamp": 0, "iSQN": {}, "Attributes": {}, "ZigateRequest": {}}}}}, "ReceiveOnIdle": "On", "Stack Version": "", "Stamp": {"time": 1668776558.0788128, "Time": "2022-11-18 14:02:38", "MsgType": "8002", "LastSeen": 1668776558}, "ZCL Version": "", "Health": "", "RoutingTable": [], "OTAClient": {"ManufacturerCode": "121b", "ImageType": "020e", "CurrentImageVersion": "10012001"}, "Epv2": {"01": {"ClusterIn": {"0000": {}, "0003": {}, "0004": {}, "0005": {}, "0006": {}, "0008": {}, "0300": {}, "1000": {}, "100f": {}}, "ClusterOut": {"0019": {}}, "ProfileID": "0104", "ZDeviceID": "010d"}, "f2": {"ClusterIn": {}, "ClusterOut": {}, "ProfileID": {}, "ZDeviceID": {}}}, "ConfigSource": "8043", "Capability": ["Full-Function Device", "Main Powered", "Receiver during Idle", "Standard security", "NwkAddr should be allocated"], "AssociatedDevices": [{"Devices": ["adcd", "58c6", "de0a"], "SQN": 0, "Status": "", "TimeStamp": 1668603440, "Time": 1668603440}], "ConfigureReporting": {"Ep": {"01": {"0006": {"TimeStamp": 1667489884.096609, "iSQN": {"0000": "01"}, "Attributes": {}, "ZigateRequest": {}}, "0008": {"TimeStamp": 1667489884.0972805, "iSQN": {"0000": "02"}, "Attributes": {}, "ZigateRequest": {}}, "0300": {"TimeStamp": 1667489884.0976818, "iSQN": {"0008": "03"}, "Attributes": {}, "ZigateRequest": {}}}}}, "Param": {}, "GroupMemberShip": {"01": {}}, "Bind": {"01": {"0000": {"Target": "0000", "Stamp": 1667489884, "Phase": "binded", "Status": "00", "i_sqn": "03"}, "0003": {"Target": "0000", "Stamp": 1667489884, "Phase": "requested", "Status": "", "i_sqn": "04"}, "0004": {"Target": "0000", "Stamp": 1667489884, "Phase": "requested", "Status": "", "i_sqn": "05"}, "0005": {"Target": "0000", "Stamp": 1667489884, "Phase": "requested", "Status": "", "i_sqn": "06"}, "0006": {"Target": "0000", "Stamp": 1667489884, "Phase": "requested", "Status": "", "i_sqn": "07"}, "0008": {"Target": "0000", "Stamp": 1667489884, "Phase": "requested", "Status": "", "i_sqn": "08"}, "0300": {"Target": "0000", "Stamp": 1667489884, "Phase": "requested", "Status": "", "i_sqn": "09"}, "1000": {"Target": "0000", "Stamp": 1667489884, "Phase": "requested", "Status": "", "i_sqn": "0a"}, "100f": {"Target": "0000", "Stamp": 1667489884, "Phase": "requested", "Status": "", "i_sqn": "0b"}}}, "WriteAttributes": {"Ep": {"01": {"0000": {"TimeStamp": 0, "iSQN": {}, "Attributes": {}, "ZigateRequest": {}}, "0003": {"TimeStamp": 0, "iSQN": {}, "Attributes": {}, "ZigateRequest": {}}, "0004": {"TimeStamp": 0, "iSQN": {}, "Attributes": {}, "ZigateRequest": {}}, "0005": {"TimeStamp": 0, "iSQN": {}, "Attributes": {}, "ZigateRequest": {}}, "0006": {"TimeStamp": 0, "iSQN": {}, "Attributes": {}, "ZigateRequest": {}}, "0008": {"TimeStamp": 0, "iSQN": {}, "Attributes": {}, "ZigateRequest": {}}, "0300": {"TimeStamp": 0, "iSQN": {}, "Attributes": {}, "ZigateRequest": {}}, "1000": {"TimeStamp": 0, "iSQN": {}, "Attributes": {}, "ZigateRequest": {}}, "100f": {"TimeStamp": 0, "iSQN": {}, "Attributes": {}, "ZigateRequest": {}}, "0019": {"TimeStamp": 0, "iSQN": {}, "Attributes": {}, "ZigateRequest": {}}}}}, "BindingTable": {"SQN": 0, "Status": "Requested", "TimeStamp": 1667514031.0932229, "BindingTableSize": 0, "Devices": []}, "NeighbourTableSize": 19, "Neighbours": [{"SQN": 0, "Status": "", "TimeStamp": 1668603440, "Time": 1668603440, "Devices": [{"de0a": {"_relationshp": "Parent", "_devicetype": "Router", "_depth": 1, "_lnkqty": 104, "_rxonwhenidl": "Rx-On", "_IEEE": "d0cf5efffe26a824", "_permitjnt": "??"}}, {"32d6": {"_relationshp": "Child", "_devicetype": "Router", "_depth": 1, "_lnkqty": 153, "_rxonwhenidl": "Rx-On", "_IEEE": "2c1165fffe22140e", "_permitjnt": "??"}}, {"9da3": {"_relationshp": "Sibling", "_devicetype": "Router", "_depth": 2, "_lnkqty": 136, "_rxonwhenidl": "Rx-On", "_IEEE": "588e81fffedb8802", "_permitjnt": "??"}}, {"9fdf": {"_relationshp": "Sibling", "_devicetype": "Router", "_depth": 1, "_lnkqty": 153, "_rxonwhenidl": "Rx-On", "_IEEE": "842e14fffefd982e", "_permitjnt": "??"}}, {"dd61": {"_relationshp": "Sibling", "_devicetype": "Rout

pipiche38, 2022-11-18:
invalid json, can you kindly drop the export.json

deennoo, 2022-11-30:
on plugin 6.4.028 : 

Device is know on certified product

fadingOff': 0 doesn't work (not sure device can do this) but if device is set to level 0 and then set to level xx move to level
moveToHueSatu' : is working
moveToColourTemp':  : is working
moveToColourRGB' : is working
moveToLevel' : is working

pipiche38, 2022-11-30:
Strange that fadingOff do not work, while the other does.

closed 2022-11-30 by pipiche38