zigbeefordomoticz/Domoticz-Zigbee #1353: write_DeviceList_txt fails with strange characters (exceptions)

Achterhoeker, 2022-11-22:
Running domoticz 2022.2 and 6.3.006

After running for several hours i get error notifications in the domoticz log. Restarting domoticz will cause missing some devices in the Devicelist.txt. Replacing a backup solves this. But after several hours the same issue occurs again (with a different character causing the issue) 

error log: 
2022-11-21 13:31:13.558 Error: zigbee: Call to function 'onHeartbeat' failed, exception details:
2022-11-21 13:31:13.558 Error: zigbee: Traceback (most recent call last):
2022-11-21 13:31:13.558 Error: zigbee: File "/home/ronald/domoticz/plugins/Domoticz-Zigbee/plugin.py", line 1597, in onHeartbeat
2022-11-21 13:31:13.558 Error: zigbee: _plugin.onHeartbeat()
2022-11-21 13:31:13.558 Error: zigbee: File "/home/ronald/domoticz/plugins/Domoticz-Zigbee/plugin.py", line 942, in onHeartbeat
2022-11-21 13:31:13.558 Error: zigbee: WriteDeviceList(self, ( (5 * 60) // HEARTBEAT) )
2022-11-21 13:31:13.558 Error: zigbee: File "/home/ronald/domoticz/plugins/Domoticz-Zigbee/Modules/database.py", line 324, in WriteDeviceList
2022-11-21 13:31:13.558 Error: zigbee: _write_DeviceList_txt(self)
2022-11-21 13:31:13.558 Error: zigbee: File "/home/ronald/domoticz/plugins/Domoticz-Zigbee/Modules/database.py", line 345, in _write_DeviceList_txt
2022-11-21 13:31:13.558 Error: zigbee: file.write(key + " : " + str(self.ListOfDevices[key]) + "\n")
2022-11-21 13:31:13.558 Error: zigbee: File "/usr/lib/python3.7/encodings/iso8859_15.py", line 19, in encode
2022-11-21 13:31:13.558 Error: zigbee: return codecs.charmap_encode(input,self.errors,encoding_table)[0]
2022-11-21 13:31:13.558 Error: zigbee: UnicodeEncodeError: 'charmap' codec can't encode character '\ݚ' in position 243: character maps to <undefined>

Yesterday the character was ‘\u075a’ today '\u0700' so it is not a consistent string it tries to write.

What is the best way of work to debug this issue, in the log file from the plugin this error is not logged at all.

pipiche38, 2022-11-22:
Are you using special caractères in the device name? If so that could explain Envoyé de mon iPhone

Achterhoeker, 2022-11-22:
> Links:

No, i've just checked it extra. Only characters I use in the device 
names are a-z A-Z and _.

Ronald.

Achterhoeker, 2022-11-22:
[download.zip](https://github.com/zigbeefordomoticz/Domoticz-Zigbee/files/10067813/download.zip)
2 files in the attached zip file. The current one (which is missing devices) and the one i place back after a restart to get all devices again.

pitchoun38, 2022-11-22:
Most -likely the error is due to the fact that the string cannot be encoded

A possible path is that one. However needs to take care of backwards compatibility. In other words, can we make the encoding to utf-8 when writing the file to disk, and leave it like it is today 

```
with open('example.txt', 'w', encoding='utf-8') as f:
    f.write(my_str)
```

So with the file provided we have a chance to duplicate the problem, and check if then it is fixed and backward compatible

pipiche38, 2022-11-22:
can you try the following (I assume that you have a good backup ). I also assume that you are on a 6.3 or 6.4 version of the plugin, otherwise you need to move to one of those branch before

then

(1) enable debugDatabase in the Debuging section of the WebUI
(2) git checkout issue/1353
(3) restart the plugin

let see what is happening.

I havent' been able to duplicate the issue on my system.
Can you share:

Your OS
Python3 version
Domoticz version

`env | grep LC`

Achterhoeker, 2022-11-23:
@pipiche38 Thx for your issue branch. I'm running it now. All seems working so far. 
I was already running a local changed version with the  encoding='utf-8' option. That one was still running without any issues. 

Only thing i didn't find is the enabling the debugDatabase option. 

[image]
This is my debug/logging screen ?

My OS:
Raspbian GNU/Linux 10 (buster)  Linux version 5.10.63-v7l+ (dom@buildbot) (arm-linux-gnueabihf-gcc-8 (Ubuntu/Linaro 8.4.0-3ubuntu1) 8.4.0, GNU ld (GNU Binutils for Ubuntu) 2.34) #1459 SMP Wed Oct 6 16:41:57 BST 2021

Domoticz:
Version: 2022.2
Build Hash: eea9db734
Compile Date: 2022-11-05 13:05:35
dzVents Version: 3.1.8
Python Version: 3.7.3 (default, Jan 22 2021, 20:04:44) [GCC 8.3.0]

pipiche38, 2022-11-23:
You have to enable Advanced ;-)

Achterhoeker, 2022-11-23:
> You have to enable Advanced ;-)

ok, stupid from me that i didn't find that myself.

pipiche38, 2022-11-23:
closed with #1355 merged on dev branch for now

closed 2022-11-28 by pipiche38