zigbeefordomoticz/Domoticz-Zigbee #1381: Translation of WebUI to German and other languages

pipiche38, 2022-12-17:
we are looking for contributors wiulling to translate an english list of items to foreign language such as German.

  The WebUI has arround 600 items ( title, short description, legend) already translated in
- French
- Spanish
- Dutch

any help would be more than welcome, you just need to request teh excel file which contains all of the items and their English version, and send it back when your localisation is over.

CC: @jp-keros

Hedda, 2022-12-21:
A tip is to consider using Weblate for managing translation online (which is free for open source projects) and use separate GNU Gettext format files (strings.po) for each language 

https://weblate.org

Kodi does this for example and documented its use for contributors in their wiki here:

https://kodi.tv/article/switching-translations-to-weblate/

https://kodi.wiki/view/Language_portal

https://kodi.wiki/view/Language_support

https://kodi.wiki/view/Translations

https://kodi.wiki/view/Translating_add-ons

https://kodi.weblate.cloud/

pipiche38, 2023-01-19:
conclusive tests with weblate. @Hedda thanks for the advice