zigbeefordomoticz/Domoticz-Zigbee #1386: Ikea Starkvind Air purifier

rhuijben, 2022-12-20:
My air purifier shows up just fine and I'm able to switch it on to specific states (auto and low, medium, high). 

But I'm unable to turn the air purifier off or use more advanced functions.

It should have a readable air quality meter, but I only see a Lux (light) report.
The box should have more programmable features like a child lock, but those are not too important to me.

* Product Name 
Ikea STARKVIND Air purifier
* Brand
Ikea
* URL on a web site
https://www.ikea.com/in/en/p/starkvind-air-purifier-black-smart-10501964/
* What do you expect
Able to turn the air purifyer off using domoticz. Able to see air quality and with that the reason why the auto option chose Low,Mid,High.

* The logs from the pairing process

* export.json file as described here above

```
{"Version":"3","ZDeviceName":"","Status":"inDB","SQN":"05","Ep":{"01":{"0000":{"4000":"1.0.033","0004":"IKEA of Sweden","0005":"STARKVIND Air purifier","0000":"3","0001":"16","0002":"103","0003":"1","0006":"20210428","0007":"1","000a":"4532303037"},"0003":{},"0004":{},"0005":{},"0202":{},"fc57":{},"fc7d":{},"0019":{},"0400":{},"042a":{},"ClusterType":{"1961":"FanControl","1962":"Lux"}},"f2":{}},"Heartbeat":"10","RIA":"4","LQI":51,"Battery":{},"Model":"STARKVIND Air purifier","ForceAckCommands":[],"MacCapa":"8e","IEEE":"0c4314fffed61adf","Type":"FanControl/Lux","ProfileID":"0104","ZDeviceID":"0007","App Version":"16","Attributes List":{"Request":{"01":{"0000":1671478067.0655785,"0003":1671478067.0659719,"0004":1671478067.0664241,"0005":1671478067.0668101,"0202":1671478067.0671961,"fc57":1671478067.0677254,"fc7d":1671478067.0682228,"0019":1671478067.0686269,"0400":1671478067.0690293,"042a":1671478067.072608},"f2":{}}},"DeviceType":"FFD","HW Version":"1","Last Cmds":[],"LogicalType":"Router","Manufacturer":"117c","Manufacturer Name":"IKEA of Sweden","NbEp":"2","PowerSource":"Main","ReadAttributes":{"Ep":{"01":{"0000":{"TimeStamp":1671529114,"iSQN":{"4000":"0b","0004":"07","0005":"07","0000":"0e","0001":"08","0002":"08","0003":"08","0006":"09","0007":"09","000a":"09","000f":"0a","0010":"0a","0015":"0a","f000":"0b"},"Attributes":{"4000":"00","0004":"00","0005":"00","0000":"00","0001":"00","0002":"00","0003":"00","0006":"00","0007":"00","000a":"00","000f":"86","0010":"86","f000":"86"},"ZigateRequest":{}},"0400":{"TimeStamp":1671471469,"iSQN":{"0000":"02"},"Attributes":{"0000":"86"},"ZigateRequest":{}},"0019":{"TimeStamp":1671471471,"iSQN":{"0001":"03","0002":"03","0003":"03","0004":"04","0005":"04","0006":"04","0007":"05","0008":"05","0009":"05","000a":"06"},"Attributes":{"0001":"86","0002":"86","0004":"86","0005":"86","0007":"86","0008":"86","000a":"86"},"ZigateRequest":{}}},"f2":{"0000":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}}}}},"ReceiveOnIdle":"On","Stack Version":"103","Stamp":{"time":1671531498.9422364,"Time":"2022-12-20 11:18:18","MsgType":"8011","LastSeen":1671531498,"LastPing":1671529114},"ZCL Version":"3","Health":"Live","ConfigureReporting":{"Ep":{"01":{"0400":{"TimeStamp":1671471465.1396747,"iSQN":{"0000":"01"},"Attributes":{"0000":"86"},"ZigateRequest":{}}}}},"Capability":["Full-Function Device","Main Powered","Receiver during Idle","Standard security","NwkAddr should be allocated"],"BindingTable":{"SQN":0,"Status":"Success","TimeStamp":1671471474.8666713,"BindingTableSize":7,"Devices":[{"0c4314fffed61adf":{"sourceEp":"01","Cluster":"0000","targetIEEE":"00124b0025e14d39","targetNickName":"Zigbee Coordinator","targetEp":"01"}},{"0c4314fffed61adf":{"sourceEp":"01","Cluster":"0003","targetIEEE":"00124b0025e14d39","targetNickName":"Zigbee Coordinator","targetEp":"01"}},{"0c4314fffed61adf":{"sourceEp":"01","Cluster":"0004","targetIEEE":"00124b0025e14d39","targetNickName":"Zigbee Coordinator","targetEp":"01"}},{"0c4314fffed61adf":{"sourceEp":"01","Cluster":"0005","targetIEEE":"00124b0025e14d39","targetNickName":"Zigbee Coordinator","targetEp":"01"}},{"0c4314fffed61adf":{"sourceEp":"01","Cluster":"0202","targetIEEE":"00124b0025e14d39","targetNickName":"Zigbee Coordinator","targetEp":"01"}},{"0c4314fffed61adf":{"sourceEp":"01","Cluster":"fc57","targetIEEE":"00124b0025e14d39","targetNickName":"Zigbee Coordinator","targetEp":"01"}},{"0c4314fffed61adf":{"sourceEp":"01","Cluster":"fc7d","targetIEEE":"00124b0025e14d39","targetNickName":"Zigbee Coordinator","targetEp":"01"}}]},"WriteAttributes":{"Ep":{"01":{"0000":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"0003":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"0004":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"0005":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"0202":{"TimeStamp":1671531497,"iSQN":{"0001":"03","0000":"04"},"Attributes":{"0001":"00","0000":"00"},"ZigateRequest":{"0001":{"Status":"fullfilled","DataType":"30","EPin":"01","EPout":"01","manuf_id":"0000","manuf_spec":"00","data":"02","ackIsDisabled":true},"0000":{"Status":"fullfilled","DataType":"30","EPin":"01","EPout":"01","manuf_id":"0000","manuf_spec":"00","data":"05","ackIsDisabled":true}}},"fc57":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"fc7d":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"0019":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"0400":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"042a":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}}}}},"RoutingTable":[{"SQN":0,"Status":"Success","TimeStamp":1671488179.2415495,"Time":1671488172,"RoutingTableTableSize":16,"Devices":[{"0000":{"Status":"Active (0)","MemoryConstrained":0,"ManyToOne":1,"RouteRecordRequired":1,"NextHopNwkId":"0000"}}]}],"NeighbourTableSize":1,"ReadConfigureReporting":{"Ep":{"01":{"0400":{"0000":{"TimeStamp":1671473647.374063,"Status":"86"}}}},"TimeStamp":1671473647.3740506},"Bind":{"01":{"0000":{"Target":"0000","Stamp":1671471465,"Phase":"binded","Status":"00","i_sqn":"03"},"0003":{"Target":"0000","Stamp":1671471465,"Phase":"binded","Status":"00","i_sqn":"04"},"0004":{"Target":"0000","Stamp":1671471466,"Phase":"binded","Status":"00","i_sqn":"05"},"0005":{"Target":"0000","Stamp":1671471466,"Phase":"binded","Status":"00","i_sqn":"06"},"0202":{"Target":"0000","Stamp":1671471466,"Phase":"binded","Status":"00","i_sqn":"07"},"fc57":{"Target":"0000","Stamp":1671471467,"Phase":"binded","Status":"00","i_sqn":"08"},"fc7d":{"Target":"0000","Stamp":1671471467,"Phase":"binded","Status":"00","i_sqn":"09"}}},"OTAClient":{"ManufacturerCode":"117c","ImageType":"110c","CurrentImageVersion":"00010033"},"SWBUILD_1":"20210428","Neighbours":[{"SQN":0,"Status":"","TimeStamp":1671488172,"Time":1671488172,"Devices":[{"0000":{"_relationshp":"Parent","_devicetype":"Coordinator","_depth":0,"_lnkqty":99,"_rxonwhenidl":"Rx-On","_IEEE":"00124b0025e14d39","_permitjnt":"??"}}]}],"ConfigSource":"8043","Epv2":{"01":{"ClusterIn":{"0000":{},"0003":{},"0004":{},"0005":{},"0202":{},"fc57":{},"fc7d":{}},"ClusterOut":{"0019":{},"0400":{},"042a":{}},"ProfileID":"0104","ZDeviceID":"0007"},"f2":{"ClusterIn":{},"ClusterOut":{},"ProfileID":{},"ZDeviceID":{}}},"SWBUILD_3":"1.0.033","ConsistencyCheck":"ok","CertifiedDevice":false,"Log_UnknowDeviceFlag":1671530502.1755261,"Max Buffer Size":"52","Max Rx":"0052","Max Tx":"0052","macapa":"8e","bitfield":"4001","server_mask":"2c00","descriptor_capability":"00","ZCLSQN":"05","RollingLQI":[48,48,48,48,48,51,51,51,51,51],"ZCL-IN-SQN":{"0008":"05"}}
```

pipiche38, 2022-12-20:
Bert, I ack your request. I'll need some time to work on that, and might need some more support from you.
At least to start with, could you kindly make sure that you are on the 'dev' branch so you are on the latest code, and most-likely I'll create a dedicated branch to work on.

From the file that you shared I see

* Illuminance
* PM 2.5
* Fan Control

rhuijben, 2022-12-21:
@pipiche38 Thanks, no problem. Feel free to reach out. I'm already on the developer branch. Not sure what other useful debug output I can produce right now.

This air purifier was the reason I bought a zigbee dongle, so not much experience there yet.
(But wrote my own python plugin variant for Domoticz to support some wifi power adapters)

Getting the purifier connected was easy. Still trying to get some legacy Ikea products to connect to my Zigbee network.

pipiche38, 2022-12-21:
can you kindly move to device/ikea/air-purifier branch

git pull
git checkout device/ikea/air-purifier

then you just restart the plugin
watch the plugin log files Domoticz-Zigbee/Log/PluginZigbee_xx.log

Normally if you do action on the air purifier you should get something on the log file. I would be glad to get those logs and eventually the corresponding action done on the device itself.

So far nothing done on teh Domoticz side, I need to understand how it behave

pipiche38, 2022-12-21:
Ok,so after few more investigation, it would be great if you can be on the latest version of the branch

```
git pull
git checkout device/ikea/air-purifier
git pull

```

- remove the domoticz devices associated to this Air Purifier
- then restart the plugin, it might eventually do some cleanup and load the most recent plugin
- then enable debugPairing and debugClusters from teh WebUI > Tools > Debug Advanced menu

then do the pairing

- the FanControl won't work, no worries, I just need for info
leave it like that for 15/20 minutes and drop me the Logs/PluginZigbee_xx.log file

rhuijben, 2022-12-21:
I'm currently not at home. 
Just switched to the branch and restarted domoticz (as that is easier than to get through the firewall at the web interface).

Will try a new pair operation tonight. Just restarting domoticz resulted in:
```
2022-12-21 16:28:23,004 INFO    : [     MainThread 0] Zigbee for Domoticz (z4d) plugin beta6-6.4.101 started
2022-12-21 16:28:23,027 INFO    : [     MainThread 0] Plugin Database: DeviceList-15.txt
2022-12-21 16:28:23,031 INFO    : [     MainThread 0] Loading ZCL Cluster definitions
2022-12-21 16:28:23,054 INFO    : [     MainThread 0]  - ZCL Cluster 0001 - Power (v1) loaded
2022-12-21 16:28:23,056 INFO    : [     MainThread 0]  - ZCL Cluster 0006 - On/Off (v1) loaded
2022-12-21 16:28:23,058 INFO    : [     MainThread 0]  - ZCL Cluster 0008 - Level (v1) loaded
2022-12-21 16:28:23,061 INFO    : [     MainThread 0]  - ZCL Cluster 0400 - Illuminance Level Sensing (v1) loaded
2022-12-21 16:28:23,065 INFO    : [     MainThread 0]  - ZCL Cluster 0402 - Temperature Measurement (v1) loaded
2022-12-21 16:28:23,066 INFO    : [     MainThread 0]  - ZCL Cluster 0403 - Pressure Measurement (v1) loaded
2022-12-21 16:28:23,076 INFO    : [     MainThread 0]  - ZCL Cluster 0405 - Relative Humidity Measurement (v1) loaded
2022-12-21 16:28:23,077 INFO    : [     MainThread 0]  - ZCL Cluster 0406 - Occupancy Sensing (v1) loaded
2022-12-21 16:28:23,094 INFO    : [     MainThread 0] DeviceConf loaded - 17 confs loaded
2022-12-21 16:28:24,562 INFO    : [     MainThread 0] DeviceConf loaded - 394 confs loaded
2022-12-21 16:28:24,563 INFO    : [     MainThread 0] load ListOfDevice
2022-12-21 16:28:24,582 INFO    : [     MainThread 0] 2 Entries loaded from /home/root/domoticz/plugins/Domoticz-Zigbee/Data/DeviceList-15.txt
2022-12-21 16:28:24,612 INFO    : [     MainThread 0] Transport mode: ZigpyZNP
2022-12-21 16:28:26,546 DEBUG   :Using selector: EpollSelector
2022-12-21 16:28:26,548 INFO    : [     MainThread 0] Start Zigpy Transport on ZNP
2022-12-21 16:28:26,555 INFO    : [    ZigpyCom_15 0] Starting radio znp port: /dev/serial/by-id/usb-Silicon_Labs_Sonoff_Zigbee_3.0_USB_Dongle_Plus_0001-if00-port0
2022-12-21 16:28:26,560 INFO    : [ZigpyForwarder_15 0] ZigpyTransport: thread_processing_and_sending Thread start.
2022-12-21 16:28:26,563 INFO    : [     MainThread 0] Start Web Server connection
2022-12-21 16:28:26,571 INFO    : [     MainThread 0] Web backend for Web User Interface started on port: 9440
2022-12-21 16:28:26,572 INFO    : [     MainThread 0] Domoticz Widgets usage is at 2.0 % (250 units free)
2022-12-21 16:28:26,931 INFO    : [    ZigpyCom_15 0] Started radio znp port: /dev/serial/by-id/usb-Silicon_Labs_Sonoff_Zigbee_3.0_USB_Dongle_Plus_0001-if00-port0
2022-12-21 16:28:34,427 INFO    : [    ZigpyCom_15 0] ZNP Radio register default Ep
2022-12-21 16:28:34,440 INFO    : [    ZigpyCom_15 0] ZNP Radio register any additional/specific Ep
2022-12-21 16:28:34,536 INFO    : [    ZigpyCom_15 0] device_initialized (0x0000 00:12:4b:00:25:e1:4d:39)
2022-12-21 16:28:34,537 INFO    : [    ZigpyCom_15 0] ZNP Configuration {'device': {'path': '/dev/serial/by-id/usb-Silicon_Labs_Sonoff_Zigbee_3.0_USB_Dongle_Plus_0001-if00-port0', 'flow_control': None, 'baudrate': 115200}, 'network': {'channel': 15, 'key': None, 'extended_pan_id': None, 'channels': <Channels.CHANNEL_25|CHANNEL_20|CHANNEL_15: 34635776>, 'update_id': 0, 'tc_address': None, 'tc_link_key': 5a:69:67:42:65:65:41:6c:6c:69:61:6e:63:65:30:39, 'pan_id': None, 'key_sequence_number': 0}, 'znp_config': {'tx_power': 0, 'connect_dtr_pin_states': [False, False, False], 'async_response_timeout': 30, 'sync_request_timeout': 15, 'auto_reconnect_retry_delay': 5, 'led_mode': <LEDMode.OFF: 0>, 'max_concurrent_requests': None, 'prefer_endpoint_1': True, 'skip_bootloader': True, 'connect_rts_pin_states': [False, True, False]}, 'topology_scan_enabled': False, 'source_routing': False, 'backup_enabled': True, 'backup_period': 1, 'validate_network_settings': False, 'ota': {'inovelli_provider': False, 'otau_directory': None, 'salus_provider': False, 'ikea_provider': False, 'ledvance_provider': False, 'sonoff_provider': False}, 'database_path': None, 'max_concurrent_requests': 8, 'topology_scan_skip_coordinator': False, 'additional_endpoints': [], 'topology_scan_period': 240}
2022-12-21 16:28:42,335 INFO    : [    ZigpyCom_15 0] ZNP Radio manufacturer: Texas Instruments
2022-12-21 16:28:42,343 INFO    : [    ZigpyCom_15 0] ZNP Radio board model: CC1352/CC2652, Z-Stack 3.30+ (build 20210708)
2022-12-21 16:28:42,348 INFO    : [    ZigpyCom_15 0] ZNP Radio version: 3.3
2022-12-21 16:28:42,349 INFO    : [    ZigpyCom_15 0] extract_versioning_for_plugin Model: CC1352/CC2652, Z-Stack 3.30+ (build 20210708) Manuf: Texas Instruments
2022-12-21 16:28:42,350 INFO    : [    ZigpyCom_15 0] extract_versioning_for_plugin 20 21 0708 Z-Stack 3.30+ (build 20210708)
2022-12-21 16:28:42,351 INFO    : [    ZigpyCom_15 0] Network settings
2022-12-21 16:28:42,352 INFO    : [    ZigpyCom_15 0]   Channel: 15

<snip>
2022-12-21 16:28:42,366 INFO    : [    ZigpyCom_15 0]   Device NWK: 0x0000
2022-12-21 16:28:42,369 INFO    : [ZigpyForwarder_15 0] Number of devices recently active in Zigate = 1
2022-12-21 16:28:42,371 INFO    : [ZigpyForwarder_15 0] [00] DevID: 00 Network addr: fac1 IEEE: 0c4314fffed61adf LQI: 000 power: ff Model: STARKVIND Air purifier ( Luchtreiniger )
2022-12-21 16:28:42,372 INFO    : [ZigpyForwarder_15 0] Decode8010 20210708Z-Stack 3.30+ (build 20210708)
[... cut ...]
2022-12-21 16:31:15,441 INFO    : [     MainThread 0] manufacturer_specific_attribute - NwkId: fac1 found attribute: 0005 Manuf Specific, return ManufCode: 117c
2022-12-21 16:31:15,442 INFO    : [     MainThread 0] manufacturer_specific_attribute - NwkId: fac1 found attribute: 0006 Manuf Specific, return ManufCode: 117c
2022-12-21 16:31:15,443 INFO    : [     MainThread 0] manufacturer_specific_attribute - NwkId: fac1 found attribute: 0007 Manuf Specific, return ManufCode: 117c
2022-12-21 16:31:15,444 INFO    : [     MainThread 0] manufacturer_specific_attribute - NwkId: fac1 found attribute: 0008 Manuf Specific, return ManufCode: 117c
```

pipiche38, 2022-12-21:
Sound good so far. (just simple logs) which proof that it has redone the Configure Reporting, let see what is happening.
If you can drop me once more a new version of the export.json this would be great

rhuijben, 2022-12-21:
```
2022-12-21 17:31:01,080 INFO    : [ZigpyForwarder_15 0] ReadCluster fc7d - fac1/01 Attribute: 0000 Type: 23 Size: 0004 Data: 00000c49
2022-12-21 17:31:01,090 INFO    : [ZigpyForwarder_15 0] ikea_air_purifier_cluster fac1/01 fc7d 0000 00000c49
2022-12-21 17:31:01,429 INFO    : [ZigpyForwarder_15 0] ReadCluster fc7d - fac1/01 Attribute: 0001 Type: 20 Size: 0001 Data: 00
2022-12-21 17:31:01,430 INFO    : [ZigpyForwarder_15 0] ikea_air_purifier_cluster fac1/01 fc7d 0001 00
2022-12-21 17:31:01,431 INFO    : [ZigpyForwarder_15 0]  -- Replace Filter: 00
2022-12-21 17:31:01,827 INFO    : [ZigpyForwarder_15 0] ReadCluster fc7d - fac1/01 Attribute: 0002 Type: 23 Size: 0004 Data: 0003f480
2022-12-21 17:31:01,827 INFO    : [ZigpyForwarder_15 0] ikea_air_purifier_cluster fac1/01 fc7d 0002 0003f480
2022-12-21 17:31:01,827 INFO    : [ZigpyForwarder_15 0]  -- Filter Life time: 0003f480
2022-12-21 17:31:02,214 INFO    : [ZigpyForwarder_15 0] ReadCluster fc7d - fac1/01 Attribute: 0003 Type: 10 Size: 0001 Data: 00
2022-12-21 17:31:02,215 INFO    : [ZigpyForwarder_15 0] ikea_air_purifier_cluster fac1/01 fc7d 0003 00
2022-12-21 17:31:02,215 INFO    : [ZigpyForwarder_15 0]  --  Led Indication: 00
2022-12-21 17:31:02,623 INFO    : [ZigpyForwarder_15 0] ReadCluster fc7d - fac1/01 Attribute: 0004 Type: 21 Size: 0002 Data: ffff
2022-12-21 17:31:02,624 INFO    : [ZigpyForwarder_15 0] ikea_air_purifier_cluster fac1/01 fc7d 0004 ffff
2022-12-21 17:31:02,625 INFO    : [ZigpyForwarder_15 0]  --  PM25: ffff --> 65535
2022-12-21 17:31:03,013 INFO    : [ZigpyForwarder_15 0] ReadCluster fc7d - fac1/01 Attribute: 0005 Type: 10 Size: 0001 Data: 00
2022-12-21 17:31:03,013 INFO    : [ZigpyForwarder_15 0] ikea_air_purifier_cluster fac1/01 fc7d 0005 00
2022-12-21 17:31:03,013 INFO    : [ZigpyForwarder_15 0]  --  Locked: 00
2022-12-21 17:31:03,412 INFO    : [ZigpyForwarder_15 0] ReadCluster fc7d - fac1/01 Attribute: 0006 Type: 20 Size: 0001 Data: 00
2022-12-21 17:31:03,412 INFO    : [ZigpyForwarder_15 0] ikea_air_purifier_cluster fac1/01 fc7d 0006 00
2022-12-21 17:31:03,412 INFO    : [ZigpyForwarder_15 0]  --  Mode: 00
2022-12-21 17:31:03,815 INFO    : [ZigpyForwarder_15 0] ReadCluster fc7d - fac1/01 Attribute: 0007 Type: 20 Size: 0001 Data: 00
2022-12-21 17:31:03,816 INFO    : [ZigpyForwarder_15 0] ikea_air_purifier_cluster fac1/01 fc7d 0007 00
2022-12-21 17:31:03,816 INFO    : [ZigpyForwarder_15 0]  --  Fan Speed: 00
2022-12-21 17:31:04,212 INFO    : [ZigpyForwarder_15 0] ReadCluster fc7d - fac1/01 Attribute: 0008 Type: 23 Size: 0004 Data: 00000c49
2022-12-21 17:31:04,212 INFO    : [ZigpyForwarder_15 0] ikea_air_purifier_cluster fac1/01 fc7d 0008 00000c49
2022-12-21 17:31:04,212 INFO    : [ZigpyForwarder_15 0]  --  Device runtime: 00000c49
(END)
```

pipiche38, 2022-12-21:
Ok. That sound promising, we are getting the right think.

Is the Device Off ? This could explain the PM24 to 0xffff ( 65535 ) and Mode is 0x00

pipiche38, 2022-12-21:
Thinking on how we will translate that into Domoticz I see

- Widget FanControl to Switch On/Off and vary the fan speed
- Widget PM2.5 to measure 
- Alarm Widget for Filter Life time and Replace Filter ( 3 level, Green Orange and red)
- Device runtime will only be visible in the export.json

for the lock and led, I'll see if and how we can manage. If this is possible this would be done over Device Parameters

rhuijben, 2022-12-21:
Yes The device is off

rhuijben, 2022-12-21:
Similar dump +- hour later
```
2022-12-21 18:30:46,560 INFO    : [ZigpyForwarder_15 0] ReadCluster fc7d - fac1/01 Attribute: 0000 Type: 23 Size: 0004 Data: 00000c49
2022-12-21 18:30:46,561 INFO    : [ZigpyForwarder_15 0] ikea_air_purifier_cluster fac1/01 fc7d 0000 00000c49
2022-12-21 18:30:46,906 INFO    : [ZigpyForwarder_15 0] ReadCluster fc7d - fac1/01 Attribute: 0001 Type: 20 Size: 0001 Data: 00
2022-12-21 18:30:46,906 INFO    : [ZigpyForwarder_15 0] ikea_air_purifier_cluster fac1/01 fc7d 0001 00
2022-12-21 18:30:46,907 INFO    : [ZigpyForwarder_15 0]  -- Replace Filter: 00
2022-12-21 18:30:47,305 INFO    : [ZigpyForwarder_15 0] ReadCluster fc7d - fac1/01 Attribute: 0002 Type: 23 Size: 0004 Data: 0003f480
2022-12-21 18:30:47,306 INFO    : [ZigpyForwarder_15 0] ikea_air_purifier_cluster fac1/01 fc7d 0002 0003f480
2022-12-21 18:30:47,306 INFO    : [ZigpyForwarder_15 0]  -- Filter Life time: 0003f480
2022-12-21 18:30:47,705 INFO    : [ZigpyForwarder_15 0] ReadCluster fc7d - fac1/01 Attribute: 0003 Type: 10 Size: 0001 Data: 00
2022-12-21 18:30:47,705 INFO    : [ZigpyForwarder_15 0] ikea_air_purifier_cluster fac1/01 fc7d 0003 00
2022-12-21 18:30:47,706 INFO    : [ZigpyForwarder_15 0]  --  Led Indication: 00
2022-12-21 18:30:48,100 INFO    : [ZigpyForwarder_15 0] ReadCluster fc7d - fac1/01 Attribute: 0004 Type: 21 Size: 0002 Data: ffff
2022-12-21 18:30:48,101 INFO    : [ZigpyForwarder_15 0] ikea_air_purifier_cluster fac1/01 fc7d 0004 ffff
2022-12-21 18:30:48,101 INFO    : [ZigpyForwarder_15 0]  --  PM25: ffff --> 65535
2022-12-21 18:30:48,499 INFO    : [ZigpyForwarder_15 0] ReadCluster fc7d - fac1/01 Attribute: 0005 Type: 10 Size: 0001 Data: 00
2022-12-21 18:30:48,500 INFO    : [ZigpyForwarder_15 0] ikea_air_purifier_cluster fac1/01 fc7d 0005 00
2022-12-21 18:30:48,500 INFO    : [ZigpyForwarder_15 0]  --  Locked: 00
2022-12-21 18:30:48,887 INFO    : [ZigpyForwarder_15 0] ReadCluster fc7d - fac1/01 Attribute: 0006 Type: 20 Size: 0001 Data: 00
2022-12-21 18:30:48,887 INFO    : [ZigpyForwarder_15 0] ikea_air_purifier_cluster fac1/01 fc7d 0006 00
2022-12-21 18:30:48,888 INFO    : [ZigpyForwarder_15 0]  --  Mode: 00
2022-12-21 18:30:49,282 INFO    : [ZigpyForwarder_15 0] ReadCluster fc7d - fac1/01 Attribute: 0007 Type: 20 Size: 0001 Data: 00
2022-12-21 18:30:49,282 INFO    : [ZigpyForwarder_15 0] ikea_air_purifier_cluster fac1/01 fc7d 0007 00
2022-12-21 18:30:49,282 INFO    : [ZigpyForwarder_15 0]  --  Fan Speed: 00
2022-12-21 18:30:49,695 INFO    : [ZigpyForwarder_15 0] ReadCluster fc7d - fac1/01 Attribute: 0008 Type: 23 Size: 0004 Data: 00000c49
2022-12-21 18:30:49,696 INFO    : [ZigpyForwarder_15 0] ikea_air_purifier_cluster fac1/01 fc7d 0008 00000c49
2022-12-21 18:30:49,696 INFO    : [ZigpyForwarder_15 0]  --  Device runtime: 00000c49
```
Then I got home and turned the purifier on, on auto.
```
2022-12-21 18:48:49,348 INFO    : [ZigpyForwarder_15 0] ReadCluster fc7d - fac1/01 Attribute: 0006 Type: 20 Size: 0001 Data: 01
2022-12-21 18:48:49,350 INFO    : [ZigpyForwarder_15 0] ikea_air_purifier_cluster fac1/01 fc7d 0006 01
2022-12-21 18:48:49,351 INFO    : [ZigpyForwarder_15 0]  --  Mode: 01
2022-12-21 18:48:49,354 INFO    : [ZigpyForwarder_15 0] ReadCluster fc7d - fac1/01 Attribute: 0007 Type: 20 Size: 0001 Data: 1e
2022-12-21 18:48:49,354 INFO    : [ZigpyForwarder_15 0] ikea_air_purifier_cluster fac1/01 fc7d 0007 1e
2022-12-21 18:48:49,355 INFO    : [ZigpyForwarder_15 0]  --  Fan Speed: 1e
2022-12-21 18:49:02,355 INFO    : [ZigpyForwarder_15 0] ReadCluster fc7d - fac1/01 Attribute: 0004 Type: 21 Size: 0002 Data: 0004
2022-12-21 18:49:02,357 INFO    : [ZigpyForwarder_15 0] ikea_air_purifier_cluster fac1/01 fc7d 0004 0004
2022-12-21 18:49:02,357 INFO    : [ZigpyForwarder_15 0]  --  PM25: 0004 --> 4
2022-12-21 18:49:07,360 INFO    : [ZigpyForwarder_15 0] ReadCluster fc7d - fac1/01 Attribute: 0004 Type: 21 Size: 0002 Data: 0007
2022-12-21 18:49:07,361 INFO    : [ZigpyForwarder_15 0] ikea_air_purifier_cluster fac1/01 fc7d 0004 0007
2022-12-21 18:49:07,361 INFO    : [ZigpyForwarder_15 0]  --  PM25: 0007 --> 7
2022-12-21 18:49:12,297 INFO    : [ZigpyForwarder_15 0] ReadCluster fc7d - fac1/01 Attribute: 0004 Type: 21 Size: 0002 Data: 0009
2022-12-21 18:49:12,298 INFO    : [ZigpyForwarder_15 0] ikea_air_purifier_cluster fac1/01 fc7d 0004 0009
2022-12-21 18:49:12,299 INFO    : [ZigpyForwarder_15 0]  --  PM25: 0009 --> 9
2022-12-21 18:49:17,277 INFO    : [ZigpyForwarder_15 0] ReadCluster fc7d - fac1/01 Attribute: 0004 Type: 21 Size: 0002 Data: 000b
2022-12-21 18:49:17,278 INFO    : [ZigpyForwarder_15 0] ikea_air_purifier_cluster fac1/01 fc7d 0004 000b
2022-12-21 18:49:17,278 INFO    : [ZigpyForwarder_15 0]  --  PM25: 000b --> 11
2022-12-21 18:49:19,175 INFO    : [ZigpyForwarder_15 0] ReadCluster fc7d - fac1/01 Attribute: 0007 Type: 20 Size: 0001 Data: 19
2022-12-21 18:49:19,176 INFO    : [ZigpyForwarder_15 0] ikea_air_purifier_cluster fac1/01 fc7d 0007 19
2022-12-21 18:49:19,177 INFO    : [ZigpyForwarder_15 0]  --  Fan Speed: 19
2022-12-21 18:49:22,256 INFO    : [ZigpyForwarder_15 0] ReadCluster fc7d - fac1/01 Attribute: 0004 Type: 21 Size: 0002 Data: 000c
2022-12-21 18:49:22,257 INFO    : [ZigpyForwarder_15 0] ikea_air_purifier_cluster fac1/01 fc7d 0004 000c
2022-12-21 18:49:22,258 INFO    : [ZigpyForwarder_15 0]  --  PM25: 000c --> 12
2022-12-21 18:49:27,235 INFO    : [ZigpyForwarder_15 0] ReadCluster fc7d - fac1/01 Attribute: 0004 Type: 21 Size: 0002 Data: 000d
[... cut ...]
2022-12-21 19:49:26,935 INFO    : [ZigpyForwarder_15 0]  --  Device runtime: 00000c85
2022-12-21 19:49:26,962 INFO    : [ZigpyForwarder_15 0] ReadCluster fc7d - fac1/01 Attribute: 0007 Type: 20 Size: 0001 Data: 0a
2022-12-21 19:49:26,963 INFO    : [ZigpyForwarder_15 0] ikea_air_purifier_cluster fac1/01 fc7d 0007 0a
2022-12-21 19:49:26,963 INFO    : [ZigpyForwarder_15 0]  --  Fan Speed: 0a
2022-12-21 19:50:26,722 INFO    : [ZigpyForwarder_15 0] ReadCluster fc7d - fac1/01 Attribute: 0000 Type: 23 Size: 0004 Data: 00000c86
```

rhuijben, 2022-12-21:
Note the new firmware message in between...

And finally I turned it back on to auto.

```
{"Version":"3","ZDeviceName":"Luchtreiniger","Status":"inDB","SQN":"47","Ep":{"01":{"0000":{"4000":"1.0.033","0004":"IKEA of Sweden","0005":"STARKVIND Air purifier","0000":"3","0001":"16","0002":"103","0003":"1","0006":"20210428","0007":"1","000a":"4532303037"},"0003":{},"0004":{},"0005":{},"0202":{},"fc57":{},"fc7d":{"0000":"00000c8c","0001":"00","0002":"0003f480","0003":"00","0004":"000a","0005":"00","0006":"01","0007":"0a","0008":"00000c8c"},"0019":{},"0400":{},"042a":{},"ClusterType":{"1961":"FanControl","1962":"Lux"}},"f2":{}},"Heartbeat":"15870","RIA":"4","LQI":39,"Battery":{},"Model":"STARKVIND Air purifier","ForceAckCommands":[],"MacCapa":"8e","IEEE":"0c4314fffed61adf","Type":"FanControl/Lux","ProfileID":"0104","ZDeviceID":"0007","App Version":"16","Attributes List":{"Request":{"01":{"0000":1671478067.0655785,"0003":1671478067.0659719,"0004":1671478067.0664241,"0005":1671478067.0668101,"0202":1671478067.0671961,"fc57":1671478067.0677254,"fc7d":1671478067.0682228,"0019":1671478067.0686269,"0400":1671478067.0690293,"042a":1671478067.072608},"f2":{}}},"DeviceType":"FFD","HW Version":"1","Last Cmds":[],"LogicalType":"Router","Manufacturer":"117c","Manufacturer Name":"IKEA of Sweden","NbEp":"2","PowerSource":"Main","ReadAttributes":{"Ep":{"01":{"0000":{"TimeStamp":1671634248,"iSQN":{"4000":"0b","0004":"07","0005":"07","0000":"21","0001":"08","0002":"08","0003":"08","0006":"09","0007":"09","000a":"09","000f":"0a","0010":"0a","0015":"0a","f000":"0b"},"Attributes":{"4000":"00","0004":"00","0005":"00","0000":"00","0001":"00","0002":"00","0003":"00","0006":"00","0007":"00","000a":"00","000f":"86","0010":"86","f000":"86"},"ZigateRequest":{}},"0400":{"TimeStamp":1671471469,"iSQN":{"0000":"02"},"Attributes":{"0000":"86"},"ZigateRequest":{}},"0019":{"TimeStamp":1671471471,"iSQN":{"0001":"03","0002":"03","0003":"03","0004":"04","0005":"04","0006":"04","0007":"05","0008":"05","0009":"05","000a":"06"},"Attributes":{"0001":"86","0002":"86","0004":"86","0005":"86","0007":"86","0008":"86","000a":"86"},"ZigateRequest":{}},"fc7d":{"TimeStamp":1671648985,"iSQN":{},"Attributes":{"0000":"00","0001":"00","0002":"00","0003":"00","0004":"00","0005":"00","0006":"00","0007":"00","0008":"00"},"ZigateRequest":{}}},"f2":{"0000":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}}}}},"ReceiveOnIdle":"On","Stack Version":"103","Stamp":{"time":1671648985.2576113,"Time":"2022-12-21 19:56:25","MsgType":"8102","LastSeen":1671648985,"LastPing":1671634248},"ZCL Version":"3","Health":"Live","Capability":["Full-Function Device","Main Powered","Receiver during Idle","Standard security","NwkAddr should be allocated"],"ConfigSource":"8043","OTAClient":{"ManufacturerCode":"117c","ImageType":"110c","CurrentImageVersion":"00010033"},"BindingTable":{"SQN":0,"Status":"Success","TimeStamp":1671471474.8666713,"BindingTableSize":7,"Devices":[{"0c4314fffed61adf":{"sourceEp":"01","Cluster":"0000","targetIEEE":"00124b0025e14d39","targetNickName":"Zigbee Coordinator","targetEp":"01"}},{"0c4314fffed61adf":{"sourceEp":"01","Cluster":"0003","targetIEEE":"00124b0025e14d39","targetNickName":"Zigbee Coordinator","targetEp":"01"}},{"0c4314fffed61adf":{"sourceEp":"01","Cluster":"0004","targetIEEE":"00124b0025e14d39","targetNickName":"Zigbee Coordinator","targetEp":"01"}},{"0c4314fffed61adf":{"sourceEp":"01","Cluster":"0005","targetIEEE":"00124b0025e14d39","targetNickName":"Zigbee Coordinator","targetEp":"01"}},{"0c4314fffed61adf":{"sourceEp":"01","Cluster":"0202","targetIEEE":"00124b0025e14d39","targetNickName":"Zigbee Coordinator","targetEp":"01"}},{"0c4314fffed61adf":{"sourceEp":"01","Cluster":"fc57","targetIEEE":"00124b0025e14d39","targetNickName":"Zigbee Coordinator","targetEp":"01"}},{"0c4314fffed61adf":{"sourceEp":"01","Cluster":"fc7d","targetIEEE":"00124b0025e14d39","targetNickName":"Zigbee Coordinator","targetEp":"01"}}]},"Neighbours":[{"SQN":0,"Status":"","TimeStamp":1671488172,"Time":1671488172,"Devices":[{"0000":{"_relationshp":"Parent","_devicetype":"Coordinator","_depth":0,"_lnkqty":99,"_rxonwhenidl":"Rx-On","_IEEE":"00124b0025e14d39","_permitjnt":"??"}}]},{"SQN":0,"Status":"","TimeStamp":1671534715,"Time":1671534715,"Devices":[{"0000":{"_relationshp":"Parent","_devicetype":"Coordinator","_depth":0,"_lnkqty":103,"_rxonwhenidl":"Rx-On","_IEEE":"00124b0025e14d39","_permitjnt":"??"}}]}],"CertifiedDevice":true,"ReadConfigureReporting":{"Ep":{"01":{"0400":{"0000":{"TimeStamp":1671636615.1829088,"Status":"86"}},"fc7d":{"0000":{"TimeStamp":1671636839.423348,"Status":"86"},"0001":{"TimeStamp":1671636839.4235697,"Status":"86"},"0002":{"TimeStamp":1671636839.423785,"Status":"86"},"0003":{"TimeStamp":1671636839.8276489,"Status":"86"},"0004":{"TimeStamp":1671636839.8278701,"Status":"86"},"0005":{"TimeStamp":1671636839.8280864,"Status":"86"},"0006":{"TimeStamp":1671636840.2252226,"Status":"86"},"0007":{"TimeStamp":1671636840.2254443,"Status":"86"},"0008":{"TimeStamp":1671636840.2256634,"Status":"86"}}}},"TimeStamp":1671636840.2256556},"SWBUILD_3":"1.0.033","WriteAttributes":{"Ep":{"01":{"0000":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"0003":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"0004":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"0005":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"0202":{"TimeStamp":1671569357,"iSQN":{"0001":"0d","0000":"0e"},"Attributes":{"0001":"00","0000":"00"},"Z [... cut ...] equired":1,"NextHopNwkId":"0000"}}]},{"SQN":0,"Status":"Success","TimeStamp":1671534723.0736632,"Time":1671534715,"RoutingTableTableSize":16,"Devices":[{"0000":{"Status":"Active (0)","MemoryConstrained":0,"ManyToOne":1,"RouteRecordRequired":1,"NextHopNwkId":"0000"}}]}],"ConsistencyCheck":"ok","ZCLSQN":"44","ZCL-IN-SQN":{"0400":"00","fc7d":"47","0019":"79"},"ZDPSQN":"08","RollingLQI":[39,39,48,48,45,39,36,39,39,39,39]}
```
This is what I get when clicking the new optimize icon in the beta website

rhuijben, 2022-12-21:
And sending max, medium, low, auto, off + off via the icon in domoticz
```
2022-12-21 20:00:52,214 INFO    : [ZigpyForwarder_15 0] ikea_air_purifier_cluster fac1/01 fc7d 0004 000a
2022-12-21 20:00:52,215 INFO    : [ZigpyForwarder_15 0]  --  PM25: 000a --> 10
2022-12-21 20:01:24,001 INFO    : [ZigpyForwarder_15 0] ReadCluster fc7d - fac1/01 Attribute: 0000 Type: 23 Size: 0004 Data: 00000c91
2022-12-21 20:01:24,002 INFO    : [ZigpyForwarder_15 0] ikea_air_purifier_cluster fac1/01 fc7d 0000 00000c91
2022-12-21 20:01:24,004 INFO    : [ZigpyForwarder_15 0] ReadCluster fc7d - fac1/01 Attribute: 0008 Type: 23 Size: 0004 Data: 00000c91
2022-12-21 20:01:24,004 INFO    : [ZigpyForwarder_15 0] ikea_air_purifier_cluster fac1/01 fc7d 0008 00000c91
2022-12-21 20:01:24,005 INFO    : [ZigpyForwarder_15 0]  --  Device runtime: 00000c91
2022-12-21 20:02:23,776 INFO    : [ZigpyForwarder_15 0] ReadCluster fc7d - fac1/01 Attribute: 0000 Type: 23 Size: 0004 Data: 00000c92
2022-12-21 20:02:23,778 INFO    : [ZigpyForwarder_15 0] ikea_air_purifier_cluster fac1/01 fc7d 0000 00000c92
2022-12-21 20:02:23,782 INFO    : [ZigpyForwarder_15 0] ReadCluster fc7d - fac1/01 Attribute: 0008 Type: 23 Size: 0004 Data: 00000c92
2022-12-21 20:02:23,783 INFO    : [ZigpyForwarder_15 0] ikea_air_purifier_cluster fac1/01 fc7d 0008 00000c92
2022-12-21 20:02:23,783 INFO    : [ZigpyForwarder_15 0]  --  Device runtime: 00000c92
2022-12-21 20:03:23,565 INFO    : [ZigpyForwarder_15 0] ReadCluster fc7d - fac1/01 Attribute: 0000 Type: 23 Size: 0004 Data: 00000c93
2022-12-21 20:03:23,567 INFO    : [ZigpyForwarder_15 0] ikea_air_purifier_cluster fac1/01 fc7d 0000 00000c93
2022-12-21 20:03:23,571 INFO    : [ZigpyForwarder_15 0] ReadCluster fc7d - fac1/01 Attribute: 0008 Type: 23 Size: 0004 Data: 00000c93
2022-12-21 20:03:23,571 INFO    : [ZigpyForwarder_15 0] ikea_air_purifier_cluster fac1/01 fc7d 0008 00000c93
2022-12-21 20:03:23,572 INFO    : [ZigpyForwarder_15 0]  --  Device runtime: 00000c93
2022-12-21 20:04:23,296 INFO    : [ZigpyForwarder_15 0] ReadCluster fc7d - fac1/01 Attribute: 0000 Type: 23 Size: 0004 Data: 00000c94
2022-12-21 20:04:23,298 INFO    : [ZigpyForwarder_15 0] ikea_air_purifier_cluster fac1/01 fc7d 0000 00000c94
2022-12-21 20:04:23,302 INFO    : [ZigpyForwarder_15 0] ReadCluster fc7d - fac1/01 Attribute: 0008 Type: 23 Size: 0004 Data: 00000c94
2022-12-21 20:04:23,302 INFO    : [ZigpyForwarder_15 0] ikea_air_purifier_cluster fac1/01 fc7d 0008 00000c94
2022-12-21 20:04:23,303 INFO    : [ZigpyForwarder_15 0]  --  Device runtime: 00000c94
2022-12-21 20:05:23,089 INFO    : [ZigpyForwarder_15 0] ReadCluster fc7d - fac1/01 Attribute: 0000 Type: 23 Size: 0004 Data: 00000c95
2022-12-21 20:05:23,090 INFO    : [ZigpyForwarder_15 0] ikea_air_purifier_cluster fac1/01 fc7d 0000 00000c95
2022-12-21 20:05:23,095 INFO    : [ZigpyForwarder_15 0] ReadCluster fc7d - fac1/01 Attribute: 0008 Type: 23 Size: 0004 Data: 00000c95
2022-12-21 20:05:23,096 INFO    : [ZigpyForwarder_15 0] ikea_air_purifier_cluster fac1/01 fc7d 0008 00000c95
2022-12-21 20:05:23,097 INFO    : [ZigpyForwarder_15 0]  --  Device runtime: 00000c95
2022-12-21 20:05:31,095 INFO    : [ZigpyForwarder_15 0] ReadCluster fc7d - fac1/01 Attribute: 0004 Type: 21 Size: 0002 Data: 0009
2022-12-21 20:05:31,096 INFO    : [ZigpyForwarder_15 0] ikea_air_purifier_cluster fac1/01 fc7d 0004 0009
2022-12-21 20:05:31,097 INFO    : [ZigpyForwarder_15 0]  --  PM25: 0009 --> 9
2022-12-21 20:05:39,605 INFO    : [ZigpyForwarder_15 0] ReadCluster fc7d - fac1/01 Attribute: 0006 Type: 20 Size: 0001 Data: 32
2022-12-21 20:05:39,606 INFO    : [ZigpyForwarder_15 0] ikea_air_purifier_cluster fac1/01 fc7d 0006 32
2022-12-21 20:05:39,607 INFO    : [ZigpyForwarder_15 0]  --  Mode: 32
2022-12-21 20:05:39,608 INFO    : [ZigpyForwarder_15 0] ReadCluster fc7d - fac1/01 Attribute: 0007 Type: 20 Size: 0001 Data: 32
2022-12-21 20:05:39,609 INFO    : [ZigpyForwarder_15 0] ikea_air_purifier_cluster fac1/01 fc7d 0007 32
2022-12-21 20:05:39,609 INFO    : [ZigpyForwarder_15 0]  --  Fan Speed: 32
2022-12-21 20:05:39,956 INFO    : [ZigpyForwarder_15 0] inRawAps fac1/01 Cluster 0008 Manuf: 117c/IKEA of Sweden Command: 0b Data: 04c3 Payload: 084a0b04c3 not processed !!!
2022-12-21 20:05:41,049 INFO    : [ZigpyForwarder_15 0] ReadCluster fc7d - fac1/01 Attribute: 0004 Type: 21 Size: 0002 Data: 0008
2022-12-21 20:05:41,049 INFO    : [ZigpyForwarder_15 0] ikea_air_purifier_cluster fac1/01 fc7d 0004 0008
2022-12-21 20:05:41,050 INFO    : [ZigpyForwarder_15 0]  --  PM25: 0008 --> 8
2022-12-21 20:05:47,237 INFO    : [ZigpyForwarder_15 0] ReadCluster fc7d - fac1/01 Attribute: 0006 Type: 20 Size: 0001 Data: 1e
2022-12-21 20:05:47,238 INFO    : [ZigpyForwarder_15 0] ikea_air_purifier_cluster fac1/01 fc7d 0006 1e
2022-12-21 20:05:47,238 INFO    : [ZigpyForwarder_15 0]  --  Mode: 1e
2022-12-21 20:05:47,240 INFO    : [ZigpyForwarder_15 0] ReadCluster fc7d - fac1/01 Attribute: 0007 Type: 20 Size: 0001 Data: 1e
2022-12-21 20:05:47,240 INFO    : [ZigpyForwarder_15 0] ikea_air_purifier_cluster fac1/01 fc7d 0007 1e
2022-12-21 20:05:47,240 INFO    : [ZigpyForwarder_15 0]  --  Fan Speed: 1e
2022-12-21 20:05:47,518 INFO    : [ZigpyForwarder_15 0] inRawAps fac1/01 Cluster 0008 Manuf: 117c/IKEA of Sweden Command: 0b Data: 04c3 Payload: 084d0b04c3 not processed !!!
2022-12-21 20:05:54,192 INFO    : [ZigpyForwarder_15 0] ReadCluster fc7d - fac1/01 Attribute: 0006 Type: 20 Size: 0001 Data: 0a
[... cut ...]
2022-12-21 20:07:22,562 INFO    : [ZigpyForwarder_15 0] ReadCluster fc7d - fac1/01 Attribute: 0000 Type: 23 Size: 0004 Data: 00000c97
2022-12-21 20:07:22,564 INFO    : [ZigpyForwarder_15 0] ikea_air_purifier_cluster fac1/01 fc7d 0000 00000c97
2022-12-21 20:07:22,568 INFO    : [ZigpyForwarder_15 0] ReadCluster fc7d - fac1/01 Attribute: 0008 Type: 23 Size: 0004 Data: 00000c97
2022-12-21 20:07:22,569 INFO    : [ZigpyForwarder_15 0] ikea_air_purifier_cluster fac1/01 fc7d 0008 00000c97
2022-12-21 20:07:22,569 INFO    : [ZigpyForwarder_15 0]  --  Device runtime: 00000c97

```

rhuijben, 2022-12-21:
And off using button, and back on (which automatically goes to auto) a few seconds later
```
2022-12-21 20:07:29,812 INFO    : [ZigpyForwarder_15 0] ReadCluster fc7d - fac1/01 Attribute: 0007 Type: 20 Size: 0001 Data: 0f
2022-12-21 20:07:29,813 INFO    : [ZigpyForwarder_15 0] ikea_air_purifier_cluster fac1/01 fc7d 0007 0f
2022-12-21 20:07:29,813 INFO    : [ZigpyForwarder_15 0]  --  Fan Speed: 0f
2022-12-21 20:07:35,593 INFO    : [ZigpyForwarder_15 0] ReadCluster fc7d - fac1/01 Attribute: 0004 Type: 21 Size: 0002 Data: 0006
2022-12-21 20:07:35,594 INFO    : [ZigpyForwarder_15 0] ikea_air_purifier_cluster fac1/01 fc7d 0004 0006
2022-12-21 20:07:35,597 INFO    : [ZigpyForwarder_15 0]  --  PM25: 0006 --> 6
2022-12-21 20:07:59,693 INFO    : [ZigpyForwarder_15 0] ReadCluster fc7d - fac1/01 Attribute: 0007 Type: 20 Size: 0001 Data: 0a
2022-12-21 20:07:59,694 INFO    : [ZigpyForwarder_15 0] ikea_air_purifier_cluster fac1/01 fc7d 0007 0a
2022-12-21 20:07:59,695 INFO    : [ZigpyForwarder_15 0]  --  Fan Speed: 0a
2022-12-21 20:08:22,320 INFO    : [ZigpyForwarder_15 0] ReadCluster fc7d - fac1/01 Attribute: 0000 Type: 23 Size: 0004 Data: 00000c98
2022-12-21 20:08:22,322 INFO    : [ZigpyForwarder_15 0] ikea_air_purifier_cluster fac1/01 fc7d 0000 00000c98
2022-12-21 20:08:22,326 INFO    : [ZigpyForwarder_15 0] ReadCluster fc7d - fac1/01 Attribute: 0008 Type: 23 Size: 0004 Data: 00000c98
2022-12-21 20:08:22,326 INFO    : [ZigpyForwarder_15 0] ikea_air_purifier_cluster fac1/01 fc7d 0008 00000c98
2022-12-21 20:08:22,327 INFO    : [ZigpyForwarder_15 0]  --  Device runtime: 00000c98
2022-12-21 20:08:24,481 INFO    : [ZigpyForwarder_15 0] ReadCluster fc7d - fac1/01 Attribute: 0004 Type: 21 Size: 0002 Data: ffff
2022-12-21 20:08:24,483 INFO    : [ZigpyForwarder_15 0] ikea_air_purifier_cluster fac1/01 fc7d 0004 ffff
2022-12-21 20:08:24,483 INFO    : [ZigpyForwarder_15 0]  --  PM25: ffff --> 65535
2022-12-21 20:08:24,487 INFO    : [ZigpyForwarder_15 0] ReadCluster fc7d - fac1/01 Attribute: 0006 Type: 20 Size: 0001 Data: 00
2022-12-21 20:08:24,497 INFO    : [ZigpyForwarder_15 0] ikea_air_purifier_cluster fac1/01 fc7d 0006 00
2022-12-21 20:08:24,498 INFO    : [ZigpyForwarder_15 0]  --  Mode: 00
2022-12-21 20:08:24,499 INFO    : [ZigpyForwarder_15 0] ReadCluster fc7d - fac1/01 Attribute: 0007 Type: 20 Size: 0001 Data: 00
2022-12-21 20:08:24,500 INFO    : [ZigpyForwarder_15 0] ikea_air_purifier_cluster fac1/01 fc7d 0007 00
2022-12-21 20:08:24,500 INFO    : [ZigpyForwarder_15 0]  --  Fan Speed: 00
2022-12-21 20:08:36,101 INFO    : [ZigpyForwarder_15 0] ReadCluster fc7d - fac1/01 Attribute: 0006 Type: 20 Size: 0001 Data: 01
2022-12-21 20:08:36,103 INFO    : [ZigpyForwarder_15 0] ikea_air_purifier_cluster fac1/01 fc7d 0006 01
2022-12-21 20:08:36,103 INFO    : [ZigpyForwarder_15 0]  --  Mode: 01
2022-12-21 20:08:36,107 INFO    : [ZigpyForwarder_15 0] ReadCluster fc7d - fac1/01 Attribute: 0007 Type: 20 Size: 0001 Data: 1e
2022-12-21 20:08:36,108 INFO    : [ZigpyForwarder_15 0] ikea_air_purifier_cluster fac1/01 fc7d 0007 1e
2022-12-21 20:08:36,108 INFO    : [ZigpyForwarder_15 0]  --  Fan Speed: 1e
2022-12-21 20:08:49,188 INFO    : [ZigpyForwarder_15 0] ReadCluster fc7d - fac1/01 Attribute: 0004 Type: 21 Size: 0002 Data: 0003
2022-12-21 20:08:49,190 INFO    : [ZigpyForwarder_15 0] ikea_air_purifier_cluster fac1/01 fc7d 0004 0003
2022-12-21 20:08:49,190 INFO    : [ZigpyForwarder_15 0]  --  PM25: 0003 --> 3
```

pipiche38, 2022-12-21:
Ok, so in summary we get the informations

PM 2.5
Fan Speed
Mode
Run Time
Lock
Led Info

From an end user perspective what do we need

Switch On/Off
Fan Speed
Mode
PM 2.5
Alarm ( Green, Yellow, Red depending on the Left duration)

Led Info and Lock would be a device parameter

What do you think ?

rhuijben, 2022-12-21:
I think the mode, speed and on/off could+should be a single widget.

Child lock is mostly a setup option. And as I only see a led on the button specifying the current state I think that is an option to just hide the state.

Showing the current speed -in auto mode, and in the/a log- might be interesting. And I would be interested in seeing the air quality over time.

The filter should be replaced every +-six months, depending on usage says the manual so this is probably the alarm interval. There is a hidden bitten for resetting this next to the connect button.

pipiche38, 2022-12-21:
Thanks very much for your time.

So the Air Purifier is configured in a way that if nothing happen it will send reports (sensors info) every hour. Now if anything change on some of the sensors then it will be immediately reported.

I just have pushed an update, which tend to update domoticz widgets from the sensors informations.
PPM, Alarm, Mode and Speed. 
For now it is just on the device to pluginn to domoticz way.

For that you need to 
remove the existing paired widgets, to remove the device
update the plugin
restart the plugin
redo the pairing

You should get 4 widgets created "PM25/AirPurifierMode/FanSpeed/AirPurifierAlarm"
I was not sure if there is a Manual mode ? in the sense that for instance you can increase the Fan Speed ?

In any cases, this is not tested, so I cannot guaranty , but would be please to see your tests.
Watch the Domoticz logs (in case there are some python related errors)

rhuijben, 2022-12-21:
When the purifier is on, you can choose auto, 1, 2, 3, 4 or 5. At startup it defaults to auto. In my experience it always starts fast and then slows down. This looks like a digital dial button. When you press it, it goes to standby (off).

rhuijben, 2022-12-21:
Re-paired with current version.
Complete log from today on 
https://lpt1.nl/f/2022/PluginZigbee_15.log

There are some interesting python stacktraces after the repairing at about 23:12

rhuijben, 2022-12-22:
```
{"Version":"3","ZDeviceName":"","Status":"inDB","SQN":"3f","Heartbeat":"5433","RIA":"0","LQI":60,"Battery":{},"Model":"STARKVIND Air purifier","ForceAckCommands":[],"MacCapa":"8e","IEEE":"0c4314fffed61adf","Type":"PM25/FanControl","ProfileID":"0104","ZDeviceID":"0007","App Version":"16","Attributes List":{},"DeviceType":"FFD","HW Version":"1","Last Cmds":[],"LogicalType":"Router","Manufacturer":"117c","Manufacturer Name":"IKEA of Sweden","NbEp":"2","PowerSource":"Main","ReadAttributes":{"Ep":{"01":{"0000":{"TimeStamp":1671660671,"iSQN":{"4000":"0c","0000":"0a","0001":"0a","0002":"0a","0003":"0b","0004":"0b","0005":"0b","0007":"0c","000a":"0c"},"Attributes":{"4000":"00","0000":"00","0001":"00","0002":"00","0003":"00","0004":"00","0005":"00","0007":"00","000a":"00"},"ZigateRequest":{}},"fc7d":{"TimeStamp":1671687804,"iSQN":{},"Attributes":{"0006":"00","0007":"00","0004":"00","0000":"00","0008":"00","0001":"00","0002":"00","0003":"00","0005":"00"},"ZigateRequest":{}}},"f2":{"0000":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}}}}},"ReceiveOnIdle":"","Stack Version":"103","Stamp":{"time":1671687804.519641,"Time":"2022-12-22 06:43:24","MsgType":"8102","LastSeen":1671687804},"ZCL Version":"3","Health":"Live","Capability":["Full-Function Device","Main Powered","Receiver during Idle","Standard security","NwkAddr should be allocated"],"ZDPSQN":"0c","Epv2":{"01":{"ClusterIn":{"0000":{},"0003":{},"0004":{},"0005":{},"0202":{},"fc57":{},"fc7d":{}},"ClusterOut":{"0019":{},"0400":{},"042a":{}},"ProfileID":"0104","ZDeviceID":"0007"},"f2":{"ClusterIn":{},"ClusterOut":{},"ProfileID":{},"ZDeviceID":{}}},"ZCLSQN":"28","RollingLQI":[57,57,57,57,57,57,54,57,60,57,60],"ConfigSource":"DeviceConf","Ep":{"01":{"0000":{"4000":"1.0.033","0004":"IKEA of Sweden","0005":"STARKVIND Air purifier","0000":"3","0001":"16","0002":"103","0003":"1","0007":"1","000a":"4532303037"},"0003":{},"0004":{},"0005":{},"0202":{},"fc57":{},"fc7d":{"0006":"01","0007":"0a","0004":"000c","0000":"00000f13","0008":"00000f13","0001":"00","0002":"0003f480","0003":"00","0005":"00"},"0019":{},"0400":{},"042a":{},"Type":"PM25/FanControl","ClusterType":{"1975":"PM25","1976":"FanControl"}},"f2":{"Type":""}},"WriteAttributes":{"Ep":{"01":{"0000":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"0003":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"0004":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"0005":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"0202":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"fc57":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"fc7d":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"0019":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"0400":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"042a":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}}}}},"PairingInProgress":false,"PairingTime":1671660662.0767016,"Bind":{"01":{"0202":{"Target":"0000","Stamp":1671660663,"Phase":"binded","Status":"00","i_sqn":"03"},"0400":{"Target":"0000","Stamp":1671660663,"Phase":"binded","Status":"00","i_sqn":"04"},"042a":{"Target":"0000","Stamp":1671660663,"Phase":"binded","Status":"00","i_sqn":"05"},"0019":{"Target":"0000","Stamp":1671660664,"Phase":"binded","Status":"00","i_sqn":"06"},"fc7d":{"Target":"0000","Stamp":1671660664,"Phase":"binded","Status":"00","i_sqn":"07"},"fc57":{"Target":"0000","Stamp":1671660665,"Phase":"binded","Status":"00","i_sqn":"08"}}},"BindingTable":{"SQN":0,"Status":"Success","TimeStamp":1671660671.611222,"BindingTableSize":6,"Devices":[{"0c4314fffed61adf":{"sourceEp":"01","Cluster":"0202","targetIEEE":"00124b0025e14d39","targetNickName":"Zigbee Coordinator","targetEp":"01"}},{"0c4314fffed61adf":{"sourceEp":"01","Cluster":"0400","targetIEEE":"00124b0025e14d39","targetNickName":"Zigbee Coordinator","targetEp":"01"}},{"0c4314fffed61adf":{"sourceEp":"01","Cluster":"042a","targetIEEE":"00124b0025e14d39","targetNickName":"Zigbee Coordinator","targetEp":"01"}},{"0c4314fffed61adf":{"sourceEp":"01","Cluster":"0019","targetIEEE":"00124b0025e14d39","targetNickName":"Zigbee Coordinator","targetEp":"01"}},{"0c4314fffed61adf":{"sourceEp":"01","Cluster":"fc7d","targetIEEE":"00124b0025e14d39","targetNickName":"Zigbee Coordinator","targetEp":"01"}},{"0c4314fffed61adf":{"sourceEp":"01","Cluster":"fc57","targetIEEE":"00124b0025e14d39","targetNickName":"Zigbee Coordinator","targetEp":"01"}}]},"ConfigureReporting":{"Ep":{"01":{"fc7d":{"TimeStamp":1671660662.083726,"iSQN":{"0000":"01","0001":"02","0002":"03","0003":"04","0004":"05","0005":"06","0006":"07","0007":"08","0008":"09"},"Attributes":{"0000":"00","0001":"00","0002":"00","0003":"00","0004":"00","0005":"00","0006":"00","0007":"00","0008":"00"},"ZigateRequest":{}}}}},"CertifiedDevice":true,"ZDeviceVersion":"0001","Max Buffer Size":"52","Max Rx":"0052","Max Tx":"0052","macapa":"8e","bitfield":"4001","server_mask":"2c00","descriptor_capability":"00","ZCL-IN-SQN":{"fc7d":"3f","0019":"34"},"SWBUILD_3":"1.0.033","ReadConfigureReporting":{"Ep":{"01":{"fc7d":{"0000":{"TimeStamp":1671660685.1107764,"Status":"86"},"0001":{"TimeStamp":1671660685.111056,"Status":"86"},"0002":{"TimeStamp":1671660685.1113136,"Status":"86"},"0003":{"TimeStamp":1671660685.5045652,"Status":"86"},"0004":{"TimeStamp":1671660685.5047858,"Status":"86"},"0005":{"TimeStamp":1671660685.5050013,"Status":"86"},"0006":{"TimeStamp":1671660685.9039292,"Status":"86"},"0007":{"TimeStamp":1671660685.904151,"Status":"86"},"0008":{"TimeStamp":1671660685.904368,"Status":"86"}}}},"TimeStamp":1671660685.9043603},"GroupMemberShip":{"01":{}},"OTAClient":{"ManufacturerCode":"117c","ImageType":"110c","CurrentImageVersion":"00010033"}}
```

pipiche38, 2022-12-22:
Please try again with the latest pushed version

rhuijben, 2022-12-22:
Did that.

New log on https://lpt1.nl//PluginZigbee-20221222.log

```
{"Version":"3","ZDeviceName":"","Status":"inDB","SQN":"37","Heartbeat":"527","RIA":"0","LQI":33,"Battery":{},"Model":"STARKVIND Air purifier","ForceAckCommands":[],"MacCapa":"8e","IEEE":"0c4314fffed61adf","Type":"PM25/AirPurifierMode/FanSpeed/AirPurifierAlarm","ProfileID":"0104","ZDeviceID":"0007","App Version":"16","Attributes List":{},"DeviceType":"FFD","HW Version":"1","Last Cmds":[],"LogicalType":"Router","Manufacturer":"117c","Manufacturer Name":"IKEA of Sweden","NbEp":"2","PowerSource":"Main","ReadAttributes":{"Ep":{"01":{"0000":{"TimeStamp":1671695984,"iSQN":{"4000":"0c","0000":"0a","0001":"0a","0002":"0a","0003":"0b","0004":"0b","0005":"0b","0007":"0c","000a":"0c"},"Attributes":{"4000":"00","0000":"00","0001":"00","0002":"00","0003":"00","0004":"00","0005":"00","0007":"00","000a":"00"},"ZigateRequest":{}},"fc7d":{"TimeStamp":1671698302,"iSQN":{},"Attributes":{"0006":"00","0007":"00","0004":"00"},"ZigateRequest":{}}},"f2":{"0000":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}}}}},"ReceiveOnIdle":"","Stack Version":"103","Stamp":{"time":1671698302.194454,"Time":"2022-12-22 09:38:22","MsgType":"8102","LastSeen":1671698302},"ZCL Version":"3","Health":"Live","Capability":["Full-Function Device","Main Powered","Receiver during Idle","Standard security","NwkAddr should be allocated"],"ZDPSQN":"0c","Epv2":{"01":{"ClusterIn":{"0000":{},"0003":{},"0004":{},"0005":{},"0202":{},"fc57":{},"fc7d":{}},"ClusterOut":{"0019":{},"0400":{},"042a":{}},"ProfileID":"0104","ZDeviceID":"0007"},"f2":{"ClusterIn":{},"ClusterOut":{},"ProfileID":{},"ZDeviceID":{}}},"ZCLSQN":"28","RollingLQI":[42,48,45,30,39,30,39,39,45,33,33],"ConfigSource":"DeviceConf","Ep":{"01":{"0000":{"4000":"1.0.033","0004":"IKEA of Sweden","0005":"STARKVIND Air purifier","0000":"3","0001":"16","0002":"103","0003":"1","0007":"1","000a":"4532303037"},"0003":{},"0004":{},"0005":{},"0202":{},"fc57":{},"fc7d":{"0006":"00","0007":"1e","0004":"ffff"},"0019":{},"0400":{},"042a":{},"Type":"PM25/AirPurifierMode/FanSpeed/AirPurifierAlarm","ClusterType":{"1979":"PM25","1980":"FanSpeed","1981":"AirPurifierAlarm","1982":"AirPurifierAlarm"}},"f2":{"Type":""}},"WriteAttributes":{"Ep":{"01":{"0000":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"0003":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"0004":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"0005":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"0202":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"fc57":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"fc7d":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"0019":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"0400":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"042a":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}}}}},"PairingInProgress":false,"PairingTime":1671695975.0952864,"Bind":{"01":{"0202":{"Target":"0000","Stamp":1671695976,"Phase":"binded","Status":"00","i_sqn":"03"},"0400":{"Target":"0000","Stamp":1671695976,"Phase":"binded","Status":"00","i_sqn":"04"},"042a":{"Target":"0000","Stamp":1671695977,"Phase":"binded","Status":"00","i_sqn":"05"},"0019":{"Target":"0000","Stamp":1671695977,"Phase":"binded","Status":"00","i_sqn":"06"},"fc7d":{"Target":"0000","Stamp":1671695977,"Phase":"binded","Status":"00","i_sqn":"07"},"fc57":{"Target":"0000","Stamp":1671695978,"Phase":"binded","Status":"00","i_sqn":"08"}}},"BindingTable":{"SQN":0,"Status":"Success","TimeStamp":1671695985.2881231,"BindingTableSize":6,"Devices":[{"0c4314fffed61adf":{"sourceEp":"01","Cluster":"0202","targetIEEE":"00124b0025e14d39","targetNickName":"Zigbee Coordinator","targetEp":"01"}},{"0c4314fffed61adf":{"sourceEp":"01","Cluster":"0400","targetIEEE":"00124b0025e14d39","targetNickName":"Zigbee Coordinator","targetEp":"01"}},{"0c4314fffed61adf":{"sourceEp":"01","Cluster":"042a","targetIEEE":"00124b0025e14d39","targetNickName":"Zigbee Coordinator","targetEp":"01"}},{"0c4314fffed61adf":{"sourceEp":"01","Cluster":"0019","targetIEEE":"00124b0025e14d39","targetNickName":"Zigbee Coordinator","targetEp":"01"}},{"0c4314fffed61adf":{"sourceEp":"01","Cluster":"fc7d","targetIEEE":"00124b0025e14d39","targetNickName":"Zigbee Coordinator","targetEp":"01"}},{"0c4314fffed61adf":{"sourceEp":"01","Cluster":"fc57","targetIEEE":"00124b0025e14d39","targetNickName":"Zigbee Coordinator","targetEp":"01"}}]},"ConfigureReporting":{"Ep":{"01":{"fc7d":{"TimeStamp":1671695975.119382,"iSQN":{"0000":"01","0001":"02","0002":"03","0003":"04","0004":"05","0005":"06","0006":"07","0007":"08","0008":"09"},"Attributes":{"0000":"00","0001":"00","0002":"00","0003":"00","0004":"00","0005":"00","0006":"00","0007":"00","0008":"00"},"ZigateRequest":{}}}}},"CertifiedDevice":true,"ZDeviceVersion":"0001","Max Buffer Size":"52","Max Rx":"0052","Max Tx":"0052","macapa":"8e","bitfield":"4001","server_mask":"2c00","descriptor_capability":"00","ZCL-IN-SQN":{"fc7d":"37","0019":"25"},"SWBUILD_3":"1.0.033","ReadConfigureReporting":{"Ep":{"01":{"fc7d":{"0000":{"TimeStamp":1671695999.0030675,"Status":"86"},"0001":{"TimeStamp":1671695999.0032916,"Status":"86"},"0002":{"TimeStamp":1671695999.0035093,"Status":"86"},"0003":{"TimeStamp":1671695999.407003,"Status":"86"},"0004":{"TimeStamp":1671695999.4072251,"Status":"86"},"0005":{"TimeStamp":1671695999.4074442,"Status":"86"},"0006":{"TimeStamp":1671695999.8128247,"Status":"86"},"0007":{"TimeStamp":1671695999.8130445,"Status":"86"},"0008":{"TimeStamp":1671695999.8132606,"Status":"86"}}}},"TimeStamp":1671695999.8132524},"GroupMemberShip":{"01":{}},"OTAClient":{"ManufacturerCode":"117c","ImageType":"110c","CurrentImageVersion":"00010033"}}
```

rhuijben, 2022-12-23:
@pipiche38 thank you for adding support. Support is now merged to the dev branch.

closed 2022-12-23 by rhuijben