zigbeefordomoticz/Domoticz-Zigbee #1392: Coordinator type tcp but serial port can be local only, should be user set for remote device socket/serial port too.

jussipo, 2022-12-22:
When coordinator is tcp, also serial port should be editable, not local.
Example using fixed Lidl ZB gateway, serial port is remote and socket redirector is used, only way to fix is just force on plugin.py code:
Parameters["SerialPort"] = "socket://192.168.1.46:8888".
So maybe used editable serialport name?

pipiche38, 2022-12-22:
That I a good point.
Would you mind proposing a pR ?

pipiche38, 2022-12-22:
Please switch to dev branch ( git checkout dev; git pull ) and make sure to restart Domoticz.

```
Coordinator Model:  Silicon Labs ...
Coordinator Type : Socket
IP: 192.168.1.46:8888
```

Available on 6.4.104
Commit: 81036318

pipiche38, 2022-12-30:
@jussipo  could you let us know if this commit (integrated on dev branch) fixe the issue ?

closed 2023-01-02 by pipiche38