zigbeefordomoticz/Domoticz-Zigbee #1394: Adeo/Lexman Smart plug 16A - LDSENK02F

Overnuts, 2022-12-22:
Adeo/Lexman Smart plug 16A - LDSENK02F

https://www.leroymerlin.fr/produits/electricite-domotique/domotique-et-objets-connectes/domotique/solutions-de-commande/prise-connectee-16a-3680w-repeteur-lexman-82428895.html

Vue dans domoticz en plusieurs devices:
Light/Switch	Switch	On
Usage	Electric	0 Watt
General	kWh	0.014 kWh
Usage	Electric	0 Watt
General	kWh	0.014 kWh
General	Current	0 A

bon courage ! (et joyeux noel :-) )

`{"Version":"3","ZDeviceName":"","Status":"inDB","SQN":"19","Ep":{"01":{"0000":{"4000":"2.6","0004":"ADEO","0005":"LDSENK02F","0000":"3","0001":"1","0002":"2","0003":"1","0006":"20200217-586","0007":"1","000a":"303130343030313332"},"0003":{},"0004":{},"0005":{},"0006":{"4001":"0","4002":"0","0000":"01"},"0702":{"0000":0,"0200":"Ok","0301":1,"0302":100,"0303":42,"0306":"Electric Metering","0400":"0.0"},"0b04":{"050b":0,"0505":0,"0508":0},"fc82":{},"000a":{},"0019":{},"ClusterType":{"852":"Switch","853":"Power","854":"Meter","855":"Power","856":"Meter","857":"Ampere"}},"f2":{}},"Heartbeat":"252","RIA":"4","LQI":78,"Battery":{},"Model":"LDSENK02F","ForceAckCommands":[],"MacCapa":"8e","IEEE":"0c4314fffe5bf435","Type":"Switch/Power/Meter/Ampere","ProfileID":"0104","ZDeviceID":"0051","App Version":"1","Attributes List":{},"DeviceType":"FFD","HW Version":"1","Last Cmds":[],"LogicalType":"Router","Manufacturer":"1277","Manufacturer Name":"ADEO","NbEp":"2","PowerSource":"Main","ReadAttributes":{"Ep":{"01":{"0000":{"TimeStamp":1671727184,"iSQN":{"4000":"10","0004":"0c","0005":"0c","0000":"0c","0001":"0d","0002":"0d","0003":"0d","0006":"0e","0007":"0e","000a":"0e","000f":"0f","0010":"0f","0015":"0f","f000":"10"},"Attributes":{"4000":"00","0004":"00","0005":"00","0000":"00","0001":"00","0002":"00","0003":"00","0006":"00","0007":"00","000a":"00","000f":"86","0010":"86","f000":"86"},"ZigateRequest":{}},"0019":{"TimeStamp":1671727177,"iSQN":{"0001":"04","0002":"04","0003":"04","0004":"05","0005":"05","0006":"05","0007":"06","0008":"06","0009":"06","000a":"07"},"Attributes":{"0001":"86","0002":"86","0004":"86","0005":"86","0007":"86","0008":"86","000a":"86"},"ZigateRequest":{}},"0702":{"TimeStamp":1671731042,"iSQN":{"0000":"08","0017":"08","0200":"08","0301":"09","0302":"09","0303":"09","0306":"0a","0400":"0a"},"Attributes":{"0000":"00","0017":"86","0200":"00","0301":"00","0302":"00","0303":"00","0306":"00","0400":"00"},"ZigateRequest":{}},"0006":{"TimeStamp":1671731045,"iSQN":{"4001":"0b","4002":"0b","0000":"29"},"Attributes":{"4001":"00","4002":"00","0000":"00"},"ZigateRequest":{}},"0b04":{"TimeStamp":1671731046,"iSQN":{"050b":"11","0505":"11","0508":"11"},"Attributes":{"050b":"00","0505":"00","0508":"00"},"ZigateRequest":{}}},"f2":{"0000":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}}}}},"ReceiveOnIdle":"On","Stack Version":"2","Stamp":{"time":1671731046.6523097,"Time":"2022-12-22 18:44:06","MsgType":"8102","LastSeen":1671731046},"ZCL Version":"3","Health":"Live","Capability":["Full-Function Device","Main Powered","Receiver during Idle","Standard security","NwkAddr should be allocated"],"ZDPSQN":"10","PairingInProgress":false,"Epv2":{"01":{"ClusterIn":{"0000":{},"0003":{},"0004":{},"0005":{},"0006":{},"0702":{},"0b04":{},"fc82":{}},"ClusterOut":{"0003":{},"000a":{},"0019":{}},"ProfileID":"0104","ZDeviceID":"0051"},"f2":{"ClusterIn":{},"ClusterOut":{},"ProfileID":{},"ZDeviceID":{}}},"ZCLSQN":"29","ZCL-IN-SQN":{"0000":"10","000a":"08","0006":"18","0702":"16","0b04":"19","0019":"09","0004":"15"},"RollingLQI":[75,78,75,78,75,78,75,81,78,78,78],"ZDeviceVersion":"0001","ConfigSource":"8043","Max Buffer Size":"52","Max Rx":"0052","Max Tx":"0052","macapa":"8e","bitfield":"4001","server_mask":"2c00","descriptor_capability":"00","SQN_000a":"08","PairingTime":1671727161.1549194,"Bind":{"01":{"0000":{"Target":"0000","Stamp":1671727163,"Phase":"binded","Status":"00","i_sqn":"03"},"0003":{"Target":"0000","Stamp":1671727165,"Phase":"binded","Status":"00","i_sqn":"04"},"0004":{"Target":"0000","Stamp":1671727165,"Phase":"binded","Status":"00","i_sqn":"05"},"0005":{"Target":"0000","Stamp":1671727166,"Phase":"binded","Status":"00","i_sqn":"06"},"0006":{"Target":"0000","Stamp":1671727167,"Phase":"binded","Status":"00","i_sqn":"07"},"0702":{"Target":"0000","Stamp":1671727167,"Phase":"binded","Status":"00","i_sqn":"08"},"0b04":{"Target":"0000","Stamp":1671727168,"Phase":"binded","Status":"00","i_sqn":"09"},"fc82":{"Target":"0000","Stamp":1671727169,"Phase":"binded","Status":"00","i_sqn":"0a"}}},"BindingTable":{"SQN":0,"Status":"Success","TimeStamp":1671727197.8296173,"BindingTableSize":8,"Devices":[{"0c4314fffe5bf435":{"sourceEp":"01","Cluster":"0000","targetIEEE":"00124b0025e14d57","targetNickName":"Zigbee Coordinator","targetEp":"01"}},{"0c4314fffe5bf435":{"sourceEp":"01","Cluster":"0003","targetIEEE":"00124b0025e14d57","targetNickName":"Zigbee Coordinator","targetEp":"01"}},{"0c4314fffe5bf435":{"sourceEp":"01","Cluster":"0004","targetIEEE":"00124b0025e14d57","targetNickName":"Zigbee Coordinator","targetEp":"01"}},{"0c4314fffe5bf435":{"sourceEp":"01","Cluster":"0005","targetIEEE":"00124b0025e14d57","targetNickName":"Zigbee Coordinator","targetEp":"01"}},{"0c4314fffe5bf435":{"sourceEp":"01","Cluster":"0006","targetIEEE":"00124b0025e14d57","targetNickName":"Zigbee Coordinator","targetEp":"01"}},{"0c4314fffe5bf435":{"sourceEp":"01","Cluster":"0702","targetIEEE":"00124b0025e14d57","targetNickName":"Zigbee Coordinator","targetEp":"01"}},{"0c4314fffe5bf435":{"sourceEp":"01","Cluster":"0b04","targetIEEE":"00124b0025e14d57","targetNickName":"Zigbee Coordinator","targetEp":"01"}},{"0c4314fffe5bf435":{"sourceEp":"01","Cluster":"fc82","targetIEEE":"00124b0025e14d57","targetNickName":"Zigbee Coordinator","targetEp":"01"}}]},"ConfigureReporting":{"Ep":{"01":{"0006":{"TimeStamp":1671727162.0629704,"iSQN":{"0000":"01"},"Attributes":{"0000":"00"},"ZigateRequest":{}},"0702":{"TimeStamp":1671727162.148108,"iSQN":{"0000":"02","0400":"02"},"Attributes":{"0000":"00","0400":"00"},"ZigateRequest":{}},"0b04":{"TimeStamp":1671727162.2723982,"iSQN":{"0505":"03","0508":"03","050b":"03"},"Attributes":{"0505":"00","0508":"00","050b":"00"},"ZigateRequest":{}}}}},"CertifiedDevice":false,"SWBUILD_1":"20200217-586","SWBUILD_3":"2.6","ReadConfigureReporting":{"Ep":{"01":{"0006":{"0000":{"TimeStamp":1671729841.2356899,"Status":"00","DataType":"10","MinInterval":"0001","MaxInterval":"012c"}},"0702":{"0000":{"TimeStamp":1671729841.4827986,"Status":"00","DataType":"25","MinInterval":"0001","MaxInterval":"fffe","Change":"000000000001"},"0400":{"TimeStamp":1671729841.5359375,"Status":"00","DataType":"2a","MinInterval":"0001","MaxInterval":"012c","Change":"00000a"}},"0b04":{"0505":{"TimeStamp":1671729841.840229,"Status":"00","DataType":"21","MinInterval":"0001","MaxInterval":"012c","Change":"0001"},"0508":{"TimeStamp":1671729841.8609362,"Status":"00","DataType":"21","MinInterval":"0001","MaxInterval":"012c","Change":"0001"},"050b":{"TimeStamp":1671729841.8932521,"Status":"00","DataType":"29","MinInterval":"0005","MaxInterval":"012c","Change":"0001"}}}},"TimeStamp":1671729841.893188},"GroupMemberShip":{"01":{}},"WriteAttributes":{"Ep":{"01":{"0000":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"0003":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"0004":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"0005":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"0006":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"0702":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"0b04":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"fc82":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"000a":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"0019":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}}}}},"OTAClient":{"ManufacturerCode":"1277","ImageType":"a135","CurrentImageVersion":"21326230"}}`

pipiche38, 2022-12-22:
integrated on dev branch
commit: 0c2a3f44
version 6.4.103

@Overnuts if you can, it would be great to be on 'dev' branch and test it. Ideally it would be good if you can remove the existing pairing devices (all of them) and then repair the device

Overnuts, 2022-12-24:
Bonjour @pipiche38 , je suis passé sur "dev", restart complet de mon serveur,
 j'ai appairé une nouvelle prise, elle est affichée en "optimisée" et dans domoticz, j'ai bien : 
- un device "wall socket" qui a été créé
- un compteur "general kWh" 
- un compteur "usage, electric"
- un compteur "general, current"   
Les compteurs en Ampères ont l'air de rester à 0

Pour la première prise, je doit virer des compteurs qui sont en doubles.

merci pour le plugin, et joeux noel   °<:-)

pipiche38, 2022-12-24:
Good, feel free to remove the non-active widgets.

closed 2022-12-24 by pipiche38

Overnuts, 2022-12-31:
Hi @pipiche38 back from leroy merlin shop (again :-) ) with a third piece.
Successfully seen from coordinator, plugin and all, 
but I noticed the "general kwh" counter is set to "computed" in "energy read" radio instead of "from device"

pipiche38, 2022-12-31:
Computed is because the device doesn't send the summation 
If you consider this is not correct pleas do open a new issue with the necessary information