zigbeefordomoticz/Domoticz-Zigbee #1397: AirBNK M500 - Zigbee

pipiche38, 2022-12-23:
Tuya Model:

[protocol_SweetHome_20220323-1_eng.pdf](https://github.com/zigbeefordomoticz/Domoticz-Zigbee/files/10294650/protocol_SweetHome_20220323-1_eng.pdf)

[export.json.txt](https://github.com/zigbeefordomoticz/Domoticz-Zigbee/files/10294628/export.json.txt)

https://developer.tuya.com/en/docs/iot/zigbee-doorlock-function-developer?id=K9fe9esofkzlf

this is the Data Point Code numbers

-« SwitchDirection » dp 16
-« AutoLockTime » dp 19
-« Auxiliary opening/locking » dp 101
-« Answer App »
-« Unlocked » dp38
-« Open Or Close » dp 8
-« App Unlock Without Password » dp 40
-« Alarm » dp 9
-« App Unlock » dp 41
-« Battery » dp 10

pipiche38, 2022-12-23:
developement going on device/tuya/smart-key-TY0A01

Polo9222, 2022-12-23:
[M500.zip](https://github.com/zigbeefordomoticz/Domoticz-Zigbee/files/10296735/M500.zip)
Voilà , je pense avoir bien suivi les instructions

Je n'ai besoin que de commander ouverture et fermeture. Eventuellement l'état des piles.
Actuellement je tourne en BlueTooth de mon téléphone avec l'appli Wehere qui a les mêmes fonctions que la Danalock 
- assistance on/off
- fermeture automatique
- calibrage
- sens d'ouverture de porte ( automatique sue la Danalock)  
Ces fonctions , je continuerai à les faire avec l'appli.
Good luck et à ta dispo

Polo9222, 2022-12-23:
J'ai oublié de te dire que j'avais fait des aller/retours sur la serrure avec l'assistance. On les aperçoit dans le log

pipiche38, 2022-12-24:
A priori y-a eu un soucis car pas ou peu d'info. 
Peux-tu activer le mode debug suivant

via WebUI > Tools > Debug Advanced
```
debugTuya
debugInput
debuginRawAPS
```

tu refais des manip sur le doorlock via to App et ou sur la serrure, et ensuite tu droppes les 2 fichiers

Polo9222, 2022-12-24:
[Domoticz_log.txt](https://github.com/zigbeefordomoticz/Domoticz-Zigbee/files/10298389/Domoticz_log.txt)
[PluginZigbee_03.log](https://github.com/zigbeefordomoticz/Domoticz-Zigbee/files/10298390/PluginZigbee_03.log)

J'ai supprimé le device, ré appairé et fait 3 A/R sur la serrure sans passer par l'appli.

Joyeux Noel

pipiche38, 2022-12-24:
Bon c'est pas mieux ... a priori le device ne cause pas.
Donc la phase suivante c'est voir comment ca fonctionne avec une passerelle Tuya et sniffer le trafic

Polo9222, 2022-12-24:
J'ai une passerelle que je n'ai pas encore réformée. Si tu me dis comment sniffer , je veux bien tenter .

pipiche38, 2022-12-30:
Il te faut une clef pour sniffer type CC2531 et utiliser un logiciel qui s'appelle wireshark

Polo9222, 2022-12-31:
J'ai une clé sonoff dongle  USB elm32, une passerelle Wi-Fi tuya, un pc, une serrure, une TV box armbian, l'appli airbnk sur mon Android.
J'ai vu que le sniffer suivait les messages wifi. Quels échanges dois-je tracer ?

Tous mes vœux pour 2023.
Amicalement
Paul

pipiche38, 2022-12-31:
et bien il faut faire une capture Zigbee des echanges entre la Gateway Zigbee et la serrure.

En gros, il faut

capturer l'appairage de la serrure sur la GW
capturer les echanges GW <-> serrure quand on agit sur la serrure
capturer chaque ordre de ton Appli AirBnk qui génére un dialiogue GW <-> Serrure

Et c'est bien le Zigbee qu'il faut capturer, pas le wifi.

Par contre je ne sais pas si on peut mettre un firmware de sniff sur la sonoff

SylvainPer, 2022-12-31:
Il y a un firmware pour différentes versions de clés : https://www.ti.com/tool/PACKET-SNIFFER

pipiche38, 2023-01-02:
please try the latest version on the branch

pipiche38, 2023-01-14:
To complex to integrate. 
I'll close it

closed 2023-01-14 by pipiche38