zigbeefordomoticz/Domoticz-Zigbee #1403: Unable to add Aqara lumi.weather device

strethill, 2022-12-28:
Domoticz \ Zigate Environment:
     Raspbian GNU/Linux 10 (buster) in RPi 3b
     Zigate USB - Firmware 0321, Plugin 6.3.008

I have a 20 Aqara Temperature and Humidity devices connected to a USB Zigate. 
I recently upgraded the Plugin from 6.3.007.
I am trying to add a new Aqara Temperature and Humidity device but without success. I have added may devices before without any issues.

The normal process I use to add a new device is:
     1 Select the "Add New Hardware" switch on the Zigate Dashboard.
     2 Verify that the blue led on the Zigate USB is flashing.
     3 Hold down the Reset button on the Zigbee device for 5 seconds or until its led flashes
     4 The Zigate Blue LED blinks rapidly three or four times.
     5 The device LED will then blink rapidly three or four times.
     6 The Dashboard will show the new device and I then configure the name.

The issue I am experiencing is when I get to steps 4 and 5. 
    - The Zigate USB will blink rapidly three of four times. 
    - The device does not blink. 
    - The Zigate USB blinks rapidly  three of four times for a second time. 
    - The Device blinks once. 
    - The Zigate USB then returns to its normal "Add New Hardware" flash.
    - No device has been added to the dashboard  

Please could you advise how I can resolve this issue.

pipiche38, 2022-12-28:
Please kindly use the forum ( https://www.domoticz.com/forum/viewforum.php?f=68 or https://easydomoticz.com/forum/viewforum.php?f=28&sid=8210a1e3252b3c3ecc2462c0eae593a8 )

I guess you are in the situation where the Zigate do not accept any more devices. 
If you have only End Devices ( battery based) I even think that the limit is 20

closed 2022-12-28 by pipiche38

strethill, 2022-12-28:
> Please kindly use the forum ( https://www.domoticz.com/forum/viewforum.php?f=68 or https://easydomoticz.com/forum/viewforum.php?f=28 <https://easydomoticz.com/forum/viewforum.php?f=28&sid=8210a1e3252b3c3ecc2462c0eae593a8> &sid=8210a1e3252b3c3ecc2462c0eae593a8 )

Many thanks for your quick response.

 

I was running with 26 battery devices but one dropped off the network and I was trying to re-add it:

 

I am aware of the 200 device limitation but was not aware of 20 Battery Device restriction.

Please can you advise where I can find the documentation about this limitation.

Is there a workaround?

If I upgrade to the Zigate+ (V2), does this device support 20+ battery devices.

 

Thanks in advance.

John

pipiche38, 2022-12-29:
Find here some information that we have tried to document .
So for sure Zigate V2 is limited to 20, Zigate V1 seem to be limited to 20 if you are using the OptiPDM firmware version.
If not I guess 40 .

Now if you are not able to pair the device any more this could be related to one of the memory corruption that the Zigate is having. And the only way to get rid of it is to erase the PDM and re-pair all devices

https://github.com/zigbeefordomoticz/wiki/blob/b985bf10e9a2c4dd7d2116323089bdab50de8cc7/en-eng/Coordinators.md#coordinators-with-nxp-chipset

strethill, 2022-12-29:
Hi Pipiche,

 

Many thanks for your reply. Once again your help is always very much appreciated.

 

My Zigate V1 has the legacy firmware – when I last did the firmware upgrade, I decided that re-pairing of all devices would be very painful, and did not upgrade to the OptiPDM version. Which was probably a good thing at the time

 

It would appear as though my Zigate has a memory corruption issue as I am getting some odd error messages (see below) and the Mesh Diagram after a scan is only showing some routers (only 3 of 5) and none of my 26 end devices.

 

2022-12-29 10:12:57.811 Error: Zigate: Detected a PDM load, result of a ZiGate reset of (crash): #1

2022-12-29 10:13:10.403 Error: Zigate: decode8000 - command miss-match 004E vs. 0530

2022-12-29 10:13:18.475 Error: Zigate: decode8000 - command miss-match 0530 vs. 0040

2022-12-29 10:13:18.583 Error: Zigate: decode8000 - command miss-match 0040 vs. 004e

 

 

 

Can you confirm that the above are further symptoms of memory corruption. If so, I will bite the bullet and erase and reflash the PDM.

 

Also, my Zigate is reporting a non-optimised device. It a Samsung SmartThings device:

Name:                   Smart Plug (2019) 

Model:                 7A-PL-Z-J3

Zigbee ID:            Zb-ONOFFPlug-D0005

 

The device is working as expected in Domoticz – I just get the Orange cross icon in the Optimize column on the Management page

I have attached the Object Info of this device, copied from the Management page.

I have not raised an issue for this device, as I wanted to make sure that this “error” is not related to the memory corruption.

 

Please let me now if you want me to create a new issue for this non-optimised device.

 

Best regards

John

 

https://github.com/zigbeefordomoticz/wiki/blob/b985bf10e9a2c4dd7d2116323089bdab50de8cc7/en-eng/Coordinators.md#coordinators-with-nxp-chipset

pipiche38, 2022-12-31:
For non-optimized devices, please have a look to the wiki, which is mentioned on the dashboard itself
The link is here:https://github.com/zigbeefordomoticz/wiki/blob/master/en-eng/Problem_Dealing-with-none-optimized-device.md