zigbeefordomoticz/Domoticz-Zigbee #1407: Error: zcl_attribute_discovery_request not implemented for RAW mode

OosieW, 2022-12-29:
Yesterday I added 2 Philips devices to my Zigbee-for-Domoticz installation. The dimmer (RWL021) was added without problems, but the ceiling lamp (LTC002 - Hue 'Fair') gave errors. The lamp is working, from Domoticz I can switch on/off and choose color (white range) and intensity, but the 'DZ dashboard' list the device as 'non-optimized'. In Domoticz logging shows this error message: 
` Error: Sonoff Zigbee 3.0 USB: zcl_attribute_discovery_request not implemented for RAW mode`
My coordinator is a Sonoff Zigbee 3.0 USB stick. Working with last (stable) firmware en plugin versions.

Who knows de cause of this errors?

pipiche38, 2022-12-29:
as stated when creating an issue, Your first place to get support is via the Forums.

    English channel : https://www.domoticz.com/forum/viewforum.php?f=68
    French Channel : https://easydomoticz.com/forum/viewforum.php?f=28&sid=a8633a9d8cb2acccb32872543b50fd47

this is a known error, we are just waiting for someone to implement this feature

closed 2022-12-29 by pipiche38

pipiche38, 2022-12-29:
For the Lamp, please follow the link in regards to Optimized device and you'll know what it is and what do