zigbeefordomoticz/Domoticz-Zigbee #1416: Ikea Tradfri lightbulb not supported

manjh, 2023-01-03:
[image]

1. Got to the Tools Menu
1. Select Plugin Raw Devices (json)
1. Export

This will create an export.json file that you can save.
   
Finally you create an issue on GitHub with the following details

* Product Name 
* Brand
* URL on a web site
* What do you expect
* The logs from the pairing process
* export.json file as described here above
[export zipped.zip](https://github.com/zigbeefordomoticz/Domoticz-Zigbee/files/10335532/export.zipped.zip)

pipiche38, 2023-01-03:
If I'm not mistaken from the IKEA Vendor it seems to be a On/Off, Level Control and White color, but not RGB color. Is that correct ?

manjh, 2023-01-03:
You are 100% correct!
Hans Man

pipiche38, 2023-01-03:
if you want to test it, just just have to switch to dev branch (6.4.119)

```
git checkout dev
git pull
```
then restart the plugin

Otherwise, you'll get it fix when the 3.6.009 will be released

closed 2023-01-03 by pipiche38

manjh, 2023-01-03:
I did as you asked, and sure enough: all devices are now green! The orange X's have disappeared.
Thank you for your quick support!
Hans Man