zigbeefordomoticz/Domoticz-Zigbee #1419: [Model Certification] - Télécommande murale Profalux Zigbee (depuis Août 2016)

Git-Bruno, 2023-01-04:
Télécommande murale Profalux Zigbee (depuis Août 2016)

https://www.profalux-pieces-detachees.com/telecommandes/429-telecommande-individuelle-profalux.html

{"Version":"3","ZDeviceName":"TC Chambre Bleue","Status":"inDB","SQN":"1a","Ep":{"01":{"0000":{"0010":"TS"},"0003":{},"0015":{},"0004":{},"0005":{},"0006":{},"0008":{},"0100":{},"fc20":{},"fc21":{},"ClusterType":{"19":"LvlControl","20":"BSO-Orientation"}}},"Heartbeat":"24111","RIA":"3","LQI":59,"Battery":{},"Model":"","ForceAckCommands":[],"MacCapa":"80","IEEE":"20918a00000f0eb1","Type":"Switch/LvlControl/BSO-Orientation","ProfileID":"0104","ZDeviceID":"0201","App Version":"","Attributes List":{},"DeviceType":"RFD","HW Version":"","Last Cmds":[],"LogicalType":"End Device","Manufacturer":"1110","Manufacturer Name":"","NbEp":"1","PowerSource":"Battery","ReadAttributes":{"Ep":{"01":{"0000":{"TimeStamp":1641578664,"iSQN":{"0004":33,"0005":34,"0010":37},"Attributes":{"0004":"86","0005":"86","0010":"00"},"ZigateRequest":{}}}}},"ReceiveOnIdle":"Off","Stack Version":"","Stamp":{"time":1670324372.3528016,"Time":"2022-12-06 11:59:32","MsgType":"8011","LastSeen":1670324372},"ZCL Version":"","Health":"","ConfigureReporting":{},"Epv2":{"01":{"ClusterIn":{"0000":{},"0003":{},"0015":{}},"ClusterOut":{"0003":{},"0004":{},"0005":{},"0006":{},"0008":{},"0100":{},"fc20":{},"fc21":{}},"ProfileID":"0104","ZDeviceID":"0201"}},"Capability":["Reduced-Function Device","Standard security","NwkAddr should be allocated"],"ReadConfigureReporting":{},"ConfigSource":"8043","Param":{},"Tuya":{"TuyaTransactionId":1},"WriteAttributes":{"Ep":{"01":{"0000":{"TimeStamp":1641581718,"iSQN":{"ffde":71},"Attributes":{"ffde":"de"},"ZigateRequest":{"ffde":{"Status":"fullfilled","DataType":"20","EPin":"01","EPout":"01","manuf_id":"0000","manuf_spec":"00","data":"13","ackIsDisabled":false}}},"0003":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"0015":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"0004":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"0005":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"0006":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"0008":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"0100":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"fc20":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"fc21":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}}}}},"Location":"TS","CertifiedDevice":false,"ConsistencyCheck":"ok","CheckParam":false}

pipiche38, 2023-01-04:
https://github.com/zigbeefordomoticz/Domoticz-Zigbee/commit/08c02d4b2f33d5cd521342c5e7aa4ca33b60157e

pipiche38, 2023-01-04:
This should be fixed with the current dev branch.
For test

```
git checkout dev
git pull

```
restart the plugin

After the restart, the Model name of the Profalux Remote should be named correctly and so becomes fully supported.

Git-Bruno, 2023-01-04:
Thanks, It is OK for most of the remotes however 2 are still shown as 'not optimized' although externally they look identical .
Below the json file for these remotes.
Bruno

{"Version":"3","ZDeviceName":"TC 0EB7","Status":"inDB","SQN":"fc","Ep":{"01":{"0000":{"0000":"1","0006":"20170309 NA 064","0007":"0","0010":"TS"},"0003":{},"0015":{},"0004":{},"0005":{},"0006":{},"0008":{},"0100":{},"fc20":{},"fc21":{},"ClusterType":{"66":"LvlControl","67":"BSO-Orientation"}}},"Heartbeat":"50548","RIA":"4","LQI":68,"Battery":{},"Model":"","ForceAckCommands":[],"MacCapa":"80","IEEE":"20918a00000f0eb7","Type":"Switch/LvlControl/BSO-Orientation","ProfileID":"0104","ZDeviceID":"0201","App Version":"","Attributes List":{},"DeviceType":"RFD","HW Version":"","Last Cmds":[],"LogicalType":"End Device","Manufacturer":"","Manufacturer Name":"","NbEp":"1","PowerSource":"Battery","ReadAttributes":{"Ep":{"01":{"0000":{"TimeStamp":1654369872,"iSQN":{"4000":8030,"0000":8020,"0001":8021,"0002":8022,"0003":8023,"0006":8024,"0007":8025,"000a":8026,"000f":8027,"0010":8028,"0015":8029,"f000":8031},"Attributes":{"4000":"86","0000":"00","0001":"86","0002":"86","0003":"86","0006":"00","0007":"00","000a":"86","000f":"86","0010":"00","0015":"86","f000":"86"},"ZigateRequest":{}},"0100":{"TimeStamp":1654369867,"iSQN":{"0000":8011,"0001":8012,"0002":8013,"0010":8014,"0011":8015},"Attributes":{"0001":"86","0002":"86","0010":"86","0011":"86"},"ZigateRequest":{}},"0006":{"TimeStamp":1654369868,"iSQN":{"4001":8017,"4002":8018,"0000":8016},"Attributes":{"4001":"86","4002":"86","0000":"86"},"ZigateRequest":{}},"0008":{"TimeStamp":1654369869,"iSQN":{"0000":8019},"Attributes":{"0000":"86"},"ZigateRequest":{}}}}},"ReceiveOnIdle":"","Stack Version":"","Stamp":{"time":1654369873.2957847,"Time":"2022-06-04 21:11:13","MsgType":"8011","LastSeen":1654369873},"ZCL Version":"1","Health":"","ConfigureReporting":{},"BindingTable":{"Devices":[{"20918a00000f0eb7":{"sourceEp":"01","Cluster":"fc20","targetIEEE":"20918a00000f0eb6","targetNickName":"Volet Salon","targetEp":"01"}}],"SQN":0,"TimeStamp":1654369872.5595763,"BindingTableTableSize":1,"Status":"Success"},"Epv2":{"01":{"ClusterIn":{"0000":{},"0003":{},"0015":{}},"ClusterOut":{"0003":{},"0004":{},"0005":{},"0006":{},"0008":{},"0100":{},"fc20":{},"fc21":{}},"ProfileID":"0104","ZDeviceID":"0201"}},"Capability":["Reduced-Function Device","Standard security","NwkAddr should be allocated"],"ReadConfigureReporting":{},"SWBUILD_1":"20170309 NA 064","ConfigSource":"8043","Param":{},"Location":"TS","CertifiedDevice":false,"ConsistencyCheck":"ok"}

pipiche38, 2023-01-04:
Thanks for the json.
For those one it is more difficult, to fix it as it seems that we are lacking informations like Manufacturer name.

I see only 2 options

(1) leave with that, it is not a big issue and it is not impacting the behaviour of the plugin as regards to that device
(2) manually EDIT the plugin database to force the Model name to the correct value

Git-Bruno, 2023-01-04:
I have not noticed any problem in the plugin related to the status of these remotes so option 1 is fine with me.
Thanks for your help.

closed 2023-01-04 by Git-Bruno