zigbeefordomoticz/Domoticz-Zigbee #1422: [Model Certification] - Aqara Smoke Detector

pipiche38, 2023-01-04:
Acheté sur AliExpress
Nom de marque: Aqara
Functions: Smoke Detector
Numéro de Modèle: JY-GZ-01AQ
Modele : lumi.sensor_smoke.acn03

{"Version":"3","ZDeviceName":"UDG_Atelier_SD","Status":"inDB","SQN":"fc","Ep":{"01":{"0000":{"4000":"2019\u0000www.","0004":"LUMI","0005":"lumi.sensor_smoke.acn03","0000":"3","0001":"15","0002":"2","0003":"1","0006":"20211227","0007":"3","000a":"0062776c"},"0500":{"0000":"01","0001":"0028","0002":"alarm1: 0, alarm2: 0, tamper: 0, batter: 0, srepor: 0, rrepor: 0, troubl: 0, acmain: 0, test: 0, batdef: 0, doorbell: 0","0010":"00124b0025e14fc2"},"0003":{},"0001":{"0020":30,"0033":1,"0035":1},"0019":{},"ClusterType":{"174":"Door","175":"Voltage"}}},"Heartbeat":"42029","RIA":"4","LQI":69,"Battery":100,"Model":"lumi.sensor_smoke.acn03","ForceAckCommands":[],"MacCapa":"84","IEEE":"54ef441000483803","Type":"Door/Voltage","ProfileID":"0104","ZDeviceID":"0402","App Version":"15","Attributes List":{"Request":{"01":{"0000":1670278138.117163,"0500":1670278138.1182501,"0003":1670278138.1193776,"0001":1670278138.1204593,"0019":1670278138.1215382}}},"DeviceType":"RFD","HW Version":"1","Last Cmds":[],"LogicalType":"End Device","Manufacturer":"1037","Manufacturer Name":"LUMI","NbEp":"1","PowerSource":"Main","ReadAttributes":{"Ep":{"01":{"0000":{"TimeStamp":1671440153,"iSQN":{"4000":"10","0004":"0c","0005":"0c","0000":"00","0001":"0d","0002":"0d","0003":"0d","0006":"0e","0007":"0e","000a":"0e","000f":"0f","0010":"0f","0015":"0f","f000":"10","ff01":"10","ff02":"11"},"Attributes":{"4000":"00","0004":"00","0005":"00","0000":"00","0001":"00","0002":"00","0003":"00","0006":"00","0007":"00","000a":"00","000f":"86","0010":"86","f000":"86","ff02":"86"},"ZigateRequest":{}},"0500":{"TimeStamp":1670274537,"iSQN":{},"Attributes":{"0000":"00","0001":"00","0002":"00","0010":"00"},"ZigateRequest":{}},"0001":{"TimeStamp":1672851021,"iSQN":{"0000":"05","0001":"05","0003":"05","0020":"06","0021":"06","0033":"06","0035":"07"},"Attributes":{"0000":"86","0001":"86","0020":"00","0021":"86","0033":"00","0035":"00"},"ZigateRequest":{}},"0019":{"TimeStamp":1670274498,"iSQN":{"0001":"08","0002":"08","0003":"08","0004":"09","0005":"09","0006":"09","0007":"0a","0008":"0a","0009":"0a","000a":"0b"},"Attributes":{},"ZigateRequest":{}}}}},"ReceiveOnIdle":"Off","Stack Version":"2","Stamp":{"time":1672851021.8795831,"Time":"2023-01-04 17:50:21","MsgType":"8102","LastSeen":1672851021,"LastPing":1671658926},"ZCL Version":"3","Health":"Live","ConfigSource":"8043","Bind":{"01":{"0000":{"Target":"0000","Stamp":1670274502,"Phase":"binded","Status":"00","i_sqn":"03"},"0003":{"Target":"0000","Stamp":1670274503,"Phase":"binded","Status":"00","i_sqn":"04"},"0001":{"Target":"0000","Stamp":1670274505,"Phase":"binded","Status":"00","i_sqn":"05"},"0500":{"Target":"0000","Stamp":1670274535,"Phase":"binded","Status":"00","i_sqn":"08"}}},"BindingTable":{"SQN":0,"Status":"Success","TimeStamp":1670274537.4006026,"BindingTableSize":4,"Devices":[{"54ef441000483803":{"sourceEp":"01","Cluster":"0000","targetIEEE":"00124b0025e14fc2","targetNickName":"Zigbee Coordinator","targetEp":"01"}},{"54ef441000483803":{"sourceEp":"01","Cluster":"0003","targetIEEE":"00124b0025e14fc2","targetNickName":"Zigbee Coordinator","targetEp":"01"}},{"54ef441000483803":{"sourceEp":"01","Cluster":"0001","targetIEEE":"00124b0025e14fc2","targetNickName":"Zigbee Coordinator","targetEp":"01"}},{"54ef441000483803":{"sourceEp":"01","Cluster":"0500","targetIEEE":"00124b0025e14fc2","targetNickName":"Zigbee Coordinator","targetEp":"01"}}]},"Capability":["Reduced-Function Device","Main Powered","Standard security","NwkAddr should be allocated"],"ReadConfigureReporting":{"Ep":{"01":{"0500":{"0000":{"TimeStamp":1671659017.844292,"Status":"8c"},"0001":{"TimeStamp":1671659017.8445632,"Status":"8c"},"0002":{"TimeStamp":1671659017.8448262,"Status":"8c"}},"0001":{"0000":{"TimeStamp":1671659018.0430963,"Status":"86"}}}},"TimeStamp":1671659018.0430834},"SWBUILD_3":"2019\u0000www.","IASBattery":100,"CertifiedDevice":false,"Param":{},"Epv2":{"01":{"ClusterIn":{"0000":{},"0500":{},"0003":{},"0001":{}},"ClusterOut":{"0019":{}},"ProfileID":"0104","ZDeviceID":"0402"}},"SWBUILD_1":"20211227","WriteAttributes":{"Ep":{"01":{"0500":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"0000":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"0003":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"0001":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"0019":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}}}}},"BatteryUpdateTime":1672851021,"IAS":{"Auto-Enrollment":{"Status":"Enrolled","Ep":{"01":{"Status":"Enrolled","TimeStamp":1670274493.0958354,"ZoneId":"00"}}},"01":{"EnrolledStatus":1,"ZoneType":40,"ZoneTypeName":"fire","ZoneStatus":{"alarm1":0,"alarm2":0,"tamper":0,"battery":0,"Support Reporting":0,"Restore Reporting":0,"trouble":0,"acmain":0,"test":0,"battdef":0,"doorbell":0,"GlobalInfos":"0;0;0;0;0;0;0;0;0;0;0","TimeStamp":1670274532}}},"ConfigureReporting":{"Ep":{"01":{"0500":{"TimeStamp":1670274498.1480043,"iSQN":{"0000":"03","0001":"03","0002":"03"},"Attributes":{"0000":"8c","0001":"8c","0002":"8c"},"ZigateRequest":{}},"0001":{"TimeStamp":1670274498.1528728,"iSQN":{"0000":"04","0020":"04","0021":"04"},"Attributes":{"0000":"86"},"ZigateRequest":{}}}}},"ConsistencyCheck":"ok","CheckParam":false,"RollingLQI":[69,69,132,66,69],"ZDPSQN":"00","_rawNodeDescriptor":"115f006400642c0000847f4002","Max Buffer Size":"7f","Max Rx":"0064","Max Tx":"0064","macapa":"84","bitfield":"4002","server_mask":"2c00","descriptor_capability":"00"}

pipiche38, 2023-01-04:
should be available on dev branch beta6-6.4.123

NewTurtle, 2023-01-05:
Pairing ok, with only one widget created (the Smoke Detector one) instead of 2 previously (1 for Smoke Detector et a second for Voltage no interest).
Thank you for help pipiche.
Anyway, the pairing for this device remains a hard job.

closed 2023-01-06 by pipiche38