zigbeefordomoticz/Domoticz-Zigbee #1425: [Model Certification] - GL-S-007ZS 12W

pabaillon, 2023-01-05:
1. Please do provide device information such as Brand and Model name.
1. Provide the URL where you have purchase the device
https://fr.aliexpress.com/item/32906939699.html?spm=a2g0o.order_list.order_list_main.30.42355e5b8wS11P&gatewayAdapt=glo2fra
1. Finaly copy-paste all information provided in the log
{"Version":"3","ZDeviceName":"gl","Status":"inDB","SQN":"56","Ep":{"0b":{"0000":{"4000":"1.0.9","0004":"GLEDOPTO","0005":"GL-S-007ZS","0000":"1","0001":"1","0002":"2","0003":"1","0006":"20131206","0007":"1"},"0003":{},"0004":{},"0005":{},"0006":{"4001":"0","4002":"0","0000":"01"},"0008":{"0000":"f7"},"0300":{"0000":"102","0001":"254","0003":"12480","0004":"47190","0007":"325","0008":"1","400a":"31"},"ClusterType":{"2533":"ColorControlRGBWW"}},"0d":{"0000":{}}},"Heartbeat":"63655","RIA":"4","LQI":69,"Battery":{},"Model":"GL-S-007ZS","ForceAckCommands":[],"MacCapa":"8e","IEEE":"00124b0020bc76dd","Type":"Switch/LvlControl/ColorControl","ProfileID":"c05e","ZDeviceID":"0210","App Version":"1","Attributes List":{},"DeviceType":"FFD","HW Version":"1","Last Cmds":[[17,"02","44f7","0100","010b000000000000010000"]],"LogicalType":"Router","Manufacturer":"0000","Manufacturer Name":"GLEDOPTO","NbEp":"2","PowerSource":"Main","ReadAttributes":{"Ep":{"0b":{"0000":{"TimeStamp":1670665550,"iSQN":{"4000":82,"0004":78,"0005":78,"0000":17,"0001":79,"0002":79,"0003":79,"0006":80,"0007":80,"000a":80,"000f":81,"0010":81,"0015":81,"f000":82},"Attributes":{"4000":"00","0004":"00","0005":"00","0000":"00","0001":"00","0002":"00","0003":"00","0006":"00","0007":"00","000a":"86","000f":"86","0010":"86","0015":"86","f000":"86"},"ZigateRequest":{}},"0006":{"TimeStamp":1671138314,"iSQN":{"4001":73,"4002":73,"0000":19509},"Attributes":{"4001":"00","4002":"00"},"ZigateRequest":{}},"0008":{"TimeStamp":1671138314,"iSQN":{"0000":19510},"Attributes":{},"ZigateRequest":{}},"0300":{"TimeStamp":1670665550,"iSQN":{"0000":75,"0001":75,"0003":75,"0004":76,"0007":76,"0008":76,"400a":77},"Attributes":{"0000":"00","0001":"00","0003":"00","0004":"00","0007":"00","0008":"00","400a":"00"},"ZigateRequest":{}}},"0d":{"0000":{"TimeStamp":1670665542,"iSQN":{"0004":58,"0005":58},"Attributes":{"0004":"86","0005":"86"},"ZigateRequest":{}}}}},"ReceiveOnIdle":"","Stack Version":"2","Stamp":{"time":1671138311.645668,"Time":"2022-12-15 21:05:11","MsgType":"  4d","LastSeen":1671138311,"LastPing":1672906169},"ZCL Version":"1","Health":"Not Reachable","BindingTable":{"SQN":0,"Status":"Not Supported (132)","TimeStamp":1670665548.265316,"BindingTableSize":0,"Devices":[]},"Bind":{"0b":{"0000":{"Target":"0000","Stamp":1670665547,"Phase":"binded","Status":"00","i_sqn":61},"0003":{"Target":"0000","Stamp":1670665547,"Phase":"binded","Status":"00","i_sqn":62},"0004":{"Target":"0000","Stamp":1670665547,"Phase":"binded","Status":"00","i_sqn":63},"0005":{"Target":"0000","Stamp":1670665547,"Phase":"binded","Status":"00","i_sqn":64},"0006":{"Target":"0000","Stamp":1670665547,"Phase":"binded","Status":"8c","i_sqn":65},"0008":{"Target":"0000","Stamp":1670665548,"Phase":"binded","Status":"8c","i_sqn":66},"0300":{"Target":"0000","Stamp":1670665548,"Phase":"binded","Status":"8c","i_sqn":67}}},"Param":{},"GroupMemberShip":{"0b":{}},"ReadConfigureReporting":{},"ConfigSource":"8043","Epv2":{"0b":{"ClusterIn":{"0000":{},"0003":{},"0004":{},"0005":{},"0006":{},"0008":{},"0300":{}},"ClusterOut":{},"ProfileID":"c05e","ZDeviceID":"0210"},"0d":{"ClusterIn":{},"ClusterOut":{},"ProfileID":{},"ZDeviceID":{}}},"SWBUILD_1":"20131206","ColorInfos":{"Hue":102,"Saturation":254,"X":"12480","Y":"47190","ColorTemperatureMireds":"325","ColorMode":1,"ColorCapabilities":"31"},"SWBUILD_3":"1.0.9","WriteAttributes":{"Ep":{"0b":{"0000":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"0003":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"0004":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"0005":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"0006":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"0008":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"0300":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}}},"0d":{"0000":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}}}}},"ConfigureReporting":{"Ep":{"0b":{"0006":{"TimeStamp":1670665547.1069233,"iSQN":{"0000":69},"Attributes":{},"ZigateRequest":{}},"0008":{"TimeStamp":1670665547.1082165,"iSQN":{"0000":70},"Attributes":{},"ZigateRequest":{}},"0300":{"TimeStamp":1670665547.1109273,"iSQN":{"0003":71,"0004":71,"0007":71,"0008":72},"Attributes":{},"ZigateRequest":{}}}}},"Log_UnknowDeviceFlag":1672906144.8813183,"CertifiedDevice":false,"Capability":["Full-Function Device","Main Powered","Receiver during Idle","Standard security","NwkAddr should be allocated"],"ConsistencyCheck":"ok","pingDeviceRetry":{"Retry":0,"TimeStamp":1672906189},"CheckParam":true}

pipiche38, 2023-01-05:
Should be ok.
If you want to test/check you'll need to move to dev branch, otherwise you can wait until the next release.

if you want to switch
```
git checkout dev
git pull
```
and restart the plugin

pabaillon, 2023-01-05:
Ok dipositif optimisé ! Merci

closed 2023-01-05 by pabaillon