zigbeefordomoticz/Domoticz-Zigbee #1433: [Model Certification] - Repeater Zigbee-JMRP01/ LOR_RP280

rcrocus, 2023-01-07:
Hello everyone,

I bought this repeater Zigbee-JMRP01/ LOR_RP280 (https://www.domotique-store.fr/domotique/modules-domotiques/repeteurs-amplificateurs-de-signal-radio/1776-repeteur-routeur-zigbee-alimente-par-usb.html?search_query=JMRP01&results=1).

The device is recognized by the plugin (in the Web Interface) but it appears like not optimized.
No device is created in Domoticz.

Please find the informations about this product (copy from the plugin) :
_{"Version":"3","ZDeviceName":"","Status":"notDB","SQN":"03","Heartbeat":"27","RIA":"1","LQI":10,"Battery":{},"Model":"TS0207-extender","ForceAckCommands":[],"MacCapa":"8e","IEEE":"1c34f1fffeb7669f","Type":"","ProfileID":"0104","ZDeviceID":"0008","App Version":"64","Attributes List":{},"DeviceType":"FFD","HW Version":"","Last Cmds":[[1170,"02","1b7f","0100","01010000000000000200040005"],[1172,"02","1b7f","0100","0101000000000000010005"]],"LogicalType":"Router","Manufacturer":"1002","Manufacturer Name":"_TZ3000_misw04hq","NbEp":"1","PowerSource":"Main","ReadAttributes":{"Ep":{"01":{"0000":{"TimeStamp":1673121795,"iSQN":{"0004":1170,"0005":1172},"Attributes":{"0004":"00","0005":"00","0001":"00","ffe2":"00","ffdf":"00"},"ZigateRequest":{}}}}},"ReceiveOnIdle":"On","Stack Version":"","Stamp":{"time":1673121814.861186,"Time":"2023-01-07 21:03:34","MsgType":"8011","LastSeen":1673121814},"ZCL Version":"","Health":"Live","Capability":["Full-Function Device","Main Powered","Receiver during Idle","Standard security","NwkAddr should be allocated"],"RollingLQI":[111,108,114,114,141,141,138,138,141,141,10],"Epv2":{"01":{"ClusterIn":{"0000":{},"000a":{},"0003":{}},"ClusterOut":{"0019":{}},"ProfileID":"0104","ZDeviceID":"0008"}},"WriteAttributes":{"Ep":{"01":{"0000":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"000a":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"0019":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}}}}},"ZDeviceVersion":"0000","ConfigSource":"8043","PairingInProgress":false,"Ep":{"01":{"0000":{"0004":"_TZ3000_misw04hq","0005":"TS0207","0001":"64","ffe2":"22","ffdf":"8\u0017+\u0013E\u0017+\u0013R\u0017+\u0013X\u0017+\u0013_\u0017+\u0013\u0005\u0005\u0005\u0005\u0005"},"000a":{},"0019":{},"Type":""}},"Max Buffer Size":"52","Max Rx":"0052","Max Tx":"0052","macapa":"8e","bitfield":"4001","server_mask":"2c00","descriptor_capability":"00","SQN_000a":"03"}_

And in Domoticz log :
_2023-01-07 21:02:41.572 Status: Zigate: Request Accepting new Hardware for 240 seconds
2023-01-07 21:02:57.326 Status: Zigate: [None] NEW OBJECT: 1b7f None
2023-01-07 21:02:57.492 Status: Zigate: [-] NEW OBJECT: 1b7f Active Endpoint Response Ep: 01 LQI: 108
2023-01-07 21:02:57.500 Status: Zigate: [-] NEW OBJECT: 1b7f Request Simple Descriptor for Ep: 01
2023-01-07 21:02:57.702 Status: Zigate: [-] NEW OBJECT: 1b7f Simple Descriptor Response EP: 0x01 LQI: 108
2023-01-07 21:02:57.703 Status: Zigate: [-] NEW OBJECT: 1b7f ProfileID 0104
2023-01-07 21:02:57.704 Status: Zigate: [-] NEW OBJECT: 1b7f ZDeviceID 0008
2023-01-07 21:02:57.704 Status: Zigate: [-] NEW OBJECT: 1b7f Application Version 0000
2023-01-07 21:02:57.705 Status: Zigate: [-] NEW OBJECT: 1b7f Cluster IN Count: 03
2023-01-07 21:02:57.706 Status: Zigate: [-] NEW OBJECT: 1b7f Cluster In 1: 0000 (Basic)
2023-01-07 21:02:57.707 Status: Zigate: [-] NEW OBJECT: 1b7f Cluster In 2: 000a (Time)
2023-01-07 21:02:57.707 Status: Zigate: [-] NEW OBJECT: 1b7f Cluster In 3: 0003 (Identify)
2023-01-07 21:02:57.708 Status: Zigate: [-] NEW OBJECT: 1b7f Cluster OUT Count: 01
2023-01-07 21:02:57.709 Status: Zigate: [-] NEW OBJECT: 1b7f Cluster Out 1: 0019 (Over-the-Air Upgrade)
2023-01-07 21:03:00.137 Status: Zigate: [None] NEW OBJECT: 1b7f Request Node Descriptor
2023-01-07 21:03:05.105 Status: Zigate: [1] NEW OBJECT: 1b7f Model Name: TS0207-extender
2023-01-07 21:03:05.108 Status: Zigate: [1] NEW OBJECT: 1b7f Manufacturer Name: _TZ3000_misw04hq
2023-01-07 21:06:31.716 Status: Zigate: Request Disabling Accepting new Hardware_ 

Tell me if you need some other information or tests.

Thanks a lot.

pipiche38, 2023-01-08:
Will investigate.
This is normal that no Devices are created in Domoticz as there is no functionality provided by the device expect the Zigbee routing capabilities.
So on one side, the device is optimized, so I feel this is much more a bug that we report as not Optimized

pipiche38, 2023-01-08:
Is the problem happening only at the time when the device was paired ? and after a restart of the plugin it was seen as Optimized ?
If yes, this should be fixed now on the latest dev branch. If you want to test

```
git pull
git checkout
```

restart of the plugin

rcrocus, 2023-01-08:
After a restart of the plugin, the device appears as optimized.

Do you want I test to the dev branch ? If yes, which test do you expect ?

pipiche38, 2023-01-08:
I was able to duplicate the issue, so I think that is solved .
Thansk for reporting it.

closed 2023-01-08 by pipiche38