zigbeefordomoticz/Domoticz-Zigbee #1435: [Model Certification] - HEIMAN Zigbee 3.0

thanks, 2023-01-09:
1. Please do provide device information such as Brand and Model name :

HEIMAN > Smart Siren > HS2WD-E

    Certification: CE,FCC,ROHS
    Origine: CN (Origine)
    Maison intelligente plate-forme: autre
    Model Number: M420
    Working Voltage: AC110-240V
    Backup Battery: 3.8V/700mA
    Alarm Sound Level: 95db(1m away)
    Wireless Network Scope: <80m(open indoors)
    Working Temperature: -10℃-50℃
    Working Humidity: <95% RH (Indicating no condensation)
    Network Mode: Zigbee 3.0
    Dimension: 80mm* 32mm
    Type: Strobe flash Siren
    Belong: Smart house / Smart home
    can be use on: Protecting family safety
    Belong2: safety precaution
    Features: Very loud alarm

1. Provide the URL where you have purchase the device :

https://fr.aliexpress.com/item/1005003615151308.html?ug_edm_item_id=1005003615151308&edm_click_module=item_detail&tracelog=rowan&rowan_id1=aeug_edm_24677_1_fr_FR_2022-02-04&rowan_msg_id=9548biz_pay_after_purchase%3A0%3A0_571171563%2405b9f9ca5c994a7e9a1f9a50dedc16a3&ck=in_edm_other&gatewayAdapt=glo2fra

1. Finaly copy-paste all information provided in the log

{"Version":"3","ZDeviceName":"Sirène Zigbee","Status":"inDB","SQN":"41","Ep":{"01":{"0000":{"0004":"HEIMAN","0005":"WarningDevice-EF-3.0","0000":"3","0001":"23","0002":"2","0003":"5","0006":"2021.2.19","0007":"129"},"0001":{"0020":41,"0021":200},"0003":{},"0004":{},"0500":{"0002":"alarm1: 0, alarm2: 0, tamper: 0, batter: 0, srepor: 0, rrepor: 0, troubl: 0, acmain: 0, test: 0, batdef: 0, doorbell: 0","0000":"01","0001":"0225","0010":"00158d0006a3fcbb"},"0502":{"0000":"65534"},"0b05":{},"0019":{},"ClusterType":{"154":"Voltage","155":"Door","156":"AlarmWD"}},"f2":{"0021":{}}},"Heartbeat":"5","RIA":"3","LQI":113,"Battery":200,"Model":"WarningDevice-EF-3.0","ForceAckCommands":[],"MacCapa":"8e","IEEE":"847127fffead4501","Type":"Voltage/Door/AlarmWD","ProfileID":"a1e0","ZDeviceID":"0061","App Version":"23","Attributes List":{"Request":{"01":{"0000":1643473948.093499,"0001":1643473948.0937638,"0003":1643473948.094021,"0004":1643473948.0942712,"0500":1643473948.0945208,"0502":1643473948.0948308,"0b05":1643473948.0950801,"0019":1643473948.0953856},"f2":{"0021":1643473948.0956497}}},"DeviceType":"FFD","HW Version":"5","Last Cmds":[],"LogicalType":"Router","Manufacturer":"120b","Manufacturer Name":"HEIMAN","NbEp":"2","PowerSource":"Main","ReadAttributes":{"Ep":{"01":{"0000":{"TimeStamp":1666628885,"iSQN":{"4000":64,"0004":52,"0005":53,"0000":10,"0001":55,"0002":56,"0003":57,"0006":58,"0007":59,"000a":60,"000f":61,"0010":62,"0015":63,"f000":65},"Attributes":{"4000":"86","0004":"00","0005":"00","0000":"00","0001":"00","0002":"00","0003":"00","0006":"00","0007":"00","000a":"86","000f":"86","0010":"86","0015":"86","f000":"86"},"ZigateRequest":{}},"0001":{"TimeStamp":1673258053,"iSQN":{"0000":34,"0001":35,"0003":36,"0020":37,"0021":38,"0033":39,"0035":40},"Attributes":{"0000":"86","0001":"86","0003":"86","0020":"00","0021":"00","0033":"86","0035":"86"},"ZigateRequest":{}},"0019":{"TimeStamp":1643459584,"iSQN":{"0001":41,"0002":42,"0003":43,"0004":44,"0005":45,"0006":46,"0007":47,"0008":48,"0009":49,"000a":50},"Attributes":{"0001":"86","0002":"86","0003":"86","0004":"86","0005":"86","0006":"86","0007":"86","0008":"86","0009":"86","000a":"86"},"ZigateRequest":{}},"0502":{"TimeStamp":1673258963,"iSQN":{"0000":51},"Attributes":{"0000":"00"},"ZigateRequest":{}},"0500":{"TimeStamp":1673259023,"iSQN":{},"Attributes":{"0000":"00","0001":"00","0002":"00","0010":"00"},"ZigateRequest":{}}},"f2":{"0000":{"TimeStamp":1643459570,"iSQN":{"0004":20,"0005":22},"Attributes":{},"ZigateRequest":{}}}}},"ReceiveOnIdle":"On","Stack Version":"2","Stamp":{"time":1673259346.5107613,"Time":"2023-01-09 11:15:46","MsgType":"8011","LastSeen":1673259346,"LastPing":1666628885},"ZCL Version":"3","Health":"Live","WriteAttributes":{"Ep":{"01":{"0500":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"0502":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"0000":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"0001":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"0003":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"0004":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"0b05":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"0019":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}}},"f2":{"0021":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}}}}},"NeighbourTableSize":1,"Capability":["Full-Function Device","Main Powered","Receiver during Idle","Standard security","NwkAddr should be allocated"],"ConfigSource":"8043","Bind":{"01":{"0000":{"Target":"0000","Stamp":1643459580,"Phase":"binded","Status":"00","i_sqn":24},"0001":{"Target":"0000","Stamp":1643459580,"Phase":"binded","Status":"00","i_sqn":25},"0003":{"Target":"0000","Stamp":1643459580,"Phase":"binded","Status":"00","i_sqn":26},"0004":{"Target":"0000","Stamp":1643459580,"Phase":"binded","Status":"00","i_sqn":27},"0500":{"Target":"0000","Stamp":1643459580,"Phase":"binded","Status":"00","i_sqn":28},"0502":{"Target":"0000","Stamp":1643459581,"Phase":"binded","Status":"00","i_sqn":29},"0b05":{"Target":"0000","Stamp":1643459581,"Phase":"binded","Status":"8c","i_sqn":30}}},"Param":{"alarmDuration":900},"IAS":{"01":{"EnrolledStatus":1,"ZoneType":549,"ZoneTypeName":"standard_warning","ZoneStatus":{"alarm1":0,"alarm2":0,"tamper":0,"battery":0,"Support Reporting":0,"Restore Reporting":0,"trouble":0,"acmain":0,"test":0,"battdef":0,"GlobalInfos":"0;0;0;0;0;0;0;0;0;0;0","TimeStamp":1673259023,"doorbell":0}}},"SWBUILD_1":"2021.2.19","ConfigureReporting":{"Ep":{"01":{"0500":{"TimeStamp":1643459580,"iSQN":{"0000":31,"0001":31,"0002":31},"Attributes":{"0000":"00","0001":"00","0002":"00"},"ZigateRequest":{}},"0502":{"TimeStamp":1643459580,"iSQN":{"0000":32},"Attributes":{"0000":"00"},"ZigateRequest":{}},"0001":{"TimeStamp":1643459580,"iSQN":{"0000":33,"0020":33,"0021":33},"Attributes":{"0000":"86","0020":"8d","0021":"8d"},"ZigateRequest":{}}}}},"BindingTable":{"SQN":0,"Status":"Requested","TimeStamp":1666631354.0539877,"BindingTableSize":0,"Devices":[]},"BatteryUpdateTime":1673258053,"IASBattery":100,"Epv2":{"01":{"ClusterIn":{"0000":{},"0001":{},"0003":{},"0004":{},"0500":{},"0502":{},"0b05":{}},"ClusterOut":{"0019":{}},"ProfileID":"0104","ZDeviceID":"0403"},"f2":{"ClusterIn":{},"ClusterOut":{"0021":{}},"ProfileID":"a1e0","ZDeviceID":"0061"}},"Neighbours":[],"ConsistencyCheck":"ok","CertifiedDevice":false,"Log_UnknowDeviceFlag":1673005363.7903628,"RollingLQI":[114,111,113,113,113,113,113,115,115,114,113],"IAS WD":{"MaxDuration":"65534"},"ZCLSQN":"04","CheckParam":false}

pipiche38, 2023-01-09:
Thanks. Which version of the plugin are you running ?

thanks, 2023-01-09:
Plugin: 6.3.008  with Firmware: Zigate - 0321

pipiche38, 2023-01-09:
This is fixed on the recent dev branch.
Planning to release it soon.

if you want to test, you can switch to the dev branch by

```
git checkout dev
git pull --recurse-submodules
```

thanks, 2023-01-09:
very good :)
Thanks for all and have a nice day

closed 2023-01-09 by pipiche38