zigbeefordomoticz/Domoticz-Zigbee #1436: Not optimized tradfri transformer 10W

waltervl, 2023-01-11:
* Product Name: Tradfri transformer 10W
* Brand: Ikea
* URL on a web site https://www.ikea.com/nl/en/p/tradfri-driver-for-wireless-control-smart-grey-50356187/
* What do you expect: To be optimezed
* The logs from the pairing process: No issues with pairing, just popups as not being optimized.
* export.json file as described here above
 [export_tradfri_transformer_10W.json.txt](https://github.com/zigbeefordomoticz/Domoticz-Zigbee/files/10396191/export_tradfri_transformer_10W.json.txt)

pipiche38, 2023-01-13:
show be ok now. if you want to check this is on dev branch
```

git checkout dev
git pull

```

and restart the plugin, otherwise it will be part of the next stable release that I hope to make soon

waltervl, 2023-01-13:
I will wait for next stable. Thanks!

closed 2023-01-14 by pipiche38