zigbeefordomoticz/Domoticz-Zigbee #1437: EBYTE - E180-DTU - Data Transceiver by ZigBee to RS485

Basile31, 2023-01-12:
Hello,
I have some problems for pairing this new device.
Thank you very much for your help or your indications for integration.

1. Please do provide device information such as Brand and Model name.
EBYTE - E180-DTU(ZG120-485)

2. Provide the URL where you have purchase the device
https://www.amazon.com/Transceiver-Transparent-Transmission-Geographical-Environment/dp/B08ZCR9297

3. Finaly copy-paste all information provided in the log
2023-01-12 09:53:37.338 Status: Domoticz V2022.2 (c)2012-2022 GizMoCuz
2023-01-12 09:53:37.340 Status: Build Hash: eea9db734, Date: 2022-11-05 13:05:35
2023-01-12 09:53:37.340 Status: Startup Path: /home/pi/domoticz/
2023-01-12 09:53:37.769 Status: PluginSystem: Started, Python version '3.9.2', 4 plugin definitions loaded.
2023-01-12 09:53:37.778 Status: WebServer(HTTP) started on address: :: with port 8080
2023-01-12 09:53:37.792 Status: WebServer(SSL) started on address: :: with port 443
2023-01-12 09:53:37.795 Status: TCPServer: shared server started...
2023-01-12 09:53:37.795 Status: RxQueue: queue worker started...
2023-01-12 09:53:39.797 Zigate: Worker thread started.
2023-01-12 09:53:39.797 Status: Zigate: Started.
2023-01-12 09:53:39.798 Status: Zigate: Entering work loop.
2023-01-12 09:53:39.799 Status: NotificationSystem: thread started...
2023-01-12 09:53:39.800 Status: EventSystem: reset all events...
2023-01-12 09:53:39.803 Status: EventSystem: reset all device statuses...
2023-01-12 09:53:39.985 Status: PluginSystem: 1 plugins started.
2023-01-12 09:53:39.998 Status: Python EventSystem: Initializing event module.
2023-01-12 09:53:39.999 Status: EventSystem: Queue thread started...
2023-01-12 09:53:39.999 Status: EventSystem: Started
2023-01-12 09:53:42.047 Zigate: Zigbee for Domoticz plugin started!
2023-01-12 09:53:42.854 Zigate: load PluginConf
2023-01-12 09:53:42.855 Zigate: Loading PluginConf - into Domoticz sqlite Db
2023-01-12 09:53:42.855 Zigate: /home/pi/domoticz/plugins/Domoticz-Zigbee/Conf/PluginConf-05.json timestamp is 1673280232.3399994
2023-01-12 09:53:42.855 Zigate: Dz PluginConf is older than Json Dz: 0 Json: 1673280232.3399994
2023-01-12 09:53:42.857 Zigate: Installing plugin custom page /home/pi/domoticz/www/templates/Zigate.html
2023-01-12 09:53:42.858 Zigate: Starting LoggingManagement thread
2023-01-12 09:53:42.860 Zigate: start_logging_thread
2023-01-12 09:53:42.860 Zigate: logging_thread - listening
2023-01-12 09:53:42.040 Status: Zigate: Initialized version 6.3, author 'pipiche38'
2023-01-12 09:53:42.047 Status: Zigate: Checking Python modules /home/pi/domoticz/plugins/Domoticz-Zigbee/requirements.txt
2023-01-12 09:53:42.756 Status: Zigate: Checking Python modules /home/pi/domoticz/plugins/Domoticz-Zigbee/requirements.txt
2023-01-12 09:53:42.857 Status: Zigate: PluginAnalytics set to 0
2023-01-12 09:53:42.857 Status: Zigate: channel set to 11
2023-01-12 09:53:42.857 Status: Zigate: filename set to /home/pi/domoticz/plugins/Domoticz-Zigbee/Conf/PluginConf-05.json
2023-01-12 09:53:42.857 Status: Zigate: pluginHome set to /home/pi/domoticz/plugins/Domoticz-Zigbee/
2023-01-12 09:53:42.857 Status: Zigate: homedirectory set to /home/pi/domoticz/plugins/Domoticz-Zigbee/
2023-01-12 09:53:42.857 Status: Zigate: pluginData set to /home/pi/domoticz/plugins/Domoticz-Zigbee/Data/
2023-01-12 09:53:42.857 Status: Zigate: pluginConfig set to /home/pi/domoticz/plugins/Domoticz-Zigbee/Conf/
2023-01-12 09:53:42.857 Status: Zigate: pluginOTAFirmware set to /home/pi/domoticz/plugins/Domoticz-Zigbee/OTAFirmware/
2023-01-12 09:53:42.857 Status: Zigate: pluginReports set to /home/pi/domoticz/plugins/Domoticz-Zigbee/Reports/
2023-01-12 09:53:42.857 Status: Zigate: pluginWWW set to /home/pi/domoticz/plugins/Domoticz-Zigbee/www/
2023-01-12 09:53:42.857 Status: Zigate: pluginLogs set to /home/pi/domoticz/plugins/Domoticz-Zigbee/Logs/
2023-01-12 09:53:42.860 Status: Zigate: Please watch plugin log into /home/pi/domoticz/plugins/Domoticz-Zigbee/Logs//PluginZigbee_05.log
2023-01-12 09:53:42.871 Status: Zigate: Zigbee for Domoticz (z4d) plugin stable6-6.3.008 started
2023-01-12 09:53:42.876 Status: Incoming connection from: 127.0.0.1
2023-01-12 09:53:42.890 Status: Zigate: DeviceConf loaded - 17 confs loaded
2023-01-12 09:53:43.323 Zigate: Set PiZigate Channels 11 and 17
2023-01-12 09:53:43.323 Zigate: + GPIO(RUN) OK
2023-01-12 09:53:43.323 Zigate: + GPIO(FLASH) OK
2023-01-12 09:53:43.258 Status: Zigate: DeviceConf loaded - 393 confs loaded
2023-01-12 09:53:43.258 Status: Zigate: load ListOfDevice
2023-01-12 09:53:43.277 Status: Zigate: 2 Entries loaded from /home/pi/domoticz/plugins/Domoticz-Zigbee/Data/DeviceList-5.txt
2023-01-12 09:53:43.286 Status: Zigate: Transport mode: V2-PI
2023-01-12 09:53:43.323 Status: Zigate: Switch PiZigate in RUN mode
2023-01-12 09:53:43.877 Status: Zigate: ZigateTransport: Serial Connection open: Serial<id=0xaad3e5b0, open=True>(port='/dev/serial0', baudrate=115200, bytesize=8, parity='N', stopbits=1, timeout=None, xonxoff=False, rtscts=False, dsrdtr=False)
2023-01-12 09:53:43.879 Status: Zigate: ZigateTransport: writer_thread Thread start.
2023-01-12 09:53:43.880 Status: Zigate: ZigateTransport: thread_processing_and_sending Thread start.
2023-01-12 09:53:43.881 Status: Zigate: Start Web Server connection
2023-01-12 09:53:43.904 Status: Zigate: Web backend for Web User Interface started on port: 9440
2023-01-12 09:53:43.904 Status: Zigate: Domoticz Widgets usage is at 5.5 % (241 units free)
2023-01-12 09:53:43.313 Error: Zigate: switchPiZigate_mode - unable to find distribution: Assuming debian, Error: module 'platform' has no attribute 'dist'
2023-01-12 09:53:52.135 Status: Zigate: ==> Transport Mode switch to full mode
2023-01-12 09:53:52.137 Status: Zigate: ZiGate+ (V2)
2023-01-12 09:53:52.138 Status: Zigate: Installer Version Number: 0321
2023-01-12 09:53:52.138 Status: Zigate: Branch Version: ==> Production <==
2023-01-12 09:54:02.205 Status: Zigate: Zigbee Coordinator ieee: 00158d000638bbec , short addr: 0000
2023-01-12 09:54:02.205 Status: Zigate: Network state UP, PANID: 2e4d extPANID: 0x42e2bdb4374ef59e Channel: 11
2023-01-12 09:54:07.113 Status: Zigate: Request Disabling Accepting new Hardware
2023-01-12 09:54:07.648 Status: Zigate: Accepting new Hardware: Disable (Off)
2023-01-12 09:54:07.752 Status: Zigate: Number of devices recently active in Zigate = 1
2023-01-12 09:54:07.753 Status: Zigate: [00] DevID: 00 Network addr: 5683 IEEE: 2c1165fffec6e55f LQI: 239 power: 01 not found in plugin database!
2023-01-12 09:54:12.079 Status: Zigate: coordinator set to Certification : 1/CE -> CE
2023-01-12 09:55:08.347 Status: Zigate: ota_scan_folder Following Firmware have been loaded
2023-01-12 09:55:08.348 Status: Zigate: Plugin with Zigate+, firmware 0321 correctly initialized

Zigate Plugin info
{"Version":"3","ZDeviceName":"","Status":"UNKNOW","SQN":"5a","Ep":{"01":{"1000":{},"0000":{}}},"Heartbeat":"12073","RIA":"4","LQI":210,"Battery":{},"Model":"","ForceAckCommands":[],"MacCapa":"8c","IEEE":"2c1165fffec6e55f","Type":{},"ProfileID":"0104","ZDeviceID":"0000","App Version":"","Attributes List":{},"DeviceType":"RFD","HW Version":"","Last Cmds":[[13,"02","5683","0100","01010000000000000200040005"]],"LogicalType":"End Device","Manufacturer":"1002","Manufacturer Name":"","NbEp":"1","PowerSource":"Main","ReadAttributes":{"Ep":{"01":{"0000":{"TimeStamp":1673447166,"iSQN":{"0004":13,"0005":13},"Attributes":{"0004":"86","0005":"86"},"ZigateRequest":{}}}}},"ReceiveOnIdle":"On","Stack Version":"","Stamp":{"time":1673447168.2000587,"Time":"2023-01-11 15:26:08","MsgType":"8011","LastSeen":1673447168},"ZCL Version":"","Health":"Live","Capability":["Reduced-Function Device","Main Powered","Receiver during Idle","Standard security","NwkAddr should be allocated"],"RollingLQI":[212,209,209,210,210,211,211,210,210],"Epv2":{"01":{"ClusterIn":{"1000":{}},"ClusterOut":{"1000":{}},"ProfileID":"0104","ZDeviceID":"0000"}},"WriteAttributes":{"Ep":{"01":{"0000":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}}}}},"ZDeviceVersion":"0001","ConfigSource":"8043","PairingInProgress":true,"Max Buffer Size":"52","Max Rx":"0052","Max Tx":"0052","macapa":"8c","bitfield":"4002","server_mask":"2c00","descriptor_capability":"00"}

User Manual for Device
[E180-DTU(ZG120-485)_UserManual_EN_v1.1.pdf](https://github.com/zigbeefordomoticz/Domoticz-Zigbee/files/10399704/E180-DTU.ZG120-485._UserManual_EN_v1.1.pdf)

pipiche38, 2023-01-13:
That is an interested product, but a tricky one.
Could you let us know what you want to do with ?

For the information that you have shared (thanks for that), the device is not providing its capabilities, that is why the plugin is not able to configure it. In other terms when the plugin detect a new device, it starts to do an interview of that device to understand what is capable to provide. In your case, the device respond with 2 clusters
0x0000 : Generic Cluster 
0x1000:  Touch Link commissioning

From the link you have provided, it looks like the device can act as 
- a Coordinateur ( like the Zigate) , not the purpose of the plugin
- a router, in such case you just have to configure it as a router and it will participate to the network mesh. 
- a device, this where it would be important to understand what you want to do with

You can contact us on https://keybase.io/pipiche/chat

Basile31, 2023-01-14:
Thank you for your quick reply.
I want to use this device as a gateway to connect a [piezo probe RS485 ](https://fr.aliexpress.com/item/1005002286044664.html?spm=a2g0o.productlist.main.19.77201184c1s4kY&algo_pvid=eda4db1a-e2ea-4a88-aa28-2b914b07d406&aem_p4p_detail=202301132326103844832760150000002608420&algo_exp_id=eda4db1a-e2ea-4a88-aa28-2b914b07d406-9&pdp_ext_f=%7B%22sku_id%22%3A%2212000020323221934%22%7D&pdp_npi=2%40dis%21EUR%2180.88%2176.83%21%21%21%21%21%402100baf316736811701916230d06d7%2112000020323221934%21sea&curPageLogUid=9g6VvxbQaB4m&ad_pvid=202301132326103844832760150000002608420_2&ad_pvid=202301132326103844832760150000002608420_2) to measure the height of a deep well. The device can participating in the mesh and being paired with the zigate coordinator.

pipiche38, 2023-01-14:
I think most of the work as to be done on the device itself, so the plugin + coordinator can do the interview.
If you can configure the device with
a Model Name : Cluster 0x0000 Attribute 0x0005 -> the plugin will be able to identfy it
a Cluster/Attribute to report the data, like 0x000c ( Analog cluster ) Attribute 0x0055 (Present Value) of Data Type 0x39

The plugin will be able to report the value to a Domoticz widget

I'm referring here to a similar device which is work. (unfortunatly in French)
https://easydomoticz.com/forum/viewtopic.php?f=17&t=11307&p=95139&hilit=analog#p95139

pipiche38, 2023-01-18:
please re-open this issue if needed, but for now we cannot do much

closed 2023-01-18 by pipiche38