zigbeefordomoticz/Domoticz-Zigbee #1438: [Model Certification] -

mamo79, 2023-01-12:
Hello,

1. Please do provide device information such as Brand and Model name.
Nodon SIN-4-RS-20 module Zigbee volet roulant
Domoticz  :
Version: 2022.2
Build Hash: eea9db734
Compile Date: 2022-11-05 13:05:35
dzVents Version: 3.1.8
Python Version: 3.8.10 (default, Nov 14 2022, 12:59:47) [GCC 9.4.0]

Zygate :
Plugin: 6.3.008 / Firmware: Zigate - 0321 

1. Provide the URL where you have purchase the device
2.https://www.espace-domotique.fr/fr/peripheriques/nodon-sin-4-rs-20-module-zigbee-volet-roulant-nodon-sin-4-rs-20-3801.html?search_query=Nodon+SIN&results=52

1. Finaly copy-paste all information provided in the log

{"Version":"3","ZDeviceName":"RDC_Volet bureau","Status":"UNKNOW","SQN":"a5","Ep":{"01":{"1000":{},"0000":{"0004":"NodOn","0005":"SIN-4-RS-20"},"0003":{},"0004":{},"0005":{},"0102":{},"fc57":{},"0019":{},"ClusterType":{}},"f2":{"0021":{}}},"Heartbeat":"169","RIA":"4","LQI":43,"Battery":{},"Model":"SIN-4-RS-20","ForceAckCommands":[],"MacCapa":"8e","IEEE":"9035eafffea1df06","Type":"WindowCovering","ProfileID":"a1e0","ZDeviceID":"0066","App Version":"","Attributes List":{},"DeviceType":"FFD","HW Version":"","Last Cmds":[[1472,"02","110b","0100","01010000000000000200040005"]],"LogicalType":"Router","Manufacturer":"128b","Manufacturer Name":"NodOn","NbEp":"2","PowerSource":"Main","ReadAttributes":{"Ep":{"01":{"0000":{"TimeStamp":1673531205,"iSQN":{"0004":1472,"0005":1472},"Attributes":{"0004":"00","0005":"00"},"ZigateRequest":{}}},"f2":{"0000":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}}}}},"ReceiveOnIdle":"On","Stack Version":"","Stamp":{"time":1673531915.7146275,"Time":"2023-01-12 14:58:35","MsgType":"8011","LastSeen":1673531915},"ZCL Version":"","Health":"Live","Capability":["Full-Function Device","Main Powered","Receiver during Idle","Standard security","NwkAddr should be allocated"],"RollingLQI":[41,42,42,41,40,43],"Epv2":{"01":{"ClusterIn":{"1000":{},"0000":{},"0003":{},"0004":{},"0005":{},"0102":{},"fc57":{}},"ClusterOut":{"0019":{}},"ProfileID":"0104","ZDeviceID":"0202"},"f2":{"ClusterIn":{"0021":{}},"ClusterOut":{"0021":{}},"ProfileID":"a1e0","ZDeviceID":"0066"}},"WriteAttributes":{"Ep":{"01":{"0000":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}}}}},"ZDeviceVersion":"0001","ConfigSource":"8043","PairingInProgress":true,"Max Buffer Size":"52","Max Rx":"0052","Max Tx":"0052","macapa":"8e","bitfield":"4001","server_mask":"2c00","descriptor_capability":"00","Param":{},"CheckParam":true}

Thanks for your help.

pipiche38, 2023-01-13:
can you switch to dev branch

```
git checkout dev
git pull
```

and redo the pairing (after checking that there is no more things left in regards to this device

mamo79, 2023-01-14:
Hello,
It's OK. Only the covering percent are strange. (60%=100%)
Thanks.

pipiche38, 2023-01-15:
Is the issue only on command from Domoticz -> Device or on Device -> Domoticz are both ?

Can you enable some debug flag via the WebUi

Cluster
Widget
[image]

then you play with the shutter, and finaly you drop here the Logs/PluginZigeeb-xx.log file

mamo79, 2023-01-16:
Hello,

The module make a calibration 

[image]

When I send 65, covering are 100%

[image]

In Domoticz the percent are more realistic but I must activate the option "Reverse Position" in domoticz for command open and close are good. 

[PluginZigbee_30.log](https://github.com/zigbeefordomoticz/Domoticz-Zigbee/files/10427641/PluginZigbee_30.log)

pipiche38, 2023-01-16:
Ok. But then I think all make sense, as the command you are using here take the input in hex . so 0x64 = 100

The most important thing is to have the Domoticz part working, and if I understood yes ? Did you try the % level also from Domoticz and is Ok ?

pipiche38, 2023-01-17:
Do you still have an issue ? My understanding that all is ok, and as the input is given in hex and not dec 65 stands for 100.

We will release soon those change in the January update

closed 2023-01-17 by pipiche38

mamo79, 2023-01-18:
Hello,
It's OK.
Thanks.