zigbeefordomoticz/Domoticz-Zigbee #1439: [Model Certification] - SceneSwitch-EM-3.0 Heiman

pabaillon, 2023-01-13:
1. Please do provide device information such as Brand and Model name.
Télécommande murale sans fil pour scènes Zigbee - Heiman
1. Provide the URL where you have purchase the device
https://www.planete-domotique.com/telecommande-murale-sans-fil-pour-scenes-zigbee-heiman.html
1. Finaly copy-paste all information provided in the log

{"Version":"3","ZDeviceName":"","Status":"inDB","SQN":"0e","Ep":{"01":{"0000":{"0004":"HEIMAN","0005":"SceneSwitch-EM-3.0","0000":"3","0001":"16","0002":"2","0003":"16","0006":"2019.3.18","0007":"3"},"0001":{"0020":29,"0021":136},"0003":{},"0004":{},"0005":{},"0b05":{},"0019":{},"fc80":{},"ClusterType":{"2601":"Voltage"}}},"Heartbeat":"145","RIA":"4","LQI":67,"Battery":136,"Model":"SceneSwitch-EM-3.0","ForceAckCommands":[],"MacCapa":"80","IEEE":"000d6f001563c8e9","Type":"Voltage","ProfileID":"0104","ZDeviceID":"0004","App Version":"16","Attributes List":{},"DeviceType":"RFD","HW Version":"16","Last Cmds":[[1453,"02","c426","0100","010100190000000003000100020003"],[1454,"02","c426","0100","010100190000000003000400050006"],[1455,"02","c426","0100","010100190000000003000700080009"],[1456,"02","c426","0100","010100190000000001000a"],[1457,"02","c426","0100","010100000000000003000400050000"],[1458,"02","c426","0100","010100000000000003000100020003"],[1459,"02","c426","0100","01010000000000000300060007000a"],[1460,"02","c426","0100","010100000000000003000f00100015"],[1461,"02","c426","0100","0101000000000000024000f000"],[1467,"02","c426","0062","010100"]],"LogicalType":"End Device","Manufacturer":"120b","Manufacturer Name":"HEIMAN","NbEp":"1","PowerSource":"Battery","ReadAttributes":{"Ep":{"01":{"0000":{"TimeStamp":1673606467,"iSQN":{"4000":1461,"0004":1457,"0005":1457,"0000":1457,"0001":1458,"0002":1458,"0003":1458,"0006":1459,"0007":1459,"000a":1459,"000f":1460,"0010":1460,"0015":1460,"f000":1461},"Attributes":{"4000":"86","0004":"00","0005":"00","0000":"00","0001":"00","0002":"00","0003":"00","0006":"00","0007":"00","000a":"86","000f":"86","0010":"86","0015":"86","f000":"86"},"ZigateRequest":{}},"0001":{"TimeStamp":1673606456,"iSQN":{"0000":1450,"0001":1450,"0003":1450,"0020":1451,"0021":1451,"0033":1451,"0035":1452},"Attributes":{"0020":"00","0021":"00","0033":"86","0035":"86","0000":"86","0001":"86","0003":"86"},"ZigateRequest":{}},"0019":{"TimeStamp":1673606454,"iSQN":{"0001":1453,"0002":1453,"0003":1453,"0004":1454,"0005":1454,"0006":1454,"0007":1455,"0008":1455,"0009":1455,"000a":1456},"Attributes":{"0001":"86","0002":"86","0003":"86","0004":"86","0005":"86","0006":"86","0007":"86","0008":"86","0009":"86","000a":"86"},"ZigateRequest":{}}}}},"ReceiveOnIdle":"Off","Stack Version":"2","Stamp":{"time":1673606751.3719196,"Time":"2023-01-13 10:45:51","MsgType":"8011","LastSeen":1673606751},"ZCL Version":"3","Health":"Live","Capability":["Reduced-Function Device","Standard security","NwkAddr should be allocated"],"PairingInProgress":false,"RollingLQI":[68,67,67,67,66,66,73,64,64,67,67],"Epv2":{"01":{"ClusterIn":{"0000":{},"0001":{},"0003":{},"0004":{},"0005":{},"0b05":{}},"ClusterOut":{"0019":{},"fc80":{}},"ProfileID":"0104","ZDeviceID":"0004"}},"ZDeviceVersion":"0000","ConfigSource":"8043","_rawNodeDescriptor":"120b005200522c000080524002","Max Buffer Size":"52","Max Rx":"0052","Max Tx":"0052","macapa":"80","bitfield":"4002","server_mask":"2c00","descriptor_capability":"00","PairingTime":1673606411.1014073,"Bind":{"01":{"0000":{"Target":"0000","Stamp":1673606411,"Phase":"binded","Status":"00","i_sqn":1442},"0001":{"Target":"0000","Stamp":1673606413,"Phase":"binded","Status":"00","i_sqn":1443},"0003":{"Target":"0000","Stamp":1673606416,"Phase":"binded","Status":"00","i_sqn":1444},"0004":{"Target":"0000","Stamp":1673606419,"Phase":"binded","Status":"00","i_sqn":1445},"0005":{"Target":"0000","Stamp":1673606422,"Phase":"binded","Status":"00","i_sqn":1446},"0b05":{"Target":"0000","Stamp":1673606425,"Phase":"binded","Status":"8c","i_sqn":1447}}},"BindingTable":{"SQN":0,"Status":"Success","TimeStamp":1673606477.7514646,"BindingTableSize":5,"Devices":[{"000d6f001563c8e9":{"sourceEp":"01","Cluster":"0000","targetIEEE":"00158d0005d2857f","targetNickName":"Zigbee Coordinator","targetEp":"01"}},{"000d6f001563c8e9":{"sourceEp":"01","Cluster":"0001","targetIEEE":"00158d0005d2857f","targetNickName":"Zigbee Coordinator","targetEp":"01"}},{"000d6f001563c8e9":{"sourceEp":"01","Cluster":"0003","targetIEEE":"00158d0005d2857f","targetNickName":"Zigbee Coordinator","targetEp":"01"}},{"000d6f001563c8e9":{"sourceEp":"01","Cluster":"0000","targetIEEE":"00158d0005d2857f","targetNickName":"Zigbee Coordinator","targetEp":"01"}},{"000d6f001563c8e9":{"sourceEp":"01","Cluster":"0001","targetIEEE":"00158d0005d2857f","targetNickName":"Zigbee Coordinator","targetEp":"01"}},{"000d6f001563c8e9":{"sourceEp":"01","Cluster":"0003","targetIEEE":"00158d0005d2857f","targetNickName":"Zigbee Coordinator","targetEp":"01"}},{"000d6f001563c8e9":{"sourceEp":"01","Cluster":"0004","targetIEEE":"00158d0005d2857f","targetNickName":"Zigbee Coordinator","targetEp":"01"}},{"000d6f001563c8e9":{"sourceEp":"01","Cluster":"0005","targetIEEE":"00158d0005d2857f","targetNickName":"Zigbee Coordinator","targetEp":"01"}},{"000d6f001563c8e9":{"sourceEp":"01","Cluster":"0004","targetIEEE":"00158d0005d2857f","targetNickName":"Zigbee Coordinator","targetEp":"01"}},{"000d6f001563c8e9":{"sourceEp":"01","Cluster":"0005","targetIEEE":"00158d0005d2857f","targetNickName":"Zigbee Coordinator","targetEp":"01"}},{"000d6f001563c8e9":{"sourceEp":"01","Cluster":"0004","targetIEEE":"00158d0005d2857f","targetNickName":"Zigbee Coordinator","targetEp":"01"}},{"000d6f001563c8e9":{"sourceEp":"01","Cluster":"0005","targetIEEE":"00158d0005d2857f","targetNickName":"Zigbee Coordinator","targetEp":"01"}},{"000d6f001563c8e9":{"sourceEp":"01","Cluster":"0004","targetIEEE":"00158d0005d2857f","targetNickName":"Zigbee Coordinator","targetEp":"01"}},{"000d6f001563c8e9":{"sourceEp":"01","Cluster":"0005","targetIEEE":"00158d0005d2857f","targetNickName":"Zigbee Coordinator","targetEp":"01"}}]},"ZDPSQN":"05","ConfigureReporting":{"Ep":{"01":{"0001":{"TimeStamp":1673606411.1052017,"iSQN":{"0000":1449,"0020":1449,"0021":1449},"Attributes":{"0000":"86","0020":"8d","0021":"8d"},"ZigateRequest":{}}}}},"CertifiedDevice":false,"BatteryUpdateTime":1673606445,"WriteAttributes":{"Ep":{"01":{"0000":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"0001":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"0003":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"0004":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"0005":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"0b05":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"0019":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"fc80":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}}}}},"SWBUILD_1":"2019.3.18","GroupMemberShip":{"01":{}}}

pipiche38, 2023-01-13:
What this device is expected to do ?
Can you enable a debug flag debugInput
then do some action on the device and check if there is anything coming in the logs ( Logs/PluginZigbee-xx.log )
if so please attached the log file and try to document the action you did on the device and the output on the log

pabaillon, 2023-01-14:
Quand j'appuie sur le bounton scène en haut
`2023-01-14 12:09:14,729 INFO    : [ZiGateSerial_12 0] zcl_decoders Unknown Command: f0 NwkId: c426 Ep: 01 Cluster: fc80 Payload:  - GlobalCommand: False, Sqn: 07, ManufacturerCode: 120b
2023-01-14 12:09:14,735 DEBUG   : [ZiGateForwarder_12 0] ZigateRead - MsgType: 8002,  Data: 000104fc80010102c426020000050b1207f0, LQI: 87
2023-01-14 12:09:14,739 INFO    : [ZiGateForwarder_12 0] zcl_decoders Unknown Command: f0 NwkId: c426 Ep: 01 Cluster: fc80 Payload:  - GlobalCommand: False, Sqn: 07, ManufacturerCode: 120b
2023-01-14 12:09:14,740 DEBUG   : [ZiGateForwarder_12 0] Reception Data indication, Source Address: c426 Destination Address: 0000 ProfilID: 0104 ClusterID: fc80 Message Payload: 050b1207f0
2023-01-14 12:09:14,741 INFO    : [ZiGateForwarder_12 0] inRawAps c426/01 Cluster fc80 Manuf: 120b/HEIMAN Command: f0 Data:  Payload: 050b1207f0 not processed !!!`

quand j'appuie sur le bouton maison (droite)

`2023-01-14 13:41:03,645 INFO    : [ZiGateSerial_12 0] zcl_decoders Unknown Command: f1 NwkId: c426 Ep: 01 Cluster: fc80 Payload:  - GlobalCommand: False, Sqn: 15, ManufacturerCode: 120b
2023-01-14 13:41:03,647 DEBUG   : [ZiGateForwarder_12 0] ZigateRead - MsgType: 8002,  Data: 000104fc80010102c426020000050b1215f1, LQI: 73
2023-01-14 13:41:03,648 INFO    : [ZiGateForwarder_12 0] zcl_decoders Unknown Command: f1 NwkId: c426 Ep: 01 Cluster: fc80 Payload:  - GlobalCommand: False, Sqn: 15, ManufacturerCode: 120b
2023-01-14 13:41:03,649 DEBUG   : [ZiGateForwarder_12 0] Reception Data indication, Source Address: c426 Destination Address: 0000 ProfilID: 0104 ClusterID: fc80 Message Payload: 050b1215f1
2023-01-14 13:41:03,650 INFO    : [ZiGateForwarder_12 0] inRawAps c426/01 Cluster fc80 Manuf: 120b/HEIMAN Command: f1 Data:  Payload: 050b1215f1 not processed !!!`

quand j'appuie sur le bouton bas (lune)

`2023-01-14 13:43:48,165 INFO    : [ZiGateSerial_12 0] zcl_decoders Unknown Command: f2 NwkId: c426 Ep: 01 Cluster: fc80 Payload:  - GlobalCommand: False, Sqn: 17, ManufacturerCode: 120b
2023-01-14 13:43:48,317 DEBUG   : [ZiGateForwarder_12 0] ZigateRead - MsgType: 8002,  Data: 000104fc80010102c426020000050b1217f2, LQI: 74
2023-01-14 13:43:48,318 INFO    : [ZiGateForwarder_12 0] zcl_decoders Unknown Command: f2 NwkId: c426 Ep: 01 Cluster: fc80 Payload:  - GlobalCommand: False, Sqn: 17, ManufacturerCode: 120b
2023-01-14 13:43:48,319 DEBUG   : [ZiGateForwarder_12 0] Reception Data indication, Source Address: c426 Destination Address: 0000 ProfilID: 0104 ClusterID: fc80 Message Payload: 050b1217f2
2023-01-14 13:43:48,319 INFO    : [ZiGateForwarder_12 0] inRawAps c426/01 Cluster fc80 Manuf: 120b/HEIMAN Command: f2 Data:  Payload: 050b1217f2 not processed !!!
`

quand j'appuie sur le bouton maison sortie(gauche)

`2023-01-14 13:44:52,337 DEBUG   : [ZiGateForwarder_12 0] Decode8102 - LQI:  83 Received Cluster:0006 Attribute: 0000 Value:   01 from (dcf7/01)
2023-01-14 13:44:52,339 DEBUG   : [ZiGateForwarder_12 0] ZigateRead - MsgType: 8011,  Data: 009a5d0100060b, LQI: 0
2023-01-14 13:44:53,091 INFO    : [ZiGateSerial_12 0] zcl_decoders Unknown Command: f3 NwkId: c426 Ep: 01 Cluster: fc80 Payload:  - GlobalCommand: False, Sqn: 18, ManufacturerCode: 120b
2023-01-14 13:44:53,092 DEBUG   : [ZiGateForwarder_12 0] ZigateRead - MsgType: 8002,  Data: 000104fc80010102c426020000050b1218f3, LQI: 73
2023-01-14 13:44:53,093 INFO    : [ZiGateForwarder_12 0] zcl_decoders Unknown Command: f3 NwkId: c426 Ep: 01 Cluster: fc80 Payload:  - GlobalCommand: False, Sqn: 18, ManufacturerCode: 120b
2023-01-14 13:44:53,093 DEBUG   : [ZiGateForwarder_12 0] Reception Data indication, Source Address: c426 Destination Address: 0000 ProfilID: 0104 ClusterID: fc80 Message Payload: 050b1218f3
2023-01-14 13:44:53,094 INFO    : [ZiGateForwarder_12 0] inRawAps c426/01 Cluster fc80 Manuf: 120b/HEIMAN Command: f3 Data:  Payload: 050b1218f3 not processed !!!
`

pipiche38, 2023-01-14:
1st iteration.

1. can you switch to the dev branche

```
git checkout dev
git pull

```

2. Remove the existing pairing as we have to redo it. Mainly you have to remove the Widget 'Voltage' which has been created

3. restart the plugin

4. check that the device doesn't exist in the Plugin device management list

5. do the pairing
=> This should create one Selector Widget

do the action on the device and see what is happening

Please cross-check the logs in Domoticz and in the PluginLog.

pabaillon, 2023-01-14:
Quand j'appuie sur les boutons rien ne se passe.
`2023-01-14 16:05:27,788 INFO    : [ZiGateSerial_12 0] zcl_decoders Unknown Command: f3 NwkId: 83bd Ep: 01 Cluster: fc80 Payload:  - GlobalCommand: False, Sqn: 4b, ManufacturerCode: 120b
2023-01-14 16:05:27,789 DEBUG   : [ZiGateForwarder_12 0] ZigateRead - MsgType: 8002,  Data: 000104fc8001010283bd020000050b124bf3, LQI: 76
2023-01-14 16:05:27,791 INFO    : [ZiGateForwarder_12 0] zcl_decoders Unknown Command: f3 NwkId: 83bd Ep: 01 Cluster: fc80 Payload:  - GlobalCommand: False, Sqn: 4b, ManufacturerCode: 120b
2023-01-14 16:05:27,792 DEBUG   : [ZiGateForwarder_12 0] Reception Data indication, Source Address: 83bd Destination Address: 0000 ProfilID: 0104 ClusterID: fc80 Message Payload: 050b124bf3
2023-01-14 16:05:27,793 INFO    : [ZiGateForwarder_12 0] heimanReadRawAPS - Nwkid: 83bd Ep: 01, Cluster: fc80, Payload: 050b124bf3
2023-01-14 16:05:27,793 INFO    : [ZiGateForwarder_12 0] heimanReadRawAPS - Nwkid: 83bd Ep: 01, command found: f3`

pipiche38, 2023-01-14:
c'est pas mal. on avance bien
peux-tu refaire une maj

git pull

puis simplement relancer le plugin 

et tu re-appuis sur des boutons

Si rien ne se passe, il faudrait ajouter dans le debug Widgets

pabaillon, 2023-01-14:
Les boutons répondent. parfaitement.

pipiche38, 2023-01-14:
cool. Peux-tu me dire ce qu'il se passe dans les logs, si tu restes appuyés sur les bouttons, ou si tu fais des double click
Je me demande si il ne peut pas y avoir plusieurs ordres par bouton

pabaillon, 2023-01-15:
- Simple click bouton movie

`15 10:27:57,261 INFO    : [ZiGateSerial_12 0] zcl_decoders Unknown Command: f0 NwkId: 83bd Ep: 01 Cluster: fc80 Payload:  - GlobalCommand: False, Sqn: 5b, ManufacturerCode: 120b
2023-01-15 10:27:57,266 DEBUG   : [ZiGateForwarder_12 0] ZigateRead - MsgType: 8002,  Data: 000104fc8001010283bd020000050b125bf0, LQI: 91
2023-01-15 10:27:57,267 INFO    : [ZiGateForwarder_12 0] zcl_decoders Unknown Command: f0 NwkId: 83bd Ep: 01 Cluster: fc80 Payload:  - GlobalCommand: False, Sqn: 5b, ManufacturerCode: 120b
2023-01-15 10:27:57,267 DEBUG   : [ZiGateForwarder_12 0] Reception Data indication, Source Address: 83bd Destination Address: 0000 ProfilID: 0104 ClusterID: fc80 Message Payload: 050b125bf0
2023-01-15 10:27:57,268 INFO    : [ZiGateForwarder_12 0] heimanReadRawAPS - Nwkid: 83bd Ep: 01, Cluster: fc80, Payload: 050b125bf0
2023-01-15 10:27:57,269 INFO    : [ZiGateForwarder_12 0] heimanReadRawAPS - Nwkid: 83bd Ep: 01, command found: f0
2023-01-15 10:27:57,270 ERROR   : [ZiGateForwarder_12] heimanReadRawAPS - Nwkid: 83bd Ep: 01, command found: f0 UNKNOW`

- Appui long

`2023-01-15 10:29:25,569 INFO    : [ZiGateSerial_12 0] zcl_decoders Unknown Command: f0 NwkId: 83bd Ep: 01 Cluster: fc80 Payload:  - GlobalCommand: False, Sqn: 5c, ManufacturerCode: 120b
2023-01-15 10:29:25,570 DEBUG   : [ZiGateForwarder_12 0] ZigateRead - MsgType: 8002,  Data: 000104fc8001010283bd020000050b125cf0, LQI: 90
2023-01-15 10:29:25,572 INFO    : [ZiGateForwarder_12 0] zcl_decoders Unknown Command: f0 NwkId: 83bd Ep: 01 Cluster: fc80 Payload:  - GlobalCommand: False, Sqn: 5c, ManufacturerCode: 120b
2023-01-15 10:29:25,573 DEBUG   : [ZiGateForwarder_12 0] Reception Data indication, Source Address: 83bd Destination Address: 0000 ProfilID: 0104 ClusterID: fc80 Message Payload: 050b125cf0
2023-01-15 10:29:25,574 INFO    : [ZiGateForwarder_12 0] heimanReadRawAPS - Nwkid: 83bd Ep: 01, Cluster: fc80, Payload: 050b125cf0
2023-01-15 10:29:25,574 INFO    : [ZiGateForwarder_12 0] heimanReadRawAPS - Nwkid: 83bd Ep: 01, command found: f0
2023-01-15 10:29:25,574 ERROR   : [ZiGateForwarder_12] heimanReadRawAPS - Nwkid: 83bd Ep: 01, command found: f0 UNKNOW`

- Double click

`
2023-01-15 10:30:35,866 INFO    : [ZiGateSerial_12 0] zcl_decoders Unknown Command: f0 NwkId: 83bd Ep: 01 Cluster: fc80 Payload:  - GlobalCommand: False, Sqn: 5d, ManufacturerCode: 120b
2023-01-15 10:30:35,868 DEBUG   : [ZiGateForwarder_12 0] ZigateRead - MsgType: 8002,  Data: 000104fc8001010283bd020000050b125df0, LQI: 90
2023-01-15 10:30:35,870 INFO    : [ZiGateForwarder_12 0] zcl_decoders Unknown Command: f0 NwkId: 83bd Ep: 01 Cluster: fc80 Payload:  - GlobalCommand: False, Sqn: 5d, ManufacturerCode: 120b
2023-01-15 10:30:35,871 DEBUG   : [ZiGateForwarder_12 0] Reception Data indication, Source Address: 83bd Destination Address: 0000 ProfilID: 0104 ClusterID: fc80 Message Payload: 050b125df0
2023-01-15 10:30:35,872 INFO    : [ZiGateForwarder_12 0] heimanReadRawAPS - Nwkid: 83bd Ep: 01, Cluster: fc80, Payload: 050b125df0
2023-01-15 10:30:35,872 INFO    : [ZiGateForwarder_12 0] heimanReadRawAPS - Nwkid: 83bd Ep: 01, command found: f0
2023-01-15 10:30:35,873 ERROR   : [ZiGateForwarder_12] heimanReadRawAPS - Nwkid: 83bd Ep: 01, command found: f0 UNKNOW
2023-01-15 10:30:36,134 INFO    : [ZiGateSerial_12 0] zcl_decoders Unknown Command: f0 NwkId: 83bd Ep: 01 Cluster: fc80 Payload:  - GlobalCommand: False, Sqn: 5e, ManufacturerCode: 120b
2023-01-15 10:30:36,137 DEBUG   : [ZiGateForwarder_12 0] ZigateRead - MsgType: 8002,  Data: 000104fc8001010283bd020000050b125ef0, LQI: 92
2023-01-15 10:30:36,140 INFO    : [ZiGateForwarder_12 0] zcl_decoders Unknown Command: f0 NwkId: 83bd Ep: 01 Cluster: fc80 Payload:  - GlobalCommand: False, Sqn: 5e, ManufacturerCode: 120b
2023-01-15 10:30:36,141 DEBUG   : [ZiGateForwarder_12 0] Reception Data indication, Source Address: 83bd Destination Address: 0000 ProfilID: 0104 ClusterID: fc80 Message Payload: 050b125ef0
2023-01-15 10:30:36,142 INFO    : [ZiGateForwarder_12 0] heimanReadRawAPS - Nwkid: 83bd Ep: 01, Cluster: fc80, Payload: 050b125ef0
2023-01-15 10:30:36,142 INFO    : [ZiGateForwarder_12 0] heimanReadRawAPS - Nwkid: 83bd Ep: 01, command found: f0
2023-01-15 10:30:36,143 ERROR   : [ZiGateForwarder_12] heimanReadRawAPS - Nwkid: 83bd Ep: 01, command found: f0 UNKNOW`

pipiche38, 2023-01-15:
Ok so we are good. I'll do a final update and let you know

pipiche38, 2023-01-15:
So I remove the action Other from the widget, and pushed and update.
If you can retrieve the latest version
`git pull`

you will have remove the existing paired information, to re-do the pairing after a restart of the plugin

pabaillon, 2023-01-15:
Ok c'est fait. Les boutons fonctionnent. Mais quand je re-clique sur le même bouton, domoticz affiche une fenêtre avec marqué " commutation Go out". Je ne sais pas si c'est un fonctionnement normal. Et l'objet reste dans le même état (Movie reste movie, home reste home etc...)

pipiche38, 2023-01-15:
Oui, le selecteur reste sur la dernière action reçue. Par contre l'action est bien prise en compte et un nouvel événement est enregistré (tu peux vérifier dans les logs)

Si tu souhaites une remise à 0, tu peux utiliser

{ "resetSwitchSelectorPushButton": 30 } comme paramètre du device,n pour qu'il y ait un reset au bout de 30 secondes

https://zigbeefordomoticz.github.io/wiki/fr-fr/Tuto_Parametrer-les-objets.html

pipiche38, 2023-01-16:
I suggest to close this issue now. It will be pushed on stable in the next release cycle

closed 2023-01-16 by pipiche38