zigbeefordomoticz/Domoticz-Zigbee #1442: Not optimized hardware: GLEDOPTO G95

RMvdBijl, 2023-01-14:
Hi, 
I succesfully installed ZiGate+V2 with Domoticz. The Inidio dimmers and a Ikea bulb are added correctly (green) but not this bulb: GLEDOPTO G95 7W LED filament bulb (Zigbee 3.0). The information that is shown is that this bulb is not optimized (orange) with the plugin 6.3008.

What is needed? What can I do?
Or does the plugin needs a update for this bulb?

Many thanks for your answers.

Kind regards, Ronald

The export (1).sjon:

Version: "3"
ZDeviceName: ""
Status: "inDB"
SQN: "71"
Ep: Object {"01":{"1000":{},"0000":{"4000":"10051203","0004":"GLEDOPTO","0005":"GL-B-004P","0000":"3","0001":"0","0002":"2","0003":"0","0006":"20211015","0007":"1"},"0003":{},"0004":{},"0005":{},"0006":{"4001":"0","4002":"0","0000":"01"},"0008":{"0000":"4a"},"0300":{"0007":"158","0008":"2","400a":"16"},"0019":{"0001":"ffffffff","0002":"13013001","0003":"0002","0004":"ffffffff","0005":"ffff","0006":"00","0007":"124f","0008":"ffff","0009":"0000"},"ClusterType":{"215":"ColorControlWW"}},"f2":{"0021":{}}}
Heartbeat: "985"
RIA: "4"
LQI: 49
Battery: Object {}
Model: "GL-B-004P"
ForceAckCommands: Array[0] []
MacCapa: "8e"
IEEE: "a4c138c3d5c59f42"
Type: "Switch/LvlControl/ColorControl"
ProfileID: "a1e0"
ZDeviceID: "0061"
App Version: "0"
Attributes List: Object {"Request":{"01":{"1000":1673700112.094324,"0000":1673700112.083158,"0003":1673700112.0836222,"0004":1673700112.0840838,"0005":1673700112.0845397,"0006":1673700112.0923824,"0008":1673700112.0933232,"0300":1673700112.0937881,"0019":1673700112.0947874},"f2":{"0021":1673700112.0952702}}}
DeviceType: "FFD"
HW Version: "0"
Last Cmds: Array[10] [[117,"02","59d0","0081","010101bf0010"],["None","02","59d0","0081","010101bf0010"],[118,"02","59d0","0081","0101019c0010"],["None","02","59d0","0081","0101019c0010"],[119,"02","59d0","0081","0101016e0010"],["None","02","59d0","0081","0101016e0010"],[120,"02","59d0","0081","0101014a0010"],["None","02","59d0","0081","0101014a0010"],[122,"07","59d0","0100","0101000600000000010000"],[123,"07","59d0","0100","0101000800000000010000"]]
LogicalType: "Router"
Manufacturer: "124f"
Manufacturer Name: "GLEDOPTO"
NbEp: "2"
PowerSource: "Main"
ReadAttributes: Object {"Ep":{"01":{"0000":{"TimeStamp":"14/01/2023 12:24:45","iSQN":{"4000":2683,"0004":2681,"0005":2681,"0000":2681,"0001":2681,"0002":2681,"0003":2682,"0006":2682,"0007":2682,"000a":2682,"000f":2682,"0010":2683,"0015":2683,"f000":2683},"Attributes":{"4000":"00","0004":"00","0005":"00","0000":"00","0001":"00","0002":"00","0003":"00","0006":"00","0007":"00","000a":"86","000f":"86","0010":"86","0015":"86","f000":"86"},"ZigateRequest":{}},"0019":{"TimeStamp":"14/01/2023 12:24:44","iSQN":{"0001":2675,"0002":2675,"0003":2675,"0004":2675,"0005":2675,"0006":2676,"0007":2676,"0008":2676,"0009":2676,"000a":2676},"Attributes":{"0001":"00","0002":"00","0003":"00","0004":"00","0005":"00","0006":"00","0007":"00","0008":"00","0009":"00","000a":"86"},"ZigateRequest":{}},"0006":{"TimeStamp":"14/01/2023 19:53:53","iSQN":{"4001":2677,"4002":2677,"0000":122},"Attributes":{"4001":"00","4002":"00","0000":"00"},"ZigateRequest":{}},"0008":{"TimeStamp":"14/01/2023 19:54:58","iSQN":{"0000":123},"Attributes":{"0000":"00"},"ZigateRequest":{}},"0300":{"TimeStamp":"14/01/2023 12:24:44","iSQN":{"0000":2679,"0001":2679,"0003":2679,"0004":2679,"0007":2679,"0008":2680,"400a":2680},"Attributes":{"0000":"86","0001":"86","0003":"86","0004":"86","0007":"00","0008":"00","400a":"00"},"ZigateRequest":{}}},"f2":{"0000":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}}}}}
ReceiveOnIdle: "On"
Stack Version: "2"
Stamp: Object {"time":1673722591.2422962,"Time":"2023-01-14 19:56:31","MsgType":"8011","LastSeen":"14/01/2023 19:56:31"}
ZCL Version: "3"
Health: "Live"
SWBUILD_1: "20211015"
BindingTable: Object {"SQN":0,"Status":"Success","TimeStamp":"14/01/2023 12:24:45","BindingTableSize":8,"Devices":[{"a4c138c3d5c59f42":{"sourceEp":"01","Cluster":"0000","targetIEEE":"00158d0005bea7d4","targetNickName":"Zigbee Coordinator","targetEp":"01"}},{"a4c138c3d5c59f42":{"sourceEp":"01","Cluster":"0003","targetIEEE":"00158d0005bea7d4","targetNickName":"Zigbee Coordinator","targetEp":"01"}},{"a4c138c3d5c59f42":{"sourceEp":"01","Cluster":"0004","targetIEEE":"00158d0005bea7d4","targetNickName":"Zigbee Coordinator","targetEp":"01"}},{"a4c138c3d5c59f42":{"sourceEp":"01","Cluster":"0005","targetIEEE":"00158d0005bea7d4","targetNickName":"Zigbee Coordinator","targetEp":"01"}},{"a4c138c3d5c59f42":{"sourceEp":"01","Cluster":"0006","targetIEEE":"00158d0005bea7d4","targetNickName":"Zigbee Coordinator","targetEp":"01"}},{"a4c138c3d5c59f42":{"sourceEp":"01","Cluster":"0008","targetIEEE":"00158d0005bea7d4","targetNickName":"Zigbee Coordinator","targetEp":"01"}},{"a4c138c3d5c59f42":{"sourceEp":"01","Cluster":"0300","targetIEEE":"00158d0005bea7d4","targetNickName":"Zigbee Coordinator","targetEp":"01"}},{"a4c138c3d5c59f42":{"sourceEp":"01","Cluster":"1000","targetIEEE":"00158d0005bea7d4","targetNickName":"Zigbee Coordinator","targetEp":"01"}}]}
ConfigSource: "8043"
Epv2: Object {"01":{"ClusterIn":{"1000":{},"0000":{},"0003":{},"0004":{},"0005":{},"0006":{},"0008":{},"0300":{}},"ClusterOut":{"0019":{}},"ProfileID":"0104","ZDeviceID":"010c"},"f2":{"ClusterIn":{},"ClusterOut":{"0021":{}},"ProfileID":"a1e0","ZDeviceID":"0061"}}
ReadConfigureReporting: Object {"Ep":{"01":{"0006":{"0000":{"TimeStamp":"14/01/2023 12:34:01","Status":"00","DataType":"10","MinInterval":"0001","MaxInterval":"012c"}},"0008":{"0000":{"TimeStamp":"14/01/2023 12:34:01","Status":"00","DataType":"20","MinInterval":"0005","MaxInterval":"012c"}}}},"TimeStamp":"14/01/2023 12:34:01"}
WriteAttributes: Object {"Ep":{"01":{"1000":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"0000":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"0003":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"0004":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"0005":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"0006":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"0008":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"0300":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"0019":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}}},"f2":{"0021":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}}}}}
CertifiedDevice: false
Bind: Object {"01":{"1000":{"Target":"0000","Stamp":"14/01/2023 12:24:43","Phase":"binded","Status":"00","i_sqn":2670},"0000":{"Target":"0000","Stamp":"14/01/2023 12:24:42","Phase":"binded","Status":"00","i_sqn":2663},"0003":{"Target":"0000","Stamp":"14/01/2023 12:24:42","Phase":"binded","Status":"00","i_sqn":2664},"0004":{"Target":"0000","Stamp":"14/01/2023 12:24:42","Phase":"binded","Status":"00","i_sqn":2665},"0005":{"Target":"0000","Stamp":"14/01/2023 12:24:42","Phase":"binded","Status":"00","i_sqn":2666},"0006":{"Target":"0000","Stamp":"14/01/2023 12:24:42","Phase":"binded","Status":"00","i_sqn":2667},"0008":{"Target":"0000","Stamp":"14/01/2023 12:24:43","Phase":"binded","Status":"00","i_sqn":2668},"0300":{"Target":"0000","Stamp":"14/01/2023 12:24:43","Phase":"binded","Status":"00","i_sqn":2669}}}
Log_UnknowDeviceFlag: 1673711779.0443385
ColorInfos: Object {"ColorTemperatureMireds":"158","ColorMode":2,"ColorCapabilities":"16"}
Capability: Array[5] ["Full-Function Device","Main Powered","Receiver during Idle","Standard security","NwkAddr should be allocated"]
SWBUILD_3: "10051203"
ConfigureReporting: Object {"Ep":{"01":{"0006":{"TimeStamp":"14/01/2023 12:24:42","iSQN":{"0000":2672},"Attributes":{"0000":"00"},"ZigateRequest":{}},"0008":{"TimeStamp":"14/01/2023 12:24:42","iSQN":{"0000":2673},"Attributes":{"0000":"00"},"ZigateRequest":{}},"0300":{"TimeStamp":"14/01/2023 12:24:42","iSQN":{"0008":2674},"Attributes":{"0008":"86"},"ZigateRequest":{}}}}}
ConsistencyCheck: "ok"
RollingLQI: Array[11] [47,47,49,47,49,50,49,49,47,48,49]
NwkID: "59d0"

pipiche38, 2023-01-14:
please could you provide the export.json in a json format. here it is a text format and not usable. ether you copy-paste into a code block or you attached as a file

RMvdBijl, 2023-01-14:
Hi, thanks for your response. 

I have difficulties in dragging & dropping from C:/ e.g. the file 'export(1).sjon' is not supported. It needs to be a different file extension. Any suggestion?

pipiche38, 2023-01-14:
That is ok. I just pushed a config file for this bulb. If you want to test/check you'll need to switch to the dev branch
```

git checkout dev
git pull
```

and restart the plugin

It would be kind of you also, if you could add this to https://zigbee.blakadder.com/z4d.html

RMvdBijl, 2023-01-14:
Hi again, I work with Putty (SSH) and changed the directory in Raspberry Pi to pi@raspberrypi:~/domoticz/plugins/Domoticz-Zigbee $ and added your command 'git checkout dev', but then this error occurs: 'fatal: .git/index: index file open failed: Permission denied'

Any suggestions?

KR, Ronald

pipiche38, 2023-01-16:
you might have to do it with the root priviliges by doing

```
sudo git checkout dev
sudo git pull
```

RMvdBijl, 2023-01-16:
Hi, the config file you pushed works perfectly! In Zigate dashboard the bulb is showing now green.

I used following commands:
cd domoticz/plugins/Domoticz-Zigbee/ 
sudo git checkout dev
sudo git pull
sudo reboot

I will add the GLEDOPTO G95 7W LED filament bulb (Zigbee 3.0) to https://zigbee.blakadder.com/z4d.html and make a PayPal donation for all the nice work  you all creating here.

pipiche38, 2023-01-16:
thanks for all.
Closing for now

closed 2023-01-16 by pipiche38

RMvdBijl, 2023-01-26:
Hi,
The current plugin version I use with Zigate+ V2 / Domoticz is 6.4.150 (red coloured).
I tried to update to the new plugin automatically as well as manually but did not succeed.
Could this be caused by the config file you pushed? 
Many thanks for your answer in advance. KR, Ronald

Herewith the log file:

Starting Zigbee for Domoticz plugin Upgrade process.
----------------------------------------------------
PWD=/home/pi/domoticz/plugins/Domoticz-Zigbee
LANG=en_GB.UTF-8
INVOCATION_ID=b97aefb8edd04060acba90652fb4795a
SHLVL=1
JOURNAL_STREAM=8:11581
PATH=/sbin:/bin:/usr/sbin:/usr/bin:/usr/local/bin
_=/bin/env
uid=0(root) gid=0(root) groups=0(root)

(1) updating Zigbee for Domoticz plugin

Setup submodule.recurse

hint: Pulling without specifying how to reconcile divergent branches is
hint: discouraged. You can squelch this message by running one of the following
hint: commands sometime before your next pull:
hint:
hint: git config pull.rebase false # merge (the default strategy)
hint: git config pull.rebase true # rebase
hint: git config pull.ff only # fast-forward only
hint:
hint: You can replace "git config" with "git config --global" to set a default
hint: preference for all repositories. You can also pass --rebase, --no-rebase,
hint: or --ff-only on the command line to override the configured default per
hint: invocation.
Fetching submodule external/bellows
Fetching submodule external/zigpy
Fetching submodule external/zigpy-deconz
Fetching submodule external/zigpy-zigate
Fetching submodule external/zigpy-znp
Your configuration specifies to merge with the ref 'refs/heads/dev'
from the remote, but no such ref was fetched.
ERROR while running command 'git pull --recurse-submodules'.
Git Status: On branch dev
Your branch is up to date with 'origin/dev'.

nothing to commit, working tree clean

pipiche38, 2023-01-26:
please use one of the Domoticz forum 

    English channel : https://www.domoticz.com/forum/viewforum.php?f=68
    French Channel : https://easydomoticz.com/forum/viewforum.php?f=28&sid=a8633a9d8cb2acccb32872543b50fd47

RMvdBijl, 2023-01-27:
Thanks for the referal.
Issue is posted here: https://www.domoticz.com/forum/viewtopic.php?p=298428#p298428