zigbeefordomoticz/Domoticz-Zigbee #1443: [Model Certification] - Zigbee profalux

cogiala, 2023-01-16:
1. Blind Profalux
1. https://www.kpark.fr/
1. 
PiZigbee - Volet-Profalux_BlindInverted-20918a00000af21c-01
{"Version":"3","ZDeviceName":"","Status":"inDB","SQN":"1b","Ep":{"01":{"0000":{"0010":"volet","0000":"1","0006":"20180723102b104","0011":"0"},"0003":{},"0004":{},"0005":{},"0006":{"0000":"01"},"0008":{"0000":"97"},"000a":{},"0015":{},"0100":{"0002":"09","0010":"0001","0011":"00"},"fc20":{},"fc21":{"0001":"3f"},"ClusterType":{"19":"LvlControl"}}},"Heartbeat":"1083","RIA":"4","LQI":60,"Battery":{},"Model":"Volet-Profalux","ForceAckCommands":[],"MacCapa":"8e","IEEE":"20918a00000af21c","Type":"BlindInverted","ProfileID":"0104","ZDeviceID":"0200","App Version":"","Attributes List":{"Request":{"01":{"0000":1673894287.0955749,"0003":1673894287.096033,"0004":1673894287.0964842,"0005":1673894287.0969343,"0006":1673894287.1050768,"0008":1673894287.1055396,"000a":1673894287.1059902,"0015":1673894287.106441,"0100":1673894287.1068926,"fc20":1673894287.1074317,"fc21":1673894287.1079428}}},"DeviceType":"FFD","HW Version":"","Last Cmds":[[409,"02","09b3","0100","01010100000000000200100011"],[410,"02","09b3","0100","0101000600000000010000"],[411,"02","09b3","0100","0101000800000000010000"],[412,"02","09b3","0100","010100000000000003000000060010"],[413,"02","09b3","0100","0101000000000000010011"],[414,"02","09b3","0100","0101fc2100011110010001"],[415,"07","09b3","00E0","09b301010000"],[418,"07","09b3","0100","0101000600000000010000"],[419,"07","09b3","0100","0101000800000000010000"],[420,"02","09b3","0062","010100"]],"LogicalType":"Router","Manufacturer":"1110","Manufacturer Name":"Profalux","NbEp":"1","PowerSource":"Main","ReadAttributes":{"Ep":{"01":{"0000":{"TimeStamp":1673890654,"iSQN":{"0004":393,"0005":393,"0000":412,"0006":412,"0010":412,"0011":413},"Attributes":{"0004":"86","0005":"86","0010":"00","0000":"00","0006":"00","0011":"00"},"ZigateRequest":{}},"0100":{"TimeStamp":1673890653,"iSQN":{"0000":408,"0001":408,"0002":408,"0010":409,"0011":409},"Attributes":{"0000":"86","0001":"86","0002":"00","0010":"00","0011":"00"},"ZigateRequest":{}},"0006":{"TimeStamp":1673890657,"iSQN":{"0000":418},"Attributes":{"0000":"00"},"ZigateRequest":{}},"0008":{"TimeStamp":1673896059,"iSQN":{"0000":419},"Attributes":{"0000":"00"},"ZigateRequest":{}},"fc21":{"TimeStamp":1673890654,"iSQN":{},"Attributes":{"0001":"00"},"ZigateRequest":{}}}}},"ReceiveOnIdle":"On","Stack Version":"","Stamp":{"time":1673896059.4857724,"Time":"2023-01-16 20:07:39","MsgType":"8102","LastSeen":1673896059},"ZCL Version":"1","Health":"Live","Capability":["Full-Function Device","Main Powered","Receiver during Idle","Standard security","NwkAddr should be allocated"],"RollingLQI":[84,60,63,57,60,51,51,51,57,60,60],"Epv2":{"01":{"ClusterIn":{"0000":{},"0003":{},"0004":{},"0005":{},"0006":{},"0008":{},"000a":{},"0015":{},"0100":{},"fc20":{},"fc21":{}},"ClusterOut":{"0003":{},"fc20":{}},"ProfileID":"0104","ZDeviceID":"0200"}},"WriteAttributes":{"Ep":{"01":{"0000":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"0003":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"0004":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"0005":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"0006":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"0008":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"000a":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"0015":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"0100":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"fc20":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"fc21":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}}}}},"ZDeviceVersion":"0000","ConfigSource":"8043","PairingInProgress":false,"Max Buffer Size":"52","Max Rx":"0052","Max Tx":"0052","macapa":"8e","bitfield":"4001","server_mask":"0000","descriptor_capability":"00","Location":"volet","PairingTime":1673890652.107803,"Bind":{"01":{"0008":{"Target":"0000","Stamp":1673890652,"Phase":"binded","Status":"00","i_sqn":405}}},"BindingTable":{"SQN":0,"Status":"Success","TimeStamp":1673890652.7864118,"BindingTableSize":3,"Devices":[{"20918a00000af21c":{"sourceEp":"01","Cluster":"fc20","targetIEEE":"20918a00000af21d","targetNickName":null,"targetEp":"01"}},{"20918a00000af21c":{"sourceEp":"01","Cluster":"fc20","targetIEEE":"20918a00000af21d","targetNickName":null,"targetEp":"01"}},{"20918a00000af21c":{"sourceEp":"01","Cluster":"0008","targetIEEE":"00158d0003af5f7a","targetNickName":"Zigbee Coordinator","targetEp":"01"}}]},"ZDPSQN":"0b","ConfigureReporting":{"Ep":{"01":{"0008":{"TimeStamp":1673890652.1159115,"iSQN":{"0000":407},"Attributes":{"0000":"00"},"ZigateRequest":{}}}}},"CertifiedDevice":false,"ZCLSQN":"00","SWBUILD_1":"20180723102b104","PhysicalEnv":"0","ReadConfigureReporting":{"Ep":{"01":{"0008":{"0000":{"TimeStamp":1673890654.4183915,"Status":"00","DataType":"20","MinInterval":"0005","MaxInterval":"012c"}}}},"TimeStamp":1673890654.4183662},"GroupMemberShip":{"01":{}},"Neighbours":[{"SQN":0,"Status":"","TimeStamp":1673895936,"Time":1673895936,"Devices":[{"0000":{"_relationshp":"Parent","_devicetype":"Coordinator","_depth":0,"_lnkqty":254,"_rxonwhenidl":"Rx-On","_IEEE":"00158d0003af5f7a","_permitjnt":"??"}}]}],"RoutingTable":[{"SQN":0,"Status":"Success","TimeStamp":1673895992.207805,"Time":1673895936,"RoutingTableTableSize":60,"Devices":[{"0000":{"Status":"Active (0)","MemoryConstrained":0,"ManyToOne":1,"RouteRecordRequired":0,"NextHopNwkId":"0000"}},{"7d70":{"Status":"Active (0)","MemoryConstrained":0,"ManyToOne":0,"RouteRecordRequired":0,"NextHopNwkId":"0000"}}]}],"NeighbourTableSize":1,"MapLQI":{"0000":51}}

cogiala, 2023-01-16:
Hi, my blind is statu not opimised.

Regards

pipiche38, 2023-01-16:
Thanks for submitting.
The profalux are a bit tricky as they do not provide model name.

I have made a fix available right now on the dev branch. If you want to test it, you need to move to the dev branch 

```
git checkout dev
git pull

```

the  restart the plugin.

cogiala, 2023-01-17:
Hi pipiche38,

Thank for prompt feedback, ok update plugin to dev and make new insert, but it's more less ok. the device is inserted in db, no opérate. The status of position inchanged but I change from command radio of blind. i joint log copy: sorry for my bad english.

Regards

Bonjour pipiche38,

merci du retour rapide, ok j'ai procéder au changement de branch et refait l'appairage. le volet entre dans le réseau, mais ne fonctionne pas, le status de position vue ne change jamais malgré les actions depuis la télécommande. ci-joint la copie du log d'appairage:

Cordialement

...
{"PluginInfos":{"HardwareID":2,"HomeFolder":"/opt/applications/Domoticz/plugins/Domoticz-Zigbee/","StartupFolder":"/opt/applications/Domoticz/","UserDataFolder":"/opt/applications/Domoticz/","WebRoot":"","Database":"/opt/applications/Domoticz/domoticz.db","Language":"fr","Version":"6.4","Author":"pipiche38","Name":"PiZigbee","Address":"0.0.0.0","Port":"9999","SerialPort":"/dev/ttyAMA0","Username":"","Password":"","Key":"Zigate","Mode1":"V1","Mode2":"PI","Mode3":"False","Mode4":"9440","Mode5":"http://127.0.0.1:8888","Mode6":"2","DomoticzVersion":"2022.2","DomoticzHash":"eea9db734","DomoticzBuildTime":"2022-11-05 13:05:35","PluginBranch":"beta6","PluginVersion":"6.4.151","TimeStamp":1673953379,"available":"6.4.150","available-firmMajor":"03","available-firmMinor":"0321","FirmwareVersion":"0321","FirmwareUpdate":false,"PluginUpdate":false,"LogLevel":7,"Zigpy":false,"CoordinatorModel":"ZiGate Classic PDM (legacy)","CoordinatorFirmwareVersion":"0321","DisplayFirmwareVersion":"Zigate - 0321","CoordinatorIEEE":"00158d0003af5f7a","NetworkSize":"Total: 7 | Routers: 1 | End Devices: 6","NetworkDevices":{"1037":{"LUMI":["lumi.weather","lumi.sens"],"unknown":["lumi.sensor_motion.aq2"]},"1110":{"Profalux":["Volet-Profalux","Telecommande-Profalux"]}}},"Analytics":1,"Device":{"Version":"3","ZDeviceName":"","Status":"inDB","SQN":"60","Ep":{"01":{"0000":{"0010":"volet","0000":"1","0006":"20180723102b104","0011":"0"},"0003":{},"0004":{},"0005":{},"0006":{"0000":"01"},"0008":{"0000":"97"},"000a":{},"0015":{},"0100":{"0002":"09","0010":"0001","0011":"00"},"fc20":{},"fc21":{"0001":"3f"},"ClusterType":{"19":"LvlControl"}}},"Heartbeat":"977","RIA":"4","LQI":66,"Battery":{},"Model":"Volet-Profalux","ForceAckCommands":[],"MacCapa":"8e","IEEE":"20918a00000af21c","Type":"BlindInverted","ProfileID":"0104","ZDeviceID":"0200","App Version":"","Attributes List":{"Request":{"01":{"0000":1673952129.0667176,"0003":1673952129.067204,"0004":1673952129.0676825,"0005":1673952129.0776222,"0006":1673952129.0781167,"0008":1673952129.0785959,"000a":1673952129.0790713,"0015":1673952129.079614,"0100":1673952129.0800893,"fc20":1673952129.0806477,"fc21":1673952129.0811894}}},"DeviceType":"FFD","HW Version":"","Last Cmds":[[43,"02","5d86","0100","01010100000000000200100011"],[44,"02","5d86","0100","0101000600000000010000"],[45,"02","5d86","0100","0101000800000000010000"],[46,"02","5d86","0100","010100000000000003000000060010"],[47,"02","5d86","0100","0101000000000000010011"],[48,"02","5d86","0100","0101fc2100011110010001"],[49,"07","5d86","00E0","5d8601010000"],[53,"07","5d86","0100","0101000600000000010000"],[54,"07","5d86","0100","0101000800000000010000"],[55,"02","5d86","0062","010100"]],"LogicalType":"Router","Manufacturer":"1110","Manufacturer Name":"Profalux","NbEp":"1","PowerSource":"Main","ReadAttributes":{"Ep":{"01":{"0000":{"TimeStamp":1673948496,"iSQN":{"0000":46,"0006":46,"0010":46,"0011":47},"Attributes":{"0004":"86","0005":"86","0010":"00","0000":"00","0006":"00","0011":"00"},"ZigateRequest":{}},"0100":{"TimeStamp":1673948495,"iSQN":{"0000":42,"0001":42,"0002":42,"0010":43,"0011":43},"Attributes":{"0000":"86","0001":"86","0002":"00","0010":"00","0011":"00"},"ZigateRequest":{}},"0006":{"TimeStamp":1673948499,"iSQN":{"0000":53},"Attributes":{"0000":"00"},"ZigateRequest":{}},"0008":{"TimeStamp":1673953299,"iSQN":{"0000":54},"Attributes":{"0000":"00"},"ZigateRequest":{}},"fc21":{"TimeStamp":1673948496,"iSQN":{},"Attributes":{"0001":"00"},"ZigateRequest":{}}}}},"ReceiveOnIdle":"On","Stack Version":"","Stamp":{"time":1673953299.2604578,"Time":"2023-01-17 12:01:39","MsgType":"8102","LastSeen":1673953299},"ZCL Version":"1","Health":"Live","Capability":["Full-Function Device","Main Powered","Receiver during Idle","Standard security","NwkAddr should be allocated"],"ZDPSQN":"02","RollingLQI":[57,60,57,63,60,60,54,75,93,51,66],"Epv2":{"01":{"ClusterIn":{"0000":{},"0003":{},"0004":{},"0005":{},"0006":{},"0008":{},"000a":{},"0015":{},"0100":{},"fc20":{},"fc21":{}},"ClusterOut":{"0003":{},"fc20":{}},"ProfileID":"0104","ZDeviceID":"0200"}},"WriteAttributes":{"Ep":{"01":{"0000":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"0003":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"0004":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"0005":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"0006":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"0008":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"000a":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"0015":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"0100":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"fc20":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"fc21":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}}}}},"ZDeviceVersion":"0000","ConfigSource":"8043","PairingInProgress [... cut ...] tNickName":"Zigbee Coordinator","targetEp":"01"}}]},"ConfigureReporting":{"Ep":{"01":{"0008":{"TimeStamp":1673948494.1374228,"iSQN":{"0000":41},"Attributes":{"0000":"00"},"ZigateRequest":{}}}}},"CertifiedDevice":true,"ZCLSQN":"00","SWBUILD_1":"20180723102b104","PhysicalEnv":"0","ReadConfigureReporting":{"Ep":{"01":{"0008":{"0000":{"TimeStamp":1673948496.7788875,"Status":"00","DataType":"20","MinInterval":"0005","MaxInterval":"012c"}}}},"TimeStamp":1673948496.778864},"GroupMemberShip":{"01":{}}}}

pipiche38, 2023-01-17:
Pourquoi avoir fait un re-appairage ? 
L'appairage des profzalux est toujours galère, donc malheuresement je n'ai pas d'autre solution que te dire de 

1) supprimer les widgets qui sont dans domoticz
2) vérifier que plus rien ne traine et quand coté plugin le volet n'existe plus
3) refaire la procédure d'appairage

Merci d'utiliser le forum https://easydomoticz.com/forum/viewforum.php?f=28&sid=0b2a5f5a90f9d63c631a5c30f018301a si le problème perssiste.

closed 2023-01-17 by pipiche38