zigbeefordomoticz/Domoticz-Zigbee #1448: [Model Certification] - AQARA TVOC Air Quality Monitor - AAQS-S01

rcrocus, 2023-01-17:
1. Please do provide device information such as Brand and Model name. 
**AQARA TVOC Air Quality Monitor - AAQS-S01**

2. Provide the URL where you have purchase the device
**_https://www.domadoo.fr/fr/peripheriques/5751-aqara-moniteur-de-qualite-d-air-zigbee-30-temperature-humidite-covt-aaqs-s01-6970504214644.html_**

3. Finaly copy-paste all information provided in the log
`{"PluginInfos":{"HardwareID":19,"HomeFolder":"/home/pi/domoticz/plugins/Domoticz-Zigate/","StartupFolder":"/home/pi/domoticz/","UserDataFolder":"/home/pi/domoticz/","WebRoot":"","Database":"/home/pi/domoticz/domoticz.db","Language":"fr","Version":"6.3","Author":"pipiche38","Name":"Zigate","Address":"0.0.0.0","Port":"9999","SerialPort":"/dev/serial/by-id/usb-Prolific_Technology_Inc._USB-Serial_Controller-if00-port0","Username":"","Password":"","Key":"Zigate","Mode1":"V1","Mode2":"USB","Mode3":"False","Mode4":"9440","Mode5":"http://domotic:rhum007@127.0.0.1:8080","Mode6":"2","DomoticzVersion":"2022.2 (build 14956)","DomoticzHash":"73fcd6d50","DomoticzBuildTime":"2023-01-11 15:17:05","PluginBranch":"beta6","PluginVersion":"6.4.152","TimeStamp":1673989951,"available":"6.4.150","available-firmMajor":"03","available-firmMinor":"0321","FirmwareVersion":"0321","FirmwareUpdate":false,"PluginUpdate":false,"LogLevel":7,"Zigpy":false,"CoordinatorModel":"ZiGate Classic PDM (legacy)","CoordinatorFirmwareVersion":"0321","DisplayFirmwareVersion":"Zigate - 0321","CoordinatorIEEE":"00158d0001ed453f","NetworkSize":"Total: 15 | Routers: 3 | End Devices: 12","NetworkDevices":{"1002":{"_TZ3000_riwp3k79":["TS0505A"],"_TZ3000_misw04hq":["TS0207-extender"]},"1037":{"LUMI":["lumi.weather","lumi.sensor_magnet.aq2","lumi.magnet.acn001","lumi.remote.b1acn01","lumi.airmonitor.acn01"]}}},"Analytics":0,"Device":{"Version":"3","ZDeviceName":"","Status":"inDB","SQN":"5d","Ep":{"01":{"0000":{"4000":"2020\u0000www.","0004":"LUMI","0005":"lumi.airmonitor.acn01","0001":"24","0000":"3","0002":"2","0003":"1","0006":"20210204","0007":"3","000a":"0062776c"},"0003":{},"0402":{"0000":1998},"0405":{"0000":54.5},"000c":{"0055":"7.0"},"0001":{"0020":32,"0033":1,"0035":1},"0019":{},"fcc0":{"00f7":"0121c40c0328130421a8130521050006249600000000082118010a2100000c20016429ce0765214c1566210e00672001","0129":"01","00fc":"00"},"ClusterType":{"323":"Temp+Hum","324":"Temp","325":"Humi","326":"Analog","327":"Voltage"}}},"Heartbeat":"4003","RIA":"4","LQI":149,"Battery":100,"Model":"lumi.airmonitor.acn01","ForceAckCommands":[],"MacCapa":"84","IEEE":"54ef441000441d7f","Type":"Temp/Humi/Analog/Voltage","ProfileID":"0104","ZDeviceID":"0302","App Version":"24","Attributes List":{"Request":{"01":{"0000":1673973462.09493,"0003":1673973462.098115,"0402":1673973462.0995784,"0405":1673973462.1008115,"000c":1673973462.1021252,"0001":1673973462.1034014,"0019":1673973462.104652,"fcc0":1673973462.1060362}}},"DeviceType":"RFD","HW Version":"1","Last Cmds":[[11,"02","6901","0100","0101000000000000010000"]],"LogicalType":"End Device","Manufacturer":"1037","Manufacturer Name":"LUMI","NbEp":"1","PowerSource":"Main","ReadAttributes":{"Ep":{"01":{"0000":{"TimeStamp":1673970928,"iSQN":{"4000":44,"0004":40,"0005":40,"0000":11,"0001":41,"0002":41,"0003":41,"0006":42,"0007":42,"000a":42,"000f":43,"0010":43,"0015":43,"f000":44,"ff01":44,"ff02":45},"Attributes":{"4000":"00","0004":"00","0005":"00","0000":"00","0001":"00","0002":"00","0003":"00","0006":"00","0007":"00","000a":"00","000f":"86","0010":"86","0015":"86","f000":"86","ff01":"86","ff02":"86"},"ZigateRequest":{}},"0402":{"TimeStamp":1673979643,"iSQN":{"0000":31},"Attributes":{"0000":"00"},"ZigateRequest":{}},"0405":{"TimeStamp":1673979633,"iSQN":{"0000":39},"Attributes":{"0000":"00"},"ZigateRequest":{}},"0001":{"TimeStamp":1673973066,"iSQN":{"0000":32,"0001":32,"0003":32,"0020":33,"0021":33,"0033":33,"0035":34},"Attributes":{"0020":"00","0000":"86","0001":"86","0003":"86","0021":"86","0033":"00","0035":"00"},"ZigateRequest":{}},"fcc0":{"TimeStamp":1673979644,"iSQN":{},"Attributes":{"00f7":"00","0129":"00","00fc":"00"},"ZigateRequest":{}},"0019":{"TimeStamp":1673969787,"iSQN":{"0001":35,"0002":35,"0003":35,"0004":36,"0005":36,"0006":36,"0007":37,"0008":37,"0009":37,"000a":38},"Attributes":{},"ZigateRequest":{}},"000c":{"TimeStamp":1673969793,"iSQN":{},"Attributes":{"0055":"00"},"ZigateRequest":{}}}}},"ReceiveOnIdle":"Off","Stack Version":"2","Stamp":{"time":1673979644.0411198,"Time":"2023-01-17 19:20:44","MsgType":"8102","LastSeen":1673979644,"LastPing":1673987880},"ZCL Version":"3","Health":"Not Reachable","BatteryUpdateTime":1673973066,"ConfigureReporting":{"Ep":{"01":{"0402":{"TimeStamp":1673969787.1589825,"iSQN":{"0000":28},"Attributes":{"0000":"00"},"ZigateRequest":{}},"0001":{"TimeStamp":1673969787.1626701,"iSQN":{"0000":29,"0020":29,"0021":29},"Attributes":{"0000":"86"},"ZigateRequest":{}},"0405":{"TimeStamp":1673969787.1692543,"iSQN":{"0000":30},"Attributes":{"0000":"00"},"ZigateRequest":{}}}}},"ConfigSource":"8043","LUMI":{"DeviceTemperature":48.7,"RSSI dB":-251,"LQI":"9600","0129":"01","00fc":"00"},"Bind":{"01":{"0000":{"Target":"0000","Stamp":1673969788,"Phase":"binded","Status":"00","i_sqn":21},"0003":{"Target":"0000","Stamp":1673969789,"Phase":"binded","Status":"00","i_sqn":22},"0402":{"Target":"0000","Stamp":1673969791,"Phase":"binded","Status":"00","i_sqn":23},"0405":{"Target":"0000","Stamp":1673969792,"Phase":"binded","Status":"00","i_sqn":24},"000c":{"Target":"0000","Stamp":1673969793,"Phase":"binded","Status":"00","i_sqn":25},"0001":{"Target":"0000","Stamp":1673969794,"Phase":"binded","Status":"00","i_sqn":26}}},"Epv2":{"01":{"ClusterIn":{"0000":{},"0003":{},"0402":{},"0405":{},"000c":{},"0001":{}},"ClusterOut":{"0019":{}},"ProfileID":"0104","ZDeviceID":"0302"}},"CertifiedDevice":false,"BindingTable":{"SQN":0,"Status":"Success","TimeStamp":1673969820.6979215,"BindingTableSize":6,"Devices":[{"54ef441000441d7f":{"sourceEp":"01","Cluster":"0000","targetIEEE":"00158d0001ed453f","targetNickName":"Zigate","targetEp":"01"}},{"54ef441000441d7f":{"sourceEp":"01","Cluster":"0003","targetIEEE":"00158d0001ed453f","targetNickName":"Zigate","targetEp":"01"}},{"54ef441000441d7f":{"sourceEp":"01","Cluster":"0402","targetIEEE":"00158d0001ed453f","targetNickName":"Zigate","targetEp":"01"}},{"54ef441000441d7f":{"sourceEp":"01","Cluster":"0405","targetIEEE":"00158d0001ed453f","targetNickName":"Zigate","targetEp":"01"}},{"54ef441000441d7f":{"sourceEp":"01","Cluster":"000c","targetIEEE":"00158d0001ed453f","targetNickName":"Zigate","targetEp":"01"}},{"54ef441000441d7f":{"sourceEp":"01","Cluster":"0001","targetIEEE":"00158d0001ed453f","targetNickName":"Zigate","targetEp":"01"}}]},"Log_UnknowDeviceFlag":1673987876.020519,"ReadConfigureReporting":{"Ep":{"01":{"0402":{"0000":{"TimeStamp":1673969818.6560254,"Status":"00","DataType":"29","MinInterval":"000a","MaxInterval":"012c"}},"0405":{"0000":{"TimeStamp":1673969819.6392274,"Status":"00","DataType":"21","MinInterval":"003c","MaxInterval":"0384"}}}},"TimeStamp":1673969819.6392066},"Capability":["Reduced-Function Device","Main Powered","Standard security","NwkAddr should be allocated"],"Param":{},"SWBUILD_3":"2020\u0000www.","SWBUILD_1":"20210204","WriteAttributes":{"Ep":{"01":{"0000":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"0003":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"0402":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"0405":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"000c":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"0001":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"0019":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"fcc0":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}}}}},"ConsistencyCheck":"ok","pingDeviceRetry":{"Retry":0,"TimeStamp":1673987890},"CheckParam":true}}` 

The device is well recognized in Domoticz for the temperature, the humidity and the battery but not for the TVOC :
[image]

pipiche38, 2023-01-18:
This is correct
Temp
Humidity
VOC
Temp+Humidity 

Voltage is not needed as you have the battery icon

please get the beta6-6.4.154 and just restart the plugin.

if nothing happen after the restart and a while , please enable debug on the cluster via the WbUI > Tools > Debug

Cluster

rcrocus, 2023-01-18:
The VOC widget (custom sensor) is not updated when others are updated.
Please find the log of the plugin after I've enabled cluster debugging.
[_PluginZigbee_19.log](https://github.com/zigbeefordomoticz/Domoticz-Zigbee/files/10451059/_PluginZigbee_19.log)

pipiche38, 2023-01-18:
leave it like no information were provided by the device in the Log. 
The report is usally done, when there is a change in the value or every hour or so.

pipiche38, 2023-01-19:
Any feedbacks ?

rcrocus, 2023-01-19:
I let an entire day and no data for Voc widget are displayefd in domoticz UI or in log. On the screen of the product I saw Voc modifications.
But I see in the same time a lot of update for the Temp and Humidity.

pipiche38, 2023-01-19:
Can you then remove everything, and then redo the pairing

rcrocus, 2023-01-20:
I will test ASAP (in few days) and I tell you back.

rcrocus, 2023-01-22:
Hello,
I removed all widget of the device and I'd make a new pairing.
4 widgets were created : 
- Zigate - lumi.airmonitor.acn01_Temp+Hum-54ef441000441d7f-01
- Zigate - lumi.airmonitor.acn01_Temp-54ef441000441d7f-01
- Zigate - lumi.airmonitor.acn01_Humi-54ef441000441d7f-01
- Zigate - lumi.airmonitor.acn01_Voc-54ef441000441d7f-01 

But only the Temp, Humi and Temp+Hum widget are updated :  
_2023-01-22 16:00:56.969 Zigate: UpdateDevice - (Zigate - lumi.airmonitor.acn01_Temp+Hum-54ef441000441d7f-01) 0:19.9;41.9;1
2023-01-22 16:01:46.762 Zigate: UpdateDevice - (Zigate - lumi.airmonitor.acn01_Temp+Hum-54ef441000441d7f-01) 0:20.0;41.9;1
2023-01-22 16:01:46.785 Zigate: UpdateDevice - (Zigate - lumi.airmonitor.acn01_Temp-54ef441000441d7f-01) 20.0:20.0
2023-01-22 16:01:56.725 Zigate: UpdateDevice - (Zigate - lumi.airmonitor.acn01_Temp+Hum-54ef441000441d7f-01) 0:20.0;41.8;1
2023-01-22 16:02:56.491 Zigate: UpdateDevice - (Zigate - lumi.airmonitor.acn01_Temp+Hum-54ef441000441d7f-01) 0:20.0;42.3;1
2023-01-22 16:02:56.502 Zigate: UpdateDevice - (Zigate - lumi.airmonitor.acn01_Humi-54ef441000441d7f-01) 42.3:1
2023-01-22 16:03:56.260 Zigate: UpdateDevice - (Zigate - lumi.airmonitor.acn01_Temp+Hum-54ef441000441d7f-01) 0:20.0;41.8;1
2023-01-22 16:03:56.280 Zigate: UpdateDevice - (Zigate - lumi.airmonitor.acn01_Humi-54ef441000441d7f-01) 41.8:1
2023-01-22 16:04:55.983 Zigate: UpdateDevice - (Zigate - lumi.airmonitor.acn01_Temp+Hum-54ef441000441d7f-01) 0:20.0;41.9;1
2023-01-22 16:05:55.753 Zigate: UpdateDevice - (Zigate - lumi.airmonitor.acn01_Temp+Hum-54ef441000441d7f-01) 0:20.0;42.5;1
2023-01-22 16:05:55.765 Zigate: UpdateDevice - (Zigate - lumi.airmonitor.acn01_Humi-54ef441000441d7f-01) 42.5:1
2023-01-22 16:06:35.649 Zigate: UpdateDevice - (Zigate - lumi.airmonitor.acn01_Temp+Hum-54ef441000441d7f-01) 0:19.9;42.5;1
2023-01-22 16:06:35.686 Zigate: UpdateDevice - (Zigate - lumi.airmonitor.acn01_Temp-54ef441000441d7f-01) 19.9:19.9
2023-01-22 16:06:45.609 Zigate: UpdateDevice - (Zigate - lumi.airmonitor.acn01_Temp+Hum-54ef441000441d7f-01) 0:20.0;42.5;1
2023-01-22 16:06:45.630 Zigate: UpdateDevice - (Zigate - lumi.airmonitor.acn01_Temp-54ef441000441d7f-01) 20.0:20.0
2023-01-22 16:06:55.616 Zigate: UpdateDevice - (Zigate - lumi.airmonitor.acn01_Temp+Hum-54ef441000441d7f-01) 0:20.0;42.8;1_

pipiche38, 2023-01-22:
can yo provide the pairing log and the export.json ( Liste des Equipements RAW).
Thanks

rcrocus, 2023-01-22:
Please find the log of the plugin (the pairing was made at 15:28) and the export.json.
[_PluginZigbee_19.zip](https://github.com/zigbeefordomoticz/Domoticz-Zigbee/files/10474845/_PluginZigbee_19.zip)
[export.zip](https://github.com/zigbeefordomoticz/Domoticz-Zigbee/files/10474846/export.zip)

pipiche38, 2023-01-22:
Strange as the Voc is at 126 now

pipiche38, 2023-01-22:
don't you see any errors in Domoticz logs ?

pipiche38, 2023-01-22:
Here is what I got in the log

`2023-01-22 15:37:34,308 DEBUG   : [ZiGateForwarder_19 0] ReadCluster - ClusterID=000C - MsgSrcEp: 01 MsgAttrID: 0055 MsgAttType: 39 MsgClusterData: 44098000
2023-01-22 15:38:04,159 DEBUG   : [ZiGateForwarder_19 0] ReadCluster - ClusterID=000C - MsgSrcEp: 01 MsgAttrID: 0055 MsgAttType: 39 MsgClusterData: 44010000
2023-01-22 15:38:04,179 DEBUG   : [ZiGateForwarder_19 0] ReadCluster - ClusterID=000C - MsgSrcEp: 01 MsgAttrID: 0055 MsgAttType: 39 MsgClusterData: 44010000
2023-01-22 15:38:34,046 DEBUG   : [ZiGateForwarder_19 0] ReadCluster - ClusterID=000C - MsgSrcEp: 01 MsgAttrID: 0055 MsgAttType: 39 MsgClusterData: 43f20000
2023-01-22 15:38:34,070 DEBUG   : [ZiGateForwarder_19 0] ReadCluster - ClusterID=000C - MsgSrcEp: 01 MsgAttrID: 0055 MsgAttType: 39 MsgClusterData: 43f20000
2023-01-22 15:39:03,930 DEBUG   : [ZiGateForwarder_19 0] ReadCluster - ClusterID=000C - MsgSrcEp: 01 MsgAttrID: 0055 MsgAttType: 39 MsgClusterData: 43e30000
2023-01-22 15:39:33,819 DEBUG   : [ZiGateForwarder_19 0] ReadCluster - ClusterID=000C - MsgSrcEp: 01 MsgAttrID: 0055 MsgAttType: 39 MsgClusterData: 43e20000
2023-01-22 15:39:33,840 DEBUG   : [ZiGateForwarder_19 0] ReadCluster - ClusterID=000C - MsgSrcEp: 01 MsgAttrID: 0055 MsgAttType: 39 MsgClusterData: 43e20000
2023-01-22 15:40:03,708 DEBUG   : [ZiGateForwarder_19 0] ReadCluster - ClusterID=000C - MsgSrcEp: 01 MsgAttrID: 0055 MsgAttType: 39 MsgClusterData: 43d30000
2023-01-22 15:40:33,591 DEBUG   : [ZiGateForwarder_19 0] ReadCluster - ClusterID=000C - MsgSrcEp: 01 MsgAttrID: 0055 MsgAttType: 39 MsgClusterData: 43d10000
2023-01-22 15:40:33,611 DEBUG   : [ZiGateForwarder_19 0] ReadCluster - ClusterID=000C - MsgSrcEp: 01 MsgAttrID: 0055 MsgAttType: 39 MsgClusterData: 43d10000
2023-01-22 15:41:03,479 DEBUG   : [ZiGateForwarder_19 0] ReadCluster - ClusterID=000C - MsgSrcEp: 01 MsgAttrID: 0055 MsgAttType: 39 MsgClusterData: 43c18000
2023-01-22 15:41:03,498 DEBUG   : [ZiGateForwarder_19 0] ReadCluster - ClusterID=000C - MsgSrcEp: 01 MsgAttrID: 0055 MsgAttType: 39 MsgClusterData: 43c18000
2023-01-22 15:41:33,401 DEBUG   : [ZiGateForwarder_19 0] ReadCluster - ClusterID=000C - MsgSrcEp: 01 MsgAttrID: 0055 MsgAttType: 39 MsgClusterData: 43b18000
2023-01-22 15:41:33,485 DEBUG   : [ZiGateForwarder_19 0] ReadCluster - ClusterID=000C - MsgSrcEp: 01 MsgAttrID: 0055 MsgAttType: 39 MsgClusterData: 43b18000
2023-01-22 15:42:03,316 DEBUG   : [ZiGateForwarder_19 0] ReadCluster - ClusterID=000C - MsgSrcEp: 01 MsgAttrID: 0055 MsgAttType: 39 MsgClusterData: 43a40000
2023-01-22 15:42:33,140 DEBUG   : [ZiGateForwarder_19 0] ReadCluster - ClusterID=000C - MsgSrcEp: 01 MsgAttrID: 0055 MsgAttType: 39 MsgClusterData: 43a28000
2023-01-22 15:43:03,065 DEBUG   : [ZiGateForwarder_19 0] ReadCluster - ClusterID=000C - MsgSrcEp: 01 MsgAttrID: 0055 MsgAttType: 39 MsgClusterData: 43a08000
2023-01-22 15:43:03,153 DEBUG   : [ZiGateForwarder_19 0] ReadCluster - ClusterID=000C - MsgSrcEp: 01 MsgAttrID: 0055 MsgAttType: 39 MsgClusterData: 43a08000
2023-01-22 15:43:32,935 DEBUG   : [ZiGateForwarder_19 0] ReadCluster - ClusterID=000C - MsgSrcEp: 01 MsgAttrID: 0055 MsgAttType: 39 MsgClusterData: 43908000
2023-01-22 15:43:33,020 DEBUG   : [ZiGateForwarder_19 0] ReadCluster - ClusterID=000C - MsgSrcEp: 01 MsgAttrID: 0055 MsgAttType: 39 MsgClusterData: 43908000
2023-01-22 15:44:02,867 DEBUG   : [ZiGateForwarder_19 0] ReadCluster - ClusterID=000C - MsgSrcEp: 01 MsgAttrID: 0055 MsgAttType: 39 MsgClusterData: 43808000
2023-01-22 15:44:02,890 DEBUG   : [ZiGateForwarder_19 0] ReadCluster - ClusterID=000C - MsgSrcEp: 01 MsgAttrID: 0055 MsgAttType: 39 MsgClusterData: 43808000
2023-01-22 15:44:32,738 DEBUG   : [ZiGateForwarder_19 0] ReadCluster - ClusterID=000C - MsgSrcEp: 01 MsgAttrID: 0055 MsgAttType: 39 MsgClusterData: 43620000
2023-01-22 15:44:32,764 DEBUG   : [ZiGateForwarder_19 0] ReadCluster - ClusterID=000C - MsgSrcEp: 01 MsgAttrID: 0055 MsgAttType: 39 MsgClusterData: 43620000
2023-01-22 15:45:02,673 DEBUG   : [ZiGateForwarder_19 0] ReadCluster - ClusterID=000C - MsgSrcEp: 01 MsgAttrID: 0055 MsgAttType: 39 MsgClusterData: 43460000
2023-01-22 15:45:32,511 DEBUG   : [ZiGateForwarder_19 0] ReadCluster - ClusterID=000C - MsgSrcEp: 01 MsgAttrID: 0055 MsgAttType: 39 MsgClusterData: 43430000
2023-01-22 15:45:32,571 DEBUG   : [ZiGateForwarder_19 0] ReadCluster - ClusterID=000C - MsgSrcEp: 01 MsgAttrID: 0055 MsgAttType: 39 MsgClusterData: 43430000
2023-01-22 15:46:02,365 DEBUG   : [ZiGateForwarder_19 0] ReadCluster - ClusterID=000C - MsgSrcEp: 01 MsgAttrID: 0055 MsgAttType: 39 MsgClusterData: 43250000
2023-01-22 15:46:32,321 DEBUG   : [ZiGateForwarder_19 0] ReadCluster - ClusterID=000C - MsgSrcEp: 01 MsgAttrID: 0055 MsgAttType: 39 MsgClusterData: 43260000
2023-01-22 15:47:02,142 DEBUG   : [ZiGateForwarder_19 0] ReadCluster - ClusterID=000C - MsgSrcEp: 01 MsgAttrID: 0055 MsgAttType: 39 MsgClusterData: 43250000
2023-01-22 15:47:32,038 DEBUG   : [ZiGateForwarder_19 0] ReadCluster - ClusterID=000C - MsgSrcEp: 01 MsgAttrID: 0055 MsgAttType: 39 MsgClusterData: 43260000
2023-01-22 15:49:31,550 DEBUG   : [ZiGateForwarder_19 0] ReadCluster - ClusterID=000C - MsgSrcEp: 01 MsgAttrID: 0055 MsgAttType: 39 MsgClusterData: 43250000
2023-01-22 15:50:31,337 DEBUG   : [ZiGateForwarder_19 0] ReadCluster - ClusterID=000C - MsgSrcEp: 01 MsgAttrID: 0055 MsgAttType: 39 MsgClusterData: 43270000
[... cut ...]
2023-01-22 16:56:56,069 DEBUG   : [ZiGateForwarder_19 0] ReadCluster - ClusterID=000C - MsgSrcEp: 01 MsgAttrID: 0055 MsgAttType: 39 MsgClusterData: 42fe0000
2023-01-22 16:57:26,011 DEBUG   : [ZiGateForwarder_19 0] ReadCluster - ClusterID=000C - MsgSrcEp: 01 MsgAttrID: 0055 MsgAttType: 39 MsgClusterData: 42fa0000
2023-01-22 16:57:55,812 DEBUG   : [ZiGateForwarder_19 0] ReadCluster - ClusterID=000C - MsgSrcEp: 01 MsgAttrID: 0055 MsgAttType: 39 MsgClusterData: 42fc0000
2023-01-22 16:58:25,814 DEBUG   : [ZiGateForwarder_19 0] ReadCluster - ClusterID=000C - MsgSrcEp: 01 MsgAttrID: 0055 MsgAttType: 39 MsgClusterData: 42fa0000`

pipiche38, 2023-01-22:
I'll pus a fix and will let you know

pipiche38, 2023-01-22:
please switch to a dedicated branch

```
git pull
git checkout device/lumi/voc
git pull
```

and simply restart the plugin

zigbeefordomoticz, 2023-01-22:
#1469

rcrocus, 2023-01-23:
Sorry, but I can't switch to the branch "device/lumi/voc".
Please find here, errors when I'll try "git pull" and "git checkout" :

`pi@raspberrypi:~/domoticz/plugins/Domoticz-Zigate $ sudo git status
Sur la branche dev
Votre branche est à jour avec 'origin/dev'.

Modifications qui ne seront pas validées :
  (utilisez "git add/rm <fichier>..." pour mettre à jour ce qui sera validé)
  (utilisez "git checkout -- <fichier>..." pour annuler les modifications dans la copie de travail)
  (valider ou annuler le contenu non suivi ou modifié dans les sous-modules)

	modifié :         Classes/ConfigureReporting.py
	modifié :         Classes/ZigateTransport/sqnMgmt.py
	supprimé :        Logs/.PRECIOUS
	modifié :         Modules/deviceAnnoucement.py
	modifié :         Modules/heartbeat.py
	modifié :         Modules/legrand_netatmo.py
	modifié :         Modules/schneider_wiser.py
	modifié :         Tools/Fedora/pi-zigate-fedora.py
	modifié :         Tools/build-Jennic.sh
	modifié :         Tools/flash_PiZiGate.py
	modifié :         Tools/flash_ZiGate-DIN.py
	modifié :         Tools/ikea-ota-download.py
	modifié :         Tools/pi-zigate.sh
	modifié :         Tools/plugin-auto-upgrade.sh
	modifié :         Tools/zigate-flasher.py
	supprimé :        bellows
	modifié :         external/zigpy-znp (contenu modifié)
	modifié :         plugin.py
	modifié :         www/z4d/robots.txt
	supprimé :        zigpy
	supprimé :        zigpy_deconz
	supprimé :        zigpy_zigate
	supprimé :        zigpy_znp

Fichiers non suivis:
  (utilisez "git add <fichier>..." pour inclure dans ce qui sera validé)

	external/dnspython/
	external/pyserial/

aucune modification n'a été ajoutée à la validation (utilisez "git add" ou "git commit -a")
pi@raspberrypi:~/domoticz/plugins/Domoticz-Zigate $ sudo git pull
remote: Enumerating objects: 208, done.
remote: Counting objects: 100% (208/208), done.
remote: Compressing objects: 100% (115/115), done.
remote: Total 208 (delta 113), reused 155 (delta 76), pack-reused 0
Réception d'objets: 100% (208/208), 1.12 MiB | 3.45 MiB/s, fait.
Résolution des deltas: 100% (113/113), complété avec 4 objets locaux.
Depuis https://github.com/zigbeefordomoticz/Domoticz-Zigbee
 * [nouvelle branche]  develop                  -> origin/develop
 * [nouvelle branche]  device/lumi/voc          -> origin/device/lumi/voc
 * [nouvelle branche]  magicplugin/zcl-handling -> origin/magicplugin/zcl-handling
   2c104448..87848042  stable6                  -> origin/stable6
 * [nouvelle étiquette] Stable5-5.1.006          -> Stable5-5.1.006
 * [nouvelle étiquette] beta6-6.4.203            -> beta6-6.4.203
 * [nouvelle étiquette] stable6-6.3.009          -> stable6-6.3.009
Fetching submodule external/bellows
Depuis https://github.com/zigbeefordomoticz/bellows
   10f94e1..2aa213e  dev        -> origin/dev
   f455ee4..5042f2c  master     -> origin/master
Fetching submodule external/zigpy
Depuis https://github.com/zigbeefordomoticz/zigpy
   ce48779..df2c114  dev        -> origin/dev
   9f99d07..49ca485  master     -> origin/master
Fetching submodule external/zigpy-deconz
Fetching submodule external/zigpy-zigate
Fetching submodule external/zigpy-znp
Votre information de configuration indique de fusionner avec la référence 'refs/heads/dev'
du serveur distant, mais cette référence n'a pas été récupérée.
pi@raspberrypi:~/domoticz/plugins/Domoticz-Zigate $ sudo git checkout device/lumi/voc
error: Vos modifications locales aux fichiers suivants seraient écrasées par l'extraction :
	plugin.py
Veuillez valider ou remiser vos modifications avant de basculer de branche.
Abandon
pi@raspberrypi:~/domoticz/plugins/Domoticz-Zigate $
`

pipiche38, 2023-01-23:
you can run the big-clean too

`sudo Tools/big-clean-git.sh`

then you do 
```
git checkout stable6
git pull
git checkout device/lumi/voc
```

rcrocus, 2023-01-23:
Thanks, it's better but now I have issues when the plugin launches :
2023-01-23 11:14:59.592 Status: Zigate: Started.
2023-01-23 11:14:59.591 Status: Zigate: Entering work loop.
2023-01-23 11:15:01.960 Zigate: Zigbee for Domoticz plugin started!
2023-01-23 11:15:01.936 Status: Zigate: Initialized version 6.4, author 'pipiche38'
2023-01-23 11:15:01.961 Error: Zigate: Call to function 'onStart' failed, exception details:
2023-01-23 11:15:01.964 Error: Zigate: Traceback (most recent call last):
2023-01-23 11:15:01.964 Error: Zigate: File "/home/pi/domoticz/plugins/Domoticz-Zigate/plugin.py", line 1533, in onStart
2023-01-23 11:15:01.964 Error: Zigate: _plugin.onStart()
2023-01-23 11:15:01.964 Error: Zigate: File "/home/pi/domoticz/plugins/Domoticz-Zigate/plugin.py", line 274, in onStart
2023-01-23 11:15:01.964 Error: Zigate: assert sys.version_info >= (3, 8) # nosec
2023-01-23 11:15:01.964 Error: Zigate: AssertionError

pipiche38, 2023-01-23:
Ok. Indeed we have put a restriction and will support only python3.8 and above.
I'll make a temporary change, to remove this restriction.

Please simply do a 
git pull
git checkout device/lumi/voc
git pull

and restart

rcrocus, 2023-01-23:
After a git pull, I see no update... (in github, I see 0 update for the branch dev or device/lumi/voc. Is it normal ?

pipiche38, 2023-01-23:
please try again

rcrocus, 2023-01-23:
Yes, it's better.
Now, I see ine domoticz log the VOC device is uptaded...but with errors : 
2023-01-23 12:20:20.735 Zigate: UpdateDevice - (Zigate - lumi.airmonitor.acn01_Temp+Hum-54ef441000441d7f-01) 0:94.0
2023-01-23 12:20:20.760 Zigate: UpdateDevice - (Zigate - lumi.airmonitor.acn01_Temp-54ef441000441d7f-01) 0:94.0
2023-01-23 12:20:20.790 Zigate: UpdateDevice - (Zigate - lumi.airmonitor.acn01_Humi-54ef441000441d7f-01) 0:94.0
2023-01-23 12:20:20.833 Zigate: UpdateDevice - (Zigate - lumi.airmonitor.acn01_Voc-54ef441000441d7f-01) 0:94.0
2023-01-23 12:20:20.734 Status: Warning: Expecting svalue with at least 2 elements separated by semicolon, 1 elements received ("94.0"), notification not sent (Hardware: 19 - Zigate, ID: 54ef441000441d7f, Unit: 45, Type: 52 - Temp + Humidity, SubType: 1 - THGN122/123/132, THGR122/228/238/268)
2023-01-23 12:20:20.840 Error: Zigate: forwarder_thread - Error while receiving a ZiGate command
2023-01-23 12:20:20.850 Error: Zigate: 'forwarder_thread' failed 'local variable 'svalue' referenced before assignment'
2023-01-23 12:20:20.876 Error: Zigate: ----> Line 35 in '/home/pi/domoticz/plugins/Domoticz-Zigate/Classes/ZigateTransport/forwarderThread.py', function forwarder_thread
2023-01-23 12:20:20.886 Error: Zigate: ----> Line 50 in '/home/pi/domoticz/plugins/Domoticz-Zigate/Classes/ZigateTransport/instrumentation.py', function f_out
2023-01-23 12:20:20.896 Error: Zigate: ----> Line 59 in '/home/pi/domoticz/plugins/Domoticz-Zigate/Classes/ZigateTransport/forwarderThread.py', function forward_message
2023-01-23 12:20:20.920 Error: Zigate: ----> Line 819 in '/home/pi/domoticz/plugins/Domoticz-Zigate/plugin.py', function processFrame
2023-01-23 12:20:20.932 Error: Zigate: ----> Line 185 in '/home/pi/domoticz/plugins/Domoticz-Zigate/Modules/input.py', function ZigateRead
2023-01-23 12:20:20.943 Error: Zigate: ----> Line 3032 in '/home/pi/domoticz/plugins/Domoticz-Zigate/Modules/input.py', function Decode8102
2023-01-23 12:20:20.956 Error: Zigate: ----> Line 3110 in '/home/pi/domoticz/plugins/Domoticz-Zigate/Modules/input.py', function scan_attribute_reponse
2023-01-23 12:20:20.978 Error: Zigate: ----> Line 3167 in '/home/pi/domoticz/plugins/Domoticz-Zigate/Modules/input.py', function read_report_attributes
2023-01-23 12:20:20.989 Error: Zigate: ----> Line 233 in '/home/pi/domoticz/plugins/Domoticz-Zigate/Modules/readClusters.py', function ReadCluster
2023-01-23 12:20:21.012 Error: Zigate: ----> Line 1437 in '/home/pi/domoticz/plugins/Domoticz-Zigate/Modules/readClusters.py', function Cluster000c
2023-01-23 12:20:21.023 Error: Zigate: ----> Line 657 in '/home/pi/domoticz/plugins/Domoticz-Zigate/Modules/domoMaj.py', function MajDomoDevice
2023-01-23 12:20:21.034 Error: Zigate: handle_error_in_thread

pipiche38, 2023-01-23:
try again ;-)

git pull
restart

rcrocus, 2023-01-23:
I've always the same error :
 2023-01-23 14:20:27.298 Zigate: UpdateDevice - (Zigate - lumi.airmonitor.acn01_Voc-54ef441000441d7f-01) 0:53.0
2023-01-23 14:20:27.158 Status: Warning: Expecting svalue with at least 2 elements separated by semicolon, 1 elements received ("53.0"), notification not sent (Hardware: 19 - Zigate, ID: 54ef441000441d7f, Unit: 45, Type: 52 - Temp + Humidity, SubType: 1 - THGN122/123/132, THGR122/228/238/268)
2023-01-23 14:20:27.301 Error: Zigate: forwarder_thread - Error while receiving a ZiGate command
2023-01-23 14:20:27.323 Error: Zigate: 'forwarder_thread' failed 'local variable 'svalue' referenced before assignment'
2023-01-23 14:20:27.351 Error: Zigate: ----> Line 35 in '/home/pi/domoticz/plugins/Domoticz-Zigate/Classes/ZigateTransport/forwarderThread.py', function forwarder_thread
2023-01-23 14:20:27.385 Error: Zigate: ----> Line 50 in '/home/pi/domoticz/plugins/Domoticz-Zigate/Classes/ZigateTransport/instrumentation.py', function f_out
2023-01-23 14:20:27.419 Error: Zigate: ----> Line 59 in '/home/pi/domoticz/plugins/Domoticz-Zigate/Classes/ZigateTransport/forwarderThread.py', function forward_message
2023-01-23 14:20:27.441 Error: Zigate: ----> Line 819 in '/home/pi/domoticz/plugins/Domoticz-Zigate/plugin.py', function processFrame
2023-01-23 14:20:27.464 Error: Zigate: ----> Line 185 in '/home/pi/domoticz/plugins/Domoticz-Zigate/Modules/input.py', function ZigateRead
2023-01-23 14:20:27.488 Error: Zigate: ----> Line 3032 in '/home/pi/domoticz/plugins/Domoticz-Zigate/Modules/input.py', function Decode8102
2023-01-23 14:20:27.512 Error: Zigate: ----> Line 3110 in '/home/pi/domoticz/plugins/Domoticz-Zigate/Modules/input.py', function scan_attribute_reponse
2023-01-23 14:20:27.537 Error: Zigate: ----> Line 3167 in '/home/pi/domoticz/plugins/Domoticz-Zigate/Modules/input.py', function read_report_attributes
2023-01-23 14:20:27.560 Error: Zigate: ----> Line 233 in '/home/pi/domoticz/plugins/Domoticz-Zigate/Modules/readClusters.py', function ReadCluster
2023-01-23 14:20:27.583 Error: Zigate: ----> Line 1437 in '/home/pi/domoticz/plugins/Domoticz-Zigate/Modules/readClusters.py', function Cluster000c
2023-01-23 14:20:27.606 Error: Zigate: ----> Line 658 in '/home/pi/domoticz/plugins/Domoticz-Zigate/Modules/domoMaj.py', function MajDomoDevice
2023-01-23 14:20:27.634 Error: Zigate: handle_error_in_thread 

But I think I'v found it (l656 to 658) :
` if ClusterType == "Analog" and WidgetType == "Voc" and Attribute_ == "":
            **sValue** = str( value )
            UpdateDevice_v2(self, Devices, DeviceUnit, 0, **svalue**, BatteryLevel, SignalLevel)`

For me 'svalue' muste be replaced by 'sValue'. What do you think ?

pipiche38, 2023-01-23:
yes. could you update it and try ?

rcrocus, 2023-01-23:
I haven't error anymore.
Tha value of VOC in well displayed in the VOC widget.
But now, there is a problem with Temp & Humidity widget : they display the value of the VOC
2023-01-23 15:34:10.801 Zigate: UpdateDevice - (Zigate - lumi.airmonitor.acn01_Temp-54ef441000441d7f-01) 19.6:19.6
2023-01-23 15:34:10.839 Zigate: UpdateDevice - (Zigate - lumi.airmonitor.acn01_Temp+Hum-54ef441000441d7f-01) 0:49.0
2023-01-23 15:34:10.856 Zigate: UpdateDevice - (Zigate - lumi.airmonitor.acn01_Temp-54ef441000441d7f-01) 0:49.0
2023-01-23 15:34:10.874 Zigate: UpdateDevice - (Zigate - lumi.airmonitor.acn01_Humi-54ef441000441d7f-01) 0:49.0
2023-01-23 15:34:10.922 Zigate: UpdateDevice - (Zigate - lumi.airmonitor.acn01_Voc-54ef441000441d7f-01) 0:49.0
2023-01-23 15:34:10.969 Zigate: UpdateDevice - (Zigate - lumi.airmonitor.acn01_Temp+Hum-54ef441000441d7f-01) 0:49.0;46.9;1
2023-01-23 15:34:10.993 Zigate: UpdateDevice - (Zigate - lumi.airmonitor.acn01_Humi-54ef441000441d7f-01) 46.9:1
**2023-01-23 15:34:10.838 Status: Warning: Expecting svalue with at least 2 elements separated by semicolon, 1 elements received ("49.0"), notification not sent (Hardware: 19 - Zigate, ID: 54ef441000441d7f, Unit: 45, Type: 52 - Temp + Humidity, SubType: 1 - THGN122/123/132, THGR122/228/238/268)** 

[image]

On the devis, the temp is : 19.5°, the humidity is 46% and VOC is 50.

pipiche38, 2023-01-23:
can you please enable debug Widget

get some cases where the Temp, Humi and Voc are updated and upload the log file here.
Thanks

pipiche38, 2023-01-23:
can you retreived the latest version and test again ?

rcrocus, 2023-01-23:
Always the same error :
`2023-01-23 17:36:49.097 Zigate: UpdateDevice - (Zigate - lumi.airmonitor.acn01_Voc-54ef441000441d7f-01) 0:474.0
2023-01-23 17:37:18.861 Zigate: UpdateDevice - (Zigate - lumi.airmonitor.acn01_Temp+Hum-54ef441000441d7f-01) 0:559.0
2023-01-23 17:37:18.859 Status: Warning: Expecting svalue with at least 2 elements separated by semicolon, 1 elements received ("559.0"), notification not sent (Hardware: 19 - Zigate, ID: 54ef441000441d7f, Unit: 45, Type: 52 - Temp + Humidity, SubType: 1 - THGN122/123/132, THGR122/228/238/268)
2023-01-23 17:37:48.770 Zigate: UpdateDevice - (Zigate - lumi.airmonitor.acn01_Temp+Hum-54ef441000441d7f-01) 0:627.0
2023-01-23 17:37:48.800 Zigate: UpdateDevice - (Zigate - lumi.airmonitor.acn01_Temp-54ef441000441d7f-01) 0:627.0
2023-01-23 17:37:48.832 Zigate: UpdateDevice - (Zigate - lumi.airmonitor.acn01_Humi-54ef441000441d7f-01) 0:627.0
2023-01-23 17:37:48.873 Zigate: UpdateDevice - (Zigate - lumi.airmonitor.acn01_Voc-54ef441000441d7f-01) 0:627.0
`
[_PluginZigbee_19.zip](https://github.com/zigbeefordomoticz/Domoticz-Zigbee/files/10481751/_PluginZigbee_19.zip)

Sometimes the display is good but some seconds after Temp & Humi are replaced by the value of VOC

pipiche38, 2023-01-23:
Should be ok now.

git pull
restart

rcrocus, 2023-01-23:
Sorry but I have no update with a 'git pull' on the branch "device/lumi/voc"

pipiche38, 2023-01-23:
try again

rcrocus, 2023-01-24:
Hello,
It seems very good. After a while, the good information are displayed in each device.
Thanks a lot !

Another thing (maybe because I am on the branch "device/lumi/voc") : from the plugin WebUI, in the graph "battery" (on the home page), this device doesn't appear but it appears well in the graph "type of objects"...

pipiche38, 2023-01-24:
Could you clarify how does this device get energy from ?

- is that a battery base device ?
- is the device connected to main power ?
- is that both ?

rcrocus, 2023-01-24:
Hello,
It's a battery base device.

pipiche38, 2023-01-24:
Ok, I'll see what I can do. The issue is the device is announcing itself as a Main powered devices :-(

pipiche38, 2023-01-24:
I did the change on my repo. Can you do the following

if you want to test, this has been integrated for the next release.

```
git remote add pipiche https://github.com/pipiche38/Domoticz-Zigbee.git
git pull pipiche device/lumi/voc
git checkout pipiche/device/lumi/voc
```

the  restart the plugin

to fall back

```
git checkout stable6
git status
git remote remove pipiche
```

Last, when completed, I'll be merging this into the develop branch. Be aware that it will required python3.8 or above

rcrocus, 2023-01-25:
With your modification, I see this device in the graph "battery" !

For python3.8, I've updated my raspeberry but for domoticz I need to update libpython and it don't work for now because I use a Raspebrry PI3+ on buster (and on this version libpython 3.8 isn't available). I'm seeing to upgrade to bullseye.

Thanks a lot for your work and your time.

zigbeefordomoticz, 2023-01-25:
Ok. 
Du coup je te recommande soit de rester sur cette branche en attendant l'upgrade python. Et surtout ne plus faire de git pull (car je vais re-activer le check de version python).

Je te propose de clore l'issue.

Les modifs ayant entrainées quelques changements dans le code je ne pourrais pas la pousser immédiatement en stable6, mais elle va etre poussée sur la branche develop

rcrocus, 2023-01-25:
Thanks again @pipiche38

closed 2023-01-25 by rcrocus