zigbeefordomoticz/Domoticz-Zigbee #1471: [Model Certification] -

pouilld, 2023-01-23:
1. KAIYE Tuya Smart Zigbee Ampoule LED E27 10W Smart Life
-----------------------------------------------------------------------------------------------------
2. {"Version":"3","ZDeviceName":"","Status":"inDB","SQN":"7b","Ep":{"01":{"0000":{"4000":"1.0.13","0004":"eWeLight","0005":"ZB-CL01","0000":"1","0001":"15","0002":"2","0003":"1","0006":"20210407","0007":"1"},"0003":{},"0004":{},"0005":{},"0006":{"4001":"0","4002":"0","0000":"00"},"0008":{"0000":"ff"},"0300":{"0000":"0","0001":"0","0003":"24939","0004":"24701","0007":"333","0008":"2","400a":"31"},"ClusterType":{"692":"ColorControlRGBWW"}},"0d":{"0000":{}}},"Heartbeat":"9999","RIA":"4","LQI":37,"Battery":{},"Model":"ZB-CL01","ForceAckCommands":[],"MacCapa":"8e","IEEE":"00124b002450ea86","Type":"Switch/LvlControl/ColorControl","ProfileID":"c05e","ZDeviceID":"0210","App Version":"15","Attributes List":{"Request":{"01":{"0000":1674158889.078024,"0003":1674158889.0785017,"0004":1674158889.0789583,"0005":1674158889.0794115,"0006":1674158889.0798736,"0008":1674158889.0803325,"0300":1674158889.0871286},"0d":{"0000":1674158889.0929108}}},"DeviceType":"FFD","HW Version":"1","Last Cmds":[[1167,"02","6421","0100","0101000000000000010000"],[1178,"02","6421","0100","0101000000000000010000"],[1183,"02","6421","0100","0101000000000000010000"],[1188,"02","6421","0100","0101000000000000010000"],[1193,"02","6421","0100","0101000000000000010000"],[1198,"02","6421","0100","0101000000000000010000"],[1203,"02","6421","0100","0101000000000000010000"],[1208,"02","6421","0100","0101000000000000010000"],[1213,"02","6421","0100","0101000000000000010000"],[1229,"02","6421","0062","010100"]],"LogicalType":"Router","Manufacturer":"0000","Manufacturer Name":"eWeLight","NbEp":"2","PowerSource":"Main","ReadAttributes":{"Ep":{"01":{"0000":{"TimeStamp":1674463344,"iSQN":{"4000":99,"0004":95,"0005":95,"0000":1213,"0001":96,"0002":96,"0003":96,"0006":97,"0007":97,"000a":97,"000f":98,"0010":98,"0015":98,"f000":99},"Attributes":{"4000":"00","0004":"00","0005":"00","0000":"00","0001":"00","0002":"00","0003":"00","0006":"00","0007":"00","000a":"86","000f":"86","0010":"86","0015":"86","f000":"86"},"ZigateRequest":{}},"0006":{"TimeStamp":1674416479,"iSQN":{"4001":90,"4002":90,"0000":1128},"Attributes":{"4001":"00","4002":"00","0000":"00"},"ZigateRequest":{}},"0008":{"TimeStamp":1674416479,"iSQN":{"0000":1129},"Attributes":{"0000":"00"},"ZigateRequest":{}},"0300":{"TimeStamp":1674153865,"iSQN":{"0000":92,"0001":92,"0003":92,"0004":93,"0007":93,"0008":93,"400a":94},"Attributes":{"0000":"00","0001":"00","0003":"00","0004":"00","0007":"00","0008":"00","400a":"00"},"ZigateRequest":{}}},"0d":{"0000":{"TimeStamp":1674153745,"iSQN":{"0004":51,"0005":51},"Attributes":{"0004":"86","0005":"86"},"ZigateRequest":{}}}}},"ReceiveOnIdle":"On","Stack Version":"2","Stamp":{"time":1674465732.9688337,"Time":"2023-01-23 10:22:12","MsgType":"8011","LastSeen":1674465732,"LastPing":1674463344},"ZCL Version":"1","Health":"Live","Capability":["Full-Function Device","Main Powered","Receiver during Idle","Standard security","NwkAddr should be allocated"],"RollingLQI":[53,52,53,53,54,53,53,53,52,53,37],"Epv2":{"01":{"ClusterIn":{"0000":{},"0003":{},"0004":{},"0005":{},"0006":{},"0008":{},"0300":{}},"ClusterOut":{},"ProfileID":"c05e","ZDeviceID":"0210"},"0d":{"ClusterIn":{},"ClusterOut":{},"ProfileID":{},"ZDeviceID":{}}},"WriteAttributes":{"Ep":{"01":{"0000":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"0003":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"0004":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"0005":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"0006":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"0008":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"0300":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}}},"0d":{"0000":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}}}}},"ZDeviceVersion":"0000","ConfigSource":"8043","PairingInProgress":false,"Max Buffer Size":"50","Max Rx":"00a0","Max Tx":"00a0","macapa":"8e","bitfield":"4001","server_mask":"0000","descriptor_capability":"00","PairingTime":1674153764.0702205,"Bind":{"01":{"0000":{"Target":"0000","Stamp":1674153764,"Phase":"requested","Status":"","i_sqn":54},"0003":{"Target":"0000","Stamp":1674153764,"Phase":"requested","Status":"","i_sqn":55},"0004":{"Target":"0000","Stamp":1674153764,"Phase":"requested","Status":"","i_sqn":56},"0005":{"Target":"0000","Stamp":1674153764,"Phase":"requested","Status":"","i_sqn":57},"0006":{"Target":"0000","Stamp":1674153764,"Phase":"requested","Status":"","i_sqn":58},"0008":{"Target":"0000","Stamp":1674153764,"Phase":"requested","Status":"","i_sqn":59},"0300":{"Target":"0000","Stamp":1674153764,"Phase":"requested","Status":"","i_sqn":60}}},"BindingTable":{"SQN":0,"Status":"Success","TimeStamp":1674153868.2785735,"BindingTableSize":4,"Devices":[{"00124b002450ea86":{"sourceEp":"01","Cluster":"0000","targetIEEE":"00158d0005bea75c","targetNickName":null,"targetEp":"01"}},{"00124b002450ea86":{"sourceEp":"01","Cluster":"0003","targetIEEE":"00158d0005bea75c","targetNickName":null,"targetEp":"01"}},{"00124b002450ea86":{"sourceEp":"01","Cluster":"0004","targetIEEE":"00158d0005bea75c","targetNickName":null,"targetEp":"01"}},{"00124b002450ea86":{"sourceEp":"01","Cluster":"0005","targetIEEE":"00158d0005bea75c","targetNickName":null,"targetEp":"01"}}]},"ZDPSQN":"07","ConfigureReporting":{"Ep":{"01":{"0006":{"TimeStamp":1674153766.2506185,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"0008":{"TimeStamp":1674153766.2523298,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"0300":{"TimeStamp":1674153766.2562718,"iSQN":{},"Attributes":{},"ZigateRequest":{}}}}},"CertifiedDevice":false,"PreviousStatus":"inDB","ReadConfigureReporting":{"Ep":{},"Request":{"Status":"Requested","Retry":0,"TimeStamp":0}},"ColorInfos":{"Hue":0,"Saturation":0,"X":"24939","Y":"24701","ColorTemperatureMireds":"333","ColorMode":2,"ColorCapabilities":"31"},"SWBUILD_1":"20210407","SWBUILD_3":"1.0.13","Neighbours":[{"SQN":0,"Status":"","TimeStamp":1674299076,"Time":1674299076,"Devices":[{"0000":{"_relationshp":"Parent","_devicetype":"Coordinator","_depth":0,"_lnkqty":32,"_rxonwhenidl":"??","_IEEE":"00158d0005bea75c","_permitjnt":"??"}},{"2fd7":{"_relationshp":"Child","_devicetype":"Router","_depth":2,"_lnkqty":0,"_rxonwhenidl":"Rx-On","_IEEE":"7cb03eaa0a040691","_permitjnt":"??"}},{"2947":{"_relationshp":"None","_devicetype":"Router","_depth":255,"_lnkqty":1,"_rxonwhenidl":"??","_IEEE":"dc8e95fffe558434","_permitjnt":"??"}},{"b780":{"_relationshp":"None","_devicetype":"Router","_depth":255,"_lnkqty":33,"_rxonwhenidl":"??","_IEEE":"f0d1b800001d36d7","_permitjnt":"??"}},{"3c93":{"_relationshp":"None","_devicetype":"Router","_depth":255,"_lnkqty":58,"_rxonwhenidl":"??","_IEEE":"00158d0002506165","_permitjnt":"??"}}]}],"RoutingTable":[{"SQN":0,"Status":"Success","TimeStamp":1674299091.741119,"Time":1674299076,"RoutingTableTableSize":2,"Devices":[{"0000":{"Status":"Active (0)","MemoryConstrained":0,"ManyToOne":0,"RouteRecordRequired":0,"NextHopNwkId":"0000"}},{"3c93":{"Status":"Active (0)","MemoryConstrained":0,"ManyToOne":0,"RouteRecordRequired":0,"NextHopNwkId":"3c93"}}]}],"NeighbourTableSize":5,"MapLQI":{"2947":48,"0000":38,"3c93":73,"b780":78},"GroupMemberShip":{"01":{}}}
---------------------------------------------------------------------------------------------------------------
3. https://fr.aliexpress.com/item/1005003903323750.html?spm=a2g0o.order_list.order_list_main.10.805f5e5bIpWmmb&gatewayAdapt=glo2fra
-------------------------------------------------------------------------------------------------------------
4. 2023-01-19 19:42:45.810  ZIGBEE: Assisted Pairing: Polling: ['138e']
2023-01-19 19:42:45.810  ZIGBEE: --> New device: 138e
2023-01-19 19:42:45.810  ZIGBEE: --> inDB , removed 138e from List
2023-01-19 19:42:45.810  ZIGBEE: ------> New Cluster: {'ClusterId': '0000', 'ClusterDesc': 'Basic'}
2023-01-19 19:42:45.810  ZIGBEE: ------> New Cluster: {'ClusterId': '0003', 'ClusterDesc': 'Identify'}
2023-01-19 19:42:45.810  ZIGBEE: ------> New Cluster: {'ClusterId': '0004', 'ClusterDesc': 'Groups'}
2023-01-19 19:42:45.811  ZIGBEE: ------> New Cluster: {'ClusterId': '0005', 'ClusterDesc': 'Scenes'}
2023-01-19 19:42:45.811  ZIGBEE: ------> New Cluster: {'ClusterId': '0006', 'ClusterDesc': 'On/Off'}
[... cut ...] terId': '0008', 'ClusterDesc': 'Level Control'}, {'ClusterId': '0300', 'ClusterDesc': 'Colour Control'}]}, {'Ep': '0d', 'Clusters': [{'ClusterId': '0000', 'ClusterDesc': 'Basic'}]}]}
2023-01-19 19:42:46.139  Status: ZIGBEE: NetworkID: 6421 is replacing 138e for object: 00124b002450ea86
2023-01-19 19:42:46.157  Status: ZIGBEE: Device Annoucement: NwkId: 6421 Ieee: 00124b002450ea86 MacCap: 8e
2023-01-19 19:42:46.613  Status: ZIGBEE: Device Annoucement: NwkId: 6421 Ieee: 00124b002450ea86 MacCap: 8e

pipiche38, 2023-01-24:
if you want to test, this has been integrated for the next release.

```
git remote add pipiche https://github.com/pipiche38/Domoticz-Zigbee.git
git pull pipiche optimized-device/6.3.010
git checkout pipiche/optimized-device/6.3.010
```

the  restart the plugin

to fall back

```
git checkout stable6
git status
git remote remove pipiche
```

pouilld, 2023-01-25:
Merci pour cette intégration rapide.

closed 2023-01-25 by pipiche38

pipiche38, 2023-02-01:
Release 6.3.010 has been released today, so you can switch to it , if not yet done
```
git checkout stable6
git pull
```
and  if you have tested and used the remote pipiche, you can simply discard it
`git remote remove pipiche`