zigbeefordomoticz/Domoticz-Zigbee #1472: [Model Certification] -

pouilld, 2023-01-23:
1.Paulmann 50052 SmartHome Zigbee LED Ampoule Soret, 8,5W, E27, 230V, Opale, 2700K, dimmable 
---------------------------------------------------------------------------------------------------------------
2. https://www.amazon.fr/gp/product/B0721G1ZNZ/ref=ppx_od_dt_b_asin_title_s00?ie=UTF8&psc=1
---------------------------------------------------------------------------------------------------------------
3. {"Version":"3","ZDeviceName":"","Status":"inDB","SQN":"e5","Ep":{"01":{"1000":{},"0000":{"4000":"1218-0909","0004":"Paulmann Licht GmbH","0005":"Dimmablelight","0000":"2","0001":"1","0002":"1","0003":"1","0006":"20160300","0007":"1","000a":"00030201"},"0004":{},"0003":{},"0006":{"4001":"0","4002":"0","0000":"00"},"0008":{"0000":"fe"},"0005":{},"0019":{},"ClusterType":{"693":"LvlControl"}}},"Heartbeat":"1166","RIA":"4","LQI":72,"Battery":{},"Model":"Dimmablelight","ForceAckCommands":[],"MacCapa":"8e","IEEE":"00158d0002506165","Type":"Switch/LvlControl","ProfileID":"0104","ZDeviceID":"0101","App Version":"1","Attributes List":{"Request":{"01":{"1000":1674158899.0840578,"0000":1674158899.0781784,"0004":1674158899.0786414,"0003":1674158899.0791023,"0006":1674158899.0795608,"0008":1674158899.0800147,"0005":1674158899.082733,"0019":1674158899.0845432}}},"DeviceType":"FFD","HW Version":"1","Last Cmds":[[1137,"07","3c93","0100","0101000800000000010000"],[1151,"02","3c93","0092","010100"],[1152,"07","3c93","0100","0101000600000000010000"],[1153,"07","3c93","0100","0101000800000000010000"],[1218,"07","3c93","0100","0101000600000000010000"],[1219,"07","3c93","0100","0101000800000000010000"],[1222,"02","3c93","0092","010100"],[1224,"07","3c93","0100","0101000600000000010000"],[1225,"07","3c93","0100","0101000800000000010000"],[1230,"02","3c93","0062","010100"]],"LogicalType":"Router","Manufacturer":"119d","Manufacturer Name":"Paulmann Licht GmbH","NbEp":"1","PowerSource":"Main","ReadAttributes":{"Ep":{"01":{"0000":{"TimeStamp":1674154112,"iSQN":{"4000":151,"0004":147,"0005":147,"0000":147,"0001":148,"0002":148,"0003":148,"0006":149,"0007":149,"000a":149,"000f":150,"0010":150,"0015":150,"f000":151},"Attributes":{"4000":"00","0004":"00","0005":"00","0000":"00","0001":"00","0002":"00","0003":"00","0006":"00","0007":"00","000a":"00","000f":"86","0010":"86","0015":"86","f000":"86"},"ZigateRequest":{}},"0019":{"TimeStamp":1674154109,"iSQN":{"0001":141,"0002":141,"0003":141,"0004":142,"0005":142,"0006":142,"0007":143,"0008":143,"0009":143,"000a":144},"Attributes":{},"ZigateRequest":{}},"0006":{"TimeStamp":1674469196,"iSQN":{"4001":145,"4002":145,"0000":1224},"Attributes":{"4001":"00","4002":"00","0000":"00"},"ZigateRequest":{}},"0008":{"TimeStamp":1674469159,"iSQN":{"0000":1225},"Attributes":{"0000":"00"},"ZigateRequest":{}}}}},"ReceiveOnIdle":"On","Stack Version":"1","Stamp":{"time":1674469271.5832975,"Time":"2023-01-23 11:21:11","MsgType":"8011","LastSeen":1674469271},"ZCL Version":"2","Health":"Live","Capability":["Full-Function Device","Main Powered","Receiver during Idle","Standard security","NwkAddr should be allocated"],"RollingLQI":[67,71,68,71,70,71,70,71,72,72,72],"Epv2":{"01":{"ClusterIn":{"1000":{},"0000":{},"0004":{},"0003":{},"0006":{},"0008":{},"0005":{}},"ClusterOut":{"0019":{}},"ProfileID":"0104","ZDeviceID":"0101"}},"WriteAttributes":{"Ep":{"01":{"1000":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"0000":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"0004":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"0003":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"0006":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"0008":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"0005":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"0019":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}}}}},"ZDeviceVersion":"0001","ConfigSource":"8043","PairingInProgress":false,"Max Buffer Size":"7f","Max Rx":"005a","Max Tx":"005a","macapa":"8e","bitfield":"4001","server_mask":"2a00","descriptor_capability":"00","PairingTime":1674154109.1054354,"Bind":{"01":{"1000":{"Target":"0000","Stamp":1674154110,"Phase":"binded","Status":"00","i_sqn":137},"0000":{"Target":"0000","Stamp":1674154109,"Phase":"binded","Status":"00","i_sqn":131},"0004":{"Target":"0000","Stamp":1674154109,"Phase":"binded","Status":"00","i_sqn":132},"0003":{"Target":"0000","Stamp":1674154109,"Phase":"binded","Status":"00","i_sqn":133},"0006":{"Target":"0000","Stamp":1674154109,"Phase":"binded","Status":"00","i_sqn":134},"0008":{"Target":"0000","Stamp":1674154110,"Phase":"binded","Status":"00","i_sqn":135},"0005":{"Target":"0000","Stamp":1674154110,"Phase":"binded","Status":"00","i_sqn":136}}},"BindingTable":{"SQN":0,"Status":"Success","TimeStamp":1674154117.0324104,"BindingTableSize":7,"Devices":[{"00158d0002506165":{"sourceEp":"01","Cluster":"0000","targetIEEE":"00158d0005bea75c","targetNickName":null,"targetEp":"01"}},{"00158d0002506165":{"sourceEp":"01","Cluster":"0004","targetIEEE":"00158d0005bea75c","targetNickName":null,"targetEp":"01"}},{"00158d0002506165":{"sourceEp":"01","Cluster":"0003","targetIEEE":"00158d0005bea75c","targetNickName":null,"targetEp":"01"}},{"00158d0002506165":{"sourceEp":"01","Cluster":"0006","targetIEEE":"00158d0005bea75c","targetNickName":null,"targetEp":"01"}},{"00158d0002506165":{"sourceEp":"01","Cluster":"0008","targetIEEE":"00158d0005bea75c","targetNickName":null,"targetEp":"01"}},{"00158d0002506165":{"sourceEp":"01","Cluster":"0005","targetIEEE":"00158d0005bea75c","targetNickName":null,"targetEp":"01"}},{"00158d0002506165":{"sourceEp":"01","Cluster":"1000","targetIEEE":"00158d0005bea75c","targetNickName":null,"targetEp":"01"}}]},"ZDPSQN":"09","ConfigureReporting":{"Ep":{"01":{"0006":{"TimeStamp":1674154109.1394415,"iSQN":{"0000":139},"Attributes":{"0000":"00"},"ZigateRequest":{}},"0008":{"TimeStamp":1674154109.1461203,"iSQN":{"0000":140},"Attributes":{"0000":"00"},"ZigateRequest":{}}}}},"CertifiedDevice":false,"ZCLSQN":"03","SWBUILD_1":"20160300","SWBUILD_3":"1218-0909","ReadConfigureReporting":{"Ep":{"01":{"0006":{"0000":{"TimeStamp":1674463455.308107,"Status":"00","DataType":"10","MinInterval":"0001","MaxInterval":"012c"}},"0008":{"0000":{"TimeStamp":1674463455.4816763,"Status":"00","DataType":"20","MinInterval":"0005","MaxInterval":"012c"}}}},"TimeStamp":1674463455.4816606},"Neighbours":[{"SQN":0,"Status":"","TimeStamp":1674299076,"Time":1674299076,"Devices":[{"0000":{"_relationshp":"Parent","_devicetype":"Coordinator","_depth":0,"_lnkqty":46,"_rxonwhenidl":"Rx-On","_IEEE":"00158d0005bea75c","_permitjnt":"??"}},{"b780":{"_relationshp":"Sibling","_devicetype":"Router","_depth":1,"_lnkqty":89,"_rxonwhenidl":"Rx-On","_IEEE":"f0d1b800001d36d7","_permitjnt":"??"}},{"6421":{"_relationshp":"Sibling","_devicetype":"Router","_depth":1,"_lnkqty":73,"_rxonwhenidl":"Rx-On","_IEEE":"00124b002450ea86","_permitjnt":"??"}},{"2947":{"_relationshp":"Sibling","_devicetype":"Router","_depth":1,"_lnkqty":22,"_rxonwhenidl":"Rx-On","_IEEE":"dc8e95fffe558434","_permitjnt":"??"}}]}],"RoutingTable":[{"SQN":0,"Status":"Success","TimeStamp":1674299084.7676225,"Time":1674299076,"RoutingTableTableSize":4,"Devices":[{"0000":{"Status":"Active (0)","MemoryConstrained":0,"ManyToOne":0,"RouteRecordRequired":0,"NextHopNwkId":"0000"}},{"ac81":{"Status":"Discovery underway (1)","MemoryConstrained":0,"ManyToOne":0,"RouteRecordRequired":0,"NextHopNwkId":"fffe"}},{"2fd7":{"Status":"Discovery underway (1)","MemoryConstrained":0,"ManyToOne":0,"RouteRecordRequired":0,"NextHopNwkId":"fffe"}},{"f16e":{"Status":"Discovery underway (1)","MemoryConstrained":0,"ManyToOne":0,"RouteRecordRequired":0,"NextHopNwkId":"fffe"}}]}],"NeighbourTableSize":4,"MapLQI":{"2947":74,"6421":58,"0000":43,"b780":136},"GroupMemberShip":{"01":{}}}
---------------------------------------------------------------------------------------------------------------
2023-01-19 19:48:06.293  ZIGBEE: Assisted Pairing: Polling: []
2023-01-19 19:48:06.293  ZIGBEE: --> Empty queue
2023-01-19 19:48:10.631  Status: ZIGBEE: [None] NEW OBJECT: 3c93 None
2023-01-19 19:48:10.683  Status: ZIGBEE: [None] NEW OBJECT: b780 None
2023-01-19 19:48:10.791  Status: ZIGBEE: [-] NEW OBJECT: 3c93 Active Endpoint Response Ep: 01 LQI: 68
2023-01-19 19:48:10.795  Status: ZIGBEE: [-] NEW OBJECT: 3c93 Request Simple Descriptor for Ep: 01
2023-01-19 19:48:10.900  Status: ZIGBEE: [-] NEW OBJECT: b780 Active Endpoint Response Ep: 01 LQI: 62
[... cut ...]
2023-01-19 19:48:34.721  ZIGBEE: UpdateDevice - (ZIGBEE - LEDVANCE DIM_LvlControl-f0d1b800001d36d7-01) 1:
2023-01-19 19:48:34.889  ZIGBEE: UpdateDevice - (ZIGBEE - LEDVANCE DIM_LvlControl-f0d1b800001d36d7-01) 1:100
2023-01-19 19:48:36.198  ZIGBEE: Disable Assisted pairing
2023-01-19 19:48:36.217  Status: ZIGBEE: Request Disabling Accepting new Hardware
2023-01-19 19:48:36.734  Status: ZIGBEE: Accepting new Hardware: Disable (Off)
2023-01-19 19:49:34.411  ZIGBEE: UpdateDevice - (ZIGBEE - Dimmablelight_LvlControl-00158d0002506165-01) 1:100

pipiche38, 2023-01-24:
if you want to test, this has been integrated for the next release.

```
git remote add pipiche https://github.com/pipiche38/Domoticz-Zigbee.git
git pull pipiche optimized-device/6.3.010
git checkout pipiche/optimized-device/6.3.010
```

the  restart the plugin

to fall back

```
git checkout stable6
git status
git remote remove pipiche
```

pouilld, 2023-01-25:
Merci.

closed 2023-01-25 by pouilld

pipiche38, 2023-02-01:
Release 6.3.010 has been released today, so you can switch to it , if not yet done
```
git checkout stable6
git pull
```
and  if you have tested and used the remote pipiche, you can simply discard it
`git remote remove pipiche`