zigbeefordomoticz/Domoticz-Zigbee #1473: [Model Certification] -

pouilld, 2023-01-23:
1. LEDVANCE Ampoule LED  E14 5W SMART+ Mini bulb Dimmable
---------------------------------------------------------------------------------------------------------------
2. https://www.amazon.fr/gp/product/B08KY8KXQH/ref=ppx_od_dt_b_asin_title_s00?ie=UTF8&psc=1
---------------------------------------------------------------------------------------------------------------
{"Version":"3","ZDeviceName":"","Status":"inDB","SQN":"35","Ep":{"01":{"0000":{"4000":"02176550","0004":"LEDVANCE","0005":"LEDVANCE DIM","0000":"3","0001":"0","0002":"0","0003":"2","0007":"1"},"0003":{},"0004":{},"0005":{},"0006":{"4001":"0","4002":"0","0000":"00"},"0008":{"0000":"ba"},"0b05":{},"0019":{},"ClusterType":{"694":"LvlControl"}},"f2":{"0021":{}}},"Heartbeat":"1420","RIA":"4","LQI":63,"Battery":{},"Model":"LEDVANCE DIM","ForceAckCommands":[],"MacCapa":"8e","IEEE":"f0d1b800001d36d7","Type":"Switch/LvlControl","ProfileID":"a1e0","ZDeviceID":"0061","App Version":"0","Attributes List":{"Request":{"01":{"0000":1674158809.079551,"0003":1674158809.0798588,"0004":1674158809.0801563,"0005":1674158809.080454,"0006":1674158809.0807636,"0008":1674158809.0810707,"0b05":1674158809.081368,"0019":1674158809.091639},"f2":{"0021":1674158809.0924916}}},"DeviceType":"FFD","HW Version":"2","Last Cmds":[[1133,"07","b780","0100","0101000800000000010000"],[1135,"02","b780","0092","010100"],[1138,"07","b780","0100","0101000600000000010000"],[1139,"07","b780","0100","0101000800000000010000"],[1220,"07","b780","0100","0101000600000000010000"],[1221,"07","b780","0100","0101000800000000010000"],[1223,"02","b780","0092","010100"],[1226,"07","b780","0100","0101000600000000010000"],[1227,"07","b780","0100","0101000800000000010000"],[1232,"02","b780","0062","010100"]],"LogicalType":"Router","Manufacturer":"1189","Manufacturer Name":"LEDVANCE","NbEp":"2","PowerSource":"Main","ReadAttributes":{"Ep":{"01":{"0000":{"TimeStamp":1674154115,"iSQN":{"4000":173,"0004":169,"0005":169,"0000":169,"0001":170,"0002":170,"0003":170,"0006":171,"0007":171,"000a":171,"000f":172,"0010":172,"0015":172,"f000":173},"Attributes":{"4000":"00","0004":"00","0005":"00","0000":"00","0001":"00","0002":"00","0003":"00","0006":"86","0007":"00","000a":"86","000f":"86","0010":"86","0015":"86","f000":"86"},"ZigateRequest":{}},"0019":{"TimeStamp":1674154114,"iSQN":{"0001":163,"0002":163,"0003":163,"0004":164,"0005":164,"0006":164,"0007":165,"0008":165,"0009":165,"000a":166},"Attributes":{"0001":"86","0002":"86","0003":"86","0004":"86","0005":"86","0006":"86","0007":"86","0008":"86","0009":"86","000a":"86"},"ZigateRequest":{}},"0006":{"TimeStamp":1674470373,"iSQN":{"4001":167,"4002":167,"0000":1226},"Attributes":{"4001":"00","4002":"00","0000":"00"},"ZigateRequest":{}},"0008":{"TimeStamp":1674470373,"iSQN":{"0000":1227},"Attributes":{"0000":"00"},"ZigateRequest":{}}},"f2":{"0000":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}}}}},"ReceiveOnIdle":"On","Stack Version":"0","Stamp":{"time":1674470373.3008904,"Time":"2023-01-23 11:39:33","MsgType":"8011","LastSeen":1674470373},"ZCL Version":"3","Health":"Live","Capability":["Full-Function Device","Main Powered","Receiver during Idle","Standard security","NwkAddr should be allocated"],"RollingLQI":[67,67,67,65,63,64,64,64,65,63,63],"Epv2":{"01":{"ClusterIn":{"0000":{},"0003":{},"0004":{},"0005":{},"0006":{},"0008":{},"0b05":{}},"ClusterOut":{"0019":{}},"ProfileID":"0104","ZDeviceID":"0101"},"f2":{"ClusterIn":{},"ClusterOut":{"0021":{}},"ProfileID":"a1e0","ZDeviceID":"0061"}},"WriteAttributes":{"Ep":{"01":{"0000":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"0003":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"0004":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"0005":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"0006":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"0008":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"0b05":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"0019":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}}},"f2":{"0021":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}}}}},"ZDeviceVersion":"0001","ConfigSource":"8043","PairingInProgress":false,"Max Buffer Size":"52","Max Rx":"0052","Max Tx":"0052","macapa":"8e","bitfield":"4001","server_mask":"2c00","descriptor_capability":"00","PairingTime":1674154109.2765958,"Bind":{"01":{"0000":{"Target":"0000","Stamp":1674154112,"Phase":"binded","Status":"00","i_sqn":153},"0003":{"Target":"0000","Stamp":1674154112,"Phase":"binded","Status":"00","i_sqn":154},"0004":{"Target":"0000","Stamp":1674154112,"Phase":"binded","Status":"00","i_sqn":155},"0005":{"Target":"0000","Stamp":1674154112,"Phase":"binded","Status":"00","i_sqn":156},"0006":{"Target":"0000","Stamp":1674154113,"Phase":"binded","Status":"00","i_sqn":157},"0008":{"Target":"0000","Stamp":1674154113,"Phase":"binded","Status":"00","i_sqn":158},"0b05":{"Target":"0000","Stamp":1674154113,"Phase":"binded","Status":"00","i_sqn":159}}},"BindingTable":{"SQN":0,"Status":"Success","TimeStamp":1674154116.8531678,"BindingTableSize":7,"Devices":[{"f0d1b800001d36d7":{"sourceEp":"01","Cluster":"0000","targetIEEE":"00158d0005bea75c","targetNickName":null,"targetEp":"01"}},{"f0d1b800001d36d7":{"sourceEp":"01","Cluster":"0003","targetIEEE":"00158d0005bea75c","targetNickName":null,"targetEp":"01"}},{"f0d1b800001d36d7":{"sourceEp":"01","Cluster":"0004","targetIEEE":"00158d0005bea75c","targetNickName":null,"targetEp":"01"}},{"f0d1b800001d36d7":{"sourceEp":"01","Cluster":"0005","targetIEEE":"00158d0005bea75c","targetNickName":null,"targetEp":"01"}},{"f0d1b800001d36d7":{"sourceEp":"01","Cluster":"0006","targetIEEE":"00158d0005bea75c","targetNickName":null,"targetEp":"01"}},{"f0d1b800001d36d7":{"sourceEp":"01","Cluster":"0008","targetIEEE":"00158d0005bea75c","targetNickName":null,"targetEp":"01"}},{"f0d1b800001d36d7":{"sourceEp":"01","Cluster":"0b05","targetIEEE":"00158d0005bea75c","targetNickName":null,"targetEp":"01"}}]},"ZDPSQN":"07","ConfigureReporting":{"Ep":{"01":{"0006":{"TimeStamp":1674154109.2977023,"iSQN":{"0000":161},"Attributes":{"0000":"00"},"ZigateRequest":{}},"0008":{"TimeStamp":1674154109.301087,"iSQN":{"0000":162},"Attributes":{"0000":"00"},"ZigateRequest":{}}}}},"CertifiedDevice":false,"ZCLSQN":"03","SWBUILD_3":"02176550","ReadConfigureReporting":{"Ep":{"01":{"0006":{"0000":{"TimeStamp":1674463457.2296515,"Status":"00","DataType":"10","MinInterval":"0001","MaxInterval":"012c"}},"0008":{"0000":{"TimeStamp":1674463457.3993442,"Status":"00","DataType":"20","MinInterval":"0005","MaxInterval":"012c"}}}},"TimeStamp":1674463457.3993344},"Neighbours":[{"SQN":0,"Status":"","TimeStamp":1674299076,"Time":1674299076,"Devices":[{"0000":{"_relationshp":"Parent","_devicetype":"Coordinator","_depth":0,"_lnkqty":95,"_rxonwhenidl":"Rx-On","_IEEE":"00158d0005bea75c","_permitjnt":"??"}},{"2947":{"_relationshp":"Sibling","_devicetype":"Router","_depth":15,"_lnkqty":59,"_rxonwhenidl":"Rx-On","_IEEE":"dc8e95fffe558434","_permitjnt":"??"}},{"3c93":{"_relationshp":"Sibling","_devicetype":"Router","_depth":15,"_lnkqty":136,"_rxonwhenidl":"Rx-On","_IEEE":"00158d0002506165","_permitjnt":"??"}},{"6421":{"_relationshp":"Sibling","_devicetype":"Router","_depth":15,"_lnkqty":78,"_rxonwhenidl":"Rx-On","_IEEE":"00124b002450ea86","_permitjnt":"??"}}]}],"RoutingTable":[{"SQN":0,"Status":"Success","TimeStamp":1674299113.4003997,"Time":1674299076,"RoutingTableTableSize":10,"Devices":[{"0000":{"Status":"Active (0)","MemoryConstrained":0,"ManyToOne":1,"RouteRecordRequired":0,"NextHopNwkId":"0000"}},{"6421":{"Status":"Active (0)","MemoryConstrained":0,"ManyToOne":0,"RouteRecordRequired":0,"NextHopNwkId":"3c93"}},{"3c93":{"Status":"Active (0)","MemoryConstrained":0,"ManyToOne":0,"RouteRecordRequired":0,"NextHopNwkId":"3c93"}},{"2947":{"Status":"Active (0)","MemoryConstrained":0,"ManyToOne":0,"RouteRecordRequired":0,"NextHopNwkId":"2947"}}]}],"NeighbourTableSize":4,"MapLQI":{"2947":88,"6421":33,"0000":48,"3c93":89},"GroupMemberShip":{"01":{}}}
---------------------------------------------------------------------------------------------------------------
2023-01-19 19:48:06.293  ZIGBEE: Assisted Pairing: Polling: []
2023-01-19 19:48:06.293  ZIGBEE: --> Empty queue
2023-01-19 19:48:10.631  Status: ZIGBEE: [None] NEW OBJECT: 3c93 None
2023-01-19 19:48:10.683  Status: ZIGBEE: [None] NEW OBJECT: b780 None
[... cut ...]
2023-01-19 19:48:34.721  ZIGBEE: UpdateDevice - (ZIGBEE - LEDVANCE DIM_LvlControl-f0d1b800001d36d7-01) 1:
2023-01-19 19:48:34.889  ZIGBEE: UpdateDevice - (ZIGBEE - LEDVANCE DIM_LvlControl-f0d1b800001d36d7-01) 1:100
2023-01-19 19:48:36.198  ZIGBEE: Disable Assisted pairing
2023-01-19 19:48:36.217  Status: ZIGBEE: Request Disabling Accepting new Hardware
2023-01-19 19:48:36.734  Status: ZIGBEE: Accepting new Hardware: Disable (Off)
2023-01-19 19:49:34.411  ZIGBEE: UpdateDevice - (ZIGBEE - Dimmablelight_LvlControl-00158d0002506165-01) 1:100

pipiche38, 2023-01-24:
if you want to test, this has been integrated for the next release.

```
git remote add pipiche https://github.com/pipiche38/Domoticz-Zigbee.git
git pull pipiche optimized-device/6.3.010
git checkout pipiche/optimized-device/6.3.010
```

the  restart the plugin

to fall back

```
git checkout stable6
git status
git remote remove pipiche
```

pouilld, 2023-01-25:
Merci

closed 2023-01-25 by pipiche38

pipiche38, 2023-02-01:
Release 6.3.010 has been released today, so you can switch to it , if not yet done
```
git checkout stable6
git pull
```
and  if you have tested and used the remote pipiche, you can simply discard it
`git remote remove pipiche`