zigbeefordomoticz/Domoticz-Zigbee #1480: [Model Certification] - TS110F

Martial83, 2023-01-24:
1. Please do provide device information such as Brand and Model name.
1. Provide the URL where you have purchase the device
1. Finaly copy-paste all information provided in the log
{"Version":"3","ZDeviceName":"Pipistrello","Status":"inDB","SQN":"3b","Ep":{"01":{"0006":{"4001":"0","4002":"0","0000":"00"},"0000":{"0004":"_TYZB01_qezuin6k","0005":"TS110F","0000":"3","0001":"65","0002":"0","0003":"1","0006":"None","0007":"1"},"0008":{"0000":"ff","fc00":"1023"},"0004":{},"0005":{},"0019":{},"000a":{},"ClusterType":{"937":"LvlControl"}}},"Heartbeat":"10270","RIA":"1","LQI":111,"Battery":{},"Model":"TS110F","ForceAckCommands":[],"MacCapa":"8e","IEEE":"804b50fffe4e7c0a","Type":"Switch/LvlControl","ProfileID":"0104","ZDeviceID":"0101","App Version":"65","Attributes List":{"Ep":{"01":{"0006":{"4001":"21","4002":"21","0000":"10"},"0000":{"0000":"20","0001":"20","0002":"20","0003":"20","0004":"42","0005":"42","0006":"42","0007":"30","ffde":"20","fffd":"21","fffe":"30","ffe0":"48","ffe1":"48","ffe2":"20","ffe3":"48"},"0008":{"0000":"20","fc00":"21","fffd":"21"},"0004":{"0000":"18","fffd":"21"},"0005":{"0000":"20","0001":"20","0002":"21","0003":"10","0004":"18","fffd":"21"}}}},"DeviceType":"FFD","HW Version":"1","Last Cmds":[[93651,"07","1813","0100","0101000800000000010000"],[94493,"02","1813","0092","010100"],[94495,"07","1813","0100","0101000600000000010000"],[94496,"07","1813","0100","0101000800000000010000"],[101912,"02","1813","0081","010101ff0010"],[101915,"07","1813","0100","0101000600000000010000"],[101916,"07","1813","0100","0101000800000000010000"],[102698,"02","1813","0092","010100"],[102699,"07","1813","0100","0101000600000000010000"],[102700,"07","1813","0100","0101000800000000010000"]],"LogicalType":"Router","Manufacturer":"1002","Manufacturer Name":"_TYZB01_qezuin6k","NbEp":"1","PowerSource":"Main","ReadAttributes":{"Ep":{"01":{"0006":{"TimeStamp":1674504094,"iSQN":{"4001":7306,"4002":7307,"0000":102699},"Attributes":{"4001":"00","4002":"00","0000":"00"},"ZigateRequest":{}},"0000":{"TimeStamp":1619944613,"iSQN":{"4000":7321,"0004":7309,"0005":7310,"0000":7311,"0001":7312,"0002":7313,"0003":7314,"0006":7315,"0007":7316,"000a":7317,"000f":7318,"0010":7319,"0015":7320,"f000":7322},"Attributes":{"4000":"86","0004":"00","0005":"00","0000":"00","0001":"00","0002":"00","0003":"00","0006":"00","0007":"00","000a":"86","000f":"86","0010":"86","0015":"86","f000":"86"},"ZigateRequest":{}},"0008":{"TimeStamp":1674504094,"iSQN":{"0000":102700},"Attributes":{"fc00":"00","0000":"00"},"ZigateRequest":{}}}}},"ReceiveOnIdle":"On","Stack Version":"0","Stamp":{"time":1674552581.7141302,"Time":"2023-01-24 10:29:41","MsgType":"8011","LastSeen":1674552581},"ZCL Version":"3","Health":"Live","SWBUILD_1":"None","RoutingTable":[{"SQN":0,"Status":"Success","TimeStamp":1657443708.0684376,"Time":1657443500,"RoutingTableTableSize":10,"Devices":[{"0000":{"Status":"Active (0)","MemoryConstrained":0,"ManyToOne":1,"RouteRecordRequired":0,"NextHopNwkId":"85af"}},{"c5bd":{"Status":"Active (0)","MemoryConstrained":0,"ManyToOne":0,"RouteRecordRequired":0,"NextHopNwkId":"f33d"}},{"e02e":{"Status":"Active (0)","MemoryConstrained":0,"ManyToOne":0,"RouteRecordRequired":0,"NextHopNwkId":"f33d"}},{"118e":{"Status":"Active (0)","MemoryConstrained":0,"ManyToOne":0,"RouteRecordRequired":0,"NextHopNwkId":"85af"}},{"140a":{"Status":"Active (0)","MemoryConstrained":0,"ManyToOne":0,"RouteRecordRequired":0,"NextHopNwkId":"140a"}},{"25f5":{"Status":"Active (0)","MemoryConstrained":0,"ManyToOne":0,"RouteRecordRequired":0,"NextHopNwkId":"7ee6"}},{"85af":{"Status":"Active (0)","MemoryConstrained":0,"ManyToOne":0,"RouteRecordRequired":0,"NextHopNwkId":"85af"}},{"1776":{"Status":"Active (0)","MemoryConstrained":0,"ManyToOne":0,"RouteRecordRequired":0,"NextHopNwkId":"1776"}}]},{"SQN":0,"Status":"Success","TimeStamp":1657532512.191694,"Time":1657532364,"RoutingTableTableSize":10,"Devices":[{"0000":{"Status":"Active (0)","MemoryConstrained":0,"ManyToOne":1,"RouteRecordRequired":0,"NextHopNwkId":"f33d"}},{"1776":{"Status":"Active (0)","MemoryConstrained":0,"ManyToOne":0,"RouteRecordRequired":0,"NextHopNwkId":"1776"}},{"25f5":{"Status":"Active (0)","MemoryConstrained":0,"ManyToOne":0,"RouteRecordRequired":0,"NextHopNwkId":"7ee6"}},{"85af":{"Status":"Active (0)","MemoryConstrained":0,"ManyToOne":0,"RouteRecordRequired":0,"NextHopNwkId":"85af"}},{"c5bd":{"Status":"Active (0)","MemoryConstrained":0,"ManyToOne":0,"RouteRecordRequired":0,"NextHopNwkId":"f33d"}},{"140a":{"Status":"Active (0)","MemoryConstrained":0,"ManyToOne":0,"RouteRecordRequired":0,"NextHopNwkId":"140a"}},{"1eea":{"Status":"Active (0)","MemoryConstrained":0,"ManyToOne":0,"RouteRecordRequired":0,"NextHopNwkId":"85af"}},{"49de":{"Status":"Active (0)","MemoryConstrained":0,"ManyToOne":0,"RouteRecordRequired":0,"NextHopNwkId":"49de"}}]},{"SQN":0,"Status":"Success","TimeStamp":1670664430.6153524,"Time":1670664324,"RoutingTableTableSize":10,"Devices":[{"0000":{"Status":"Active (0)","MemoryConstrained":0,"ManyToOne":1,"RouteRecordRequired":1,"NextHopNwkId":"85af"}},{"85af":{"Status":"Active (0)","MemoryConstrained":0,"ManyToOne":0,"RouteRecordRequired":0,"NextHopNwkId":"85af"}},{"1eea":{"Status":"Active (0)","MemoryConstrained":0,"ManyToOne":0,"RouteRecordRequired":0,"NextHopNwkId":"85af"}},{"0a9b":{"Status":"Active (0)","MemoryConstrained":0,"ManyToOne":0,"RouteRecordRequired":0,"NextHopNwkId":"b90f"}},{"49de":{"Status":"Active (0)","MemoryConstrained":0,"ManyToOne":0,"RouteRecordRequired":0,"NextHopNwkId":"49de"}},{"96fa":{"Status":"Active (0)","MemoryConstrained":0,"ManyToOne":0,"RouteRecordRequired":0,"NextHopNwkId":"96fa"}},{"4131":{"Status":"Active (0)","MemoryConstrained":0,"ManyToOne":0,"RouteRecordRequired":0,"NextHopNwkId":"4131"}},{"5ad2":{"Status":"Active (0)","MemoryConstrained":0,"ManyToOne":0,"RouteRecordRequired":0,"NextHopNwkId":"85af"}}]},{"SQN":0,"Status":"Success","TimeStamp":1673451643.1405425,"Time":1673451483,"RoutingTableTableSize":10,"Devices":[{"c13f":{"Status":"Active (0)","MemoryConstrained":0,"ManyToOne":0,"RouteRecordRequired":0,"NextHopNwkId":"c13f"}},{"0000":{"Status":"Active (0)","MemoryConstrained":0,"ManyToOne":1,"RouteRecordRequired":0,"NextHopNwkId":"85af"}},{"e02e":{"Status":"Discovery underway (1)","MemoryConstrained":0,"ManyToOne":0,"RouteRecordRequired":0,"NextHopNwkId":"c13f"}},{"128a":{"Status":"Active (0)","MemoryConstrained":0,"ManyToOne":0,"RouteRecordRequired":0,"NextHopNwkId":"c13f"}},{"6eb5":{"Status":"Active (0)","MemoryConstrained":0,"ManyToOne":0,"RouteRecordRequired":0,"NextHopNwkId":"6eb5"}},{"85af":{"Status":"Active (0)","MemoryConstrained":1,"ManyToOne":1,"RouteRecordRequired":1,"NextHopNwkId":"85af"}},{"9afc":{"Status":"Active (0)","MemoryConstrained":0,"ManyToOne":0,"RouteRecordRequired":0,"NextHopNwkId":"9afc"}},{"2512":{"Status":"Active (0)","MemoryConstrained":0,"ManyToOne":0,"RouteRecordRequired":0,"NextHopNwkId":"2512"}},{"0a9b":{"Status":"Active (0)","MemoryConstrained":0,"ManyToOne":0,"RouteRecordRequired":0,"NextHopNwkId":"c13f"}}]},{"SQN":0,"Status":"Success","TimeStamp":1674488585.668174,"Time":1674488323,"RoutingTableTableSize":10,"Devices":[{"0000":{"Status":"Active (0)","MemoryConstrained":0,"ManyToOne":0,"RouteRecordRequired":0,"NextHopNwkId":"85af"}},{"26a3":{"Status":"Active (0)","MemoryConstrained":0,"ManyToOne":0,"RouteRecordRequired":0,"NextHopNwkId":"26a3"}},{"5f1f":{"Status":"Active (0)","MemoryConstrained":0,"ManyToOne":0,"RouteRecordRequired":0,"NextHopNwkId":"85af"}},{"c13f":{"Status":"Active (0)","MemoryConstrained":0,"ManyToOne":0,"RouteRecordRequired":0,"NextHopNwkId":"c13f"}},{"7ee6":{"Status":"Active (0)","MemoryConstrained":0,"ManyToOne":0,"RouteRecordRequired":0,"NextHopNwkId":"c13f"}},{"579e":{"Status":"Active (0)","MemoryConstrained":0,"ManyToOne":0,"RouteRecordRequired":0,"NextHopNwkId":"c13f"}},{"0a9b":{"Status":"Active (0)","MemoryConstrained":0,"ManyToOne":0,"RouteRecordRequired":0,"NextHopNwkId":"c13f"}}]}],"Bind":{"01":{"0019":{"Target":"0000","Stamp":1619944608,"Phase":"binded","Status":"00","i_sqn":7299},"0000":{"Target":"0000","Stamp":1619944609,"Phase":"binded","Status":"00","i_sqn":7302},"0006":{"Target":"0000","Stamp":1657270170,"Phase":"binded","Status":"00","i_sqn":17},"0008":{"Target":"0000","Stamp":1657270170,"Phase":"binded","Status":"00","i_sqn":19}}},"BindingTable":{"SQN":0,"Status":"Success","TimeStamp":1657228351.0014832,"BindingTableSize":4,"Devices":[{"804b50fffe4e7c0a":{"sourceEp":"01","Cluster":"0019","targetIEEE":"00158d00021e1 [... cut ...] Log_UnknowDeviceFlag":1673450868.0768542,"RollingLQI":[105,84,84,81,78,99,75,75,90,111,111],"SQN_000a":"3b","ZDPSQN":"03","_rawNodeDescriptor":"1002005200522c00008e524001","Max Buffer Size":"52","Max Rx":"0052","Max Tx":"0052","macapa":"8e","bitfield":"4001","server_mask":"2c00","descriptor_capability":"00","MapLQI":{"3456":55,"1eea":0,"d727":46,"85af":53,"579e":1,"9afc":255,"c292":92,"5f1f":0,"48ce":54,"71b5":144,"7b24":96},"CheckParam":false,"PreviousStatus":"inDB","GroupMemberShip":{"01":{}}}

Martial83, 2023-01-24:
Je cherche sa marque et modèle, mais il va falloir ,que j'ouvre une boîte murale...

pipiche38, 2023-01-24:
https://zigbee.blakadder.com/Lonsonho_QS-Zigbee-D02-TRIAC-LN.html

pipiche38, 2023-01-24:
if you want to test, this has been integrated for the next release.

```
git remote add pipiche https://github.com/pipiche38/Domoticz-Zigbee.git
git pull pipiche optimized-device/6.3.010
git checkout pipiche/optimized-device/6.3.010
```

the  restart the plugin

to fall back

```
git checkout stable6
git status
git remote remove pipiche
```

Martial83, 2023-01-31:
Merci Patrick et désolé d'avoir mis du temps à tester, mais je n'y arrive pas:
pi@raspslave:~/domoticz/plugins/Domoticz-Zigate $ git pull pipiche optimized-device/6.3.010
fatal: Couldn't find remote ref optimized-device/6.3.010
pi@raspslave:~/domoticz/plugins/Domoticz-Zigate $ git checkout pipiche/optimized-device/6.3.010
error: le spécificateur de chemin 'pipiche/optimized-device/6.3.010' ne correspond à aucun fichier connu de git

pipiche38, 2023-01-31:
Indeed with the time the branch has been removed
so you can try 

`git checkout optimized-device/6.3.010`

Martial83, 2023-01-31:
le spécificateur de chemin 'hotfix/6.3.010' ne correspond à aucun fichier connu de git
it does not matter...

pipiche38, 2023-01-31:
```
git checkout stable6
git pull
git checkout hotfix/6.3.010
```

Martial83, 2023-01-31:
It's OK, but the device is still non-optimized

pipiche38, 2023-01-31:
Unfortunately you need to remove it and then re-pair it

closed 2023-02-06 by pipiche38