zigbeefordomoticz/Domoticz-Zigbee #1481: [Model Certification] - TRADFRI bulb E27 CWS 806lm

Chaussette23, 2023-01-24:
Ampoule IKEA Tradfri RGB (multicolor), achetée chez IKEA.

Fonctionne parfaitement, aucune difficulté pour l'appairer (même méthode que les autres ampoules Tradfri. La gestion des couleurs, température, puissance d'éclairage est parfaite.

Dump du plugin : 
`{"PluginInfos":{"HardwareID":8,"HomeFolder":"/","StartupFolder":"/","UserDataFolder":"/","WebRoot":"","Database":"domoticz.db","Language":"fr","Version":"6.3","Author":"pipiche38","Name":"ZiGate+","Address":"0.0.0.0","Port":"9999","SerialPort":"/dev/ttyUSB-ZiGate+","Username":"","Password":"","Key":"Zigate","Mode1":"V2","Mode2":"USB","Mode3":"False","Mode4":"9440","Mode5":"","Mode6":"2","DomoticzVersion":"2022.2","DomoticzHash":"eea9db734","DomoticzBuildTime":"2022-11-05 13:05:35","PluginBranch":"stable6","PluginVersion":"6.3.009","TimeStamp":1674591339,"available":"6.3.009","available-firmMajor":"05","available-firmMinor":"0321","FirmwareVersion":"0322","FirmwareUpdate":false,"PluginUpdate":false,"LogLevel":7,"Zigpy":false,"CoordinatorModel":"ZiGate+ (V2)","CoordinatorFirmwareVersion":"0322","DisplayFirmwareVersion":"Zigate - 0322","CoordinatorIEEE":"00158d0005be903c","NetworkSize":"Total: 39 | Routers: 21 | End Devices: 18","NetworkDevices":{"1002":{"SONOFF":["01MINIZB"]},"1037":{"LUMI":["lumi.sensor_magnet.aq2","lumi.weather","lumi.sensor_wleak.aq1"]},"1110":{"Profalux":["BSO-Profalux","VoletBSO-Profalux","Volet-Profalux","Telecommande-Profalux"]},"1286":{"SONOFF":["01MINIZB"]},"117c":{"IKEA of Sweden":["TRADFRI bulb E27 WS opal 1000lm","TRADFRI bulb E27 CWS 806lm"]},"0000":{"eWeLink":["MS01","TH01"]}}},"Analytics":0,"Device":{"Version":"3","ZDeviceName":"Ampoule","Status":"inDB","SQN":"08","Ep":{"01":{"1000":{},"0000":{"4000":"1.0.021","0004":"IKEA of Sweden","0005":"TRADFRI bulb E27 CWS 806lm","0000":"3","0001":"80","0002":"101","0003":"1","0006":"2019062886LEL08N","0007":"1","000a":"4c4544313932344739"},"0003":{},"0004":{"0001":"00000000"},"0005":{"0000":"0","0001":"0","0002":"0","0003":"00"},"0006":{"4001":"0","4002":"0","0000":"00"},"0008":{"0000":"fe"},"0300":{"0000":"26","0001":"182","0003":"28513","0004":"26452","0007":"335","0008":"2","400a":"31"},"0b05":{},"fc7c":{},"0019":{},"ClusterType":{"308":"ColorControlRGBWW"}},"f2":{"0021":{}}},"Heartbeat":"158","RIA":"4","LQI":59,"Battery":{},"Model":"TRADFRI bulb E27 CWS 806lm","ForceAckCommands":[],"MacCapa":"8e","IEEE":"003c84fffe1a696f","Type":"Switch/LvlControl/ColorControl","ProfileID":"a1e0","ZDeviceID":"0061","App Version":"80","Attributes List":{"Request":{"01":{"1000":1660072135.099826,"0000":1660072135.097609,"0003":1660072135.0978768,"0004":1660072135.0981336,"0005":1660072135.0983877,"0006":1660072135.0986402,"0008":1660072135.0988972,"0300":1660072135.0991478,"0b05":1660072135.0993977,"fc7c":1660072135.1001241,"0019":1660072135.1003804},"f2":{"0021":1660072135.100647}}},"DeviceType":"FFD","HW Version":"1","Last Cmds":[],"LogicalType":"Router","Manufacturer":"117c","Manufacturer Name":"IKEA of Sweden","NbEp":"2","PowerSource":"Main","ReadAttributes":{"Ep":{"01":{"0019":{"TimeStamp":1669203444,"iSQN":{"0001":566,"0002":566,"0003":566,"0004":567,"0005":567,"0006":567,"0007":568,"0008":568,"0009":568,"000a":569},"Attributes":{"0001":"86","0002":"86","0003":"86","0004":"86","0005":"86","0006":"86","0007":"86","0008":"86","0009":"86","000a":"86"},"ZigateRequest":{}},"0006":{"TimeStamp":1674591095,"iSQN":{"4001":570,"4002":570,"0000":358},"Attributes":{"4001":"00","4002":"00","0000":"00"},"ZigateRequest":{}},"0008":{"TimeStamp":1674591095,"iSQN":{"0000":359},"Attributes":{"0000":"00"},"ZigateRequest":{}},"0300":{"TimeStamp":1674591331,"iSQN":{"0000":572,"0001":572,"0003":572,"0004":573,"0007":573,"0008":573,"400a":574},"Attributes":{"0000":"00","0001":"00","0003":"00","0004":"00","0007":"00","0008":"00","400a":"00"},"ZigateRequest":{}},"0000":{"TimeStamp":1669203445,"iSQN":{"4000":579,"0004":575,"0005":575,"0000":575,"0001":576,"0002":576,"0003":576,"0006":577,"0007":577,"000a":577,"000f":578,"0010":578,"0015":578,"f000":579},"Attributes":{"4000":"00","0004":"00","0005":"00","0000":"00","0001":"00","0002":"00","0003":"00","0006":"00","0007":"00","000a":"00","000f":"86","0010":"86","0015":"86","f000":"86"},"ZigateRequest":{}},"0004":{"TimeStamp":1674562736,"iSQN":{},"Attributes":{"0001":"00"},"ZigateRequest":{}},"0005":{"TimeStamp":1674562736,"iSQN":{},"Attributes":{"0000":"00","0001":"00","0002":"00","0003":"00"},"ZigateRequest":{}}}}},"ReceiveOnIdle":"On","Stack Version":"101","Stamp":{"time":1674591331.9915352,"Time":"2023-01-24 21:15:31","MsgType":"8011","LastSeen":1674591331,"LastPing":1659861931},"ZCL Version":"3","Health":"Live","RoutingTable":[{"SQN":0,"Status":"Success","TimeStamp":1669479859.2322435,"Time":1669479696,"RoutingTableTableSize":10,"Devices":[{"0000":{"Status":"Active (0)","MemoryConstrained":0,"ManyToOne":1,"RouteRecordRequired":0,"NextHopNwkId":"d64a"}},{"281e":{"Status":"Active (0)","MemoryConstrained":0,"ManyToOne":0,"RouteRecordRequired":0,"NextHopNwkId":"281e"}},{"c556":{"Status":"Active (0)","MemoryConstrained":0,"ManyToOne":0,"RouteRecordRequired":0,"NextHopNwkId":"c556"}},{"4866":{"Status":"Active (0)","MemoryConstrained":0,"ManyToOne":0,"RouteRecordRequired":0,"NextHopNwkId":"ea5c"}}]}],"Param":{},"Capability":["Full-Function Device","Main Powered","Receiver during Idle","Standard security","NwkAddr should be allocated"],"NeighbourTableSize":14,"Bind":{"01":{"1000":{"Target":"0000","Stamp":1669203443,"Phase":"binded","Status":"00","i_sqn":560},"0000":{"Target":"0000","Stamp":1669203441,"Phase":"binded","Status":"00","i_sqn":552},"0003":{"Target":"0000","Stamp":1669203442,"Phase":"binded","Status":"00","i_sqn":553},"0004":{"Target":"0000","Stamp":1669203442,"Phase":"binded","Status":"00","i_sqn":554},"0005":{"Target":"0000","Stamp":1669203442,"Phase":"binded","Status":"00","i_sqn":555},"0006":{"Target":"0000","Stamp":1669203442,"Phase":"binded","Status":"00","i_sqn":556},"0008":{"Target":"0000","Stamp":1669203442,"Phase":"binded","Status":"00","i_sqn":557},"0300":{"Target":"0000","Stamp":1669203443,"Phase":"binded","Status":"00","i_sqn":558},"0b05":{"Target":"0000","Stamp":1669203443,"Phase":"binded","Status":"00","i_sqn":559},"fc7c":{"Target":"0000","Stamp":1669203443,"Phase":"binded","Status":"00","i_sqn":561}}},"GroupMemberShip":{"01":{}},"Epv2":{"01":{"ClusterIn":{"1000":{},"0000":{},"0003":{},"0004":{},"0005":{},"0006":{},"0008":{},"0300":{},"0b05":{},"fc7c":{}},"ClusterOut":{"0019":{}},"ProfileID":"0104","ZDeviceID":"010d"},"f2":{"ClusterIn":{},"ClusterOut":{"0021":{}},"ProfileID":"a1e0","ZDeviceID":"0061"}},"ReadConfigureReporting":{"Ep":{"01":{"0006":{"0000":{"TimeStamp":1669203445.9150283,"Status":"00","DataType":"10","MinInterval":"0001","MaxInterval":"012c"}},"0008":{"0000":{"TimeStamp":1669203446.0288138,"Status":"00","DataType":"20","MinInterval":"0005","MaxInterval":"012c"}},"0300":{"0008":{"TimeStamp":1669203446.1433063,"Status":"00","DataType":"30","MinInterval":"0001","MaxInterval":"012c"}}}},"TimeStamp":1669203446.1432898},"ConfigureReporting":{"Ep":{"01":{"0006":{"TimeStamp":1669203440.293883,"iSQN":{"0000":563},"Attributes":{"0000":"00"},"ZigateRequest":{}},"0008":{"TimeStamp":1669203440.320347,"iSQN":{"0000":564},"Attributes":{"0000":"00"},"ZigateRequest":{}},"0300":{"TimeStamp":1669203440.3418834,"iSQN":{"0008":565},"Attributes":{"0008":"00"},"ZigateRequest":{}}}}},"BindingTable":{"SQN":0,"Status":"Success","TimeStamp":1669203446.423552,"BindingTableSize":10,"Devices":[{"003c84fffe1a696f":{"sourceEp":"01","Cluster":"0000","targetIEEE":"00158d0005be903c","targetNickName":"Zigate","targetEp":"01"}},{"003c84fffe1a696f":{"sourceEp":"01","Cluster":"0003","targetIEEE":"00158d0005be903c","targetNickName":"Zigate","targetEp":"01"}},{"003c84fffe1a696f":{"sourceEp":"01","Cluster":"0004","targetIEEE":"00158d0005be903c","targetNickName":"Zigate","targetEp":"01"}},{"003c84fffe1a696f":{"sourceEp":"01","Cluster":"0005","targetIEEE":"00158d0005be903c","targetNickName":"Zigate","targetEp":"01"}},{"003c84fffe1a696f":{"sourceEp":"01","Cluster":"0006","targetIEEE":"00158d0005be903c","targetNickName":"Zigate","targetEp":"01"}},{"003c84fffe1a696f":{"sourceEp":"01","Cluster":"0008","targetIEEE":"00158d0005be903c","targetNickName":"Zigate","targetEp":"01"}},{"003c84fffe1a696f":{"sourceEp":"01","Cluster":"0300","targetIEEE":"00158d0005be903c","targetNickName":"Zigate","targetEp":"01"}},{"003c84fffe1a696f":{"sourceEp":"01","Cluster":"0b05","targetIEEE":"00158d0005be903c","targetNickName":"Zigate", [... cut ...] mitjnt":"??"}},{"592a":{"_relationshp":"Child","_devicetype":"End Device","_depth":2,"_lnkqty":146,"_rxonwhenidl":"Rx-Off","_IEEE":"00124b002515cd70","_permitjnt":"Off"}}]}],"ConfigSource":"8043","SWBUILD_1":"2019062886LEL08N","SWBUILD_3":"1.0.021","ColorInfos":{"Hue":26,"Saturation":182,"X":"28513","Y":"26452","ColorTemperatureMireds":"335","ColorMode":2,"ColorCapabilities":"31"},"ConsistencyCheck":"ok","RollingLQI":[56,64,59],"CertifiedDevice":false,"Log_UnknowDeviceFlag":1674591115.0566273}}`

pipiche38, 2023-01-24:
The change has been done, and will be ready for the next release.

if you want to test, in between

```
git remote add pipiche https://github.com/pipiche38/Domoticz-Zigbee.git
git pull pipiche optimized-device/6.3.010
git checkout optimized-device/6.3.010
```

the  restart the plugin

to fall back

```
git checkout stable6
git status
git remote remove pipiche
```

Please kindly consider enabling the analytics part. This will help us to know who is using the plugin and having more insights to define our priorities.

closed 2023-02-01 by pipiche38

Chaussette23, 2023-02-12:
I confirm it now have a green check mark !