zigbeefordomoticz/Domoticz-Zigbee #1484: SUNRICHER - Télécommande 4 zones Zigbee 3.0

mamo79, 2023-01-25:
Hello,

Is-it possible to use this commande to open and close covers ?

1. Please do provide device information such as Brand and Model name.

SUNRICHER - Télécommande 4 zones Zigbee 3.0

1. Provide the URL where you have purchase the device

https://www.domadoo.fr/fr/peripheriques/5246-sunricher-telecommande-4-zones-zigbee-30.html

1. Finaly copy-paste all information provided in the log
 
{"PluginInfos":{"HardwareID":30,"HomeFolder":"/config/plugins/Domoticz-Zigate/","StartupFolder":"/usr/share/domoticz/","UserDataFolder":"/config/","WebRoot":"/","Database":"/config/domoticz.db","Language":"fr","Version":"6.4","Author":"pipiche38","Name":"Zigate","Address":"192.168.44.136","Port":"9999","SerialPort":"/dev/ttyUSB0","Username":"","Password":"","Key":"Zigate","Mode1":"V2","Mode2":"Wifi","Mode3":"False","Mode4":"9440","Mode5":"http://192.168.44.12:8080","Mode6":"2","DomoticzVersion":"2022.2","DomoticzHash":"eea9db734","DomoticzBuildTime":"2022-11-05 13:05:35","PluginBranch":"stable6","PluginVersion":"6.3.009","TimeStamp":1674680816,"available":"6.3.009","available-firmMajor":"05","available-firmMinor":"0321","FirmwareVersion":"0321","FirmwareUpdate":false,"PluginUpdate":false,"LogLevel":4,"Zigpy":false,"CoordinatorModel":"ZiGate+ (V2)","CoordinatorFirmwareVersion":"0321","DisplayFirmwareVersion":"Zigate - 0321","CoordinatorIEEE":"00158d0004ff96dc","NetworkSize":"Total: 14 | Routers: 7 | End Devices: 7","NetworkDevices":{"1037":{"LUMI":["lumi.weather","lumi.plug.maeu01"],"LiXee":["ZLinky_TIC-standard-mono"]},"1224":{"Sunricher":["ZGRC-KEY-013"]},"110c":{"OSRAM":["Plug 01"]},"128b":{"NodOn":["SIN-4-RS-20"]}}},"Analytics":1,"Device":{"Version":"3","ZDeviceName":"","Status":"inDB","SQN":"3a","Ep":{"01":{"0001":{"0021":200,"0020":33,"0033":0},"0000":{"4000":"2.7.6_r25","0004":"Sunricher","0005":"ZGRC-KEY-013","0000":"3","0001":"0","0002":"0","0003":"0","0006":"NULL","0007":"3","000a":"3030"},"ClusterType":{"1258":"Voltage"}},"02":{"1000":{},"0001":{"0021":200,"0020":33,"0033":0},"0000":{"4000":"2.7.6_r25","0004":"Sunricher","0005":"ZGRC-KEY-013","0000":"3","0001":"0","0002":"0","0003":"0","0006":"NULL","0007":"3","000a":"3030"},"0003":{},"0b05":{},"0004":{},"0005":{},"0006":{},"0008":{},"0019":{},"0300":{},"ClusterType":{"1259":"Voltage","1260":"ColorControlFull"}},"03":{"0000":{"0004":"Sunricher","0005":"ZGRC-KEY-013"}},"04":{"0000":{"0004":"Sunricher","0005":"ZGRC-KEY-013"}}},"Heartbeat":"19427","RIA":"4","LQI":117,"Battery":200,"Model":"ZGRC-KEY-013","ForceAckCommands":[],"MacCapa":"80","IEEE":"04cd15fffe901d21","Type":"Voltage/Switch/LvlControl/ColorControl","ProfileID":"0104","ZDeviceID":"0001","App Version":"0","Attributes List":{},"DeviceType":"RFD","HW Version":"0","Last Cmds":[[51,"02","6f1c","0100","010200190000000003000100020003"],[52,"02","6f1c","0100","010200190000000003000400050006"],[53,"02","6f1c","0100","010200190000000003000700080009"],[54,"02","6f1c","0100","010200190000000001000a"],[55,"02","6f1c","0100","010200060000000003000040014002"],[56,"02","6f1c","0100","0102000800000000010000"],[57,"02","6f1c","0100","010203000000000003000000010003"],[58,"02","6f1c","0100","010203000000000003000400070008"],[59,"02","6f1c","0100","010203000000000001400a"],[60,"07","6f1c","00E0","6f1c01020000"]],"LogicalType":"End Device","Manufacturer":"1224","Manufacturer Name":"Sunricher","NbEp":"4","PowerSource":"Battery","ReadAttributes":{"Ep":{"01":{"0001":{"TimeStamp":1674670371,"iSQN":{"0000":35,"0001":35,"0003":35,"0020":36,"0021":36,"0033":36,"0035":37},"Attributes":{"0000":"86","0001":"86","0003":"86","0020":"00","0021":"00","0033":"00","0035":"86"},"ZigateRequest":{}},"0000":{"TimeStamp":1674583702,"iSQN":{"4000":45,"0004":41,"0005":41,"0000":41,"0001":42,"0002":42,"0003":42,"0006":43,"0007":43,"000a":43,"000f":44,"0010":44,"0015":44,"f000":45},"Attributes":{"4000":"00","0004":"00","0005":"00","0000":"00","0001":"00","0002":"00","0003":"00","0006":"00","0007":"00","000a":"00","000f":"86","0010":"86","0015":"86","f000":"86"},"ZigateRequest":{}}},"02":{"0000":{"TimeStamp":1674583708,"iSQN":{"4000":50,"0004":46,"0005":46,"0000":46,"0001":47,"0002":47,"0003":47,"0006":48,"0007":48,"000a":48,"000f":49,"0010":49,"0015":49,"f000":50},"Attributes":{"4000":"00","0004":"00","0005":"00","0000":"00","0001":"00","0002":"00","0003":"00","0006":"00","0007":"00","000a":"00","000f":"86","0010":"86","0015":"86","f000":"86"},"ZigateRequest":{}},"0001":{"TimeStamp":1674659531,"iSQN":{"0000":38,"0001":38,"0003":38,"0020":39,"0021":39,"0033":39,"0035":40},"Attributes":{"0000":"86","0001":"86","0003":"86","0020":"00","0021":"00","0033":"00","0035":"86"},"ZigateRequest":{}},"0019":{"TimeStamp":1674583712,"iSQN":{"0001":51,"0002":51,"0003":51,"0004":52,"0005":52,"0006":52,"0007":53,"0008":53,"0009":53,"000a":54},"Attributes":{"0001":"86","0002":"86","0003":"86","0004":"86","0005":"86","0006":"86","0007":"86","0008":"86","0009":"86","000a":"86"},"ZigateRequest":{}},"0006":{"TimeStamp":1674583713,"iSQN":{"4001":55,"4002":55,"0000":55},"Attributes":{"4001":"86","4002":"86","0000":"86"},"ZigateRequest":{}},"0008":{"TimeStamp":1674583714,"iSQN":{"0000":56},"Attributes":{"0000":"86"},"ZigateRequest":{}},"0300":{"TimeStamp":1674583718,"iSQN":{"0000":57,"0001":57,"0003":57,"0004":58,"0007":58,"0008":58,"400a":59},"Attributes":{"0000":"86","0001":"86","0003":"86","0004":"86","0007":"86","0008":"86","400a":"86"},"ZigateRequest":{}}},"03":{"0000":{"TimeStamp":1674583664,"iSQN":{"0004":20,"0005":20},"Attributes":{"0004":"00","0005":"00"},"ZigateRequest":{}}},"04":{"0000":{"TimeStamp":1674583665,"iSQN":{"0004":21,"0005":21},"Attributes":{"0004":"00","0005":"00"},"ZigateRequest":{}}}}},"ReceiveOnIdle":"Off","Stack Version":"0","Stamp":{"time":1674678454.8953342,"Time":"2023-01-25 21:27:34","MsgType":"8011","LastSeen":1674678454},"ZCL Version":"3","Health":"Live","Capability":["Reduced-Function Device","Standard security","NwkAddr should be allocated"],"ZDPSQN":"08","RollingLQI":[119,120,120,120,119,120,117,117,117,118,117],"BatteryUpdateTime":1674670371,"WriteAttributes":{"Ep":{"01":{"0001":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"0000":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}}},"02":{"1000":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"0001":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"0000":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"0003":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"0b05":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"0004":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"0005":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"0006":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"0008":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"0019":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"0300":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}}},"03":{"0000":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}}},"04":{"0000":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}}}}},"Epv2":{"01":{"ClusterIn":{},"ClusterOut":{},"ProfileID":{},"ZDeviceID":{}},"02":{"ClusterIn":{"1000":{},"0000":{},"0001":{},"0003":{},"0b05":{}},"ClusterOut":{"1000":{},"0003":{},"0004":{},"0005":{},"0006":{},"0008":{},"0019":{},"0300":{}},"ProfileID":"0104","ZDeviceID":"0001"},"03":{"ClusterIn":{},"ClusterOut":{},"ProfileID":{},"ZDeviceID":{}},"04":{"ClusterIn":{},"ClusterOut":{},"ProfileID":{},"ZDeviceID":{}}},"PairingInProgress":false,"ZDeviceVersion":"0001","ConfigSource":"8043","_rawNodeDescriptor":"1224005200522c000080524002","Max Buffer Size":"52","Max Rx":"0052","Max Tx":"0052","macapa":"80","bitfield":"4002","server_mask":"2c00","descriptor_capability":"00","PairingTime":1674583678.0982347,"Bind":{"02":{"1000":{"Target":"0000","Stamp":1674583684,"Phase":"binded","Status":"00","i_sqn":28},"0000":{"Target":"0000","Stamp":1674583679,"Phase":"binded","Status":"00","i_sqn":24},"0001":{"Target":"0000","Stamp":1674583681,"Phase":"binded","Status":"00","i_sqn":25},"0003":{"Target":"0000","Stamp":1674583682,"Phase":"binded","Status":"00","i_sqn":26},"0b05":{"Target":"0000","Stamp":1674583683,"Phase":"binded","Status":"00","i_sqn":27}}},"BindingTable":{"SQN":0,"Status":"Success","TimeStamp":1674583721.3347702,"BindingTableSize":10,"Devices":[{"04cd15fffe901d21":{"sourceEp":"01","Cluster":"0000","targetIEEE":"00158d0004ff96dc","targetNickName":"Zigate","targetEp":"01"}},{"04cd1 [... cut ...] ":{"0000":31,"0020":31,"0021":31},"Attributes":{"0000":"86","0020":"8d","0021":"8d"},"ZigateRequest":{}},"0006":{"TimeStamp":1674583678.1068847,"iSQN":{"0000":32},"Attributes":{"0000":"86"},"ZigateRequest":{}},"0008":{"TimeStamp":1674583678.1076589,"iSQN":{"0000":33},"Attributes":{"0000":"86"},"ZigateRequest":{}},"0300":{"TimeStamp":1674583678.1086385,"iSQN":{"0008":34},"Attributes":{"0008":"86"},"ZigateRequest":{}}}}},"CertifiedDevice":false,"SWBUILD_1":"NULL","SWBUILD_3":"2.7.6_r25"}}

Thanks.

pipiche38, 2023-01-26:
If the question is could I bind the remote with Blind/covers, the answer is no, as this remote provide Cluster 0x0006, 0x0008 and 0x0300 so would most-likely manage On/Off, Level Control and Color Control.

What could be done, is an integration in the plugin, where when doing action on the remote we report those actions on a Switch Selector in Domoticz and then you can do some Lua/dzVents scripting.

For the integration, i would need more information such as:

enable debug "Inputs", "Cluster" from the WebUI, and then get the logs generated by pressing each of the remote button.
The log file is available under Logs/PluginZigbee-xx.log

pipiche38, 2023-02-04:
Not able to do anything with the logs, as I didn't see any messages coming from the remote except the Battery information.
Yes this remote does work with non-Zigate coordinator.

Zigate has many limitation and that one is one of the biggest one.

You might want to enable via the WebUI > Settings

* enable Groups
* Create Group 0000

mamo79, 2023-02-10:
Hello,

I enabled Groups and create group 0000 but I did'nt recieve anything from the telecommande in domoticz and nothing in log of pluggin.

[image]

Thanks for your help.

pipiche38, 2023-02-10:
I'm trying to contact Sunricher, but without the device, I cannot do much.
The main issue is the Zigate which is preventing to see the action generated by the Remote and we need to identify on which Zigbee group id the remote is acting.

mamo79, 2023-02-11:
Hello, it's really nice to look more on my problem. If you go through Nantes I could lend you the remote control 😉.
I find this :
https://easydomoticz.com/forum/viewtopic.php?t=11778&sid=7f0a1c95ad3cc57eb55562f5b8bfd76b

pipiche38, 2023-05-19:
There hasn't been any activity on this issue recently. Due to the high number of incoming GitHub notifications, we have to clean some of the old issues, as many of them have already been resolved with the latest updates. Please make sure to update to the latest version and check if that solves the issue. Let us know if that works for you by adding a comment 👍 This issue has now been marked as stale and will be closed if no further activity occurs. Thank you for your contributions.

closed 2023-05-19 by pipiche38