zigbeefordomoticz/Domoticz-Zigbee #1485: SonOff SNZB-02D Non optimized plugin

IW2FAG-11, 2023-01-26:
[export.zip](https://github.com/zigbeefordomoticz/Domoticz-Zigbee/files/10511712/export.zip)
[image]

1. Got to the Tools Menu
1. Select Plugin Raw Devices (json)
1. Export

This will create an export.json file that you can save.
   
Finally you create an issue on GitHub with the following details

* Product Name 
* Brand
* URL on a web site
* What do you expect
* The logs from the pairing process
* export.json file as described here above

pipiche38, 2023-01-26:
The change has been done, and will be ready for the next release.

if you want to test, in between

```
git remote add pipiche https://github.com/pipiche38/Domoticz-Zigbee.git
git pull pipiche optimized-device/6.3.010
git checkout optimized-device/6.3.010
```

the  restart the plugin

to fall back

```
git checkout stable6
git status
git remote remove pipiche
```

IW2FAG-11, 2023-01-26:
Thanks. 
Is ok, now is working.

closed 2023-01-26 by IW2FAG-11