zigbeefordomoticz/Domoticz-Zigbee #1486: SmartAir TS0601 | _TZE200_dwcarsat

jon-wrennall, 2023-01-26:
[SmartAir TS0601](https://zigbee.blakadder.com/Tuya_DCR-KQG.html) 

Non Optimized Device Export:
[{"Version":"3","ZDeviceName":"","Status":"CreateDB","SQN":"c3","Ep":{"01":{"0000":{"0004":"_TZE200_dwcarsat","0000":"3","0001":"70","0005":"TS0601","0007":"1","fffe":"0","ffe2":"54","ffe4":"0","ffdf":"oe+ipe+iqe+iqe+ide+ide+iee+iee+i"}},"f2":{}},"Heartbeat":"591","RIA":"6","LQI":75,"Battery":{},"Model":"TS0601","ForceAckCommands":[],"MacCapa":"8e","IEEE":"a4c138b669ef1e73","Type":{},"ProfileID":{},"ZDeviceID":{},"App Version":"70","Attributes List":{},"DeviceType":"FFD","HW Version":"","Last Cmds":[],"LogicalType":"Router","Manufacturer":"1141","Manufacturer Name":"_TZE200_dwcarsat","NbEp":"2","PowerSource":"Main","ReadAttributes":{"Ep":{"01":{"0000":{"TimeStamp":"26/01/2023 20:46:07","iSQN":{},"Attributes":{"0001":"00","ffe2":"00","ffe4":"00","ffdf":"00"},"ZigateRequest":{}}}}},"ReceiveOnIdle":"On","Stack Version":"","Stamp":{"time":1674765987.3478434,"Time":"2023-01-26 20:46:27","MsgType":"8011","LastSeen":"26/01/2023 20:46:27"},"ZCL Version":"3","Health":"Live","_rawNodeDescriptor":"1141004200422a00008e424001","Max Buffer Size":"42","macapa":"8e","descriptor_capability":"00","Max Rx":"0042","server_mask":"2a00","Capability":["Full-Function Device","Main Powered","Receiver during Idle","Standard security","NwkAddr should be allocated"],"bitfield":"4001","Max Tx":"0042","Epv2":{"01":{"ClusterIn":{},"ClusterOut":{},"ProfileID":{},"ZDeviceID":{}},"f2":{"ClusterIn":{},"ClusterOut":{},"ProfileID":{},"ZDeviceID":{}}},"ConsistencyCheck":"","ZCL-IN-SQN":{"ef00":"c3","0000":"b3"},"RollingLQI":[75,75,75,75,75,72,72,72,75,75,75],"WriteAttributes":{"Ep":{"01":{"0000":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}}}}},"CertifiedDevice":false,"Log_UnknowDeviceFlag":1674764802.7196484,"NwkID":"7bb3"}]

pipiche38, 2023-01-26:
Can you kindly check on my dev. You need to remove what you have been paired so far. Make sure that nothing left

```
git remote add pipiche https://github.com/pipiche38/Domoticz-Zigbee.git
git pull pipiche optimized-device/6.3.010
git checkout optimized-device/6.3.010
```

the  restart the plugin
and then check again, that the device is not there and do the pairing

to fall back

```
git checkout stable6
git status
git remote remove pipiche
```

Please kindly consider enabling the analytics part. This will help us to know who is using the plugin and having more insights to define our priorities.

jon-wrennall, 2023-01-27:
Dev version works - thanks ! (Though noticed version number didn’t change - assume you only update on release)

pipiche38, 2023-01-27:
yes, that is correct.
I'll release on stable in the first weeks of February.

jon-wrennall, 2023-02-01:
Having properly checked at least one of the parameters returned is not there and one not identified correctly 

There should be 6 - as can be seen in [https://www.zigbee2mqtt.io/devices/TS0601_smart_air_house_keeper.html](https://www.zigbee2mqtt.io/devices/TS0601_smart_air_house_keeper.html)

Currently 6 widgets are created - one which has humidity and temp, and then one each.  The one with Humidity and temp (the 6th) should be pm25 (which currently looks like it is missing).

PS I switched to stable but got 
“ 
Switched to branch 'stable6'
Your branch is ahead of 'origin/stable6' by 16 commits.
  (use "git push" to publish your local commits)
”
So I’ve stayed on your dev for now

pipiche38, 2023-02-01:
probably you need to do
```
git reset --hard
git checkout stable6
git pull
```

Let's re-open for the pm25

pipiche38, 2023-03-04:
If you can switch to 'develop' branch

```
git checkout develop
git pull
sudo python3 -m pip install -r requirements.txt --upgrade
```

Then , the best would be to re-pair it and let see how it works

jon-wrennall, 2023-03-07:
Sorry for the noob esq q - but must have a dependency issue - even though have Python 3.9.15…
??
***@***.***:~/domoticz/plugins/Domoticz-Zigbee $ git pull
Already up to date.
***@***.***:~/domoticz/plugins/Domoticz-Zigbee $ sudo python3 -m pip install -r requirements.txt --upgrade
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
ERROR: Ignored the following versions that require a different python version: 0.47.3 Requires-Python >=3.8; 0.48.0 Requires-Python >=3.8; 0.49.0 Requires-Python >=3.8; 0.49.1 Requires-Python >=3.8; 0.50.0 Requires-Python >=3.8; 0.52.0 Requires-Python >=3.8; 0.52.1 Requires-Python >=3.8; 0.52.2 Requires-Python >=3.8; 0.52.3 Requires-Python >=3.8; 0.53.0 Requires-Python >=3.8; 0.53.1 Requires-Python >=3.8; 0.53.2 Requires-Python >=3.8
ERROR: Could not find a version that satisfies the requirement zigpy==0.53.2 (from versions: 0.0.1, 0.0.2, 0.0.3, 0.1.0, 0.2.0, 0.20.0a1, 0.20a2, 0.20.0, 0.20.1a1, 0.20.1a2, 0.20.1a3, 0.20.1, 0.20.2, 0.20.3, 0.20.4, 0.21.0, 0.22.0, 0.22.1, 0.22.2, 0.23.0, 0.23.1, 0.23.2, 0.24.0, 0.24.1, 0.24.2, 0.24.3, 0.25.0, 0.26.0, 0.27.0, 0.27.1, 0.28.0, 0.28.1, 0.28.2, 0.29.0, 0.30.0, 0.30.1, 0.31.0, 0.32.0, 0.33.0, 0.34.0, 0.35.0, 0.35.1, 0.35.2, 0.36.0, 0.36.1, 0.36.2, 0.37.0, 0.37.1, 0.37.2, 0.38.0, 0.39.0, 0.40.0, 0.41.0, 0.42.0, 0.43.0, 0.44.1, 0.44.2, 0.45.0.dev0, 0.45.0, 0.45.1, 0.46.0, 0.47.0, 0.47.1, 0.47.2, 0.50.1, 0.50.2, 0.50.3, 0.51.0, 0.51.1, 0.51.2, 0.51.3, 0.51.4, 0.51.5, 0.51.6)
ERROR: No matching distribution found for zigpy==0.53.2

[notice] A new release of pip available: 22.3.1 -> 23.0.1
[notice] To update, run: python3 -m pip install --upgrade pip
***@***.***:~/domoticz/plugins/Domoticz-Zigbee $ python --version
Python 3.9.15

pipiche38, 2023-04-03:
@jon-wrennall any update ?

closed 2023-06-09 by github-actions[bot]