zigbeefordomoticz/Domoticz-Zigbee #1487: Bug in 0x0702 Summation and Instant Power calculation

pipiche38, 2023-01-26:
The code here 
https://github.com/zigbeefordomoticz/Domoticz-Zigbee/blob/878480424f4d031787959029e335a9852e26836e/Modules/readClusters.py#L3557

Doesn't reflect what is in the Zigbee norm

10.4.2.2.4.2 Multiplier Attribute 0x0301
Multiplier provides a value to be multiplied against a raw or uncompensated sensor count of Energy, Gas, or Water being measured by the metering device. If present, this attribute must be applied against all sum- mation, consumption and demand values to derive the delivered and received values expressed in the unit of measure specified. This attribute must be used in conjunction with the Divisor attribute.

10.4.2.2.4.3 Divisor Attribute 0x0302
Divisor provides a value to divide the results of applying the Multiplier Attribute against a raw or uncom- pensated sensor count of Energy, Gas, or Water being measured by the metering device. If present, this attribute must be applied against all summation, consumption and demand values to derive the delivered and received values expressed in the unit of measure specified. This attribute must be used in conjunction with the Multiplier attribute.

So the code should be more like that

```
def compute_conso(self, MsgSrcAddr, MsgSrcEp, MsgClusterId, MsgAttrID, raw_value):

    conso = raw_value  # Raw value
    
    multiplier = (
        self.ListOfDevices[MsgSrcAddr]["Ep"][MsgSrcEp][MsgClusterId]["0301"] 
            if (
                MsgSrcEp in self.ListOfDevices[MsgSrcAddr]["Ep"] 
                and MsgClusterId in self.ListOfDevices[MsgSrcAddr]["Ep"][MsgSrcEp] 
                and "0301" in self.ListOfDevices[MsgSrcAddr]["Ep"][MsgSrcEp][MsgClusterId])
            else 1
    )
    divisor = (
        self.ListOfDevices[MsgSrcAddr]["Ep"][MsgSrcEp][MsgClusterId]["0302"] 
            if (
                MsgSrcEp in self.ListOfDevices[MsgSrcAddr]["Ep"] 
                and MsgClusterId in self.ListOfDevices[MsgSrcAddr]["Ep"][MsgSrcEp] 
                and "0302" in self.ListOfDevices[MsgSrcAddr]["Ep"][MsgSrcEp][MsgClusterId] 
            )
            else 1
    )

    return round( conso * multiplier / divisor )
```

pipiche38, 2023-01-26:
@SylvainPer @badzz  feel free to comment. Would be good if you could cross-check with

- [ ] BMS
- [ ] HACT
- [ ] LMACT

zigbeefordomoticz, 2023-01-27:
Provides Overwrite device configuration attributs
* For Instant Power
  * Device Configuration `PowerMeteringMultiplier` can overwrite the Multiplier
  * Device Configuration `PowerMeteringDivisor` can overwrite the Divisor
* For Summation 
  * Device Configuration `SummationMeteringMultiplier` can overwrite the Multiplier
  * Device Configuration `SummationMeteringDivisor` can overwrite the Divisor

pipiche38, 2023-01-28:
Patrick tested
- [ ] Innr SP 120
- [ ] Tuya TS011F Plug
- [ ]

closed 2023-01-29 by pipiche38