zigbeefordomoticz/Domoticz-Zigbee #1503: Aqara Smart Wall Switch H1 EU No neutral, double rocker WS-EUK02

lemonacid, 2023-01-30:
*  Smart Wall Switch H1 EU No neutral, double rocker WS-EUK02
* Aqara
* https://www.aqara.com/eu/product/smart-wall-switch-h1-eu-no-neutral
* What do you expect - two switches, both  left and right, are working. Currently only one (left) monitored and controlled.
* The logs from the pairing process

2023-01-28 17:05:59,674 INFO    : [   MainThread 541] Request Accepting new Hardware for 240 seconds
2023-01-28 17:05:59,750 INFO    : [   ZigpyCom_2 683] PERMIT-TO-JOIN: {'cmd': 'PERMIT-TO-JOIN', 'datas': {'Duration': 240, 'targetRouter': 'FFFC'}, 'NwkId': None, 'TimeStamp': 1674914759.6658692, 'ACKIsDisable': False, 'Sqn': None}
2023-01-28 17:05:59,753 INFO    : [   ZigpyCom_2 683] PERMIT-TO-JOIN: duration: 240 for Radio: znp for node: None
2023-01-28 17:05:59,770 INFO    : [   ZigpyCom_2 683] Calling self.app.permit(time_s=240, node=None )
2023-01-28 17:05:59,904 INFO    : [   ZigpyCom_2 683] returning from the self.app.permit(time_s=240, node=None )
2023-01-28 17:06:30,878 INFO    : [ZigpyForwarder_2 684] --> Adding device 5b44 in self.DevicesInPairingMode
2023-01-28 17:06:30,894 INFO    : [ZigpyForwarder_2 684] --> ['5b44']
2023-01-28 17:06:30,941 INFO    : [ZigpyForwarder_2 684] [None] NEW OBJECT: 5b44 None
2023-01-28 17:06:31,479 INFO    : [ZigpyForwarder_2 684] Decode8046 - Match Descriptor response: SQN: 00 Status: 00 Nwkid: 445b Lenght: 00 List:
2023-01-28 17:06:31,582 INFO    : [ZigpyForwarder_2 684] [-] NEW OBJECT: 5b44 Active Endpoint Response Ep: 01 LQI: 0
2023-01-28 17:06:31,594 INFO    : [ZigpyForwarder_2 684] [-] NEW OBJECT: 5b44 Active Endpoint Response Ep: 02 LQI: 0
2023-01-28 17:06:31,610 INFO    : [ZigpyForwarder_2 684] [-] NEW OBJECT: 5b44 Active Endpoint Response Ep: f2 LQI: 0
2023-01-28 17:06:31,629 INFO    : [ZigpyForwarder_2 684] --> Not full scope
2023-01-28 17:06:31,645 INFO    : [ZigpyForwarder_2 684] --> Build list of Attributes
2023-01-28 17:06:31,665 INFO    : [ZigpyForwarder_2 684] --> ListOfEp with 0x0000: []
2023-01-28 17:06:31,714 INFO    : [ZigpyForwarder_2 684] --> Build list Eps for Cluster Basic ['01', '02', 'f2']
2023-01-28 17:06:31,749 INFO    : [ZigpyForwarder_2 684] Request Basic  Manufacturer via Read Attribute request: 0004
2023-01-28 17:06:31,783 INFO    : [ZigpyForwarder_2 684] Request Basic  Model Name via Read Attribute request: 0005
2023-01-28 17:06:31,785 INFO    : [ZigpyForwarder_2 684] EP: {'01': {}, '02': {}, 'f2': {}} Attributes: [4, 5]
2023-01-28 17:06:31,841 INFO    : [ZigpyForwarder_2 684] [-] NEW OBJECT: 5b44 Request Simple Descriptor for Ep: 01
2023-01-28 17:06:32,231 INFO    : [ZigpyForwarder_2 684] check_found_plugin_model - lumi.switch.l2aeu1 LUMI  None
2023-01-28 17:06:32,386 INFO    : [ZigpyForwarder_2 684] check_found_plugin_model - lumi.switch.l2aeu1 LUMI  None
2023-01-28 17:06:32,620 INFO    : [ZigpyForwarder_2 684] [-] NEW OBJECT: 5b44 Simple Descriptor Response EP: 0x01 LQI: 0
2023-01-28 17:06:32,631 INFO    : [ZigpyForwarder_2 684] [-]    NEW OBJECT: 5b44 ProfileID 0104
2023-01-28 17:06:32,639 INFO    : [ZigpyForwarder_2 684] [-]    NEW OBJECT: 5b44 ZDeviceID 0100
2023-01-28 17:06:32,646 INFO    : [ZigpyForwarder_2 684] [-]    NEW OBJECT: 5b44 Application Version 0001
2023-01-28 17:06:32,651 INFO    : [ZigpyForwarder_2 684] [-]    NEW OBJECT: 5b44 Cluster IN Count: 07
2023-01-28 17:06:32,657 INFO    : [ZigpyForwarder_2 684] [-]       NEW OBJECT: 5b44 Cluster In 1: 0000 (Basic)
2023-01-28 17:06:32,664 INFO    : [ZigpyForwarder_2 684] [-]       NEW OBJECT: 5b44 Cluster In 2: 0002
2023-01-28 17:06:32,669 INFO    : [ZigpyForwarder_2 684] [-]       NEW OBJECT: 5b44 Cluster In 3: 0003 (Identify)
2023-01-28 17:06:32,675 INFO    : [ZigpyForwarder_2 684] [-]       NEW OBJECT: 5b44 Cluster In 4: 0004 (Groups)
2023-01-28 17:06:32,677 INFO    : [ZigpyForwarder_2 684] [-]       NEW OBJECT: 5b44 Cluster In 5: 0005 (Scenes)
2023-01-28 17:06:32,680 INFO    : [ZigpyForwarder_2 684] [-]       NEW OBJECT: 5b44 Cluster In 6: 0006 (On/Off)
2023-01-28 17:06:32,682 INFO    : [ZigpyForwarder_2 684] [-]       NEW OBJECT: 5b44 Cluster In 7: 0009 (Alarms)
2023-01-28 17:06:32,685 INFO    : [ZigpyForwarder_2 684] [-]    NEW OBJECT: 5b44 Cluster OUT Count: 02
2023-01-28 17:06:32,687 INFO    : [ZigpyForwarder_2 684] [-]       NEW OBJECT: 5b44 Cluster Out 1: 000a (Time)
2023-01-28 17:06:32,690 INFO    : [ZigpyForwarder_2 684] [-]       NEW OBJECT: 5b44 Cluster Out 2: 0019 (Over-the-Air Upgrade)
2023-01-28 17:06:36,128 INFO    : [   MainThread 541] [1] NEW OBJECT: 5b44 Model Name: lumi.switch.l2aeu1
2023-01-28 17:06:36,143 INFO    : [   MainThread 541] [1] NEW OBJECT: 5b44 Manufacturer Name: LUMI
2023-01-28 17:06:36,169 INFO    : [   MainThread 541] [None] NEW OBJECT: 5b44 Request Node Descriptor
2023-01-28 17:06:41,098 INFO    : [   MainThread 541] [2] NEW OBJECT: 5b44 Model Name: lumi.switch.l2aeu1
2023-01-28 17:06:41,106 INFO    : [   MainThread 541] [2] NEW OBJECT: 5b44 Manufacturer Name: LUMI
2023-01-28 17:06:46,076 INFO    : [   MainThread 541] [3] NEW OBJECT: 5b44 Model Name: lumi.switch.l2aeu1
2023-01-28 17:06:46,141 INFO    : [   MainThread 541] [3] NEW OBJECT: 5b44 Manufacturer Name: LUMI
2023-01-28 17:06:47,297 ERROR   : [       MainThread] zcl_attribute_discovery_request not implemented for RAW mode
2023-01-28 17:06:47,393 ERROR   : [       MainThread] zcl_attribute_discovery_request not implemented for RAW mode
2023-01-28 17:06:47,469 ERROR   : [       MainThread] zcl_attribute_discovery_request not implemented for RAW mode
2023-01-28 17:06:47,587 ERROR   : [       MainThread] zcl_attribute_discovery_request not implemented for RAW mode
2023-01-28 17:06:47,660 ERROR   : [       MainThread] zcl_attribute_discovery_request not implemented for RAW mode
2023-01-28 17:06:47,748 ERROR   : [       MainThread] zcl_attribute_discovery_request not implemented for RAW mode
2023-01-28 17:06:47,804 ERROR   : [       MainThread] zcl_attribute_discovery_request not implemented for RAW mode
2023-01-28 17:06:47,882 ERROR   : [       MainThread] zcl_attribute_discovery_request not implemented for RAW mode
2023-01-28 17:06:48,065 ERROR   : [       MainThread] zcl_attribute_discovery_request not implemented for RAW mode
2023-01-28 17:06:48,148 ERROR   : [       MainThread] zcl_attribute_discovery_request not implemented for RAW mode
2023-01-28 17:06:51,240 INFO    : [ZigpyForwarder_2 684] check_found_plugin_model - lumi.switch.l2aeu1 LUMI 115f 0100
2023-01-28 17:09:38,153 INFO    : [   MainThread 541] Request Disabling Accepting new Hardware
2023-01-28 17:09:38,204 INFO    : [   ZigpyCom_2 683] PERMIT-TO-JOIN: {'cmd': 'PERMIT-TO-JOIN', 'datas': {'Duration': 0, 'targetRouter': 'FFFC'}, 'NwkId': None, 'TimeStamp': 1674914978.1631665, 'ACKIsDisable': False, 'Sqn': None}
2023-01-28 17:09:38,225 INFO    : [   ZigpyCom_2 683] PERMIT-TO-JOIN: duration: 0 for Radio: znp for node: None
2023-01-28 17:09:38,227 INFO    : [   ZigpyCom_2 683] Calling self.app.permit(time_s=0, node=None )
2023-01-28 17:09:38,357 INFO    : [ZigpyForwarder_2 684] Accepting new Hardware: Disable (Off)
2023-01-28 17:09:38,487 INFO    : [   ZigpyCom_2 683] returning from the self.app.permit(time_s=0, node=None )

* export.json
[{"Status":"","RIA":"10","LogicalType":"Coordinator","ZDeviceName":"Zigbee Coordinator","Version":"3","Ep":{"00":{},"02":{"0000":{}},"01":{"0000":{},"0006":{},"000a":{},"0019":{},"0501":{},"0001":{},"0020":{},"0500":{},"0502":{}}},"PowerSource":"Main","RoutingTable":[{"Devices":[],"SQN":0,"Status":"Success","TimeStamp":"28/01/2023 17:20:04","Time":"28/01/2023 17:20:04","RoutingTableTableSize":0}],"Neighbours":[{"SQN":0,"Status":"","TimeStamp":"28/01/2023 17:20:04","Time":"28/01/2023 17:20:04","Devices":[{"5b44":{"_relationshp":"Child","_devicetype":"End Device","_depth":1,"_lnkqty":94,"_rxonwhenidl":"Rx-Off","_IEEE":"54ef441000446e47","_permitjnt":"??"}}]}],"IEEE":"00124b002a2e8849","Health":"Live","ConsistencyCheck":"","LQI":{},"_rawNodeDescriptor":"000000a000a02c01008f504000","Max Buffer Size":"50","Max Rx":"00a0","Max Tx":"00a0","macapa":"8f","bitfield":"4000","server_mask":"2c01","descriptor_capability":"00","Manufacturer":"0000","DeviceType":"FFD","ReceiveOnIdle":"On","Stamp":{"LasteSeen":{},"Time":"2023-01-28 17:20:10","MsgType":"8041","time":1674915610.2608724,"LastSeen":"28/01/2023 17:20:10"},"SQN":"00","Model":"CC1352/CC2652, Z-Stack 3.30+ (znp)","ZDPSQN":"03","AssociatedDevices":[{"Devices":[],"SQN":0,"Status":"","TimeStamp":"28/01/2023 17:20:04","Time":"28/01/2023 17:20:04"}],"NeighbourTableSize":1,"NwkID":"0000"} [... cut ...] 004":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"0005":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"0006":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"0009":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"000a":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"0019":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}}},"02":{"0000":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}}}}},"NwkID":"8952"}]

pipiche38, 2023-02-01:
can switch to the following branch, and redo the pairing from scratch

```
git pull
git checkout hotfix-6.3.011
```

I'm not sure it will be enough but let see
please enable debug 

- [ ] Input
- [ ] Cluster
- [ ] Lumi

[image]

then try to press the left button check the logs ( Logs/PluginZigbee-xx.log )
press the right button check the logs

and then drop the logs here if the right button still do not work

pipiche38, 2023-02-04:
any update , results ?

lemonacid, 2023-02-05:
hello,
after switching to  hotfix-6.3.011 and re-pairing from scratch I have two switches in DomoticZ and can control both of them but DomoticZ does not properly display status of a right switch when changed by button press (not from Domoticz).

Logs with debug Lumi, cluster and input enabled foloows.

Left buttow pressed (load switched on) and pressed again (load switched off):

`2023-02-05 21:50:36,264 DEBUG   : [ZigpyForwarder_2 13504] ZigateRead - MsgType: 8002,  Data: 00010400060101025a5f020000187d0a00001001, LQI: 81
2023-02-05 21:50:36,272 DEBUG   : [ZigpyForwarder_2 13504] Decode8102 - Attribute Reports: [5a5f:01] MsgSQN: 7d ClusterID: 0006 AttributeID: 0000 Status: 00 Type: 10 Size: 0001 ClusterData: >01<
2023-02-05 21:50:36,288 DEBUG   : [ZigpyForwarder_2 13504] scan_attribute_reponse - Sqn: 7d i_sqn: 7d Nwkid: 5a5f Ep: 01 Cluster: 0006 MsgData: 7d5a5f01000600000010000101 Type: 8102
2023-02-05 21:50:36,292 DEBUG   : [ZigpyForwarder_2 13504] scan_attribute_reponse - 8102 idx: 26 Read Attribute Response: [5a5f:01] ClusterID: 0006 MsgSQN: 7d, i_sqn: 7d, AttributeID: 0000 Status: 00 Type: 10 Size: 0001 ClusterData: >01<
2023-02-05 21:50:36,305 DEBUG   : [ZigpyForwarder_2 13504] Decode8102 - LQI:  81 Received Cluster:0006 Attribute: 0000 Value:   01 from (5a5f/01)
2023-02-05 21:50:36,308 DEBUG   : [ZigpyForwarder_2 13504] ReadCluster - 0006 - 5a5f/01 AttrId: 0000 AttrType: 10 Attsize: 0001 Status: 00 AttrValue: 01
2023-02-05 21:50:36,380 DEBUG   : [ZigpyForwarder_2 13504] ReadCluster - ClusterId=0006 - reception General: On/Off: 01
2023-02-05 21:50:38,869 DEBUG   : [ZigpyForwarder_2 13504] ZigateRead - MsgType: 8002,  Data: 00010400060101025a5f02000018810a00001000, LQI: 69
2023-02-05 21:50:38,879 DEBUG   : [ZigpyForwarder_2 13504] Decode8102 - Attribute Reports: [5a5f:01] MsgSQN: 81 ClusterID: 0006 AttributeID: 0000 Status: 00 Type: 10 Size: 0001 ClusterData: >00<
2023-02-05 21:50:38,896 DEBUG   : [ZigpyForwarder_2 13504] scan_attribute_reponse - Sqn: 81 i_sqn: 81 Nwkid: 5a5f Ep: 01 Cluster: 0006 MsgData: 815a5f01000600000010000100 Type: 8102
2023-02-05 21:50:38,899 DEBUG   : [ZigpyForwarder_2 13504] scan_attribute_reponse - 8102 idx: 26 Read Attribute Response: [5a5f:01] ClusterID: 0006 MsgSQN: 81, i_sqn: 81, AttributeID: 0000 Status: 00 Type: 10 Size: 0001 ClusterData: >00<
2023-02-05 21:50:38,906 DEBUG   : [ZigpyForwarder_2 13504] Decode8102 - LQI:  69 Received Cluster:0006 Attribute: 0000 Value:   00 from (5a5f/01)
2023-02-05 21:50:38,909 DEBUG   : [ZigpyForwarder_2 13504] ReadCluster - 0006 - 5a5f/01 AttrId: 0000 AttrType: 10 Attsize: 0001 Status: 00 AttrValue: 00
2023-02-05 21:50:38,966 DEBUG   : [ZigpyForwarder_2 13504] ReadCluster - ClusterId=0006 - reception General: On/Off: 00
`

RightButton buttow pressed (load switched on) and pressed again (load switched off):

`2023-02-05 21:51:32,104 DEBUG   : [ZigpyForwarder_2 13504] ZigateRead - MsgType: 8002,  Data: 00010400060101025a5f02000018fb0a00001000, LQI: 72
2023-02-05 21:51:32,112 DEBUG   : [ZigpyForwarder_2 13504] Decode8102 - Attribute Reports: [5a5f:01] MsgSQN: fb ClusterID: 0006 AttributeID: 0000 Status: 00 Type: 10 Size: 0001 ClusterData: >00<
2023-02-05 21:51:32,138 DEBUG   : [ZigpyForwarder_2 13504] scan_attribute_reponse - Sqn: fb i_sqn: fb Nwkid: 5a5f Ep: 01 Cluster: 0006 MsgData: fb5a5f01000600000010000100 Type: 8102
2023-02-05 21:51:32,141 DEBUG   : [ZigpyForwarder_2 13504] scan_attribute_reponse - 8102 idx: 26 Read Attribute Response: [5a5f:01] ClusterID: 0006 MsgSQN: fb, i_sqn: fb, AttributeID: 0000 Status: 00 Type: 10 Size: 0001 ClusterData: >00<
2023-02-05 21:51:32,155 DEBUG   : [ZigpyForwarder_2 13504] Decode8102 - LQI:  72 Received Cluster:0006 Attribute: 0000 Value:   00 from (5a5f/01)
2023-02-05 21:51:32,158 DEBUG   : [ZigpyForwarder_2 13504] ReadCluster - 0006 - 5a5f/01 AttrId: 0000 AttrType: 10 Attsize: 0001 Status: 00 AttrValue: 00
2023-02-05 21:51:32,175 DEBUG   : [ZigpyForwarder_2 13504] ReadCluster - ClusterId=0006 - reception General: On/Off: 00
`

pipiche38, 2023-02-05:
The logs doesn't looks enough, as they are the same despite the button pressed.
Can you please provide the full logs so I can see if there is not something else

lemonacid, 2023-02-05:
`2023-02-05 21:44:51,797 DEBUG   : [ZigpyForwarder_2 13504] ZigateRead - MsgType: 8002,  Data: 00010400060101025a5f02000018750a00001000, LQI: 63
2023-02-05 21:44:51,811 DEBUG   : [ZigpyForwarder_2 13504] Decode8102 - Attribute Reports: [5a5f:01] MsgSQN: 75 ClusterID: 0006 AttributeID: 0000 Status: 00 Type: 10 Size: 0001 ClusterData: >00<
2023-02-05 21:44:51,824 DEBUG   : [ZigpyForwarder_2 13504] scan_attribute_reponse - Sqn: 75 i_sqn: 75 Nwkid: 5a5f Ep: 01 Cluster: 0006 MsgData: 755a5f01000600000010000100 Type: 8102
2023-02-05 21:44:51,827 DEBUG   : [ZigpyForwarder_2 13504] scan_attribute_reponse - 8102 idx: 26 Read Attribute Response: [5a5f:01] ClusterID: 0006 MsgSQN: 75, i_sqn: 75, AttributeID: 0000 Status: 00 Type: 10 Size: 0001 ClusterData: >00<
2023-02-05 21:44:51,837 DEBUG   : [ZigpyForwarder_2 13504] Decode8102 - LQI:  63 Received Cluster:0006 Attribute: 0000 Value:   00 from (5a5f/01)
2023-02-05 21:44:51,840 DEBUG   : [ZigpyForwarder_2 13504] ReadCluster - 0006 - 5a5f/01 AttrId: 0000 AttrType: 10 Attsize: 0001 Status: 00 AttrValue: 00
2023-02-05 21:44:51,852 DEBUG   : [ZigpyForwarder_2 13504] ReadCluster - ClusterId=0006 - reception General: On/Off: 00
2023-02-05 21:45:45,319 DEBUG   : [ZigpyForwarder_2 13504] ZigateRead - MsgType: 8002,  Data: 00010400060101025a5f02000018ef0a00001001, LQI: 87
2023-02-05 21:45:45,328 DEBUG   : [ZigpyForwarder_2 13504] Decode8102 - Attribute Reports: [5a5f:01] MsgSQN: ef ClusterID: 0006 AttributeID: 0000 Status: 00 Type: 10 Size: 0001 ClusterData: >01<
2023-02-05 21:45:45,353 DEBUG   : [ZigpyForwarder_2 13504] scan_attribute_reponse - Sqn: ef i_sqn: ef Nwkid: 5a5f Ep: 01 Cluster: 0006 MsgData: ef5a5f01000600000010000101 Type: 8102
2023-02-05 21:45:45,356 DEBUG   : [ZigpyForwarder_2 13504] scan_attribute_reponse - 8102 idx: 26 Read Attribute Response: [5a5f:01] ClusterID: 0006 MsgSQN: ef, i_sqn: ef, AttributeID: 0000 Status: 00 Type: 10 Size: 0001 ClusterData: >01<
2023-02-05 21:45:45,367 DEBUG   : [ZigpyForwarder_2 13504] Decode8102 - LQI:  87 Received Cluster:0006 Attribute: 0000 Value:   01 from (5a5f/01)
2023-02-05 21:45:45,370 DEBUG   : [ZigpyForwarder_2 13504] ReadCluster - 0006 - 5a5f/01 AttrId: 0000 AttrType: 10 Attsize: 0001 Status: 00 AttrValue: 01
2023-02-05 21:45:45,467 DEBUG   : [ZigpyForwarder_2 13504] ReadCluster - ClusterId=0006 - reception General: On/Off: 01
2023-02-05 21:45:47,503 DEBUG   : [ZigpyForwarder_2 13504] ZigateRead - MsgType: 8002,  Data: 00010400060101025a5f02000018f30a00001000, LQI: 96
2023-02-05 21:45:47,516 DEBUG   : [ZigpyForwarder_2 13504] Decode8102 - Attribute Reports: [5a5f:01] MsgSQN: f3 ClusterID: 0006 AttributeID: 0000 Status: 00 Type: 10 Size: 0001 ClusterData: >00<
2023-02-05 21:45:47,530 DEBUG   : [ZigpyForwarder_2 13504] scan_attribute_reponse - Sqn: f3 i_sqn: f3 Nwkid: 5a5f Ep: 01 Cluster: 0006 MsgData: f35a5f01000600000010000100 Type: 8102
2023-02-05 21:45:47,533 DEBUG   : [ZigpyForwarder_2 13504] scan_attribute_reponse - 8102 idx: 26 Read Attribute Response: [5a5f:01] ClusterID: 0006 MsgSQN: f3, i_sqn: f3, AttributeID: 0000 Status: 00 Type: 10 Size: 0001 ClusterData: >00<
2023-02-05 21:45:47,542 DEBUG   : [ZigpyForwarder_2 13504] Decode8102 - LQI:  96 Received Cluster:0006 Attribute: 0000 Value:   00 from (5a5f/01)
2023-02-05 21:45:47,545 DEBUG   : [ZigpyForwarder_2 13504] ReadCluster - 0006 - 5a5f/01 AttrId: 0000 AttrType: 10 Attsize: 0001 Status: 00 AttrValue: 00
2023-02-05 21:45:47,606 DEBUG   : [ZigpyForwarder_2 13504] ReadCluster - ClusterId=0006 - reception General: On/Off: 00
2023-02-05 21:46:41,671 DEBUG   : [ZigpyForwarder_2 13504] ZigateRead - MsgType: 8002,  Data: 00010400060101025a5f020000186d0a00001000, LQI: 63
2023-02-05 21:46:41,678 DEBUG   : [ZigpyForwarder_2 13504] Decode8102 - Attribute Reports: [5a5f:01] MsgSQN: 6d ClusterID: 0006 AttributeID: 0000 Status: 00 Type: 10 Size: 0001 ClusterData: >00<
2023-02-05 21:46:41,687 DEBUG   : [ZigpyForwarder_2 13504] scan_attribute_reponse - Sqn: 6d i_sqn: 6d Nwkid: 5a5f Ep: 01 Cluster: 0006 MsgData: 6d5a5f01000600000010000100 Type: 8102
2023-02-05 21:46:41,690 DEBUG   : [ZigpyForwarder_2 13504] scan_attribute_reponse - 8102 idx: 26 Read Attribute Response: [5a5f:01] ClusterID: 0006 MsgSQN: 6d, i_sqn: 6d, AttributeID: 0000 Status: 00 Type: 10 Size: 0001 ClusterData: >00<
2023-02-05 21:46:41,703 DEBUG   : [ZigpyForwarder_2 13504] Decode8102 - LQI:  63 Received Cluster:0006 Attribute: 0000 Value:   00 from (5a5f/01)
2023-02-05 21:46:41,707 DEBUG   : [ZigpyForwarder_2 13504] ReadCluster - 0006 - 5a5f/01 AttrId: 0000 AttrType: 10 Attsize: 0001 Status: 00 AttrValue: 00
2023-02-05 21:46:41,724 DEBUG   : [ZigpyForwarder_2 13504] ReadCluster - ClusterId=0006 - reception General: On/Off: 00
2023-02-05 21:47:18,939 DEBUG   : [ZigpyForwarder_2 13504] ZigateRead - MsgType: 8002,  Data: 00010400060101025a5f02000018c30a00001001, LQI: 45
2023-02-05 21:47:18,946 DEBUG   : [ZigpyForwarder_2 13504] Decode8102 - Attribute Reports: [5a5f:01] MsgSQN: c3 ClusterID: 0006 AttributeID: 0000 Status: 00 Type: 10 Size: 0001 ClusterData: >01<
2023-02-05 21:47:18,955 DEBUG   : [ZigpyForwarder_2 13504] scan_attribute_reponse - Sqn: c3 i_sqn: c3 Nwkid: 5a5f Ep: 01 Cluster: 0006 MsgData: c35a5f01000600000010000101 Type: 8102
2023-02-05 21:47:18,958 DEBUG   : [ZigpyForwarder_2 13504] scan_attribute_reponse - 8102 idx: 26 Read Attribute Response: [5a5f:01] ClusterID: 0006 MsgSQN: c3, i_sqn: c3, AttributeID: 0000 Status: 00 Type: 10 Size: 0001 ClusterData: >01<
[... cut ...]
2023-02-05 21:55:05,899 DEBUG   : [ZigpyForwarder_2 13504] scan_attribute_reponse - 8102 idx: 26 Read Attribute Response: [5a5f:01] ClusterID: 0006 MsgSQN: e3, i_sqn: e3, AttributeID: 0000 Status:
00 Type: 10 Size: 0001 ClusterData: >00<
2023-02-05 21:55:05,902 DEBUG   : [ZigpyForwarder_2 13504] Decode8102 - LQI:  51 Received Cluster:0006 Attribute: 0000 Value:   00 from (5a5f/01)
2023-02-05 21:55:05,915 DEBUG   : [ZigpyForwarder_2 13504] ReadCluster - 0006 - 5a5f/01 AttrId: 0000 AttrType: 10 Attsize: 0001 Status: 00 AttrValue: 00
2023-02-05 21:55:05,930 DEBUG   : [ZigpyForwarder_2 13504] ReadCluster - ClusterId=0006 - reception General: On/Off: 00

`

pipiche38, 2023-02-05:
Maybe an idea. Give me 10 minutes

pipiche38, 2023-02-05:
I pushed an update, so you update with a `git pull`
restart the plugin, then  wait that the startup is completed

pipiche38, 2023-02-06:
In the Device management can you change the Parameter

"AqaraOpple_switch_operation_mode_opple": 1   (from 0 to 1 ), you have to save it twice  (in the edit pop-up and then in the top left window) 
See here
https://zigbeefordomoticz.github.io/wiki/en-eng/HowTo_Device-parameters.html

pipiche38, 2023-02-06:
just pushed an update, would appreciate if you could test it

simply load the update with git pull
restart the plugin
leaves it starting, then wait about 10 more minutes before doing the tests

finally I'm interested about the log file and the export.json

lemonacid, 2023-02-06:
First test performed without changeing Opple mode setting.
pre-test state: both channels switched on.
Stopped Domoticz;
git pull;
removed logfile;
started Domoticz;
wait 10 minutes;
pressed left button, load switched off, domoticz widget changed on->off;
pressed left button again, load switched on, domotic
z widget changed off->on;
pressed right button, load switched off, domoticz widget not changed (still on);
pressed right button again, load switched on, domoticz widget widget not changed (still on).
Complete logfile from domoticz startup and exprt.json attached.
[PluginZigbee_02.zip](https://github.com/zigbeefordomoticz/Domoticz-Zigbee/files/10667991/PluginZigbee_02.zip)

will return soon with same test with Opple mode changed.

will return soon with

pipiche38, 2023-02-06:
Thanks for the export.json. This looks bad as there are some expected actions which are not happening

try to play with the parameter 
"AqaraOppleMode": 1,    and set it to 0 once and then to 1

please enable zclCommand as debug, and you need to wait for a log message starting with  "zcl_write_attribute

lemonacid, 2023-02-06:
Testing with {  'AqaraOpple_switch_operation_mode_opple': 1} and  zclCommand debug enabled.
Sad but there is no zcl_write_attribute messages in log.
pre-test conditions: both channels switched off.
pressed left button, load switched on;
pressed left button again, load switched off;
pressed right button, load switched on;
pressed right button again, load switched off.
[PluginZigbee_02.zip](https://github.com/zigbeefordomoticz/Domoticz-Zigbee/files/10668482/PluginZigbee_02.zip)
i'm switching to work now, be able to continue testing tomorrow.

pipiche38, 2023-02-06:
But you are on version  Zigbee for Domoticz (z4d) plugin beta6-6.4.221 started

I was asking you to switch to a development branch https://github.com/zigbeefordomoticz/Domoticz-Zigbee/issues/1503#issuecomment-1412635612

So please do remote the existing paired device

the update and switch to the branch

```
git pull
git checkout config-release-6.3.011
```

restart the plugin and do the pairing

pipiche38, 2023-02-07:
There is an error in the log, which could explain that some commands are not send to the device.
Need to understand what is happening.

pipiche38, 2023-02-07:
updated pushed.
Simply on the same branch `git pull `and then restart the plugin

pipiche38, 2023-02-07:
If it doesn't work. Where are you based ? Do you have a way to ship it to me in France ?

lemonacid, 2023-02-08:
Now everything is working properly. I ran some tests - statuses of both widgets changing in correct way in any scenario. Many thanx, great work!

pipiche38, 2023-02-08:
Thanks also for your patience.
I'll be merging that branch in stable6. When done you will have simply to switch with the following commands

```
git checkout stable6
git pull
```