zigbeefordomoticz/Domoticz-Zigbee #1505: [Model Certification] - namron-zigbee-touch-termostat-16a-hvit

RoaldNakstad, 2023-01-30:
namron-zigbee-touch-termostat-16a-hvit - modelname: 4512737

Product page:  https://www.namron.com/products/namron-zigbee-touch-termostat-16a-hvit/
Manual:  https://www.elektroimportoren.no/docs/lib/4512737-Brukerveiledning-5.pdf

[export.json.txt](https://github.com/zigbeefordomoticz/Domoticz-Zigbee/files/10539891/export.json.txt)

Have tested this device with plugin 6.4.211 (beta), and it works kind of 'out-of-the-box' with some small issues.

pipiche38, 2023-01-30:
Thanks for the doc, looks usefull

RoaldNakstad, 2023-01-30:
Also coul'd have been nice to do OTA since they have a new firmware for download.

pipiche38, 2023-01-30:
Started here
https://github.com/pipiche38/Domoticz-Zigbee/tree/device/power-adjustement-kaNor

pipiche38, 2023-01-31:
I have created a specific developement branch

[zigbeefordomoticz:device/narom-thermostat](https://github.com/zigbeefordomoticz/Domoticz-Zigbee/tree/device/narom-thermostat)

```
git pull
git checkout device/narom-thermostat
```

should put you on it.

pipiche38, 2023-02-04:
Did you try the branch ?
How does it work ?
Is the OTA enabled now ?

RoaldNakstad, 2023-02-06:
Yes, have been running device/narom-thermostat for some time now. Measured values are correct :-)
Did a 'Recreate widget(s) of existing device', but the selector switch still shows wrong labels. 
I will do some more testing to see witch value actually setts witch mode and give you an update.

I placed the firmware in a new NAMRON folder, but did'nt show up. So I placed it inside SCHNEIDER, and was able to select it:
[image]

But after hitting update, nothing seems to happend before this error was logged:

Time: "06/02/2023 21:17:48"
Module: "OTA"
nwkid: "None"
PluginHealth: "Ready"
message: "Ota detects Timeout while notifying device 22c2"
PermitToJoin:
Duration: 240
Starttime: 1675717263
Thread: "MainThread"

pipiche38, 2023-02-07:
For the OTA, it looks like the plugin is proposing Ep: 01, while the device is expecting 02 for OTA. Could be an issue on the plugin side. I'll investigate.

For the Widget, yes nothing done for now.
Any suggestion of what you would like to have for the Thermostat Mode ?

pipiche38, 2023-02-07:
please update the branch in order to get the latest changes. The OTA should be better.
You should get an NAMRON AS Manufacturer now

RoaldNakstad, 2023-02-07:
Did a git pull and copied tke firmware-file into the new folder 'NAMRON AS'
But after a plugin restart, the file does'nt show up here:
[image]

Maybee some issues with foldername with a space in it? Try just NAMRON instead?

pipiche38, 2023-02-07:
Where can I find the firmware ? 
What do you have in the log,

pipiche38, 2023-02-07:
It works for me 

[image]

Can you check when starting the plugin you have a similar message in the log ?

> Feb 07 21:56:20 rasp jemalloc.sh[15933]: 2023-02-07 21:56:20.099  Status: Zigpy-Elelabs: Available Firmware - ManufCode: 1224 ImageType: 0x1771 FileVersion: 0x      12 Size:   247634 Bytes Filename: 4512737-Firmware-35.ota

closed 2023-06-15 by github-actions[bot]