zigbeefordomoticz/Domoticz-Zigbee #1515: Unrecognised TS0601 Temperature/Humidity sensor

stevedh42, 2023-02-01:
* Product Name : Tenky Tuya ZigBee Temperature Humidity Sensor
* Brand : Tenky
* URL on a web site : https://www.aliexpress.com/item/1005005012094422.html
* What do you expect : To be able to get temperature and humidity values from it.

* ZigBeeforDomoticZ version : 6.3.010

*Other info : This is a cheap Chinese zigbee temperature sensor with a display bought from aliexpress. Before I updated to 6.3.010 it was detected but didn't work.. old log also included just in case it is of any use.

* The logs from the pairing process
* export.json file as described here above
[pairing.log](https://github.com/zigbeefordomoticz/Domoticz-Zigbee/files/10561319/pairing.log)
[export.json.txt](https://github.com/zigbeefordomoticz/Domoticz-Zigbee/files/10561325/export.json.txt)
[pairing_old_6_3_005.log](https://github.com/zigbeefordomoticz/Domoticz-Zigbee/files/10561438/pairing_old_6_3_005.log)

stevedh42, 2023-02-01:
I did try just adding 
    "_TZE200_whkgqxse"
to
TUYA_TEMP_HUMI = (
in tuyaConst.py

it was detected, but readings were zero, and got this in the log.

2023-02-01 21:07:17.107 Error: Sonoff Zigbee: Aberrant Temp: 1.8180266169757779e+78 (below -50 or above 100) for device: 8a34
2023-02-01 21:07:18.089 Error: Sonoff Zigbee: Aberrant Temp: 1.8180266169757779e+78 (below -50 or above 100) for device: 8a34

pipiche38, 2023-02-02:
Which version of the plugin are you using?Envoyé de mon iPhone

pipiche38, 2023-02-02:
could you please go to the hotfix-6.3.011 branch

```
git pull
git checkout hotfix-6.3.011
```

remove the device from domoticz et the plugin
restart the plugin

enable debug via the 'WebUI' > Debug

Input
Tuya

[image]

Then do the pairing, leave it like that during 5 or 10 minutes

finaly drop me the log file 
- Log/PluginZigbee-xx.log
and the export.json 'Plugin Device List RAW'

stevedh42, 2023-02-02:
thanks
Please find attached log and raw plugin list. Hope these help. From googling looks like this device is usually described as Temperature/Humidity sensor with clock... although I think the clock is unreadable.

[PluginZigbee_04.log](https://github.com/zigbeefordomoticz/Domoticz-Zigbee/files/10567277/PluginZigbee_04.log)
[export2.json.txt](https://github.com/zigbeefordomoticz/Domoticz-Zigbee/files/10567286/export2.json.txt)

pipiche38, 2023-02-02:
The data transmitted looks totally un-expected.

Is this values correct ?

Temp: 0xc2 -> 19.4
Humi: 0x37 -> 55
Battery: 0x64 -> 100
0a: 0x68 -> 104 ???
0b: 0xc8 -> 200 ???

stevedh42, 2023-02-02:
Yes those values look correct, I'm pretty sure it was around 19 degrees at that time..
Be interesting to know what those other values are, pity I didn't set the clock correctly as maybe they might have had something to do with the time..

pipiche38, 2023-02-02:
I beleive you can set some Alarm, so it might be related to a Temp Min / Max for example

Can you switch to a new branch (untested)

```
git pull
git checkout tuya-cluster/recursive-scan
git pull
```

simply restart the plugin

and let see . Please cross check if any errors can occures in the Domoticz Logs and the plugin logs
leave the debug flag as they are

stevedh42, 2023-02-02:
On the plus side... it set the date and time automatically, which surprised me.

On the minus side It is still reading the data incorrectly, I checked and as far as I can tell I updated correctly.

2023-02-02 18:10:38.540 Error: Sonoff Zigbee: Aberrant Temp: 2.0496107954504102e+78 (below -50 or above 100) for device: db4f
2023-02-02 18:10:42.503 Error: Sonoff Zigbee: Aberrant Temp: 2.0496107954504102e+78 (below -50 or above 100) for device: db4f

stevedh42, 2023-02-02:
Log just in case it is helpful
[PluginZigbee_04.log](https://github.com/zigbeefordomoticz/Domoticz-Zigbee/files/10572007/PluginZigbee_04.log)

stevedh42, 2023-02-02:
If you want me to check a file or something I can.

pipiche38, 2023-02-02:
Please provides the output of command

git status

stevedh42, 2023-02-02:
steve@ubuntu1:~/domoticz/plugins/Domoticz-Zigbee$ sudo git status
On branch tuya-cluster/recursive-scan
Your branch is up to date with 'origin/tuya-cluster/recursive-scan'.

nothing to commit, working tree clean

stevedh42, 2023-02-03:
Done.. 
still has same error but thanks for looking at this.

2023-02-03 09:49:35.108 Error: Sonoff Zigbee: Aberrant Temp: 2.2464573471538477e+78 (below -50 or above 100) for device: db4f

I've attached log.
[PluginZigbee_04.log](https://github.com/zigbeefordomoticz/Domoticz-Zigbee/files/10577808/PluginZigbee_04.log)

BTW the screen of db4f is showing
temp  19.4
humidity 57%

And Git status

On branch tuya-cluster/recursive-scan
Your branch is up to date with 'origin/tuya-cluster/recursive-scan'.

nothing to commit, working tree clean

pipiche38, 2023-02-03:
not sure I understand, is the temp/humi correctly reported ?

stevedh42, 2023-02-03:
No sorry, still getting 0 for temp and 50 for humid which never change.
still getting this error in the log.
2023-02-03 09:49:35.108 Error: Sonoff Zigbee: Aberrant Temp: 2.2464573471538477e+78 (below -50 or above 100) for device: db4f

I just included the temperature read directly from the display of the device incase that helped.

pipiche38, 2023-02-03:
Ok, try again

git pull (on the same branch)

stevedh42, 2023-02-03:
no :(

2023-02-03 11:31:34.981 Error: Sonoff Zigbee: Aberrant Temp: 2.3043533917725058e+78 (below -50 or above 100) for device: db4f
2

Log attached

[PluginZigbee_04.log](https://github.com/zigbeefordomoticz/Domoticz-Zigbee/files/10578461/PluginZigbee_04.log)

pipiche38, 2023-02-03:
No clue what is happening, just added some debug , can you update and restart

stevedh42, 2023-02-03:
Done.. log attached
[PluginZigbee_04.log](https://github.com/zigbeefordomoticz/Domoticz-Zigbee/files/10578809/PluginZigbee_04.log)

stevedh42, 2023-02-03:
ALMOST :) I was watching domoticz log waiting for the error to pop up and it didn't

Temperature is correct but humidity is wrong

Temp + Humidity	THGN122/123/132, THGR122/228/238/268	20.8 C, 5 %

Humidity should be 56%

log attached.

[PluginZigbee_04.log](https://github.com/zigbeefordomoticz/Domoticz-Zigbee/files/10578999/PluginZigbee_04.log)

pipiche38, 2023-02-03:
please switch back to hotfix-6.3.011

```
git pull
git checkout hotfix-6.3.011
```

and then restart the plugin

stevedh42, 2023-02-03:
Done thanks,
Humidity still show 5% instead of now 55%

Log attached

[PluginZigbee_04.log](https://github.com/zigbeefordomoticz/Domoticz-Zigbee/files/10579322/PluginZigbee_04.log)
[PluginZigbee_log_error_history.json.txt](https://github.com/zigbeefordomoticz/Domoticz-Zigbee/files/10579323/PluginZigbee_log_error_history.json.txt)

pipiche38, 2023-02-03:
indeed they were an error, try again

stevedh42, 2023-02-03:
YAY!!
all appears to be working, thanks, I'm guessing this patch was a bit more complicated then normal.

pipiche38, 2023-02-03:
Good, so it will merged on stable6 at a point of time.

pipiche38, 2023-02-03:
Can you kindly switch to the branch config-release-6.3.011

```
git pull
git checkout config-release-6.3.011

```
and redo the all test from pairing

stevedh42, 2023-02-03:
everything looks good.

Did the pull and checkout.
removed it from Domoticz and the plugin
rebooted plugin.
repaired it.
everything working well. (also confirmed battery level by using uncharged battery)

pipiche38, 2023-02-04:
Ok, thanks for the cross-check.
You can remain on that branch until we published release 6.3.011

thanks for your support

closed 2023-02-04 by pipiche38