zigbeefordomoticz/Domoticz-Zigbee #1523: TUYA - PJ-mini-ZSW01 - seen as TS011F switch only (no power meter)

Overnuts, 2023-02-06:
Hi, 
I bought a TUYA "PJ-mini-ZSW01" 16A module for monitoring : 
https://fr.aliexpress.com/item/1005004128359494.html

Received today, it pairs to my system, device is "certified" as TS011F by plugin, but only as a switch.

This device should have power meter functions.

thanks for your work :-)

Note, while pairing, device was only wired to power, nothing after switched circuit

Here are plugin logs:

> 2023-02-06 18:16:48,679 INFO    : [     MainThread 0] Request Accepting new Hardware for 240 seconds 
2023-02-06 18:16:48,778 INFO    : [    ZigpyCom_16 0] PERMIT-TO-JOIN: {'cmd': 'PERMIT-TO-JOIN', 'datas': {'Duration': 240, 'targetRouter': 'FFFC'}, 'NwkId': None, 'TimeStamp': 1675703808.6694882, 'ACKIsDisable': False, 'Sqn': None}
2023-02-06 18:16:48,792 INFO    : [    ZigpyCom_16 0] PERMIT-TO-JOIN: duration: 240 for Radio: znp for node: None
2023-02-06 18:16:48,817 INFO    : [    ZigpyCom_16 0] Calling self.app.permit(time_s=240, node=None )
2023-02-06 18:16:49,425 INFO    : [    ZigpyCom_16 0] returning from the self.app.permit(time_s=240, node=None )
2023-02-06 18:16:50,171 INFO    : [ZigpyForwarder_16 0] Accepting new Hardware: Enable (On)
2023-02-06 18:18:28,503 INFO    : [ZigpyForwarder_16 0] --> Adding device 903d in self.DevicesInPairingMode
2023-02-06 18:18:28,522 INFO    : [ZigpyForwarder_16 0] --> ['903d']
2023-02-06 18:18:28,543 INFO    : [ZigpyForwarder_16 0] [None] NEW OBJECT: 903d None
2023-02-06 18:18:28,563 INFO    : [ZigpyForwarder_16 0] ReadAttributeRequest_0000_for_tuya 903d
2023-02-06 18:18:29,192 INFO    : [ZigpyForwarder_16 0] check_found_plugin_model - TS011F _TZ3000_cjrngdr3  None
2023-02-06 18:18:29,444 INFO    : [ZigpyForwarder_16 0] Decode8046 - Match Descriptor response: SQN: 00 Status: 00 Nwkid: 3d90 Lenght: 00 List: 
2023-02-06 18:18:29,571 INFO    : [ZigpyForwarder_16 0] [-] NEW OBJECT: 903d Active Endpoint Response Ep: 01 LQI: 0
2023-02-06 18:18:29,576 INFO    : [ZigpyForwarder_16 0] [-] NEW OBJECT: 903d Active Endpoint Response Ep: f2 LQI: 0
2023-02-06 18:18:32,371 INFO    : [     MainThread 0] [0] NEW OBJECT: 903d Model Name: TS011F
2023-02-06 18:18:32,447 INFO    : [     MainThread 0] [0] NEW OBJECT: 903d Manufacturer Name: _TZ3000_cjrngdr3
2023-02-06 18:18:32,453 INFO    : [     MainThread 0] [None] NEW OBJECT: 903d Request Node Descriptor
2023-02-06 18:18:35,354 INFO    : [     MainThread 0] ----> Tuya Hardware: fffe
2023-02-06 18:18:59,692 INFO    : [ZigpyForwarder_16 0] check_found_plugin_model - TS011F _TZ3000_cjrngdr3 1141 None
2023-02-06 18:19:09,319 INFO    : [     MainThread 0] Request Disabling Accepting new Hardware
2023-02-06 18:19:09,427 INFO    : [    ZigpyCom_16 0] PERMIT-TO-JOIN: {'cmd': 'PERMIT-TO-JOIN', 'datas': {'Duration': 0, 'targetRouter': 'FFFC'}, 'NwkId': None, 'TimeStamp': 1675703949.307765, 'ACKIsDisable': False, 'Sqn': None}
2023-02-06 18:19:09,443 INFO    : [    ZigpyCom_16 0] PERMIT-TO-JOIN: duration: 0 for Radio: znp for node: None
2023-02-06 18:19:09,465 INFO    : [    ZigpyCom_16 0] Calling self.app.permit(time_s=0, node=None )
2023-02-06 18:19:09,783 INFO    : [ZigpyForwarder_16 0] Accepting new Hardware: Disable (Off)
2023-02-06 18:19:10,127 INFO    : [    ZigpyCom_16 0] returning from the self.app.permit(time_s=0, node=None )

and the json export    (due to github issue limitation to 64ko text, I removed other working devices 3x lexman 16 power / 1x Zlinky / 1x temp/humidity sensor):

> [{"Status":"","RIA":"10","AssociatedDevices":[{"Devices":[],"SQN":0,"Status":"","TimeStamp":"25/01/2023 18:43:12","Time":"25/01/2023 18:43:12"},{"Devices":[],"SQN":0,"Status":"","TimeStamp":"31/01/2023 18:48:15","Time":"31/01/2023 18:48:15"}],"LogicalType":"Coordinator","PowerSource":"Main","Neighbours":[{"SQN":0,"Status":"","TimeStamp":"16/01/2023 08:23:05","Time":"16/01/2023 08:23:05","Devices":[{"003c":{"_relationshp":"Child","_devicetype":"End Device","_depth":1,"_lnkqty":170,"_rxonwhenidl":"Rx-Off","_IEEE":"84fffe21662bc8f9","_permitjnt":"??"}},{"9434":{"_relationshp":"Child","_devicetype":"End Device","_depth":1,"_lnkqty":170,"_rxonwhenidl":"Rx-Off","_IEEE":"69fffe8149185088","_permitjnt":"??"}},{"3aa2":{"_relationshp":"Sibling","_devicetype":"Router","_depth":255,"_lnkqty":106,"_rxonwhenidl":"??","_IEEE":"0c4314fffe5bf51f","_permitjnt":"??"}},{"7bcb":{"_relationshp":"Sibling","_devicetype":"Router","_depth":255,"_lnkqty":131,"_rxonwhenidl":"??","_IEEE":"0c4314fffe5bf435","_permitjnt":"??"}},{"01d1":{"_relationshp":"Sibling","_devicetype":"Router","_depth":255,"_lnkqty":26,"_rxonwhenidl":"??","_IEEE":"84ba20fffeceafe4","_permitjnt":"??"}},{"f041":{"_relationshp":"Sibling","_devicetype":"Router","_depth":255,"_lnkqty":125,"_rxonwhenidl":"??","_IEEE":"00158d0005bea862","_permitjnt":"??"}}]},{"SQN":0,"Status":"","TimeStamp":"17/01/2023 17:12:24","Time":"17/01/2023 17:12:24","Devices":[{"003c":{"_relationshp":"Child","_devicetype":"End Device","_depth":1,"_lnkqty":170,"_rxonwhenidl":"Rx-Off","_IEEE":"84fffe21662bc8f9","_permitjnt":"??"}},{"9434":{"_relationshp":"Child","_devicetype":"End Device","_depth":1,"_lnkqty":170,"_rxonwhenidl":"Rx-Off","_IEEE":"69fffe8149185088","_permitjnt":"??"}},{"7bcb":{"_relationshp":"Sibling","_devicetype":"Router","_depth":255,"_lnkqty":127,"_rxonwhenidl":"??","_IEEE":"0c4314fffe5bf435","_permitjnt":"??"}},{"01d1":{"_relationshp":"Sibling","_devicetype":"Router","_depth":255,"_lnkqty":54,"_rxonwhenidl":"??","_IEEE":"84ba20fffeceafe4","_permitjnt":"??"}},{"3aa2":{"_relationshp":"Sibling","_devicetype":"Router","_depth":255,"_lnkqty":113,"_rxonwhenidl":"??","_IEEE":"0c4314fffe5bf51f","_permitjnt":"??"}},{"f041":{"_relationshp":"Sibling","_devicetype":"Router","_depth":255,"_lnkqty":126,"_rxonwhenidl":"??","_IEEE":"00158d0005bea862","_permitjnt":"??"}}]},{"SQN":0,"Status":"","TimeStamp":"20/01/2023 09:34:56","Time":"20/01/2023 09:34:56","Devices":[{"003c":{"_relationshp":"Child","_devicetype":"End Device","_depth":1,"_lnkqty":170,"_rxonwhenidl":"Rx-Off","_IEEE":"84fffe21662bc8f9","_permitjnt":"??"}},{"9434":{"_relationshp":"Child","_devicetype":"End Device","_depth":1,"_lnkqty":170,"_rxonwhenidl":"Rx-Off","_IEEE":"69fffe8149185088","_permitjnt":"??"}},{"f041":{"_relationshp":"Sibling","_devicetype":"Router","_depth":255,"_lnkqty":126,"_rxonwhenidl":"??","_IEEE":"00158d0005bea862","_permitjnt":"??"}},{"3aa2":{"_relationshp":"Sibling","_devicetype":"Router","_depth":255,"_lnkqty":124,"_rxonwhenidl":"??","_IEEE":"0c4314fffe5bf51f","_permitjnt":"??"}},{"7bcb":{"_relationshp":"Sibling","_devicetype":"Router","_depth":255,"_lnkqty":116,"_rxonwhenidl":"??","_IEEE":"0c4314fffe5bf435","_permitjnt":"??"}},{"01d1":{"_relationshp":"Sibling","_devicetype":"Router","_depth":255,"_lnkqty":37,"_rxonwhenidl":"??","_IEEE":"84ba20fffeceafe4","_permitjnt":"??"}}]},{"SQN":0,"Status":"","TimeStamp":"25/01/2023 18:43:12","Time":"25/01/2023 18:43:12","Devices":[{"003c":{"_relationshp":"Child","_devicetype":"End Device","_depth":1,"_lnkqty":170,"_rxonwhenidl":"Rx-Off","_IEEE":"84fffe21662bc8f9","_permitjnt":"??"}},{"9434":{"_relationshp":"Child","_devicetype":"End Device","_depth":1,"_lnkqty":170,"_rxonwhenidl":"Rx-Off","_IEEE":"69fffe8149185088","_permitjnt":"??"}},{"3aa2":{"_relationshp":"Sibling","_devicetype":"Router","_depth":255,"_lnkqty":123,"_rxonwhenidl":"??","_IEEE":"0c4314fffe5bf51f","_permitjnt":"??"}},{"7bcb":{"_relationshp":"Sibling","_devicetype":"Router","_depth":255,"_lnkqty":125,"_rxonwhenidl":"??","_IEEE":"0c4314fffe5bf435","_permitjnt":"??"}},{"01d1":{"_relationshp":"Sibling","_devicetype":"Router","_depth":255,"_lnkqty":67,"_rxonwhenidl":"??","_IEEE":"84ba20fffeceafe4","_permitjnt":"??"}},{"f041":{"_relationshp":"Sibling","_devicetype":"Router","_depth":255,"_lnkqty":126,"_rxonwhenidl":"??","_IEEE":"00158d0005bea862","_permitjnt":"??"}}]},{"SQN":0,"Status":"","TimeStamp":"31/01/2023 18:48:15","Time":"31/01/2023 18:48:15","Devices":[{"003c":{"_relationshp":"Child","_devicetype":"End Device","_depth":1,"_lnkqty":170,"_rxonwhenidl":"Rx-Off","_IEEE":"84fffe21662bc8f9","_permitjnt":"??"}},{"9434":{"_relationshp":"Child","_devicetype":"End Device","_depth":1,"_lnkqty":170,"_rxonwhenidl":"Rx-Off","_IEEE":"69fffe8149185088","_permitjnt":"??"}},{"3aa2":{"_relationshp":"Sibling","_devicetype":"Router","_depth":255,"_lnkqty":119,"_rxonwhenidl":"??","_IEEE":"0c4314fffe5bf51f","_permitjnt":"??"}},{"7bcb":{"_relationshp":"Sibling","_devicetype":"Router","_depth":255,"_lnkqty":128,"_rxonwhenidl":"??","_IEEE":"0c4314fffe5bf435","_permitjnt":"??"}},{"01d1":{"_relationshp":"S [... cut ...] ,"CertifiedDevice":true,"SQN_000a":"16","_rawNodeDescriptor":"1141004200422a00008e424001","Max Buffer Size":"42","Max Rx":"0042","Max Tx":"0042","macapa":"8e","bitfield":"4001","server_mask":"2a00","descriptor_capability":"00","ReadConfigureReporting":{"Ep":{"01":{"0006":{"0000":{"TimeStamp":"06/02/2023 18:19:06","Status":"00","DataType":"10","MinInterval":"0001","MaxInterval":"012c"}}}},"TimeStamp":"06/02/2023 18:19:06"},"GroupMemberShip":{"01":{}},"Param":{},"CheckParam":false,"NwkID":"903d"}]

pipiche38, 2023-02-06:
which plugin version are you ?

Overnuts, 2023-02-06:
running plugin 6.3.009

pipiche38, 2023-02-06:
So please make sure to run on the latest plugin version 6.3.010
You'll have to remove the existing paired device and re-pair it