zigbeefordomoticz/Domoticz-Zigbee #1539: [Model Certification] - AQSZB-110

filippocomparin, 2023-02-17:
Brand: Frient - Develco 
Model: AQSZB-110 - Air Quality Sensor
Tecnical data: https://www.develcoproducts.com/media/1674/aqszb-110-technical-manual-air-quality-sensor-04-08-20.pdf

I tried to add this config:
```
{
    "_source": "https://github.com/zigbeefordomoticz/Domoticz-Zigbee/issues/1448",
    "_blakadder": "https://zigbee.blakadder.com/Aqara_AAQS-S01.html",
    "Ep": {
        "01": {
			"Type": ""
		},
		"26": {
			"0000": "",
			"0001": "",
			"0003": "",
			"0020": "",
			"0402": "",
			"0405": "",
			"042E": "",
			"FC03": "",
			"Type": "Temp/Humi/Voc"
		}
    },
    "Type": "",
    "ClusterToBind": [ "0402", "0405", "FC03" ],
    "ConfigureReporting": {
        "0402": { "Attributes": { "0000": { "DataType": "29", "MinInterval": "000a", "MaxInterval": "012C", "TimeOut": "0000", "Change": "0001" } } },
        "0405": { "Attributes": { "0000": { "DataType": "21", "MinInterval": "000a", "MaxInterval": "012C", "TimeOut": "0000", "Change": "0001" } } },
	"FC03": { "Attributes": { "0000": { "DataType": "21", "MinInterval": "000a", "MaxInterval": "012C", "TimeOut": "0000", "Change": "0001" } } }
    },
    "ReadAttributes": {
       	"0000": [ "0000", "0004", "0005", "0006", "0007" ],
		"0001": [ "0020" ],
        "0402": [ "0000" ],
        "0405": [ "0000" ],
		"042E": [ "0000" ],
        "FC03": [ "0000" ]
    },
    "BatteryDevice": 1,
	"VoltageConverter": 10,
	"Param": {
		"ReadAttributeChunk": 5
	}
}

```

The pairing logs:
2023-02-17 09:40:02.337 Status: Gateway: [-] NEW OBJECT: 18a6 Active Endpoint Response Ep: 26 LQI: 123
2023-02-17 09:40:02.340 Status: Gateway: [-] NEW OBJECT: 18a6 Request Simple Descriptor for Ep: 01
2023-02-17 09:40:04.099 Status: Gateway: [0] NEW OBJECT: 18a6 Manufacturer Name: frient A/S
2023-02-17 09:40:04.100 Status: Gateway: [0] NEW OBJECT: 18a6 Model Name: AQSZB-110
2023-02-17 09:40:04.106 Status: Gateway: [-] NEW OBJECT: 18a6 Cluster In 1: 0003 (Identify)
2023-02-17 09:40:04.107 Status: Gateway: [-] NEW OBJECT: 18a6 Cluster In 2: 0005 (Scenes)
2023-02-17 09:40:04.108 Status: Gateway: [-] NEW OBJECT: 18a6 Cluster In 3: 0006 (On/Off)
2023-02-17 09:40:04.109 Status: Gateway: [-] NEW OBJECT: 18a6 Application Version 0001
2023-02-17 09:40:04.110 Status: Gateway: [-] NEW OBJECT: 18a6 Cluster IN Count: 03
2023-02-17 09:40:04.111 Status: Gateway: [-] NEW OBJECT: 18a6 Cluster OUT Count: 00
2023-02-17 09:40:04.112 Status: Gateway: [-] NEW OBJECT: 18a6 ProfileID c0c9
2023-02-17 09:40:04.113 Status: Gateway: [-] NEW OBJECT: 18a6 ZDeviceID 0001
2023-02-17 09:40:04.114 Status: Gateway: [-] NEW OBJECT: 18a6 Simple Descriptor Response EP: 0x01 LQI: 132
2023-02-17 09:40:09.104 Status: Gateway: [1] NEW OBJECT: 18a6 Manufacturer Name: frient A/S
2023-02-17 09:40:09.105 Status: Gateway: [1] NEW OBJECT: 18a6 Model Name: AQSZB-110
2023-02-17 09:40:09.105 Status: Gateway: [None] NEW OBJECT: 18a6 Request Node Descriptor
2023-02-17 09:40:10.127 Gateway: Assisted Pairing: Polling: ['18a6']
2023-02-17 09:40:10.127 Gateway: --> New device: 18a6
2023-02-17 09:40:14.109 Status: Gateway: [2] NEW OBJECT: 18a6 Manufacturer Name: frient A/S
2023-02-17 09:40:14.110 Status: Gateway: [2] NEW OBJECT: 18a6 Model Name: AQSZB-110
2023-02-17 09:40:19.112 Status: Gateway: [3] NEW OBJECT: 18a6 Manufacturer Name: frient A/S
2023-02-17 09:40:19.114 Status: Gateway: [3] NEW OBJECT: 18a6 Model Name: AQSZB-110
2023-02-17 09:40:19.111 Error: Gateway: Call to function 'onHeartbeat' failed, exception details:
2023-02-17 09:40:19.120 Error: Gateway: Traceback (most recent call last):
2023-02-17 09:40:19.120 Error: Gateway: File "C:\Domoticz\plugins\Domoticz-Zigbee\plugin.py", line 1577, in onHeartbeat
2023-02-17 09:40:19.120 Error: Gateway: _plugin.onHeartbeat()
2023-02-17 09:40:19.121 Error: Gateway: File "C:\Domoticz\plugins\Domoticz-Zigbee\plugin.py", line 966, in onHeartbeat
2023-02-17 09:40:19.121 Error: Gateway: processListOfDevices(self, Devices)
2023-02-17 09:40:19.122 Error: Gateway: File "C:\Domoticz\plugins\Domoticz-Zigbee\Modules\heartbeat.py", line 891, in processListOfDevices
2023-02-17 09:40:19.122 Error: Gateway: processNotinDBDevices(self, Devices, NWKID, status, RIA)
2023-02-17 09:40:19.122 Error: Gateway: File "C:\Domoticz\plugins\Domoticz-Zigbee\Modules\pairingProcess.py", line 116, in processNotinDBDevices
2023-02-17 09:40:19.122 Error: Gateway: interview_state_createDB(self, Devices, NWKID, RIA, status)
2023-02-17 09:40:19.123 Error: Gateway: File "C:\Domoticz\plugins\Domoticz-Zigbee\Modules\pairingProcess.py", line 339, in interview_state_createDB
2023-02-17 09:40:19.124 Error: Gateway: full_provision_device(self, Devices, NWKID, RIA, status)
2023-02-17 09:40:19.125 Error: Gateway: File "C:\Domoticz\plugins\Domoticz-Zigbee\Modules\pairingProcess.py", line 386, in full_provision_device
2023-02-17 09:40:19.125 Error: Gateway: zigbee_provision_device(self, Devices, NWKID, RIA, status)
2023-02-17 09:40:19.125 Error: Gateway: File "C:\Domoticz\plugins\Domoticz-Zigbee\Modules\pairingProcess.py", line 426, in zigbee_provision_device
2023-02-17 09:40:19.126 Error: Gateway: self.configureReporting.processConfigureReporting(NwkId=NWKID)
2023-02-17 09:40:19.126 Error: Gateway: AttributeError: 'NoneType' object has no attribute 'processConfigureReporting'
2023-02-17 09:40:20.145 Gateway: Assisted Pairing: Polling: ['18a6']
2023-02-17 09:40:20.146 Gateway: --> New device: 18a6
2023-02-17 09:40:20.146 Gateway: --> inDB , removed 18a6 from List
2023-02-17 09:40:20.147 Gateway: ------> New Cluster: {'ClusterId': '0003', 'ClusterDesc': 'Identify'}
2023-02-17 09:40:20.147 Gateway: ------> New Cluster: {'ClusterId': '0005', 'ClusterDesc': 'Scenes'}
2023-02-17 09:40:20.147 Gateway: ------> New Cluster: {'ClusterId': '0006', 'ClusterDesc': 'On/Off'}
2023-02-17 09:40:20.147 Gateway: ----> New Ep: {'Ep': '01', 'Clusters': [{'ClusterId': '0003', 'ClusterDesc': 'Identify'}, {'ClusterId': '0005', 'ClusterDesc': 'Scenes'}, {'ClusterId': '0006', 'ClusterDesc': 'On/Off'}]}
2023-02-17 09:40:20.148 Gateway: ------> New Cluster: {'ClusterId': '0000', 'ClusterDesc': 'Basic'}
2023-02-17 09:40:20.148 Gateway: ----> New Ep: {'Ep': '26', 'Clusters': [{'ClusterId': '0000', 'ClusterDesc': 'Basic'}]}
2023-02-17 09:40:20.148 Gateway: --> New Device: {'NwkId': '18a6', 'ProvisionStatus': 'inDB', 'ProvisionStatusDesc': 'inDB', 'IEEE': '0015bc003600107a', 'ProfileId': 'c0c9', 'ProfileIdDesc': 'Unknow', 'ZDeviceID': '0001', 'ZDeviceIDDesc': 'Unknow', 'Model': 'AQSZB-110', 'PluginCertified': 'no', 'Ep': [{'Ep': '01', 'Clusters': [{'ClusterId': '0003', 'ClusterDesc': 'Identify'}, {'ClusterId': '0005', 'ClusterDesc': 'Scenes'}, {'ClusterId': '0006', 'ClusterDesc': 'On/Off'}]}, {'Ep': '26', 'Clusters': [{'ClusterId': '0000', 'ClusterDesc': 'Basic'}]}]}
2023-02-17 09:40:20.710 Error: Gateway: Decode8000 - PacketType: 0530 TypeSqn: None sqn_app: 00 sqn_aps: None Status: [ZigBee Error Code APS - [a6] A parameter value was invalid or out of range]

pipiche38, 2023-02-17:
you might try the following to update as this is a manufacturer specific cluster/attribute

"FC03": { "Attributes": { "0000": { "ManufSpecific": "1015", "DataType": "21", "MinInterval": "000a", "MaxInterval": "012C", "TimeOut": "0000", "Change": "0001" } } }

Ideally, try to switch to the develop branche, which provided changes to make the plugin much more open.
Work is in progress, https://github.com/zigbeefordomoticz/wiki/blob/master/en-eng/HowTo_Device-Customization.md

filippocomparin, 2023-02-17:
Thanks for the reply.
I tried to add ManufSpecific and lowercase, but with no results.

I don't understand about the default reporting place. On logs I found only temperature and humidity (0402 cluster address 0x000 and 0x003) for that device, no VOC.

I've to develop a DevicesModules?

pipiche38, 2023-02-17:
This is why I think moving to the 'develop' branch might be easier.

(1) as it is a customer specific cluster, the plugin doesn't handle such cluster
(2) with the develop branch and the new way to handle cluster, it is more open to add new clusters

if it is easier for you , I'm available on keybase https://keybase.io/pipiche/chat

pipiche38, 2023-02-18:
Attached is a potential config file which should be good on, develop branch

[AQSZB-110.json.zip](https://github.com/zigbeefordomoticz/Domoticz-Zigbee/files/10775122/AQSZB-110.json.zip)

pipiche38, 2023-04-03:
@filippocomparin any update ?

closed 2023-06-09 by github-actions[bot]