zigbeefordomoticz/Domoticz-Zigbee #1541: Aqara Motion Sensor P1 MS-S02

pabaillon, 2023-02-17:
1. Please do provide device information such as Brand and Model name.
Aqara Motion Sensor P1 MS-S02
1. Provide the URL where you have purchase the device
https://www.domadoo.fr/fr/peripheriques/6138-aqara-detecteur-de-mouvement-et-luminosite-zigbee-30-aqara-motion-sensor-p1-ms-s02-6970504215979.html
1. Finaly copy-paste all information provided in the log
{"Version":"3","ZDeviceName":"","Status":"inDB","SQN":"3b","Ep":{"01":{"0000":{"4000":"0.0.0_0006","0001":"6","0005":"lumi.motion.ac02","0004":"LUMI","0000":"3","0003":"1"},"0001":{"0020":31,"0000":3217},"0003":{},"fcc0":{"0112":"00010038","00f7":"0121910c0328140421000005210100082106010a212e840c20011320001420006410006521450069201e6a20026b2000"},"0019":{},"ClusterType":{"2648":"Voltage"},"0405":{"0000":0.7},"0006":{"0000":"00"}}},"Heartbeat":"3739","RIA":"4","LQI":100,"Battery":100,"Model":"lumi.motion.ac02","ForceAckCommands":[],"MacCapa":"80","IEEE":"54ef4410005b46bf","Type":"Voltage","ProfileID":"0104","ZDeviceID":"0107","App Version":"6","Attributes List":{},"DeviceType":"RFD","HW Version":"1","Last Cmds":[[608287,"02","e983","0100","010100190000000003000100020003"],[608288,"02","e983","0100","010100190000000003000400050006"],[608289,"02","e983","0100","010100190000000003000700080009"],[608290,"02","e983","0100","010100190000000001000a"],[608291,"02","e983","0100","010100000000000003000400050000"],[608292,"02","e983","0100","010100000000000003000100020003"],[608293,"02","e983","0100","01010000000000000300060007000a"],[608294,"02","e983","0100","010100000000000003000f00100015"],[608295,"02","e983","0100","0101000000000000034000f000ff01"],[608296,"02","e983","0100","010100000000000001ff02"]],"LogicalType":"End Device","Manufacturer":"115f","Manufacturer Name":"LUMI","NbEp":"1","PowerSource":"Battery","ReadAttributes":{"Ep":{"01":{"0000":{"TimeStamp":1676628987,"iSQN":{"4000":608295,"0004":608291,"0005":608291,"0000":608291,"0001":608292,"0002":608292,"0003":608292,"0006":608293,"0007":608293,"000a":608293,"000f":608294,"0010":608294,"0015":608294,"f000":608295,"ff01":608295,"ff02":608296},"Attributes":{"4000":"00","0004":"00","0005":"00","0000":"00","0001":"00","0002":"86","0003":"00","000f":"86","0010":"86","0015":"86","f000":"86","ff01":"86","ff02":"86"},"ZigateRequest":{}},"0001":{"TimeStamp":1676628300,"iSQN":{"0000":608284,"0001":608284,"0003":608284,"0020":608285,"0021":608285,"0033":608285,"0035":608286},"Attributes":{"0020":"00","0021":"86","0033":"86","0035":"86"},"ZigateRequest":{}},"0019":{"TimeStamp":1676628286,"iSQN":{"0001":608287,"0002":608287,"0003":608287,"0004":608288,"0005":608288,"0006":608288,"0007":608289,"0008":608289,"0009":608289,"000a":608290},"Attributes":{},"ZigateRequest":{}},"fcc0":{"TimeStamp":1676644155,"iSQN":{},"Attributes":{"0112":"00","00f7":"00"},"ZigateRequest":{}}}}},"ReceiveOnIdle":"Off","Stack Version":"","Stamp":{"time":1676644155.6244545,"Time":"2023-02-17 15:29:15","MsgType":"8102","LastSeen":1676644155},"ZCL Version":"3","Health":"Live","Capability":["Reduced-Function Device","Standard security","NwkAddr should be allocated"],"ZDPSQN":"04","RollingLQI":[55,77,78,77,77,77,82,71,63,100,100],"Epv2":{"01":{"ClusterIn":{"0000":{},"0001":{},"0003":{},"fcc0":{}},"ClusterOut":{"0003":{},"0019":{},"fcc0":{}},"ProfileID":"0104","ZDeviceID":"0107"}},"WriteAttributes":{"Ep":{"01":{"0000":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"0001":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"0003":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"fcc0":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"0019":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"0405":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"0006":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}}}}},"PairingInProgress":false,"ZDeviceVersion":"0001","ConfigSource":"8043","_rawNodeDescriptor":"115f006400642c0000807f4002","Max Buffer Size":"7f","Max Rx":"0064","Max Tx":"0064","macapa":"80","bitfield":"4002","server_mask":"2c00","descriptor_capability":"00","PairingTime":1676628286.0725918,"Bind":{"01":{"0000":{"Target":"0000","Stamp":1676628286,"Phase":"binded","Status":"00","i_sqn":608278},"0001":{"Target":"0000","Stamp":1676628287,"Phase":"binded","Status":"00","i_sqn":608279},"0003":{"Target":"0000","Stamp":1676628288,"Phase":"binded","Status":"00","i_sqn":608280},"fcc0":{"Target":"0000","Stamp":1676628286,"Phase":"requested","Status":"","i_sqn":608281}}},"BindingTable":{"SQN":0,"Status":"Success","TimeStamp":1676628316.4191868,"BindingTableSize":4,"Devices":[{"54ef4410005b46bf":{"sourceEp":"01","Cluster":"0000","targetIEEE":"00158d0005d2857f","targetNickName":"Zigbee Coordinator","targetEp":"01"}},{"54ef4410005b46bf":{"sourceEp":"01","Cluster":"0001","targetIEEE":"00158d0005d2857f","targetNickName":"Zigbee Coordinator","targetEp":"01"}},{"54ef4410005b46bf":{"sourceEp":"01","Cluster":"0003","targetIEEE":"00158d0005d2857f","targetNickName":"Zigbee Coordinator","targetEp":"01"}},{"54ef4410005b46bf":{"sourceEp":"01","Cluster":"fcc0","targetIEEE":"00158d0005d2857f","targetNickName":"Zigbee Coordinator","targetEp":"01"}}]},"ConfigureReporting":{"Ep":{"01":{"0001":{"TimeStamp":1676628286.076344,"iSQN":{"0000":608283,"0020":608283,"0021":608283},"Attributes":{"0000":"86","fffe":"86","0020":"8c"},"ZigateRequest":{}}}}},"CertifiedDevice":false,"BatteryUpdateTime":1676644155,"LUMI":{"0112":"00010038","DeviceTemperature":51.2,"RSSI dB":-255,"BatteryVoltage":3217},"SWBUILD_3":"0.0.0_0006","Param":{},"CheckParam":false}

pipiche38, 2023-02-17:
Please clarify which version of the plugin you are using.

I suggest that you switch to develop branch and you redo a plugin from scratch.

This motion was integrated by @deennoo

pabaillon, 2023-02-18:
Je suis en version dev  Plugin: 6.4.228

pipiche38, 2023-02-18:
I suggest that you switch on the release candidate 6.3.011 branch. I tried to do a first trial

```
git checkout config-release-6.3.011
git pull
```

remove the existing pairing (remove the voltage widgets)

restart the plugin

and redo the pairing

pabaillon, 2023-02-19:
Ok done,
Le device est reconnu pas, d'action dans les logs. Et la mesure de lumière reste à 0

2023-02-19 10:07:08 |   |  
-- | -- | --
2023-02-19 10:06:12 |   |  
2023-02-19 10:06:09 |   |  
2023-02-19 10:06:06 | Sound 0 |  
2023-02-19 10:06:02 |   |  
2023-02-19 10:05:59 |   |  
2023-02-19 10:05:53 |   |  
2023-02-19 10:05:50 |   |  
2023-02-19 10:05:39 |   |  
2023-02-19 10:05:27 |

pipiche38, 2023-02-19:
Thanks to enabled debug via the WebUI>Tools > Debug
Cluster
Lumi

And then do some activities around the sensor and finally provides here the Log file Logs/PluginZigbee-xx.log

pabaillon, 2023-02-19:
[_PluginZigbee_12(2).log](https://github.com/zigbeefordomoticz/Domoticz-Zigbee/files/10777563/_PluginZigbee_12.2.log)

pipiche38, 2023-02-21:
I just purchased one so it will be easier for integration.
Will keep you posted when I receive it

pipiche38, 2023-02-24:
The device is delivered, but on vacations for now, will work on it next week

pipiche38, 2023-02-26:
So I got the device, and I somehow disappointed, as this motion is only capable to provide the Lux information when a motion is detected.

So pretty weak device

pipiche38, 2023-02-26:
if you want to try the latest version of the plugin - still on branch config-release-6.3.011 -
this should work.

Please do note that 
- the Motion must be automatically reset to Off, as the device do not send a Off itself.
- the Motion provides Lux only if a motion is detected. In case of no motion you'll get the lux value from the last motion

pipiche38, 2023-02-26:
This will be released in the up-coming 6.3.011 release

closed 2023-02-27 by pipiche38